geerlingguy/raspberry-pi-pcie-devices #343: Test Home Assistant Yellow (formerly 'Amber')

geerlingguy, 2021-12-10:
I added the Amber in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/253, but the name has been changed to 'Home Assistant Yellow' (see: https://www.home-assistant.io/blog/2021/09/13/home-assistant-yellow/).

[image]

I just got the one pictured above in the mail and would like to test it, especially with my existing Hue bulbs, my AirGradient DIY air quality sensors (maybe using ESPHome?), etc.

geerlingguy, 2021-12-10:
Technically it's on the site... I just need to update the listing to read 'Yellow' instead of 'Amber'.

punnypenguins, 2021-12-22:
You want to close this now since you added the file updates for it in Issue #253 the other day?

geerlingguy, 2021-12-22:
@punnypenguins - No, still need to test it :)

closed 2022-03-14 by geerlingguy

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/home-assistant-yellow.html