geerlingguy/raspberry-pi-pcie-devices #344: Test Quectel EC25-A LTE Cat 4 mini PCIe module

geerlingguy, 2021-12-12:
I just bought a [Quectel EC25-A](https://www.quectel.com/product/lte-ec25-mini-pcie-series) LTE mini PCIe module. I have some plans for it—assuming I can get it to work!

[image]

From the spec sheet:

> Quectel EC25 Mini PCIe is a series of LTE category 4 module adopting standard PCI Express® Mini Card form factor (Mini PCIe). It is optimized specially for M2M and IoT applications, and delivers 150 Mbps downlink and 50 Mbps uplink data rates.

Supposedly it has "USB serial drivers for Windows 7/8/8.1/10, Linux, Android" and some other Pi users have reported success with these models (the EU version).

There's a [tutorial for using cellular modems with the Pi 4](https://www.switchdoc.com/2021/05/tutorial-using-cellular-modems-with-the-raspberry-pi-4b/) that I think should apply equally well for the CM4. We'll see. I also have a SIM from SixFab, but I will try to see if I can also get an AT&T SIM working later, since I may have a bead on an unlimited data plan for a 'decent-ish' price, which might be nice for a project I want to do.

TeoTN, 2021-12-16:
Subscribing to this issue, I'd love to have RPi with backup connection over LTE/5G modem and UPS, that'd be crazy useful! Especially when _your ISP is lying_ 😉

geerlingguy, 2021-12-16:
@TeoTN - Note that the data plans for 4G/5G are pretty expensive if your Pi does anything more than a trickle of data... that's something I'm trying to figure out before I get deeper into this project!

pheetr, 2021-12-17:
Plan prices are getting better, depending on where you live, I suppose. In Ireland we have 3 providers claiming to have unlimited 5G plans for €45/month. If you look at their terms a bit closer, two have fair use policies of 750GB and 1TB per month and the third claims to be fully unlimited but reserves the right to restrict based on network conditions. I'm not sure how the latter works in practice, the wired ISPs I've been with haven't even enforced the stated fair use limits so far. :) 
A fourth provider has an unlimited 4G for €25 per month with no fair use policy limit, but also reserves the right to restrict.
So yeah, it's still not a perfect situation, but I'd say 4G/5G are starting to become a viable alternative even for primary access here.

TeoTN, 2021-12-17:
It's similar in Poland with 500GB over 5G available for €26/mo ($29), and since for me this would be only as a backup for short periods of time, I'd be happy even with 50GB :)

geerlingguy, 2022-01-04:
Step 1: Try to use the card on the Turing Pi. Inserted SixFab SIM and card in Mini PCIe slot for node 1 and booted. Neither `lspci` nor `lsusb` return anything. Hmm...

danmanners, 2022-01-04:
Wondering if this may be of any assistance? Sixfab seems to have a ZIP in their repo [here](https://github.com/sixfab/Sixfab_RPi_3G-4G-LTE_Base_Shield) for their Base Shield. Obviously not using that, but wondering if it may still help out on discovery.

https://github.com/sixfab/Sixfab_RPi_3G-4G-LTE_Base_Shield/blob/master/tutorials/QMI_tutorial/src/quectel-CM.zip

geerlingguy, 2022-01-04:
@danmanners - It seems like that presumes a device is appearing in `/dev/tty*` already, which is not happening at least on my board. I wonder if maybe the Turing Pi 2's USB connections to the first PCIe slot might not be correct? I will be ordering a USB to Mini PCIe WWAN adapter and I'll see if it shows up connected through there. I hope I don't have a broken LTE modem!

geerlingguy, 2022-01-04:
Some more debugging steps:

  1. Tried the card in the mPCIe slot for node 2 on the Turing Pi, nothing in `lspci`, `lsusb`, or in `/dev/tty*` either.
  2. Tried installing modem manager (`sudo apt install modemmanager`) on node 1, then running `mmcli -m 0`, but that gives:

```
pi@turing-node-1:~ $ mmcli -m 0
error: couldn't find modem
```

I've ordered both a [USB to mini PCIe WWAN/LTE adapter with SIM tray](https://amzn.to/3mXBQYb), and an [Ableconn PEX-MP117 Mini PCI-E to PCI-E Adapter Card](https://amzn.to/3zoffcu) and will test with both to see if I can get the card recognized via USB or PCIe on another Pi.

geerlingguy, 2022-01-07:
On a Pi 4 model B with the [USB to mini PCIe adapter](https://amzn.to/3mXBQYb), I get the following:

```
pi@lte:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@lte:~ $ ls /dev/tty*
/dev/tty    /dev/tty15  /dev/tty22  /dev/tty3   /dev/tty37  /dev/tty44  /dev/tty51  /dev/tty59  /dev/tty9
/dev/tty0   /dev/tty16  /dev/tty23  /dev/tty30  /dev/tty38  /dev/tty45  /dev/tty52  /dev/tty6   /dev/ttyAMA0
/dev/tty1   /dev/tty17  /dev/tty24  /dev/tty31  /dev/tty39  /dev/tty46  /dev/tty53  /dev/tty60  /dev/ttyprintk
/dev/tty10  /dev/tty18  /dev/tty25  /dev/tty32  /dev/tty4   /dev/tty47  /dev/tty54  /dev/tty61  /dev/ttyUSB0
/dev/tty11  /dev/tty19  /dev/tty26  /dev/tty33  /dev/tty40  /dev/tty48  /dev/tty55  /dev/tty62  /dev/ttyUSB1
/dev/tty12  /dev/tty2   /dev/tty27  /dev/tty34  /dev/tty41  /dev/tty49  /dev/tty56  /dev/tty63  /dev/ttyUSB2
/dev/tty13  /dev/tty20  /dev/tty28  /dev/tty35  /dev/tty42  /dev/tty5   /dev/tty57  /dev/tty7   /dev/ttyUSB3
/dev/tty14  /dev/tty21  /dev/tty29  /dev/tty36  /dev/tty43  /dev/tty50  /dev/tty58  /dev/tty8

pi@lte:~ $ sudo apt install -y modemmanager
...
pi@lte:~ $ sudo mmcli -m 0
error: couldn't find modem
pi@lte:~ $ sudo mmcli -L
No modems were found
```

geerlingguy, 2022-01-07:
Following [SixFab's instructions](https://docs.sixfab.com/page/internet-connection-with-quectel-ec25-by-using-ecm-mode), I communicated with the modem via `minicom -D /dev/ttyUSB2 -b 115200` (after installing `minicom`), and here's the transaction history (the last command triggered a modem reboot, then it came back with `AT` / `OK`:

[image]

geerlingguy, 2022-01-07:
```
pi@lte:~ $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e4:5f:01:45:fe:b2 brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:45:fe:b4 brd ff:ff:ff:ff:ff:ff
    inet 10.0.100.159/24 brd 10.0.100.255 scope global dynamic noprefixroute wlan0
       valid_lft 85700sec preferred_lft 74900sec
    inet6 fe80::4a4c:fd06:5dd4:52e9/64 scope link 
       valid_lft forever preferred_lft forever
5: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether c2:a2:8a:5b:3d:f9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.225.23/24 brd 192.168.225.255 scope global dynamic noprefixroute usb0
       valid_lft 43028sec preferred_lft 37628sec
    inet6 fe80::7ffa:e040:7995:99d8/64 scope link 
       valid_lft forever preferred_lft forever
pi@lte:~ $ ping -I usb0 sixfab.com -c 5
PING sixfab.com (172.66.40.115) from 192.168.225.23 usb0: 56(84) bytes of data.
64 bytes from 172.66.40.115 (172.66.40.115): icmp_seq=1 ttl=49 time=153 ms
64 bytes from 172.66.40.115 (172.66.40.115): icmp_seq=2 ttl=49 time=193 ms
64 bytes from 172.66.40.115 (172.66.40.115): icmp_seq=3 ttl=49 time=152 ms
64 bytes from 172.66.40.115 (172.66.40.115): icmp_seq=4 ttl=49 time=192 ms
64 bytes from 172.66.40.115 (172.66.40.115): icmp_seq=5 ttl=49 time=148 ms

--- sixfab.com ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4002ms
rtt min/avg/max/mdev = 148.419/167.943/193.195/20.305 ms
```

geerlingguy, 2022-01-07:
Testing the speed; first with all interfaces going:

```
pi@lte:~ $ sudo apt-get install speedtest-cli
...
pi@lte:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Amazon.com (3.239.194.27)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by eero (Ashburn, VA) [0.81 km]: 173.036 ms
Testing download speed................................................................................
Download: 8.17 Mbit/s
Testing upload speed......................................................................................................
Upload: 0.57 Mbit/s
```

This was in suboptimal conditions, with a tiny cheap antenna located in my basement :P

Monitored the traffic through the interface (to confirm it was going out over LTE and not the LAN) with `sudo apt install -y iftop` then `sudo iftop -i usb0`:

[image]

geerlingguy, 2022-01-07:
A few other notes concerning routing—SixFab seems to use a mobile network that's CG-NAT'ed behind AWS, as the public IP address in front of the LTE connection is:

```
pi@lte:~ $ curl icanhazip.com
3.239.194.27
```

And my private IP on the LTE modem is `192.168.225.23`.

geerlingguy, 2022-01-07:
Learning more about interacting with the modem via minicom:

```
# Signal strength: https://m2msupport.net/m2msupport/atcsq-signal-quality/
AT+CSQ
+CSQ: 15,99

# Network information: https://m2msupport.net/m2msupport/network-information-automaticmanual-selection/
AT+COPS?
+COPS: 0,0,"AT&T Twilio",7

# Get clock: https://m2msupport.net/m2msupport/m2m-module-diagnostics/
AT+CCLK?
+CCLK: "22/01/07,05:19:00-24"
```

geerlingguy, 2022-01-07:
More exploration: I tried swapping out a separate AT&T SIM from a cradlepoint unit to see what would happen. Even after 10 minutes or so I'm seeing in the logs via minicom:

```
+QLWURC: "pdp active","failed","attm2mglobal"
+QLWURC: "deregister",0

+QLWURC: "pdp active","failed","attm2mglobal"
+QLWURC: "deregister",0

+QLWURC: "pdp active","failed","nxtgenphone"
+QLWURC: "deregister",0

+QLWURC: "pdp active","failed","nxtgenphone"
+QLWURC: "deregister",0

+QLWURC: "pdp active","failed","nxtgenphone"
+QLWURC: "deregister",0
```

Seems like it's not able to connect to LTE towers in my basement on this SIM, not sure if it's the data plan, the signal, or what... but the other SIM said it was on ATT / Twilio, and my cell phone plan is AT&T (and I get 2-3 bars in my basement), so...

geerlingguy, 2022-01-07:
Just as a comparison, testing through a Cradlepoint router in the same location with AT&T (though it has a nicer antenna configuration):

```
pi@lte:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from AT&T Wireless (107.77.206.52)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Xiber LLC (Indianapolis, IN) [262.61 km]: 58.766 ms
Testing download speed................................................................................
Download: 11.70 Mbit/s
Testing upload speed......................................................................................................
Upload: 3.55 Mbit/s
```

geerlingguy, 2022-01-09:
Testing upstairs (instead of downstairs) also using LAN port on the cradlepoint:

```
pi@lte:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from AT&T Wireless (107.77.208.18)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Consolidated Cooperative (Mount Gilead, OH) [426.59 km]: 47.725 ms
Testing download speed................................................................................
Download: 12.07 Mbit/s
Testing upload speed......................................................................................................
Upload: 12.05 Mbit/s
```

geerlingguy, 2022-01-09:
Tested the Quectel with the SixFab SIM upstairs too:

```
pi@lte:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Amazon.com (3.239.194.21)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Rackdog (Ashburn, VA) [0.81 km]: 146.569 ms
Testing download speed................................................................................
Download: 9.15 Mbit/s
Testing upload speed......................................................................................................
Upload: 10.88 Mbit/s
```

And testing signal strength:

```
# Signal strength: https://m2msupport.net/m2msupport/atcsq-signal-quality/
AT+CSQ
+CSQ: 22,99
```

geerlingguy, 2022-01-11:
I swapped in my own iPhone AT&T SIM (though on a limited data plan), and it seems to have connected immediately, no fuss. With a different antenna, I'm getting signal of 16-19, with speedtest result:

```
# Network information: https://m2msupport.net/m2msupport/network-information-automaticmanual-selection/
AT+COPS?
+COPS: 0,0,"AT&T",7

pi@lte:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from AT&T Wireless (107.77.207.227)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Misaka Network, Inc. (Kansas City, MO) [664.74 km]: 58.119 ms
Testing download speed................................................................................
Download: 11.63 Mbit/s
Testing upload speed......................................................................................................
Upload: 5.10 Mbit/s
```

Wish I had a T-Mobile SIM to compare the two.

Doing a speed test with my Mac tethered directly to my phone, with the same SIM, I get:

```
$ speedtest-cli
Retrieving speedtest.net configuration...
Testing from AT&T Wireless (107.77.207.5)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Pavlov Media (Champaign, IL) [203.48 km]: 44.745 ms
Testing download speed................................................................................
Download: 69.11 Mbit/s
Testing upload speed......................................................................................................
Upload: 3.30 Mbit/s
```

Over multiple tests I see anywhere from 45-70 Mbps down, 3-5 Mbps up. So it's interesting the Quectel seems to be able to more consistently put up better upload numbers (and the Cradlepoint blows them both away, getting its 12 Mbps up pretty much every test!).

I currently don't have another CM4 board available to test the SIM tray and 4G card in a mini PCIe slot, but I think it's safe to say the card will work fine with a Pi CM4, assuming the USB data lines in the mini PCIe slot are wired in correctly.

I do want to make sure it works in that configuration on some board, though, before I close out this issue.

danmanners, 2022-01-11:
Since we know the mPCIe lanes work for other devices (at least on my board), is it worth looking at picking up something like this and trying it on slot two? More than happy to send you one if you'd like :)

[image]

https://www.amazon.com/dp/B07T33NQPF?th=1

geerlingguy, 2022-01-11:
@danmanners - My guess is the USB D+/- lanes on slot two are also wired into the BMC's USB hub, therefore I'd hit the same issue on that slot :/

geerlingguy, 2022-01-12:
The Turing Pi 2's slot is now working, thanks to the removal of this tiny resistor: https://twitter.com/geerlingguy/status/1481027568052711425

I also just received another CM4 board with a mini PCIe slot and SIM card tray... so I might test it in there too at some point.

geerlingguy, 2022-01-12:
So new question / problem:

  - I have `eth0` and `usb0` both providing Internet access on different subnets/networks.
  - I would like to have things like `speedtest-cli` route only through the `usb0` interface.
  - I try things like `speedtest-cli --source 192.168.225.57` to force `usb0`, but that doesn't work.

Do I really have to mess with routing tables in Linux to basically 'set a service priority order' like I can do on macOS in the Network system preference pane?

I want to temporarily switch `usb0` to be the primary network interface (but not disable `eth0`, since that's the interface through which I'm locally interacting with the machine).

```
pi@turing-node-1:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.0.100.1      0.0.0.0         UG    202    0        0 eth0
0.0.0.0         192.168.225.1   0.0.0.0         UG    203    0        0 usb0
10.0.100.0      0.0.0.0         255.255.255.0   U     202    0        0 eth0
192.168.225.0   0.0.0.0         255.255.255.0   U     203    0        0 usb0
```

geerlingguy, 2022-01-12:
Ah... trying out `ifmetric` as suggested in [this answer](https://superuser.com/a/1260798):

```
pi@turing-node-1:~ $ sudo apt install ifmetric

pi@turing-node-1:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.0.100.1      0.0.0.0         UG    202    0        0 eth0
0.0.0.0         192.168.225.1   0.0.0.0         UG    203    0        0 usb0
10.0.100.0      0.0.0.0         255.255.255.0   U     202    0        0 eth0
192.168.225.0   0.0.0.0         255.255.255.0   U     203    0        0 usb0
```

That... didn't work.

geerlingguy, 2022-01-12:
So, looks like I may need to do my own route table modifications...

```
pi@turing-node-1:~ $ ip route list
default via 10.0.100.1 dev eth0 proto dhcp src 10.0.100.250 metric 202 
default via 192.168.225.1 dev usb0 proto dhcp src 192.168.225.57 metric 203 mtu 1360 
10.0.100.0/24 dev eth0 proto dhcp scope link src 10.0.100.250 metric 202 
192.168.225.0/24 dev usb0 proto dhcp scope link src 192.168.225.57 metric 203 mtu 1360
```

Both connections are set as default.

geerlingguy, 2022-01-12:
AHA! I found it... after reading through [](https://netbeez.net/blog/linux-set-route-priorities/), I found the key—since Raspberry Pi OS uses `dhcpcd`, I had to set the metric in that configuration.

So, basically:

```
# Edit dhcpcd's configuration.
pi@turing-node-1:~ $ sudo nano /etc/dhcpcd.conf

# Add the following to the end of the file and save it:
interface usb0
metric 0

# Restart dhcpcd.
pi@turing-node-1:~ $ sudo systemctl restart dhcpcd

# Et voila!
pi@turing-node-1:~ $ ip route list
default via 192.168.225.1 dev usb0 proto dhcp src 192.168.225.57 mtu 1360 
default via 10.0.100.1 dev eth0 proto dhcp src 10.0.100.250 metric 202 
10.0.100.0/24 dev eth0 proto dhcp scope link src 10.0.100.250 metric 202 
192.168.225.0/24 dev usb0 proto dhcp scope link src 192.168.225.57 mtu 1360
```

geerlingguy, 2022-01-13:
And testing through the Turing Pi 2:

```
pi@turing-node-1:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Amazon.com (3.239.194.16)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Xiber LLC (Kansas City, MO) [1472.95 km]: 172.994 ms
Testing download speed................................................................................
Download: 8.91 Mbit/s
Testing upload speed......................................................................................................
Upload: 2.78 Mbit/s
```

geerlingguy, 2022-01-13:
Blog post based on findings with routing metric priority: [Network interface routing priority on a Raspberry Pi](https://www.jeffgeerling.com/blog/2022/network-interface-routing-priority-on-raspberry-pi).

geerlingguy, 2022-01-16:
Reopening for a bit—I am wondering if I can eke out more performance. I know I won't hit the 150/50 Mbps performance that's theoretically possible, but I've heard from others ([example](https://twitter.com/AndrewDurrant10/status/1482774109499109376)) who are getting more like 75/25.

Never getting more than 10-12 Mbps makes it seem like maybe something is getting stuck in USB 1.x speeds (which are limited to 12 Mbps).

But looking at the modem with `lsusb` it seems to be in `480M` mode (USB 2.0) at least on a hardware level:

```
pi@turing-node-1:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@turing-node-1:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 2, If 1, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 2, If 2, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 2, If 3, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 2, If 4, Class=Communications, Driver=cdc_ether, 480M
    |__ Port 1: Dev 2, If 5, Class=CDC Data, Driver=cdc_ether, 480M
```

And `ethtool` just shows:

```
pi@turing-node-1:~ $ sudo ethtool usb0
Settings for usb0:
        Current message level: 0x00000007 (7)
                               drv probe link
	Link detected: yes
```

geerlingguy, 2022-01-16:
SixFab has a couple guides for QMI mode, notably:

  - [QMI Interface Internet Connection Setup Using Sixfab Shield/HAT](https://docs.sixfab.com/page/qmi-interface-internet-connection-setup-using-sixfab-shield-hat)
  - [Setting up a data connection over QMI interface using libqmi](https://docs.sixfab.com/page/setting-up-a-data-connection-over-qmi-interface-using-libqmi)

The latter is more useful to me, since I already have the modem in USB mode and just need to turn that off to _hopefully_ get to QMI/wwan mode. We'll see if it's any faster.

geerlingguy, 2022-01-16:
Since `lsusb -t` returns `cdc_ether`, I need to reset the modem in QMI mode:

```
$ minicom -D /dev/ttyUSB2 -b 115200

AT+QCFG="usbnet"  # returns 1
AT+QCFG="usbnet",0  # disables USB mode
AT+CFUN=1,1  # reset the modem (takes a minute or so)
```

geerlingguy, 2022-01-16:
Another attempt:

```
pi@turing-node-1:~ $ sudo ip link set wwan0 down
pi@turing-node-1:~ $ echo 'Y' | sudo tee /sys/class/net/wwan0/qmi/raw_ip
Y
pi@turing-node-1:~ $ sudo ip link set wwan0 up
pi@turing-node-1:~ $ sudo qmicli -d /dev/cdc-wdm0 --dms-set-operating-mode='online'
[16 Jan 2022, 15:59:03] -Warning ** Error reading from istream: Resource temporarily unavailable
error: operation failed: endpoint hangup
error: couldn't release client: Transaction timed out
```

I also noticed `ModemManager` is running:

```
pi@turing-node-1:~ $ ps ax | grep -i modem
    498 ?        Ssl    0:01 /usr/sbin/ModemManager
```

So...

```
$ sudo systemctl stop ModemManager

$ sudo qmicli -d /dev/cdc-wdm0 --dms-set-operating-mode='online'
[/dev/cdc-wdm0] Operating mode set successfully

pi@turing-node-1:~ $ sudo qmicli -d /dev/cdc-wdm0 --wda-get-data-format
error: couldn't create client for the 'wda' service: QMI protocol error (3): 'Internal'
```

Then to connect:

```
pi@turing-node-1:~ $ sudo qmicli -p -d /dev/cdc-wdm0 --device-open-net='net-raw-ip|net-no-qos-header' --wds-start-network="apn='super',ip-type=4" --client-no-release-cid
[/dev/cdc-wdm0] Network started
	Packet data handle: '2267458816'
[/dev/cdc-wdm0] Client ID not released:
	Service: 'wds'
	    CID: '20'
```

Then get an IP address / set up route:

```
pi@turing-node-1:~ $ sudo udhcpc -q -f -i wwan0
udhcpc: started, v1.30.1
No resolv.conf for interface wwan0.udhcpc
udhcpc: sending discover
udhcpc: sending select for 100.80.158.214
udhcpc: lease of 100.80.158.214 obtained, lease time 7200
```

Got an IP:

```
pi@turing-node-1:~ $ ip a show dev wwan0
12: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/none 
    inet 100.80.158.214/30 scope global wwan0
       valid_lft forever preferred_lft forever
```

And ping works:

```
pi@turing-node-1:~ $ ping -I wwan0 -c 5 www.midwesternmac.com
PING www.midwesternmac.com (45.55.136.152) from 100.80.158.214 wwan0: 56(84) bytes of data.
64 bytes from static.midwesternmac.com (45.55.136.152): icmp_seq=1 ttl=29 time=158 ms
64 bytes from static.midwesternmac.com (45.55.136.152): icmp_seq=2 ttl=29 time=118 ms
64 bytes from static.midwesternmac.com (45.55.136.152): icmp_seq=3 ttl=29 time=157 ms
64 bytes from static.midwesternmac.com (45.55.136.152): icmp_seq=4 ttl=29 time=125 ms
64 bytes from static.midwesternmac.com (45.55.136.152): icmp_seq=5 ttl=29 time=156 ms

--- www.midwesternmac.com ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4003ms
rtt min/avg/max/mdev = 118.345/142.730/158.297/17.485 ms
```

geerlingguy, 2022-01-16:
After [changing the interface routing priority](https://www.jeffgeerling.com/blog/2022/network-interface-routing-priority-on-raspberry-pi), I ran `speedtest-cli` from the basement (non-ideal, but just for testing):

```
pi@turing-node-1:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Amazon.com (3.239.194.10)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Rackdog (Ashburn, VA) [0.81 km]: 226.382 ms
Testing download speed................................................................................
Download: 8.87 Mbit/s
Testing upload speed......................................................................................................
Upload: 2.67 Mbit/s
```

That's faster than the earlier test in the exact same location, so now I need to bring the setup upstairs and see how it fares up there, with better antenna placement.

geerlingguy, 2022-01-16:
To temporarily disable the link, `sudo ip link set wwan0 down` — I need to figure out what `udhcpcd` does—it seems like I can't tweak normal dhcpcd.conf options on that interface since I presume `udhcpcd` kinda controls the device?

geerlingguy, 2022-01-20:
Startup configuration is basically:

```
# Create the interface file.
pi@lte:~ $ sudo nano /etc/network/interfaces.d/wwan0

# Paste in the following contents (replacing with your own APN).
iface wwan0 inet manual
     pre-up ifconfig wwan0 down
     pre-up echo Y > /sys/class/net/wwan0/qmi/raw_ip
     pre-up for _ in $(seq 1 10); do /usr/bin/test -c /dev/cdc-wdm0 && break; /bin/sleep 1; done
     pre-up for _ in $(seq 1 10); do /usr/bin/qmicli -d /dev/cdc-wdm0 --nas-get-signal-strength && break; /bin/sleep 1; done
     pre-up sudo qmicli -p -d /dev/cdc-wdm0 --device-open-net='net-raw-ip|net-no-qos-header' --wds-start-network="apn='nxtgenphone',ip-type=4" --client-no-release-cid
     pre-up udhcpc -i wwan0
     post-down /usr/bin/qmi-network /dev/cdc-wdm0 stop
```

Add the line `auto wwan0` above that configuration block to have the interface brought up automatically during system boot. Otherwise you can run `sudo ifup wwan0` to bring it up, and `sudo ifdown wwan0` to bring it down.

geerlingguy, 2022-01-20:
Running four speedtests upstairs in QMI mode on my AT&T SIM:

- 19.95 Mbps down (avg)
- 0.36 Mbps up (avg)

geerlingguy, 2022-01-20:
And after resetting the modem into ECM mode on the same SIM in the same location, here are four more speedtests:

  - 19.74 Mbps down (avg)
  - 15.65 Mbps up (avg)
  - 44ms ping (avg)

geerlingguy, 2022-01-20:
Posted a bit of a compendium: [Using 4G LTE wireless modems on a Raspberry Pi](https://www.jeffgeerling.com/blog/2022/using-4g-lte-wireless-modems-on-raspberry-pi).

joyshmitz, 2022-01-23:
could not leave a comment on the blog where you solve the problem of how to contact the device without a permanent address
I used to suffer from the cost of a dedicated IP address, but later I found a great, and most importantly free tool - `zerotire`. Delighted with the opportunity, I picked up the always free server in the Oracle cloud to distribute IP addresses under my own control, and then dropped it, because 50 free addresses are more than enough to test any hardware, as you will have time - try. Everything is put in a minute, `curl -s https://install.zerotier.com | sudo bash`, and configured for another minute. The management site is easy to carry https://my.zerotire.com 
Just forgot what it is like to configure ddns or other remote access tools.

Firefishy, 2022-02-24:
The Quectel EC25-A modem also supports a PPP based mode, this allows using the "external" IP without involving NAT (on the modem). The PPP session is terminated on the modem. Basic tutorial: https://wiki.teltonika-networks.com/view/Setting_up_mobile_interface_on_raspberry_pi

paulwratt, 2022-02-24:
@geerlingguy looking at what you did to get `metric` to work, and knowing its like setting the SWAP Priority, you should just be able to `down` the interface, and change the `metric` then (or _maybe_ bring it `up` with the new `metric`).

(without testing it) its set at creation time (which is why it works as a DHCP setting), but while it is "unusable" (ie. `down`) you should also be able to change various parts of the interface, incl. the `metric`.

Also `metric 0` means "no metric", as opposed to "metric == 0", so you wont get any `metric` output on any queries (like `route`) - use `metric 1` to maintain its output in any queries, and still get the same priority over-ride.

Nice work with the various testing and getting it working too BTW.

NOTE: the only reason you _might_ want to use routing is if you want all HTTP/S traffic to go _in and out_ over one interface, _OR_ you want _output traffic_ to go over a different interface than _input traffic_ (but this setup _might_ have to be on the same sub-net due to upstream return paths) - an experiment for another day ..

thecokeguy, 2023-04-05:
> The Turing Pi 2's slot is now working, thanks to the removal of this tiny resistor: https://twitter.com/geerlingguy/status/1481027568052711425

Could you share which one you removed?

geerlingguy, 2024-04-24:
This one:

[image]

Kasperror, 2025-02-19:
This test could be reopened since it's "outdated" now - new Pi OS Bookworm primarily uses `NetworkManager` so setting up Pi to work with both eth0 and eth1 (EC25 is now assigned this interface) interchangeably is a nightmare and requires some more digging. I'm happy to help if such thread/issue is to be opened.

geerlingguy, 2025-02-19:
Reopened!

Kasperror, 2025-02-20:
To start off, in following days I will try to present two separate approaches to connecting Bookworm-running Pi to a network using EC25 modem

1. ECM - much simpler to implement, much harder to maintain as reliable multi-interface connection
2. QMI - tricky to understand but much easier to use after successful setup

I will present solutions based on mini-PCIe EC25 connected to `reTerminal DM` (marketed to me by [JeffGeerling yt channel](https://www.youtube.com/@JeffGeerling)) which we now use as a kiosk-a-like solution BUT for all those using USB LTE EC25 dongle  setup should be just the same as with mini-PCIe modem because it works as USB device. I will welcome any tips on how I can improve my setup and I'm happy to help with any questions

aklein24, 2025-03-02:
@Kasperror 
I started my journey with Raspberry Pi modems about 12 months ago using Jeff's blog post ( https://www.jeffgeerling.com/blog/2022/using-4g-lte-wireless-modems-on-raspberry-pi ) as my starting point. Like Jeff says in that article, there's a whole lot of how to guides and not much in the way of how to troubleshoot. At that point in time I had a lot of trouble and was only able to get my modems (quectel EC25 and BG96) to start reliably using USB mode. 

Recently found this article: https://andino.systems/andino-4g-modem/setup-via-modem-manager 

It was a lot easier and more straightforward, using NetworkManager like you (@Kasperror) mentioned above.

These are the commands I ran in this order:
```
sudo apt install modemmanager libqmi-utils network-manager
sudo mmcli -m 0 --simple-connect="apn=my custom apn"
sudo mmcli -m 0 --enable --timeout=180
sudo systemctl enable NetworkManager
sudo systemctl start NetworkManager
sudo systemctl enable ModemManager
sudo systemctl start ModemManager
```

It works great for both the EC25 and BG96 in QMI mode. It's just a few steps and worked perfectly without having to update modes or anything. I'm using these on a reComputer R1000 which is similar to the the reTerminal you're using. 

Also: beware of bad modem hardware. I had one modem that went bad and it was hard to tell. It would respond to AT commands but never connect. So having a spare on hand was a lifesaver. Good luck!

commit bc8ff9bd (2022-01-13): Fixes #344: Quectel EC25-A LTE card works great.