geerlingguy/raspberry-pi-pcie-devices #345: Add blicube's bliKVM CM4 IO Board for Pi-KVM

geerlingguy, 2021-12-15:
blicube is making a [bliKVM CM4 IO Board for Pi-KVM](https://www.aliexpress.com/item/1005003262886521.html?spm=a2g0o.store_pc_allProduct.8148356.5.39cd62bejpZaWF).

[image]

[image]

[image]

The bliKVM integrates seemingly everything from the PiKVM v3 HAT into one compact package that fits in an even smaller space, and doesn't require as complicated an assembly process. Currently the board is priced at $80, the case at $25, or a combo pack with case, fan, OLED display, and board at $100. Pretty good deal!

geerlingguy, 2022-01-13:
It's on the site!

closed 2022-01-13 by geerlingguy

bryanvaz, 2022-01-25:
@geerlingguy any plans on testing this?

I'm eyeing this very much. Apparently the next version will have POE. I also left them a suggestion to make it rack-friendly (1U and all ports on one side).

Also are you aware of any other PiKVM boards supporting the CM4? I couldn't find any with a web search, but maybe another developer dropped you a note?

Cheers,
Bryan

geerlingguy, 2022-01-26:
So far nothing else for Pi-KVM itself, but I do have a couple units of BliKVM and will be testing them soon—I even give the thing a shout out at the end of the video I'll be posting tomorrow morning at 9 on my YouTube channel ;)

geerlingguy, 2022-01-30:
FYI here's a link to a github repo for the board: https://github.com/ThomasVon2021/pikvm-CM4-Board

PhilipTS, 2022-02-03:
> So far nothing else for Pi-KVM itself, but I do have a couple units of BliKVM and will be testing them soon—I even give the thing a shout out at the end of the video I'll be posting tomorrow morning at 9 on my YouTube channel ;)

@geerlingguy waiting patiently for your video (or should I say impatiently)

geerlingguy, 2022-02-03:
Heh... I have it all put together and was going to record something today but the snowstorm through the midwest may delay that :(

PhilipTS, 2022-02-03:
@geerlingguy well I am a very impatient guy so I pressed the BUY button. So now I can patiently wait for you video :)

acarbonaro, 2022-02-17:
Can you use a pair of these devices to control each other?  I could see this being useful for folks who run redundant Pi-Hole and/or PiVPN installs.  It would be nice to be able to recover (or at least diagnose) an instance from the other.

paulwratt, 2022-02-21:
> @geerlingguy well I am a very impatient guy so I pressed the BUY button. So now I can patiently wait for you video :)

Can't remember if I heard the _bliKVM_ mentioned in the video, but I decided "5 left" was enough of an incentive (ooh my 1st CM4)

geerlingguy, 2022-02-21:
Forgot to mention, the video is up on the Geerling Engineering channel: https://www.youtube.com/watch?v=3OPd7svT3bE

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/blikube-blikvm-cm4-io-board.html