geerlingguy/raspberry-pi-pcie-devices #357: Testing Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon] - missing sas2ircu for arm

mi-hol, 2022-01-11:
Inspired by Jeff's post on HW RAID I'm testing a Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon] HBA.
Drivers for card are included in OS but I couldn't find the CLI for RAID configuration (sas2ircu) for arm64/aarch64.

Any hints how to configure it on a RPi?

`
pi@nas64-6:~ $ sudo lspci -vk
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
        Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@1,0
        Flags: fast devsel
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00000000-00000fff [size=4K]
        Memory behind bridge: c0000000-c00fffff [size=1M]
        Prefetchable memory behind bridge: [disabled]
        Capabilities: [48] Power Management version 3
        Capabilities: [ac] Express Root Port (Slot-), MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
        Capabilities: [240] L1 PM Substates

01:00.0 Serial Attached SCSI controller: Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon] (rev 03)
        Subsystem: Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon]
        Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@1,0/usb@1,0
        Flags: fast devsel
        I/O ports at <unassigned> [disabled]
        Memory at 6000c0000 (64-bit, non-prefetchable) [disabled] [size=16K]
        Memory at 600080000 (64-bit, non-prefetchable) [disabled] [size=256K]
        Expansion ROM at 600000000 [virtual] [disabled] [size=512K]
        Capabilities: [50] Power Management version 3
        Capabilities: [68] Express Endpoint, MSI 00
        Capabilities: [d0] Vital Product Data
        Capabilities: [a8] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable- Count=15 Masked-
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [138] Power Budgeting <?>
        Capabilities: [150] Single Root I/O Virtualization (SR-IOV)
        Capabilities: [190] Alternative Routing-ID Interpretation (ARI)

pi@nas64-6:~ $dmesg | grep mpt
[    0.000000]   Normal   empty
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
pi@nas64-6:~ $
`

mi-hol, 2022-01-19:
@DanaGoyette in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18#issuecomment-955112870 you mentioned   "my SAS2008 died" die you flash it to IT mode?

mi-hol, 2022-01-19:
@guibom @PBXForums researching a bit about experience with SAS2008  I found your posts.
In summary it seems that no SAS2008 card worked on a CM4.
Is this correct understanding?
 
@guibom do you perhaps remember what kernel version you used in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18#issuecomment-907697637 ?
kernel 5.10.63 shows more capabilities for the card and  was most likely used by PBXForums post https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18#issuecomment-756940366 too.
The differences are:
```
Capabilities: [150] Single Root I/O Virtualization (SR-IOV)
Capabilities: [190] Alternative Routing-ID Interpretation (ARI)
```

DanaGoyette, 2022-01-27:
> @DanaGoyette in [#18 (comment)](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18#issuecomment-955112870) you mentioned "my SAS2008 died" did you flash it to IT mode?

I had the 2308 in IT mode, but after I used it in a secondhand Supermicro X10 board, the card failed to the point where none of the LEDs would light up.  It was definitely the fuse, because bridging it with solder made it work again, but the solder bridge gets warm enough that I don't dare keep using it.  That Supermicro board itself later failed with a short-circuit near the ATX power connector!

Speaking of SR-IOV, I wonder if the 2008 is just lying about supporting it, due to a bug?  I was never able to find any documentation on how to enable SR-IOV on it, and I couldn't find any articles about VF drivers for it either.  With ESXi, you can try to enable SR-IOV on the 2008, but after rebooting to enable it, it just tells you again to reboot to enable it.  The 2308 doesn't seem to claim SR-IOV support.

Given the stock shortage of Pi4s, it's probably easier for somebody who already has the CM4 to test a 2308.  I have a Home Assistant Yellow on order, so I can try with its M.2 slot, but I don't know when that's supposed to ship.

closed 2023-08-19 by mi-hol