geerlingguy/raspberry-pi-pcie-devices #364: Test Innodisk EGPL-T101 M.2 10 GbE adapter

geerlingguy, 2022-01-19:
The [Innodisk EGPL-T101](https://www.innodisk.com/en/products/embedded-peripheral/communication/egpl-t101) is a 10 GbE M.2 adapter in a tiny form factor. It has an Aquantia chip inside, and should be well supported under Linux.

TODO: Image here.

I ordered one, would like to see how it performs. Could be a quick way to add 10 Gigabit Ethernet to a Pi project, in a very tiny form factor (would only get 3-4 Gbps on it, of course).

ZhenmingYu, 2022-03-02:
Where can you order one of these?

geerlingguy, 2022-03-02:
I bought mine from here: https://www.wdlsystems.com/Innodisk-EGPL-T101-M.2-2280-to-single-10GbE-LAN-Module-0C-to-55C

ZhenmingYu, 2022-03-03:
> I bought mine from here: https://www.wdlsystems.com/Innodisk-EGPL-T101-M.2-2280-to-single-10GbE-LAN-Module-0C-to-55C

Thanks a lot!

i0ntempest, 2022-09-25:
Could you please tell me what chipset it has? I see it supports 10Mbps up to 10Gbps so probably an AQC113? I would love to put one of these into my tiny hackintosh server.

ZhenmingYu, 2022-09-25:
> Could you please tell me what chipset it has? I see it supports 10Mbps up to 10Gbps so probably an AQC113? I would love to put one of these into my tiny hackintosh server. Also the shipping price to Canada on that wdlsystems website is outrageous (nearly the same as the card itself!)

Yeah it's an AQC113

i0ntempest, 2022-09-25:
> Yeah it's an AQC113
Awesome then it's gonna work with a bit of kext patching. Problem tho is how to get one in Canada or China for reasonable price...

ZhenmingYu, 2022-09-25:
> 

You found an supplier in China with this thing available? I tried out at Taobao and Alibaba, but no luck there.

i0ntempest, 2022-09-25:
> You found an supplier in China with this thing available? I tried out at Taobao and Alibaba, but no luck there.

Not yet. I found [this](https://www.anewtech.net/products/flash-storage/embedded-peripheral/dram-module/embedded-peripheral/communication/id-egpl-t101) which they say they are based in Singapore, so it might be possible for them to ship to China relatively cheap. I haven't asked yet, and right now waiting for reply from Innodisk and wdlsystems.

i0ntempest, 2022-09-26:
Got a reply from WDS! Their automatic shipping estimate is indeed bogus and I was told it will take just $50 for shipping to Canada. So anyone looking into buying there should request a quote before pulling the trigger.

ZhenmingYu, 2022-09-27:
> Got a reply from WDL! Their automatic shipping estimate is indeed bogus and I was told it will take just $50 for shipping to Canada. So anyone looking into buying there should request a quote before pulling the trigger. EDIT: And later got a reply from Innodisk themself, pointing me to WDL as their NA distributor. So most likely that's where I'll buy it.

Mostly the same story for me. Innodisk point me to their EU distributor, who later told me that it's not available due to chip shortage.

i0ntempest, 2022-10-04:
Just got it today! Total cost is 174 USD+50 USD shipping+42.35 CAD import tax, which is really high for a 10GbE card. That's SFF tax I guess. It BARELY fits under the cover of my mini hack, I would not be able to screw the cover back on if it was another .5 mm higher. Works fine with a bit of kext patching and property injection, except this [airplay problem](https://www.insanelymac.com/forum/topic/330614-marvell-aquantia-10-gb-ethernet-support-thread/?do=findComment&comment=2793866) which I think is present on all AQC113 chip with ID 14c0 (tried Gigabyte GC-AQC113C as well).
[image]
[image]
[image]
[image]
[image]

TinkerTurtle, 2023-02-12:
I like these logs in GH issues!

I installed the card about a month ago and experienced a network dropout today. The NIC reported an overheat event in Windows Event Viewer and required a reboot to recover.

I noticed the heatsink was a bit loose and the stock thermal pad (it's 1mm thick) wasn't making good contact with the chip. I replaced it with some thermal putty that I had on hand, and hope it helps. I didn't touch the thermal pad on the rear as it looked OK.

If it still overheats I might try installing a thicker thermal pad.

chocmake, 2024-03-09:
> Trying to see if they'll do USPS, 160-180 USD for shipping to Canada is outrageous, it should be 25$.

Thought I'd add to this that since late 2023 there are now Chinese sellers on Alibaba and eBay selling Aquantia AQC107 based equivalents (older NIC chipset) a lot more affordably. Two variants I've seen: one using ribbon cable connects and another using a similar corded style connect as the InnoDisk.

geerlingguy, 2024-11-22:
Finally plugged it into a Pi (with PCIe Gen 3, no less!).

```
0000:01:00.0 Ethernet controller: Aquantia Corp. AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig] (rev 03)
	Subsystem: Innodisk Corporation AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 1b80400000 (64-bit, non-prefetchable) [disabled] [size=512K]
	Region 2: Memory at 1b804a0000 (64-bit, non-prefetchable) [disabled] [size=4K]
	Region 4: Memory at 1b80000000 (64-bit, non-prefetchable) [disabled] [size=4M]
	Expansion ROM at 1b80480000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/32 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 512 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x2, ASPM not supported
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: Upstream Port
	Capabilities: [b0] MSI-X: Enable- Count=32 Masked-
		Vector table: BAR=2 offset=00000000
		PBA: BAR=2 offset=00000200
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed+ WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [168 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [178 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [198 v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [1bc v1] Lane Margining at the Receiver <?>
	Capabilities: [1d4 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [1dc v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=14us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [1ec v1] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
	Capabilities: [2ec v1] Data Link Feature <?>
	Capabilities: [2f8 v1] Precision Time Measurement
		PTMCap: Requester:+ Responder:- Root:-
		PTMClockGranularity: Unimplemented
		PTMControl: Enabled:- RootSelected:-
		PTMEffectiveGranularity: Unknown
	Capabilities: [304 v1] Vendor Specific Information: ID=0003 Rev=1 Len=054 <?>
```

I built a custom Linux kernel following Raspberry Pi's instructions on their 6.6.y branch, with the following driver selected in `make menuconfig`:

```
Device Drivers
  > Network device support
    > PHY Device support and infrastructure
      > Aquantia PHYs
```

geerlingguy, 2024-11-22:
Works now:

```
$ dmesg | grep atlantic
[    4.342036] atlantic 0000:01:00.0: enabling device (0000 -> 0002)
[    4.379197] atlantic: Detect ATL2FW 1030013
[   13.660108] atlantic 0000:01:00.0 eth1: atlantic: link change old 0 new 10000
```

ethtool output:

```
$ sudo ethtool eth1
Settings for eth1:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Full
	                        100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Full
	                        100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	                                     10000baseT/Full
	                                     2500baseT/Full
	                                     5000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: No
	Link partner advertised FEC modes: Not reported
	Speed: 10000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: pg
	Wake-on: g
        Current message level: 0x00000005 (5)
                               drv link
	Link detected: yes
```

geerlingguy, 2024-11-22:
Some performance tests:

```
$ iperf3 -c 10.0.2.15
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.228 port 52404 connected to 10.0.2.15 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   719 MBytes  6.03 Gbits/sec    0   1.75 MBytes       
[  5]   1.00-2.00   sec   719 MBytes  6.03 Gbits/sec    0   1.88 MBytes       
[  5]   2.00-3.00   sec   721 MBytes  6.05 Gbits/sec    0   1.99 MBytes       
[  5]   3.00-4.00   sec   721 MBytes  6.05 Gbits/sec    0   1.99 MBytes       
[  5]   4.00-5.00   sec   715 MBytes  6.00 Gbits/sec    0   2.67 MBytes       
[  5]   5.00-6.00   sec   719 MBytes  6.03 Gbits/sec    0   2.82 MBytes       
[  5]   6.00-7.00   sec   721 MBytes  6.05 Gbits/sec    0   2.98 MBytes       
[  5]   7.00-8.00   sec   720 MBytes  6.04 Gbits/sec    0   3.16 MBytes       
[  5]   8.00-9.00   sec   721 MBytes  6.05 Gbits/sec    0   3.16 MBytes       
[  5]   9.00-10.00  sec   720 MBytes  6.04 Gbits/sec    0   3.32 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  7.03 GBytes  6.04 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  7.03 GBytes  6.03 Gbits/sec                  receiver

iperf Done.

$ iperf3 -c 10.0.2.15 --reverse
Connecting to host 10.0.2.15, port 5201
Reverse mode, remote host 10.0.2.15 is sending
[  5] local 10.0.2.228 port 39524 connected to 10.0.2.15 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   771 MBytes  6.47 Gbits/sec                  
[  5]   1.00-2.00   sec   775 MBytes  6.50 Gbits/sec                  
[  5]   2.00-3.00   sec   794 MBytes  6.66 Gbits/sec                  
[  5]   3.00-4.00   sec   796 MBytes  6.67 Gbits/sec                  
[  5]   4.00-5.00   sec   794 MBytes  6.66 Gbits/sec                  
[  5]   5.00-6.00   sec   784 MBytes  6.57 Gbits/sec                  
[  5]   6.00-7.00   sec   786 MBytes  6.59 Gbits/sec                  
[  5]   7.00-8.00   sec   795 MBytes  6.67 Gbits/sec                  
[  5]   8.00-9.00   sec   804 MBytes  6.75 Gbits/sec                  
[  5]   9.00-10.00  sec   792 MBytes  6.64 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  7.71 GBytes  6.62 Gbits/sec                  sender
[  5]   0.00-10.00  sec  7.70 GBytes  6.62 Gbits/sec                  receiver

iperf Done.

$ iperf3 -c 10.0.2.15 --bidir
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.228 port 33940 connected to 10.0.2.15 port 5201
[  7] local 10.0.2.228 port 33956 connected to 10.0.2.15 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   626 MBytes  5.25 Gbits/sec    0   1.73 MBytes       
[  7][RX-C]   0.00-1.00   sec   562 MBytes  4.71 Gbits/sec                  
[  5][TX-C]   1.00-2.00   sec   608 MBytes  5.09 Gbits/sec    0   1.84 MBytes       
[  7][RX-C]   1.00-2.00   sec   594 MBytes  4.98 Gbits/sec                  
[  5][TX-C]   2.00-3.00   sec   629 MBytes  5.28 Gbits/sec    0   2.67 MBytes       
[  7][RX-C]   2.00-3.00   sec   550 MBytes  4.62 Gbits/sec                  
[  5][TX-C]   3.00-4.00   sec   618 MBytes  5.18 Gbits/sec    0   2.97 MBytes       
[  7][RX-C]   3.00-4.00   sec   593 MBytes  4.97 Gbits/sec                  
[  5][TX-C]   4.00-5.00   sec   606 MBytes  5.09 Gbits/sec    0   3.13 MBytes       
[  7][RX-C]   4.00-5.00   sec   587 MBytes  4.92 Gbits/sec                  
[  5][TX-C]   5.00-6.00   sec   606 MBytes  5.09 Gbits/sec    0   3.13 MBytes       
[  7][RX-C]   5.00-6.00   sec   594 MBytes  4.98 Gbits/sec                  
[  5][TX-C]   6.00-7.00   sec   602 MBytes  5.05 Gbits/sec    0   3.13 MBytes       
[  7][RX-C]   6.00-7.00   sec   582 MBytes  4.88 Gbits/sec                  
[  5][TX-C]   7.00-8.00   sec   589 MBytes  4.94 Gbits/sec    0   3.64 MBytes       
[  7][RX-C]   7.00-8.00   sec   577 MBytes  4.84 Gbits/sec                  
[  5][TX-C]   8.00-9.00   sec   595 MBytes  4.99 Gbits/sec    0   3.64 MBytes       
[  7][RX-C]   8.00-9.00   sec   594 MBytes  4.98 Gbits/sec                  
[  5][TX-C]   9.00-10.00  sec   592 MBytes  4.97 Gbits/sec    0   3.82 MBytes       
[  7][RX-C]   9.00-10.00  sec   593 MBytes  4.98 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  5.93 GBytes  5.09 Gbits/sec    0             sender
[  5][TX-C]   0.00-10.00  sec  5.93 GBytes  5.09 Gbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  5.69 GBytes  4.89 Gbits/sec                  sender
[  7][RX-C]   0.00-10.00  sec  5.69 GBytes  4.89 Gbits/sec                  receiver
```

geerlingguy, 2024-11-22:
This adapter handles 5+ Gbps bidirectional no problem. It tops out around 6 Gbps, could maybe eke out more with jumbo frames, but it's good to know it works, and works better (at least currently) than the R8126 5 Gbps NIC from Realtek at 5 Gbps (the one only got 400-500 Mbps in reverse, see #675).

geerlingguy, 2024-11-22:
The card is up on the site here: https://pipci.jeffgeerling.com/cards_network/innodisk-egpl-t101.html

closed 2024-11-22 by geerlingguy