geerlingguy/raspberry-pi-pcie-devices #365: Add UUGear PiGear Nano

geerlingguy, 2022-01-20:
UUGear's PiGear Nano (link TBD) is a CM4 carrier board that is basically the king of USB.

[image]

[image]

[image]

In case it's not apparent in the pictures, there's not one but _three_ VLI USB 3 hubs on this board. One is a VLI VL805 like is found on the Raspberry Pi 4 model B:

```
pi@pigear-nano:~ $ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)
```

And then there are two additional VLI VL817 chips (USB 3.1 gen 1) that drive all those USB 3 ports on the rear. It also has an M.2 NVMe slot, though that is bridged through a RTL9210 M.2 NVMe to USB adapter. And it has a mini PCIe slot on the topside suitable for 4G LTE modules (with a SIM tray on the underside).

And there's a bevy of IO through a pluggable header, along with a red, yellow, and green LED stack, a gigabit Ethernet port, a USB-C power input, a 2-pin pluggable power input, and a few other bits and bobs that I'm still sorting out.

It has an HDMI port, power and reset buttons, and a microSD card slot, so you can boot Lite CM4 modules from it, as well as a buzzer, a CAM0 and DISP0 CSI and DSI port, and what I'm guessing is the same 14-pin header as is on the CM4 IO board.

geerlingguy, 2022-01-20:
A few quick notes on first bringup:

  - USB-C power is not adequate to power the whole board and a Samsung 970 EVO Plus; halfway through boot the system just dies as it tries to initialize the NVMe drive.
  - UUGear is currently finishing up their launch documentation. I received permission from them to work on adding the board to this site, so I figured I'd start doing so. I'll be covering the board more in my next video on YouTube next week.
  - The board has LEDs for each of the USB ports, to show if a device is connected. Nice.

geerlingguy, 2022-01-20:
Default USB layout with nothing plugged in:

```
pi@pigear-nano:~ $ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
```

That NVMe adapter being listed as 480M seems a bit weird though...

```
Bus 001 Device 003: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210 M.2 NVME Adapter
```

geerlingguy, 2022-01-20:
I plugged in all the USB flash drives I owned, and:

```
pi@pigear-nano:~ $ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 3: Dev 9, If 0, Class=Mass Storage, Driver=uas, 5000M
        |__ Port 4: Dev 10, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 8, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 3: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
```

Although, after about 10 seconds having them all plugged in, the board powered down, likely because that's just _too darn much_ power.

geerlingguy, 2022-01-20:
Also testing some other onboard features. There's a buzzer attached to GPIO 25, so:

```
pi@pigear-nano:~ $ cat buzz.py 
from gpiozero import LED
from time import sleep

buzz = LED(25)

while True:
    buzz.on()
    sleep(1)
    buzz.off()
    sleep(1)

pi@pigear-nano:~ $ python3 buzz.py 
^C
```

Results in BEEEEEP! BEEEEEP!

geerlingguy, 2022-01-20:
I switched to using a 12V 5A power adapter using the barrel plug to terminal adapter it came with:

[image]

And now my NVMe drive shows up correctly (thankfully, as a 5000M superspeed device):

```
pi@pigear-nano:~ $ lsusb -t
...
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M

pi@pigear-nano:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 232.9G  0 disk 
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 118.8G  0 part /
```

geerlingguy, 2022-01-20:
I wanted to test NVMe boot (via USB of course), so I ran:

```
# Install rpi-clone
git clone https://github.com/billw2/rpi-clone.git
sudo cp rpi-clone/rpi-clone-setup /usr/local/sbin

# Set hostname
sudo rpi-clone-setup -t pigear-nano

# Do the clone
sudo rpi-clone sda
(this takes a few minutes)
```

geerlingguy, 2022-01-20:
After pulling the microSD card and a reboot, I got:

```
$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 232.9G  0 disk 
├─sda1   8:1    0   256M  0 part /boot
└─sda2   8:2    0 232.6G  0 part /
```

Though the NVMe over USB boot is taking quite some time as it seemed the clone must've confused Pi OS a bit. I see a few pauses like:

```
[    3.097725] usb 1-1.4: Manufacturer: VIA Labs, Inc.         
[    3.102759] hub 1-1.4:1.0: USB hub found
[    3.106288] hub 1-1.4:1.0: 4 ports detected
[   33.758715] vcc-sd: disabling
[   33.770701] sd 0:0:0:0: [sda] tag#3 uas_eh_abort_handler 0 uas-tag 1 inflight: IN 
[   33.773757] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x28 28 00 09 0e 5c a0 00 00 28 00
[   33.790706] scsi host0: uas_eh_device_reset_handler start
[   33.924265] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   34.379202] usb 2-1: Enable of device-initiated U1 failed.
[   34.383438] usb 2-1: Enable of device-initiated U2 failed.
[   34.389561] scsi host0: uas_eh_device_reset_handler success

...

[   35.593486] systemd[1]: Started Journal Service.
[   35.749797] EXT4-fs (sda2): re-mounted. Opts: (null)
[   35.806898] systemd-journald[142]: Received client request to flush runtime journal.
[   37.934716] random: crng init done
[   37.934738] random: 7 urandom warning(s) missed due to ratelimiting
[   66.026707] sd 0:0:0:0: [sda] tag#25 uas_eh_abort_handler 0 uas-tag 1 inflight: IN 
[   66.026733] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x28 28 00 08 8e 5b 60 00 00 48 00
[   66.042714] scsi host0: uas_eh_device_reset_handler start
[   66.172199] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   66.618443] usb 2-1: Enable of device-initiated U1 failed.
[   66.619803] usb 2-1: Enable of device-initiated U2 failed.
[   66.622978] scsi host0: uas_eh_device_reset_handler success

...

[   99.162732] scsi host0: uas_eh_device_reset_handler start
[   99.292214] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   99.737672] usb 2-1: Enable of device-initiated U1 failed.
[   99.739059] usb 2-1: Enable of device-initiated U2 failed.
[   99.742286] scsi host0: uas_eh_device_reset_handler success
[   99.770848] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   99.773810] bcm2835_audio bcm2835_audio: there is not valid maps for state default
[   99.780663] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 0
[  129.838713] sd 0:0:0:0: [sda] tag#16 uas_eh_abort_handler 0 uas-tag 1 inflight: IN 
[  129.838735] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 08 d0 87 78 00 01 08 00
[  129.854730] scsi host0: uas_eh_device_reset_handler start
[  129.855086] sd 0:0:0:0: [sda] tag#19 uas_zap_pending 0 uas-tag 4 inflight: 
[  129.855103] sd 0:0:0:0: [sda] tag#19 CDB: opcode=0x28 28 00 08 93 85 80 00 00 88 00
[  129.984121] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  130.436958] usb 2-1: Enable of device-initiated U1 failed.
[  130.438328] usb 2-1: Enable of device-initiated U2 failed.
[  130.441609] scsi host0: uas_eh_device_reset_handler success
```

geerlingguy, 2022-01-20:
Running my [disk-benchmark.sh](https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/setup/benchmarks/disk-benchmark.sh) script:

```
fio 1M sequential: 368 MB/sec
iozone 1M seq read: 321 MB/sec
iozone 1M seq write: 294 MB/sec
iozone 4K rand read: 21.27 MB/sec
iozone 4K rand write: 34.64 MB/sec
```

geerlingguy, 2022-01-20:
lol

```
pi@pigear-nano:~ $ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 9, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 2: Dev 8, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 3: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 4: Dev 10, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 1: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
```

And

```
pi@pigear-nano:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sdb           8:16   1 114.6G  0 disk 
sdc           8:32   1  28.7G  0 disk 
sdd           8:48   1  28.7G  0 disk 
sde           8:64   1  14.3G  0 disk 
sdf           8:80   1  14.3G  0 disk 
sdg           8:96   1  28.7G  0 disk 
sdh           8:112  1  28.7G  0 disk 
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 118.8G  0 part /
```

geerlingguy, 2022-01-20:
[image]

heh... I set up a RAID 0 array with mdadm (following my [mdadm guide here](https://www.jeffgeerling.com/blog/2021/htgwa-create-raid-array-linux-mdadm)):

```
pi@pigear-nano:~ $ sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=7 /dev/sd[b-h]1
mdadm: chunk size defaults to 512K
mdadm: Defaulting to version 1.2 metadata
mdadm: array /dev/md0 started.

pi@pigear-nano:~ $ sudo mdadm --detail /dev/md0
/dev/md0:
           Version : 1.2
     Creation Time : Thu Jan 20 17:41:11 2022
        Raid Level : raid0
        Array Size : 269908480 (257.40 GiB 276.39 GB)
      Raid Devices : 7
     Total Devices : 7
       Persistence : Superblock is persistent

       Update Time : Thu Jan 20 17:41:11 2022
             State : clean 
    Active Devices : 7
   Working Devices : 7
    Failed Devices : 0
     Spare Devices : 0

            Layout : original
        Chunk Size : 512K

Consistency Policy : none

              Name : pigear-nano:0  (local to host pigear-nano)
              UUID : 7f6dcd01:9c16e597:50bfc071:a701fad1
            Events : 0

    Number   Major   Minor   RaidDevice State
       0       8       17        0      active sync   /dev/sdb1
       1       8       33        1      active sync   /dev/sdc1
       2       8       49        2      active sync   /dev/sdd1
       3       8       65        3      active sync   /dev/sde1
       4       8       81        4      active sync   /dev/sdf1
       5       8       97        5      active sync   /dev/sdg1
       6       8      113        6      active sync   /dev/sdh1

pi@pigear-nano:~ $ sudo mkfs.ext4 -m 0 -E lazy_itable_init=0,lazy_journal_init=0 /dev/md0
mke2fs 1.46.2 (28-Feb-2021)
Creating filesystem with 67477120 4k blocks and 16875520 inodes
Filesystem UUID: 8b9be6ca-35b8-4ea9-b8ed-8e67cd8d7536
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done
```

geerlingguy, 2022-01-21:
```
$ sudo DEVICE_UNDER_TEST=/dev/md0 DEVICE_MOUNT_PATH=/mnt/raid0 ./disk-benchmark.sh

fio 1M sequential: 291 MB/sec
iozone 1M seq read: 136 MB/sec
iozone 1M seq write: 236 MB/sec
iozone 4K rand read: 6.50 MB/sec
iozone 4K rand write: 2.56 MB/sec
```

geerlingguy, 2022-01-21:
UUGear also maintains a `pgnano` application (instructions for installing it are in their user guide), and it includes a CLI interface as well as a web interface (named `uwi` for UUGear Web Interface):

[image]

geerlingguy, 2022-01-25:
Testing complete, it's on the site, and it will be featured in tomorrow morning's video!

closed 2022-01-25 by geerlingguy

5ft24dave, 2022-01-26:
Try adding storage quirks for the drive. Those resets are what I get with a JMicro usb3 to sata adapter before I add storage quirks to cmdline.txt. Though if that fixes it I would be puzzled as to why UGear put a controller on the board that isn't properly supported

uugear, 2022-02-01:
About the NVMe boot delay (usually 90 seconds, can also be 30/60/120 seconds), I have dig into the issue and confirm it is a power-related issue.

These messages show up in the boot log when the problem happens:
> [ 2.271169] usb 2-1: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
> [ 2.272447] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 2.273718] usb 2-1: Product: RTL9210
> [ 2.275023] usb 2-1: Manufacturer: Realtek
> [ 2.276283] usb 2-1: SerialNumber: 012345678909
> [ 2.281256] usb 2-1: Enable of device-initiated U1 failed.
> [ 2.283657] usb 2-1: Enable of device-initiated U2 failed.

When this error happens, the system will try to reset the hub and then retry, which will take about 30 seconds. If this error happen again, it will repeat the process and take another 30 seconds. When powering via USB type C, it usually repeats this three times before it eventually succeeds. But sometimes it luckily succeeds after only one attempt and sometimes it takes 4 attempts, that’s why I also saw 30 seconds and 120 seconds delay.

I then used 15V power supply to test it and the problem disappeared: it always finish the boot without delay.

I also used adjustable power supply to test it, and I noticed that: the higher voltage I give it, the smaller chance it will get stuck by this error.

I also tried different kinds of power supplies, and it turned out that the max output current of power supply also matters. I have a 12V/5A power supply who never met that error, while my another 12V/3A power supply sometimes got hit by that error.

Considering the board draws smaller current when using higher voltage power supply, it seems to me the initialisation of SSD will draw peak (spike) current from the power supply, which the power supply may not provide instantly and then that error happens. By increasing the input voltage, the average and spike current goes down and the chance to get hit by this error will be reduced.

It is true that disabling USAP on RTL9210 (adding “usb-storage.quirks=0bda:9210:u” to /boot/cmdline.txt) can also workaround this problem. By doing so I can avoid this error on boot even when I power the board with 5V via USB type C. However but the SSD I/O performance will go down (about 30% I think).

I am not sure if this issue is specific to RTL9210. By coincidence I am using the same SSD like Jeff's (Samsung 970 EVO Plus), so it is also possible to be specific to this SSD. Anyone who shares the experience with other SSDs will be appreciated. We will also test other SSDs when possible.

5ft24dave, 2022-02-01:
You need to see what current the SSD is trying to draw from the Pi's USB port
And if there is anything else connected to the USB ports, what total draw is.  I believe the Pi4 can supply 1.2 amps max via USB but that is shared by all 4 ports.

https://support.thepihut.com/hc/en-us/articles/360015272218-How-much-current-can-I-draw-from-a-Raspberry-Pi-USB-port-#:~:text=If%20you're%20adding%20USB,ports%20at%20the%20same%20time.

⁣Get BlueMail for Android ​

***@***.***>

geerlingguy, 2022-02-01:
@5ft24dave - Note that the PiGear Nano has it's own USB power circuit, so the limitations may be radically different than the one on the Pi 4 model B.