geerlingguy/raspberry-pi-pcie-devices #366: Test Sierra Wireless EM7565 4G LTE NGFF modem

geerlingguy, 2022-01-21:
I'm planning on testing the [Sierra Wireless EM7565 4G LTE modem](https://amzn.to/3nKDnBe) soon; perhaps using Waveshare's 'Dual Gigabit Ethernet 5G/4G Mini-Computer' board.

TODO: Image goes here.

The modem is a Cat-12 4G modem in the M.2 form factor (not like the mini PCIe modems I'm used to), and supposedly delivers up to 600 Mbps down, 150 Mbps up. It has GNSS and seems to be popular in more high-end industrial devices.

Shipping is insane for these cards, so I may get it anytime between now and early March.

Toomoch, 2022-02-05:
This card does NOT use PCIE, in fact most M.2 or miniPCIE modems use USB. I'm not sure the board you listed provides USB over the M.2.

geerlingguy, 2022-02-06:
@Toomoch - Indeed it does not, but it is still a card that I would like to validate as working (or not) and how to get it working, and since the physical connector is shared with other PCIe devices, it still has a place on this site (IMO).

It will always feel weird to plug a device into a mini PCIe or M.2 slot and it shows up as USB :P

geerlingguy, 2022-04-01:
```
$ sudo lsusb -v
Bus 001 Device 003: ID 1199:90c3 Sierra Wireless, Inc. Sierra Wireless EM7565 with Qualcomm Snapdragon X16 LTE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1199 Sierra Wireless, Inc.
  idProduct          0x90c3 
  bcdDevice            0.06
  iManufacturer           1 Sierra Wireless, Incorporated
  iProduct                2 Sierra Wireless EM7565 with Qualcomm Snapdragon X16 LTE
  iSerial                 3 2af40a0a
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0082
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      ** UNRECOGNIZED:  05 24 00 10 01
      ** UNRECOGNIZED:  05 24 01 00 00
      ** UNRECOGNIZED:  04 24 02 02
      ** UNRECOGNIZED:  05 24 06 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval               9
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      ** UNRECOGNIZED:  05 24 00 10 01
      ** UNRECOGNIZED:  05 24 01 00 00
      ** UNRECOGNIZED:  04 24 02 02
      ** UNRECOGNIZED:  05 24 06 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval               9
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

geerlingguy, 2022-04-01:
Just checking; don't see anything loading in `dmesg`, and:

```
pi@pi-router:~ $ sudo apt install -y modemmanager
pi@pi-router:~ $ mmcli -L
No modems were found
```

It looks like I'll have to compile the driver / GobiSerial?

See:

  - [Integrating LTE Device EM7565 into Raspberry Pi 3](https://techfortalk.co.uk/2020/02/27/integrating-lte-device-em7565-into-raspberry-pi-3/)
  - [EM7565: Cross compiling GobiSerial for Raspberry Pi 3 B](https://forum.sierrawireless.com/t/em7565-cross-compiling-gobiserial-for-raspberry-pi-3-b/19468)

geerlingguy, 2022-04-01:
On the Waveshare board, to control the modem's power:

```
pi@pi-router:~ $ echo "6" > /sys/class/gpio/export
pi@pi-router:~ $ echo "out" > /sys/class/gpio/gpio6/direction
pi@pi-router:~ $ echo "1" > /sys/class/gpio/gpio6/value
pi@pi-router:~ $ echo "0" > /sys/class/gpio/gpio6/value
pi@pi-router:~ $ echo "6" > /sys/class/gpio/unexport
```

After setting GPIO 6 high, the red STA LED turned off. Then when I set it low, the LED turned back on.

geerlingguy, 2022-04-03:
I just registered an account at https://login.sierrawireless.io/session/signup — I didn't find any way through the website itself to sign up, I had to find that by googling 'Sierra Wireless Registration'.

geerlingguy, 2022-04-13:
Haven't had a chance to finish that yet, I'm going to put that off a bit. One thing I've notice is there are a _lot_ of different Sierra modems out there with different capabilities. It seems some come preconfigured to work with USB TTY out of the box, others don't.

shayne, 2022-05-08:
@geerlingguy I built out a Waveshare Dual Gigabit Ethernet 5G/4G Base Board using a SIM8200EA-M2. After a bit of work I have built and OpenWrt image with the required kernel changes. Connection over QMI and seeing great results! [example speedtest running on pi!](https://www.speedtest.net/result/c/ab2c4d36-4f14-46f7-b4dc-174b4ce042ad)

Interested in documenting it here or somewhere else?

geerlingguy, 2022-05-09:
@shayne - Could you open a new issue for it so we can track how you got it working there? I'm assuming you mean [this card](https://www.simcom.com/product/SIM8200EA_M2.html)?

shayne, 2022-05-09:
Created #442

geerlingguy, 2022-08-21:
For OpenWRT, I went into `menuconfig` and selected:

```
Kernel modules
  -> USB support
    -> Support for Sierra Wireless devices (kmod-usb-net-sierrawireless)
```

(`kmod-usb-net` and `kmod-usb-net-lan78xx` were already selected by default, for the built-in USB-to-Ethernet adapter on my Waveshare board.)

geerlingguy, 2022-08-21:
Built an OpenWRT image using [these instructions](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem):

```
make -j $(nproc) kernel_menuconfig
```

TODO.

geerlingguy, 2022-08-21:
Working further on this in https://github.com/geerlingguy/pi-router