geerlingguy/raspberry-pi-pcie-devices #367: Add Waveshare Dual Gigabit Ethernet 5G/4G Base Board

geerlingguy, 2022-01-21:
Waveshare has a new [Dual Gigabit Ethernet 5G/4G Base Board](TODO) with dual gigabit Ethernet (2nd port is USB RTL8153 chip), RTC, dual HDMI, USB 3.2 Gen 1, and an M.2 B-key slot.

[image]

The board seems ideal for use as a router with something like OpenWRT, with a WAN connection and 4G/5G failover/backup.

ThomasKaiser, 2022-01-22:
The link in 1st paragraph most probably should end up here https://www.waveshare.com/cm4-dual-eth-4g-5g-base.htm instead?

Let's hope it's an RTL8153B since a lot of people reported problems with early chip revisions. And as a side note: RTL8153B attached to the VL805 USB3 host controller is a surprisingly but really solid choice.

geerlingguy, 2022-01-22:
Heh, yes, that's the right link... updating now.

closed 2022-01-25 by geerlingguy

geerlingguy, 2022-04-04:
One more note: for the RTL8153 off the USB bus:

```
pi@pi-router:~ $ sudo ethtool eth1
Settings for eth1:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Supports Wake-on: d
	Wake-on: d
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
```

geerlingguy, 2022-04-04:
Using the Pi's built-in Gigabit NIC:

```
pi@pi-router:~ $ iperf3 -c 10.0.100.15 --bidir
Connecting to host 10.0.100.15, port 5201
[  5] local 10.0.100.114 port 46496 connected to 10.0.100.15 port 5201
[  7] local 10.0.100.114 port 46498 connected to 10.0.100.15 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   113 MBytes   944 Mbits/sec    0    403 KBytes       
[  7][RX-C]   0.00-1.00   sec  95.6 MBytes   802 Mbits/sec                  
[  5][TX-C]   1.00-2.00   sec   111 MBytes   932 Mbits/sec    0    420 KBytes       
[  7][RX-C]   1.00-2.00   sec   100 MBytes   841 Mbits/sec                  
[  5][TX-C]   2.00-3.00   sec   107 MBytes   894 Mbits/sec    0    438 KBytes       
[  7][RX-C]   2.00-3.00   sec  89.1 MBytes   747 Mbits/sec                  
[  5][TX-C]   3.00-4.00   sec   111 MBytes   933 Mbits/sec    0    438 KBytes       
[  7][RX-C]   3.00-4.00   sec  96.9 MBytes   813 Mbits/sec                  
[  5][TX-C]   4.00-5.00   sec   110 MBytes   924 Mbits/sec    0    438 KBytes       
[  7][RX-C]   4.00-5.00   sec  95.6 MBytes   802 Mbits/sec                  
[  5][TX-C]   5.00-6.00   sec   111 MBytes   931 Mbits/sec    0    332 KBytes       
[  7][RX-C]   5.00-6.00   sec  97.2 MBytes   815 Mbits/sec                  
[  5][TX-C]   6.00-7.00   sec   111 MBytes   927 Mbits/sec    0    368 KBytes       
[  7][RX-C]   6.00-7.00   sec  96.7 MBytes   811 Mbits/sec                  
[  5][TX-C]   7.00-8.00   sec   110 MBytes   927 Mbits/sec    0    400 KBytes       
[  7][RX-C]   7.00-8.00   sec  96.9 MBytes   813 Mbits/sec                  
[  5][TX-C]   8.00-9.00   sec   110 MBytes   923 Mbits/sec    0    414 KBytes       
[  7][RX-C]   8.00-9.00   sec  96.9 MBytes   813 Mbits/sec                  
[  5][TX-C]   9.00-10.00  sec   110 MBytes   925 Mbits/sec    0    414 KBytes       
[  7][RX-C]   9.00-10.00  sec  96.0 MBytes   805 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  1.08 GBytes   926 Mbits/sec    0             sender
[  5][TX-C]   0.00-10.01  sec  1.08 GBytes   924 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec   961 MBytes   806 Mbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec   961 MBytes   806 Mbits/sec                  receiver

iperf Done.
```

And using the RTL8153:

```
pi@pi-router:~ $ iperf3 -c 10.0.100.15 --bidir
Connecting to host 10.0.100.15, port 5201
[  5] local 10.0.100.143 port 46748 connected to 10.0.100.15 port 5201
[  7] local 10.0.100.143 port 46750 connected to 10.0.100.15 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   111 MBytes   928 Mbits/sec    0    380 KBytes       
[  7][RX-C]   0.00-1.00   sec  22.0 MBytes   185 Mbits/sec                  
[  5][TX-C]   1.00-2.00   sec   112 MBytes   940 Mbits/sec    0    380 KBytes       
[  7][RX-C]   1.00-2.00   sec  23.8 MBytes   200 Mbits/sec                  
[  5][TX-C]   2.00-3.00   sec   112 MBytes   942 Mbits/sec    0    380 KBytes       
[  7][RX-C]   2.00-3.00   sec  23.3 MBytes   195 Mbits/sec                  
[  5][TX-C]   3.00-4.00   sec   112 MBytes   944 Mbits/sec    0    380 KBytes       
[  7][RX-C]   3.00-4.00   sec  22.5 MBytes   189 Mbits/sec                  
[  5][TX-C]   4.00-5.00   sec   112 MBytes   936 Mbits/sec    0    380 KBytes       
[  7][RX-C]   4.00-5.00   sec  24.7 MBytes   207 Mbits/sec                  
[  5][TX-C]   5.00-6.00   sec   112 MBytes   940 Mbits/sec    0    380 KBytes       
[  7][RX-C]   5.00-6.00   sec  25.4 MBytes   213 Mbits/sec                  
[  5][TX-C]   6.00-7.00   sec   112 MBytes   940 Mbits/sec    0    403 KBytes       
[  7][RX-C]   6.00-7.00   sec  26.1 MBytes   219 Mbits/sec                  
[  5][TX-C]   7.00-8.00   sec   113 MBytes   945 Mbits/sec    0    423 KBytes       
[  7][RX-C]   7.00-8.00   sec  25.3 MBytes   213 Mbits/sec                  
[  5][TX-C]   8.00-9.00   sec   112 MBytes   941 Mbits/sec    0    423 KBytes       
[  7][RX-C]   8.00-9.00   sec  25.7 MBytes   215 Mbits/sec                  
[  5][TX-C]   9.00-10.00  sec   112 MBytes   940 Mbits/sec    0    423 KBytes       
[  7][RX-C]   9.00-10.00  sec  24.5 MBytes   206 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  1.09 GBytes   940 Mbits/sec    0             sender
[  5][TX-C]   0.00-10.01  sec  1.09 GBytes   938 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec   244 MBytes   204 Mbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec   243 MBytes   204 Mbits/sec                  receiver

iperf Done.
```

Just running one way (Pi to Mac), I got 941 Mbps. Running in reverse one-way (Mac to Pi), I got 885 Mbps.

Running bidir continously, and monitoring `atop`, I could see:

[image]

So it seems the packets in were dramatically reduced if packets out were high (20,000 in vs 80,000 out).

Running the same test with the CM4's built-in NIC:

[image]

For the built-in NIC, whatever bottleneck USB imposes is gone, and IRQs are maxing out giving an aggregate of around 140,000 pps (vs 100,000 via USB 3.0). The bandwidth on the built-in NIC seems far more symmetrical over long periods.

suhongok, 2023-08-10:
I love this work. Thank you. But I couldn't get similer result from you. I tested my notebook and io board directly, using iperf3. 
can you explain how you tested?

newkton1, 2024-04-18:
Jeff this is reviving an old thread but might there be a way of using the M.2 B-key connector on this board to run an M.2 NGFF SATA (B-key) SSD as non-booting storage instead of a 4/5G wirless board?  It tried and the SSD is powered but does not show in sudo fdisk -l.
Waveshare support is hopless BTW.