geerlingguy/raspberry-pi-pcie-devices #386: Add Sharp / NEC MPi4 NEC MediaPlayer Kit

geerlingguy, 2022-02-11:
Sharp / NEC displays has a [MPi4 NEC MediaPlayer Kit](https://www.sharpnecdisplays.eu/p/uk/en/products/accessories/details/t/options/computing/rp/raspberrypi-compute-module4.xhtml) available that connects a Raspberry Pi directly into one of their display solutions using their modular display system:

[image]

Meant for digital signage applications, the board comes preinstalled with NEC MediaPlayer, and there's a version with WiFi and an external antenna connector for wireless use.

geerlingguy, 2022-02-11:
Here's a video on the display's full feature set (mentions the Pi integration at 2:23): https://www.youtube.com/watch?v=zr4WQ8dhhG8

closed 2022-02-25 by geerlingguy

bam80, 2024-02-07:
I couldn't find info about the NEC MediaPlayer, is it open-source?