geerlingguy/raspberry-pi-pcie-devices #394: Add WiFi HaLow (802.11ah) ALFA Networks Raspberry PI products

FlyingHavoc, 2022-02-16:
https://www.alfa.com.tw/collections/wifi-halow

AHPI7292S is the world's first WiFi HaLow™ Raspberry Pi™ HAT module The AHPI7292S module series allows building long range, ultra-low power WiFi networks in sub 1 GHz license-exempt bands

APIMB is a simple mikroBUS™ sockets adapter for Raspberry Pi™ Dedicated for ALFA Network's 802.11ah WiFi module, the AHMB7292S, can be used also with many other click boards

geerlingguy, 2022-02-16:
The AHM27292U would probably be the most appropriate device to test. Interesting little device. First I've seen of IEEE 802.11ah.

punnypenguins, 2022-02-16:
@geerlingguy would you like all of them added, or just the AHM27292U for now?

geerlingguy, 2022-02-25:
Closing this issue—but if anyone gets a chance to test out the card (I don't have a HaLow network to test on...), I'd be happy to update the card's status.

closed 2022-02-25 by geerlingguy

manikmakki, 2023-06-03:
Hey @geerlingguy !
I've been working on a project using 802.11ah stuff on Raspberry Pis recently. The AHM27292U module is interesting. It exposes a serial connection over the M.2 interface, its essentially a standalone device running OpenWRT, requiring no drivers to run on the "host" device.
I've got some stuff documented here: https://github.com/manikmakki/Alfa-Tube-AH-Info

Demo code for the PiHat version is here, it can be used as a standard network interface:
https://github.com/manikmakki/batman-over-halow

kocoman1, 2023-11-16:
> Hey @geerlingguy ! I've been working on a project using 802.11ah stuff on Raspberry Pis recently. The AHM27292U module is interesting. It exposes a serial connection over the M.2 interface, its [...]

Hi can you look at your google drive link (on your page) for the firmware its not working. thanks

manikmakki, 2023-11-17:
> Hi can you look at your google drive link (on your page) for the firmware its not working. thanks

That link was provided by Alfa, unfortunately I don't have a copy of the firmware anymore. Their support team is reasonably responsive though!