geerlingguy/raspberry-pi-pcie-devices #397: Add the mnt reform RCM4 board

NiklausHofer, 2022-02-21:
Currently still in the prototype state, but quite definitely a thing:

* Source: https://source.mnt.re/reform/mnt-reform-raspberry-pi-cm4-som
* Social media updates:
    * https://mastodon.social/@mntmn/107581480356172805
    * https://mastodon.social/@mntmn/107837330047564462

geerlingguy, 2023-11-21:
Just received a complete kit in the mail for testing. I plan on doing so in December and posting some in-progress thoughts on the whole package in this issue!

geerlingguy, 2023-12-27:
I have been testing, wanted to document steps here for getting a CM4 going in a build that had an A311D BPI-CM4 shipped.

First, I booted up a CM4 in a CM4 IO Board separately, and logged into it over SSH to configure it for the MNT Reform—I had tried booting directly on the Reform but for some reason I was getting:

```
mmc0: error -110 whilst initializing SD card
```

That would happen after the Pi OS splash screen displayed for a while, and it would drop me back out to the `initramfs` shell.

So I logged into a fresh install on an IO board and ran:

```
cd Downloads
git clone https://source.mnt.re/reform/mnt-reform-raspberry-pi-cm4-som.git
cd mnt-reform-raspberry-pi-cm4-som/
sudo cp rcm4-overlay.dtbo /boot/overlays/
sudo nano /etc/modules

# Add the following to the end and save:
snd-soc-wm8960

sudo nano /etc/rc.local

# Add the following before `exit 0` and save:
gpioset 0 10=0 # Assert USB hub reset
gpioset 0 10=1 # Deassert USB hub reset

# Enable the backlight
echo 0 > /sys/class/backlight/backlight/bl_power

sudo nano /boot/config.txt

# Add the recommended settings at the end, save, and reboot (or shut down):
# https://source.mnt.re/reform/mnt-reform-raspberry-pi-cm4-som#recommended-changes-to-boot-config
```

geerlingguy, 2023-12-27:
The internal display works now, but I'm still getting the dreaded `mmc0: error -110 whilst initializing SD card` error. So I'm going to switch to an eMMC module instead, and see if I can get it booting. Currently twiddling my thumbs waiting for Imager to write the disk image to the CM4 :)

My current theory is the SD card slot's extension that reaches its way over to the RCM4 board somehow causes the signal to not be perfect, thus causing SD card read errors. I've seen similar on the Mame Gear when I was using the SD card using the microSD card extension to the expansion board...

I tried the same card and CM4 in my CM4 IO board and I don't get that error when booting.

NiklausHofer, 2023-12-27:
Wanted to point you to this [post over on Mastodon](https://tenforward.social/@artemis/111651313036967158). That person is trying to get yet another CM4 clone, a SOQuartz module, running in the reform and making some good progress by the look of it.

geerlingguy, 2023-12-27:
I can confirm that running the eMMC CM4 worked a treat—so it seems like the SDIO signaling from the SD card slot to the CM4 might be a little finicky, not sure if it was from the particular install on the MNT Reform I was testing or if it could be an issue on all models with RCM4... will have to mention it to MNT and see what they think.

But I got in some more testing, and USB worked fine, though I had a little trouble with audio—probably just needed to mess with Alsa mixer a bit more or something :P — it's Linux, audio's never cut and dry!

closed 2024-01-09 by geerlingguy