geerlingguy/raspberry-pi-pcie-devices #405: BIGTREETECH Raspberry Pad 5 - 5" IPS Touch Display CM4 Board

42CrMo4, 2022-03-13:
The BIGTREETECH Raspberry Pad 5 migth be an intresting CM4 Board to Test. 
It is available under [BIGTREETECH Raspberry Pad 5](https://www.biqu.equipment/products/bigtreetech-raspberry-pad-5?variant=39610695516258)
Main features according to the Website:
* 5-inch IPS Touch Display Screen
* Multi-point Touch
* 800*480 Pixels HD
* Interface: HDMI, 40 pin GPIO, RJ45, Type-C, 3* USB 2.0, CSI Port

[image]

*No affiliation with the brand or product.

geerlingguy, 2022-03-13:
Ha, it's like you read my mind! One showed up on my desk last week, and I'm going to be testing it soon :)

42CrMo4, 2022-03-13:
What a coincidence ^^ I'm looking forward to it.

geerlingguy, 2022-03-14:
leepspvideo already has a video up on the board! https://www.youtube.com/watch?v=KX53vMdJ3Ek — I'll hopefully cover it in the next few weeks as well, but check out his video for a good overview.

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/bigtreetech-raspberry-pad-5.html

geerlingguy, 2022-04-11:
Testing complete, this board will be in a video probably coming next Wednesday!

closed 2022-04-11 by geerlingguy

jaymcd, 2022-10-25:
Hey Jeff, I have tried everything with the Pad and my CM4 and can't get wifi or USB . I get a brcm_pci link down error. If I disable otg and put in the dwc2 overlay I can do lsusb -t and it shows a port but none of my peripherals actually work. No keyboard or mouse, etc. Any help would be appreciated. I am thinking maybe my Pad is faulty as I have put the CM in a CMIO board and usb works but not the wifi (same error). Thanks Jay

sl1pkn07, 2023-04-10:
Hi. anyone have test this pad with the pine64 soquartz?

greetings

kbenyous, 2024-03-28:
Hello. Until now, the only drawback I have identified is the lack of backlight control : https://github.com/bigtreetech/Raspberry-Pad/issues/3 . The firmware of the MCU that controls the backlight does not handle these actions over DSI or I2C.
I am trying a hardware hack : plugging the pull-down buttons that control the brightness, directly on some GPIOs of the RPI.
But I must find how to control this with DPMS.