geerlingguy/raspberry-pi-pcie-devices #418: Website link for RPi4-RTC-PoE broken on the Compute Module 4 Boards and Projects page

mactijn, 2022-03-28:
It looks like the page for the RPi4-RTC-PoE has been removed from Tindie, resulting in a 404 "Tindie must have eaten this page!" message.

I couldn't find an appropriate replacement link, so I guess it's just not happening any more?

geerlingguy, 2022-03-28:
My guess is, sadly, the project's availability in the West is yet another casualty of the war in Ukraine.

The group behind the RPi4-RTC-PoE is Russian (see https://ab-log.ru), and my guess is they aren't allowed (or are just taking the risk off the table) to sell through non-Russian sites at this point.

geerlingguy, 2022-03-28:
I think we should put a note to that effect on that page, and also update the link to this one:

https://ab-log.ru/forum/viewtopic.php?f=1&t=1895 ([translated](https://1-ab--log-ru.translate.goog/forum/viewtopic.php?f=1&t=1895&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp&_x_tr_enc=1))

mactijn, 2022-03-28:
Sounds good to me. Thanks for the link :)

FYI, Gmail is warning me about the link being unsafe:

> ### Suspicious link
> This link leads to an untrusted site. Are you sure you want to proceed to ab-log.ru?

paulwratt, 2022-03-29:
thats because RU has to produce its own SSL certs now, so there is no "master" to verify against (unless its blanket "protecting" .ru addresses)

mactijn, 2022-03-29:
The certificate that's being used (at least from my perspective) is a simple LetsEncrypt certificate, so I don't think that's it.

geerlingguy, 2022-03-31:
Ah, it looks like [Tindie may have put the link offline](https://blog.tindie.com/2022/03/ukrainian-tindie-sellers-tell-their-story-of-invasion/):

> In line with our parent company Siemens, Tindie has stopped new business in Russia as of last week, and Belarus as of this week. Stores in these countries have been put on hold, and Tindie is unable to take orders to these destinations.

closed 2022-04-06 by geerlingguy

WimLeflere, 2022-04-29:
The internet archive has a copy: https://web.archive.org/web/20211121105603/https://www.tindie.com/products/megad-2561/rpi4-rtc-poe/

commit ec7d9965 (2022-04-06): Fixes #418: Update link for ab-log RPi4-RTC-PoE computer.