geerlingguy/raspberry-pi-pcie-devices #425: Add EVB board CM4 Cluster Mini-ITX board

geerlingguy, 2022-04-18:
I just heard about the [CM4 Cluster Mini ITX board](https://dwmzone.com/en/raspberry-pi/1045-raspberry-pi-cm4-cluster-mini-itx-board-6-rpi-cm4-supported.html) on DWMzone:

[image]

[image]

It looks like it supports up to 6 CM4 modules, with an M.2 slot and microSD card slot available to each of the boards. It uses a 4-pin ATX CPU power plug, and has two RJ45 Gigabit Ethernet ports connected through an onboard switch. It looks like there is a USB 2.0 hub and two HDMI ports connected through to the first Pi, which is nice for cluster management / debugging purposes.

I just ordered one and will post testing notes here or to a new issue later.

geerlingguy, 2022-04-19:
Board is up on the site now: https://pipci.jeffgeerling.com/boards_cm/cm4-cluster-mini-itx.html

Changed the title for this issue so I can test it out soon.

geerlingguy, 2022-04-27:
Guess what just arrived?

[image]

I believe @danmanners is getting one today too...

geerlingguy, 2022-04-27:
Some notes from a tour around the board:

[image]

  - Ethernet switch is a [Realtek RTL8370N](https://www.realtek.com/en/products/communications-network-ics/item/rtl8370n-vb-cg).
  - Each Pi seems to have its own 12-pin header with GND/3V3 on one side, and nRPiBOOT, BT_nDisable, WL_nDisable, EEPROM, nWP, 5V, and GPIO18 down the left side.
  - Each Pi also has its own microSD jack for flashing on each slot
  - Each Pi also has a set of four LEDs. I'm guessing power/act, then Ethernet link/act, but not sure yet.
  - There's an 'Always ON' switch that seems to allow the board to always power all the Pis, or not... though I'm not sure. I'll have to see what that switch does exactly.
  - There are built in PWR_BTN and RESET momentary buttons.
  - There is a built-in front panel header, with PWR_LED connections, PWR_BTN, and RESET pins available
  - There are three fan headers—_looks_ like they're PWM (they're all 4 pin)
  - There are two 4-pin POE1/POE2 headers near the Ethernet jacks, so I'm guessing there would be a way to add a PoE header. Not sure.
  - On the back side, there is a space for an M.2 2280 M-key device (e.g. NVMe SSD) and microSD card for each pi (though it doesn't follow a 1-6 order, so make sure you note which slot goes where!)
  - There's a chip labeled "6679GM 946118" on the back near the power input. Not sure what it is for.
  - The USB 2.0 ports are driven through a [CoreChips SL2.1A USB 2.0 HUB TT0122A21](https://lcsc.com/product-detail/USB-ICs_CoreChips-SL2-1A_C192893.html).
  - Near the power/reset buttons, there's an [STC Micro 15W408AS](http://stcmicro.com/cn/stc/stc15w408as.html), with a 'Enhanced 8051 Central Processing Unit' inside. I'm guessing this is the MCU that controls power on the board.
  - On the top, near an oscillator and the Ethernet switch chip, there's an IC labeled "ATMCL908 24C08N SU27 D", not sure what it is though.

danmanners, 2022-04-27:
[image]

I can confirm that the Chassis-connector power button and LED work as expected, and I've had one node boot up successfully.

Power supply shipped with my unit is a 19V 5.26A, or 100W (99.94w, but whatever) and my one Pi CM4 boots up. I haven't hooked up HDMI or network yet, but that's next on the list.

Can also confirm that the two Noctua Fans connected or operating as expected, albeit at full speed.

geerlingguy, 2022-04-27:
@danmanners - Hmm... that's the same one I'm using. It looks like the little flat screws that hold the heatsink on can't but up against the risers in each of the four corners :(

D'oh!

danmanners, 2022-04-27:
[image]

Migrated over my MicroSD and NVMe from one of my [Waveshare BaseBoard (B)](https://www.waveshare.com/cm4-io-base-b.htm) and have TalosOS up and operational, as well as HDMI via capture card.

[image]

At least according to my Ubnt EdgeSwitch, it doesn't see any additional MAC Addresses or management IPs, so my immediate guess is that Realtek chip is unmanaged.

tltangliang, 2022-04-28:
> Some notes from a tour around the board:
>
> [image]
>
> * Ethernet switch is a [Realtek RTL8370N](https://www.realtek.com/en/products/communications-network-ics/item/rtl8370n-vb-cg).
> * Each Pi seems to [...]

https://wiki.wisdpi.com/raspberry-pi-cm4/rpi-cm4-itx-cluster

geerlingguy, 2022-04-28:
@tltangliang - Oh excellent! Thanks for posting that. Looks like the switch is basically unmanaged, and the two Ethernet ports are just hanging off it in addition to the 6 internal ports then.

That site also answers some other questions I had, like:

  1. The six green LEDs are 'ACT' LEDs for each of the individual Pis.
  2. The pinout of the ATX 12V input
  3. LED mapping for each of the Pis
  4. The 4-pin header by CM #1 is actually ARGB (controlled via GPIO pin 18). That's cool!

As @danmanners, myself, and some others get into testing this board, we may have more questions too. Is there a specific support forum or email we can contact? I know I got an email directly from one of the devs, and I'll share feedback there, but it would also be nice to know a central place so other users could also provide feedback or get guidance.

tltangliang, 2022-04-29:
> @tltangliang - Oh excellent! Thanks for posting that. Looks like the switch is basically unmanaged, and the two Ethernet ports are just hanging off it in addition to the 6 internal ports then.
> 
> That site also answers some other questions I had, like:
> 
> 1. The six green LEDs are 'ACT' LEDs for each of the individual Pis.
> 2. The pinout of the ATX 12V input
> 3. LED mapping for each of the Pis
> 4. The 4-pin header by CM [Test SATA adapter (I/O Crest 4 port Marvell 9215) #1](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1) is actually ARGB (controlled via GPIO pin 18). That's cool!
> 
> As @danmanners, myself, and some others get into testing this board, we may have more questions too. Is there a specific support forum or email we can contact? I know I got an email directly from one of the devs, and I'll share feedback there, but it would also be nice to know a central place so other users could also provide feedback or get guidance.
> 
> Edit: Looks like there's a [forum](https://forum.wisdpi.com) but it's still being set up a bit. Is that the best place to direct users?

I think the forum can be working now.
https://forum.wisdpi.com/

geerlingguy, 2022-05-15:
Hmm... this looks _awfully_ similar to [52Pi's DeskPi Super6C](https://deskpi.com/collections/deskpi-super6c/products/deskpi-super6c-raspberry-pi-cm4-cluster-mini-itx-board-6-rpi-cm4-supported), which retails for $199.

The Cluster Mini-ITX board is $200 on DWMZone and shipping was $40ish. Shipping is about the same on 52Pi. So the question is, who actually makes the board, and what's the board's most 'official' name? :D

Pi-Mania, 2022-05-15:
Great Question.  Let's find out..

> Hmm... this looks _awfully_ similar to [52Pi's DeskPi Super6C](https://deskpi.com/collections/deskpi-super6c/products/deskpi-super6c-raspberry-pi-cm4-cluster-mini-itx-board-6-rpi-cm4-supported), [...]

Pi-Mania, 2022-05-15:
The ''official'' name is "DeskPi Super6C" DWMZone is an authorised agent

geerlingguy, 2022-05-16:
Thanks for clarifying that! I guess we should update the name on the site, then, and also add a reference to both locations in the description.

drifterdave, 2022-06-18:
Hey guys, would either of you by chance have a 3d Printer? I got this board, and designed an IO shield for the Super6c but don't have a 3D printer to try to print it out just yet. 

Would you be willing to make sure all of the holes line up? As always, great work, I really appreciate the work you put into your videos. 

[thin_ITX_IO_Shield_Super6c.scad.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/8932826/thin_ITX_IO_Shield_Super6c.scad.txt)
> (It is labeled _thin_ ITX, but there is a customizer that you can toggle Standard ATX or Thin ATX. The case I happen to be using is a Thin ITX case.)

adamfowleruk, 2022-07-20:
FYI I've purchased a DeskPi Super6C board and am using it to test the SOQuartz module. Booted Manjaro ok, but need to rejig the USB driver. Initial details and picture on the SOQuartz issue here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/336#issuecomment-1190668987

danmanners, 2022-07-23:
> Hey guys, would either of you by chance have a 3d Printer? I got this board, and designed an IO shield for the Super6c but don't have a 3D printer to try to print it out just yet.
>
> Would you be [...]

Hey @drifterdave; my apologies for the delay. I don't have a functional 3D printer, but I'm hoping to get one working in the next 2-3 months, so I should be able to validate everything for you then!

punnypenguins, 2022-07-26:
@geerlingguy can we remove the not-on-site label from this one?

geerlingguy, 2022-07-26:
Original URL was: https://pipci.jeffgeerling.com/boards_cm/cm4-cluster-mini-itx.html

New URL is: https://pipci.jeffgeerling.com/boards_cm/deskpi-super6c.html

geerlingguy, 2022-07-29:
@drifterdave - I'm printing the shield as we speak (STL file attached). I'll let you know how it lines up!

[DeskPi_Super6c_IO_Shield.stl.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/9222924/DeskPi_Super6c_IO_Shield.stl.txt)
.

drifterdave, 2022-07-29:
@geerlingguy @danmanners Thank you so much! Hopefully it fits well and I can order one to be made :).

geerlingguy, 2022-07-29:
@drifterdave - It's close but needs more tolerances; on my board one of the vertical HDMI ports is slanted a bit, and otherwise, almost everything looks like it needs to go down a bit to fit correctly. The power port is the furthest off though:

[image]

I might mess with OpenSCAD and see if I can get the tolerances dialed in better. I feel like the QC on the finished port locations is not going to be as precise as their mechanical drawings!

drifterdave, 2022-07-29:
Hmm very interesting, on my board the power jack is much closer to the edge than it seems yours is, I actually measured off the board as opposed to the mechanical drawings. I think you are right, tolerances are gonna be tough with the inconsistency, lol.

icaesarv, 2022-07-30:
It doesnt have a GPIO as i can see? Is that correct?

icaesarv, 2022-07-30:
@geerlingguy.. No video for it in a fully working Mode on your channel yet? 🥇

geerlingguy, 2022-07-31:
@icaesarv - Working on that currently ;)

geerlingguy, 2022-07-31:
> Hmm very interesting, on my board the power jack is much closer to the edge than it seems yours is, I actually measured off the board as opposed to the mechanical drawings. I think you are right, tolerances are gonna be tough with the inconsistency, lol.

Heh... I noticed the solder joints on some of the points were a bit haphazard. All are good, but some of the ports are a little high/low—even the LEDs are all over the place. So yeah, I'm guessing everyone will need a little tweaking. I might just add a bunch of extra space (like +/-1mm in each dimension) and live with 'panel gaps'.

icaesarv, 2022-08-02:
You should build the maximum out of that board, put 6 Raspberry CM 8Gb with eMMC & 6x 1TB SSDs, Watercooling all Processors or using 6 ICE Towers, power it with a decent power supply, clock the heck out of it and test its limits.
That should be a very beasty NAS Server.
Give the manufacturer your feedback on that setup and let them produce a casing for it, that can be used both as a  desktop and a 19" server unit using Adapters.
If it passes that test, it would be very useful as a NAS and in Home Automation ;)

geerlingguy, 2022-08-02:
Working on a cluster configuration here: https://github.com/geerlingguy/deskpi-super6c-cluster

@drifterdave - After four iterations, I have the ports all lined up:

[image]

Here are the relevant files:

  - [DeskPi_Super6c_IO_Shield.stl.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/9243924/DeskPi_Super6c_IO_Shield.stl.txt)
  - [thin_ITX_IO_Shield_Super6c.scad.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/9243925/thin_ITX_IO_Shield_Super6c.scad.txt)

drifterdave, 2022-08-02:
Awesome thank you! I'm going to see if I can get a buddy to print it and see how well it fits mine.

drifterdave, 2022-08-09:
V4 fits my case as well! Although the sides could be a little taller, it doesn't fit tightly without the motherboard behind it.

geerlingguy, 2022-08-10:
@drifterdave - Woah, that's a neat little setup! I was thinking of adding some sort of tiny retainer clip by adding a triangular shape that juts out on the frame that holds the io shield in the rectangle, just didn't get around to trying it out. It would be nice if it kind of 'popped in' instead of resting in place, because yes, mine drops down if you don't have the motherboard pressing it in.

icaesarv, 2022-08-10:
> V4 fits my case as well! Although the sides could be a little taller, it doesn't fit tightly without the motherboard behind it. Thank you for making the tweaks to make everything line up @geerlingguy ! [image]

Are those compute moduls heat thinks?

drifterdave, 2022-08-10:
> > V4 fits my case as well! Although the sides could be a little taller, it doesn't fit tightly without the motherboard behind it. Thank you for making the tweaks to make everything line up @geerlingguy ! [image]
> 
> 
> 
> Are those compute moduls heat thinks? 

It is yes, this one [here](https://a.co/d/eWCdifY)

icaesarv, 2022-08-10:
Watching what both of you are doing.. I know what my next Project is :D

fhemberger, 2022-08-10:
@drifterdave What case are you using? Looks like the perfect fit …

drifterdave, 2022-08-10:
@fhemberger It's the [akasa Cypher](https://www.akasa.com.tw/update.php?tpl=product/product.detail.tpl&no=181&type=Chassis&type_sub=Mini%20ITX&model=AK-ITX04-BK) it's pretty hard to find, but is such a great case, I have 3 of them from when intel was making Thin Mini ITX motherboards.

fhemberger, 2022-08-10:
@drifterdave Thanks!

drifterdave, 2022-08-11:
> @drifterdave Thanks!

If you wanted one, looks like FrozenCPU has a few in stock, and I have ordered from them before. 

[](https://www.frozencpu.com/products/21372/cst-1440/Akasa_Cypher_Ultra-Compact_Thin_Mini-ITX_Case_-_VESA_Mountable_AK-ITX04-BK.html )

geerlingguy, 2022-08-17:
Posted a video today: https://www.youtube.com/watch?v=ecdm3oA-QdQ

And here's the IO shield on Thingiverse: https://www.thingiverse.com/thing:5465766

closed 2022-08-17 by geerlingguy

paulwratt, 2022-08-31:
> Akasa Cypher Ultra-Compact Thin Mini-ITX Case - VESA Mountable (AK-ITX04-BK)

@drifterdave can you post your files for the 3D printed back-plate for this Mini-ITX case, thanks

With the dimentions being different for @geerlingguy (regular) sized back-plate, and me not having a 3D printer (and hence no software installed), I'd like to be able to upload it to PCB / 3D printing service ..

Cheers

Paul
PS great to see both you guys with a nice little box for your Super6c - looking forward to mine ..

drifterdave, 2022-08-31:
@paulwratt, @geerlingguy's thingiverse link actually includes the Thin-ITX version of the back plate. If you go to Customizer on thingaverse, there will be a dropdown for Standard or Thin-ITX.

gambcl, 2022-09-19:
Does anyone know if this thing will auto power on after a power loss, or do you have to hit the power button every time?

gambcl, 2022-09-19:
> V4 fits my case as well! Although the sides could be a little taller, it doesn't fit tightly without the motherboard behind it. Thank you for making the tweaks to make everything line up @geerlingguy ! [image]

It looks like you have swapped out the stock 40mm fan with some Noctuas right? Can just about see them.
How are they working out?
Does the case have extra screw holes for the extra fans?

filipnavara, 2022-09-19:
> Does anyone know if this thing will auto power on after a power loss, or do you have to hit the power button every time?

There's a HW switch for that, so it's possible.

gambcl, 2022-09-19:
> 

Good to know, thanks.

gambcl, 2022-09-19:
> > Does anyone know if this thing will auto power on after a power loss, or do you have to hit the power button every time?
> 
> There's a HW switch for that, so it's possible.

Good to know, thanks.

drifterdave, 2022-09-19:
[image]

No screws in the case, but the fans fit very snugly under the edge of the case, so they don't move or vibrate at all.

paulwratt, 2022-09-24:
right around 24days ago (a reply above) I was watching Wendel talk about 5 & 1/4 bays in his "IcyDock Stuff" video, and that night I got a Super6, which arived the day after my Axzez carrier board, and ended up look at 2 Inwin mini-desktop (BL040 & BL631) with 5 & 1/4 bay and 3x 3.5" internals

I ended up getting the Imwin BL631 which comes with 300W TFX powersupply, and IcyDock Quinto MB344SPO (4x SSD + slim DVD) 5.25" bay, 2x SSD connected to Axzez, and one of the 4-pin connect to Super6C - I have the front off now with intent to add "second front panel (stuff)" ("stuff" because there can be only _one_ power switch)

I plan to put 2 Exxos 10Gb or 12Gb 3.5" paired with the slim DVD + 2x SSD all connected tot the Axzez, and the other 2 SSD on the Super6C via internal USB adapters (and that still leaves 1x 3.5" bay open .. for something ..)

I'll post pictures, but I cant use either atm, unless I take the 2Gb CM4 out of a PiKVM I got on special (for a similar AMD based multi-motherboard mATX setup)   :)

michaelarmstrong, 2022-09-28:
I finally have one of these and wanted to share three very early experiences with the unit (1 hour of use) for others who are considering exploring it:

- The tolerances aren't great, my PCB even looked like it had impact on one of the corners, still works though.
- The NVMe drives (in linux anyway) seem to indeed come over PCIe however on the 3 CM4's with 3 different NVMe drives I tried, none have performance anything like they should and none exceed this as a quick sample with `hdparm`:

`/dev/nvme0n1:
 Timing buffered disk reads: 1102 MB in  3.01 seconds = 366.66 MB/sec`
 
 - The RPIBoot jumper (which usually doubles up as a way to disable eMMC boot... doesn't). Current plan is to dd /dev/zero to the eMMC devices on each Pi that I'd like to straight boot from NVMe and try again. 
 
 Once I've spent a bit more time with this device (I'm replacing a K3s Pi4 Cluster with this), I'll update the thread with more experiences.

michaelarmstrong, 2022-09-30:
- managed to get CM4's with eMMC to default to NVMe using the `BOOT_ORDER=0xf25416` in the boot loader and the RPiBOOT jumper config/locations are clearer to me now.
- still suffering poor NVMe performance, will begin that investigation shortly.

joshuatam, 2022-10-01:
Just checked DeskPi website, they have released an official case for Super6c.

https://deskpi.com/collections/deskpi-super6c/products/deskpi-itx-case-kit-for-deskpi-super6c-raspberry-pi-cm4-cluster-mini-itx-board

[image]

joshuatam, 2022-10-01:
> * The tolerances aren't great, my PCB even looked like it had impact on one of the corners, still works though.

I can confirm this, my PCB also looked like that, still can use btw.

[image]

michaelarmstrong, 2022-10-01:
> Just checked DeskPi website, they have released an official case for Super6c.
> 
> https://deskpi.com/collections/deskpi-super6c/products/deskpi-itx-case-kit-for-deskpi-super6c-raspberry-pi-cm4-cluster-mini-itx-board
> 
> [image]

Have just ordered it! Bit pricey for what it is, but will report back.

1201am, 2022-10-04:
Looks like there is a new revision of DeskPi board....with LED light on network ports and USB header finally the way it should be.
[image]
[image]

geerlingguy, 2022-10-04:
Nice! Looks like they're doing a great job taking feedback and adjusting a bit.

michaelarmstrong, 2022-11-02:
My case arrived and it looks like the USB header is the wrong way around on my board! Anyone hazard a guess to confirm or denyy?
[image]

michaelarmstrong, 2022-11-03:
Additionally I'm seeing packet loss of around 25% on all CM4's... wondering if anyone else experienced this.

althas316, 2022-11-05:
Has anyone ran EMMC on this board.  I have had nothing but problems

Flashed 64-bit piOS lite - regularly fails to boot reporting end kernel panic - not syncing asynchronous error interrupt

Tried ubuntu lite and sticks on the rainbow boot screen

Spent the day flashing and re-flashing the emmc on this board and a carrier board and all the same results.  Have no issues on the carrier board, they boot each and every time
I am not a hugely knowledgeable person on this (that is half the reason to get this) so if anyone has encountered these issue would appreciate any feedback

rr4444, 2023-01-19:
@geerlingguy @michaelarmstrong @joshuatam 

Amazing post, video and thread! I am now really thinking about one of these...

Do you guys have an idea whether:

1) there is space for NVME heatsinks underneath in the official or unofficial cases?

2) can the micro USBs used for mass storage, like SSDs, and would that be USB3?

3) any idea if Intel Optane NVME is RPI compatible? Struggling to find this out, as it seems like a very particular storage arch...

4) is the Deskpi AliExpress store legit? How on earth do problems get assessed? I'm worried I'll end up with an old board with upside USB problem or some other issue....

Any insights appreciated!!!!

danmanners, 2023-01-19:
@rr4444 responding in order:

1. Considering motherboard standoff screws are generally 6mm in height, as long as you have a heatsink that is no taller than ~2-3mm then yes. That being said, you're not going to have a ton of options for heatsinks that make any difference at that thickness, IMO.
2. No, MicroUSB cannot be use for mass storage. Definitely USB2/3.
3. No idea, but [quick googling does not seem promising](https://forums.raspberrypi.com/viewtopic.php?t=339293).
4. No idea; I would buy directly from [DeskPi's website](https://deskpi.com/collections/deskpi-super6c/products/deskpi-super6c-raspberry-pi-cm4-cluster-mini-itx-board-6-rpi-cm4-supported) personally, unless they don't ship to your location.

[27 more replies]