geerlingguy/raspberry-pi-pcie-devices #429: Testing ASM1083 PCIe to PCI Bridge and get legacy PCI Devices to work

t5b6de, 2022-04-22:
I'm currently trying to get the ASM1083 PCIe to PCI bridge to work but i had no success so far.

easiest card i tried to get to work is a simple USB 2.0 EHCI PCI Card.
I recompiled the kernel to add EHCI, OHCI and some other PCI related stuff into the kernel.

All what I get is probe failed with -12 (ENOMEM). So i added printk. a huge TON of printk. I ended up in kernel/dma/direct.c

within that function static struct page *__dma_direct_alloc_pages
it is checking if page is coherent (in function dma_coherent_ok). but that fails. everytime. so i tried to determine the reason.
Somehow the dma address is everytime way bigger then the coherent_dma_mask (which is 2^32-1)
That if statement is always false.

i tried some other PCI cards, one of them is a multistandard tv tuner, same problem, a gigabit ethernet pci card -> same problem. every time this function has its addresses 
in that Check-function i added a printk, which prints me all values related to that dma allocation:
dma_addr: 0x501ade000, size: 4096, mask: 0xffffffff, bus_dma_limit: 25769803775

there you see that dma address is too high for that mask.
If i modify that function so that will only check with bus_dma_limit then the initialization works, and the card works somehow and detects add and removal from usb devices successfully. But the devices can not be accessed. (error reading device descriptor)

then i've read about a bug in the dma controllers with dma adresses above 3GiB in die BCM-Chip. Maybe this could be the problem? after days of testing and digging further to the kernel, i do not want to give up but at current point its somehow wasting time...

Does anyone had more success with that adapter?
You can find that PCIe to PCI adapter if you googling for PCE1PCI-A03.

PS: That adapter is working perfectly within a x64 based PC or within a x64 windows Machine, with all my cards i have. My Cards all do NOT require any IO BAR since these are not available on ARM based machines.

(excuse my english I'm a German speaking person ;))

mi-hol, 2022-04-24:
> i've read about a bug in the dma controllers with dma adresses above 3GiB in die BCM-Chip.

This bug (from what I remember but could not find a reference quickly) affects version of BCM2711 before CM4 and RPi 400 got introduced. Please supply the hardware details (chipset revision  and firmware version) you are using

mi-hol, 2022-04-26:
Are you aware of [detailed BCM2711 processor documentation](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf)  (ie section "1.2. Address map")?

6by9, 2022-04-26:
Already partially covered by https://forums.raspberrypi.com/viewtopic.php?p=1772997#p1772997

PCI cards frequently use I/O BAR space, which isn't supported on ARM and so don't work.
Find the driver and if it uses ```outb``` or ```inb```, then it's not going to work on ARM.

You can use ```dtoverlay=pcie-32bit-dma``` to drop the PCIe window down into the bottom 4GB so that it can be addressed via 32bit addresses only.

t5b6de, 2022-04-26:
Hi,

thanks for reply. i will test that. 

proc cpuinfo shows that:
Hardware        : BCM2835
Revision        : d03140
Serial          : 100000002fe4xxxx
Model           : Raspberry Pi Compute Module 4 Rev 1.0

on that chip itself is marked:
Broadcom
2711ZPKFSB06C0T
TE2050
01X-05 B3 W

Cards do not use any kind of io bar. (at least there's no indication in dmesg for that)

i tried dtoverlay with 32 bit pcie dma, and if freaking works!
all my old legacy PCI cards i have are working flawless on a Raspberry Pi! this is kinda creepy :-D

Thanks!

walterav1984, 2022-04-26:
> i tried some other PCI cards, one of them is a multistandard tv tuner

So besides some USB 2.0 or 100Mbit/1Gbit LAN adapter, does the tvtuner happen to be DVB based device working? Just curious.

t5b6de, 2022-04-27:
i did a test of these cards on a linux PC and a windows PC. none of them are using io bar. In Windows you can see it, when there is a IO address given in the resources tab. (properties of device in device manager)

but none are there. so only memory mapped devices.
the dvbs2 card is a tt budget s2-3200, which is working with ttpci kernel driver. i tested it with w_scan and satpi (small SAT>IP server for linux) and a Windows PC with DVBViewer as a client. it worked without any issues.

walterav1984, 2022-04-27:
Great to hear dvb-s2 is working, was curious since similar experience with budget-ci model and minisatip on x86. Does the ASM1083 also supports 5 volt pci devices besides 3,3v?

t5b6de, 2022-04-28:
This particular card has a jumper. it can be configured to 5V and 3.3V. But that slot is a 5Volt one.
so if you have a 3.3V only card, it will not fit in that slot (key is mirrored to the other side)

i jumpered it to 3.3V anyway. but i do not have any 5v only cards, all my pci cards i own are newer ones which do support 3.3v and 5v (two keys on card edge)

walterav1984, 2022-04-28:
Great detail about the voltage jumper. The key's actually don't matter that much since the skystar pcicard itself has 2 holes and will fit fine in a modern 3,3 or old 5,5 motherboard however it would only work with 3,3. Do you happen to have a specific brand/part/item ebay/amazon/aliexpress link?

t5b6de, 2022-04-28:
i bought it from aliexpress shipped from france
https://de.aliexpress.com/item/1005003222943451.html

but you'll find it, if you search for PCE1PCI-A03
this specific adapter is marked with PCE1PCI-A03 VER006S, see attached photo.
[image]

PixlRainbow, 2022-05-20:
wouldn't it be funny if somehow modern PCIE graphics card were difficult to get working on the CM4, but older PCI graphics cards like 3dfx's Voodoos worked just fine

t5b6de, 2022-05-20:
This would be funny, but i think that old cards would use the IO Bar which is not available on ARM.

i selled my old 3dfx voodoo cards some years ago on ebay for only 2 bucks... such a shame.

PixlRainbow, 2022-05-20:
I mentioned it as a joke, but it turns out several PCI GPUs have Linux drivers for RISC-based architectures such as PowerPC or SPARC because they predate Intel's market dominance. No out-of-the-box ARM support though, because although ARM-based desktops existed at the time, they didn't have any PCI slots.

xjustbmw, 2023-03-04:
Hi there
I also upgraded to the new AM5 platform without any legacy pci slots
I ordered this board (still didnt arrive),and I have a tt budget s2-3200/skystar hd2 lying out at the moment

what is the proper jumper configuration for this board on the adapter ? Does it matter which one is selected?

Also, do you use your legacy pci card outside the pc now ? Or how did you mount it ?

thanks