geerlingguy/raspberry-pi-pcie-devices #430: Add Dell AMD Radeon 7470 to site

geerlingguy, 2022-04-22:
Today I received an AMD Radeon 7470 (C26411), which was marked on eBay as being a 'Dell Radeon HD 6450', but reports in the system as "Dell Radeon HD 7470":

TODO: Picture here.

TODO: Description here.

PCIe info:

```
0a:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Caicos XT [Radeon HD 7470/8470 / R5 235/310 OEM] (prog-if 00 [VGA controller])
	Subsystem: Dell Radeon HD 7470
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 47
	Region 0: Memory at 600000000 (64-bit, prefetchable) [size=256M]
	Region 2: Memory at 610000000 (64-bit, non-prefetchable) [size=128K]
	Region 4: I/O ports at <unassigned> [disabled]
	Expansion ROM at 610020000 [disabled] [size=128K]
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [58] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #8, Speed 5GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fffffffc  Data: 6540
	Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP+ BadDLLP+ Rollover- Timeout- AdvNonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: radeon
	Kernel modules: radeon

0a:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Caicos HDMI Audio [Radeon HD 6450 / 7450/8450/8490 OEM / R5 230/235/235X OEM]
	Subsystem: Dell Caicos HDMI Audio [Radeon HD 6450 / 7450/8450/8490 OEM / R5 230/235/235X OEM]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin B routed to IRQ 255
	Region 0: Memory at 610040000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [58] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #8, Speed 5GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP+ BadDLLP+ Rollover- Timeout- AdvNonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
```

Neofetch:

```
pi@radeon10:~ $ neofetch
       _,met$$$$$gg.          pi@radeon10 
    ,g$$$$$$$$$$$$$$$P.       ----------- 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 11 (bullseye) aarch64 
 ,$$P'              `$$$.     Host: Raspberry Pi Compute Module 4 Rev 1.0 
',$$P       ,ggs.     `$$b:   Kernel: 5.10.17-v8+ 
`d$$'     ,$P"'   .    $$$    Uptime: 1 min 
 $$P      d$'     ,    $$P    Packages: 1755 (dpkg) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.1.4 
 $$;      Y$b._   _,d$P'      Resolution: 1920x1080 
 Y$$.    `.`"Y$$$$P"'         Terminal: /dev/pts/0 
 `$$b      "-.__              CPU: BCM2835 (4) 
  `Y$$                        GPU: AMD ATI Radeon HD 7470/8470 / R5 235/310 OEM 
   `Y$$.                      Memory: 114MiB / 3795MiB 
     `$$b.
       `Y$$b.                                         
          `"Y$b._                                     
              `"""
```

geerlingguy, 2022-04-22:
I'm using the same branch from @Coreforge for 5.10.y / AMD Radeon that he's testing on the 6450 and I've also tested on my 5450.

`startx` results in a working display at the start, though it jumps up and down a few pixels here and there. (This is over DisplayPort to DVI on my monitor).

`glxgears` gives a similar style to the 5450 I've been testing, with the screen blinking on each gear turn, until Xorg eventually stops rendering. I can kill Xorg with `kill -9 -1 [pid]`.

The fan on the 7470 can get quite loud during reboots! I'm glad it uses some sort of PWM though, to quiet down when it's initialized and idling...

geerlingguy, 2022-04-22:
GLMark2 (`glmark2-drm`) ran all the way through—a task I've been unable to complete on my 5450.

```
=======================================================
    glmark2 2021.12
=======================================================
    OpenGL Information
    GL_VENDOR:      X.Org
    GL_RENDERER:    AMD CAICOS (DRM 2.50.0 / 5.10.17-v8+, LLVM 11.0.1)
    GL_VERSION:     3.1 Mesa 20.3.5
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0
    Surface Size:   1920x1080 fullscreen
=======================================================
[build] use-vbo=false: FPS: 19 FrameTime: 52.632 ms
[build] use-vbo=true: FPS: 59 FrameTime: 16.949 ms
[texture] texture-filter=nearest: FPS: 59 FrameTime: 16.949 ms
[texture] texture-filter=linear: FPS: 59 FrameTime: 16.949 ms
[texture] texture-filter=mipmap: FPS: 59 FrameTime: 16.949 ms
[shading] shading=gouraud: FPS: 59 FrameTime: 16.949 ms
[shading] shading=blinn-phong-inf: FPS: 59 FrameTime: 16.949 ms
[shading] shading=phong: FPS: 59 FrameTime: 16.949 ms
[shading] shading=cel: FPS: 59 FrameTime: 16.949 ms
[bump] bump-render=high-poly: FPS: 59 FrameTime: 16.949 ms
[bump] bump-render=normals: FPS: 59 FrameTime: 16.949 ms
[bump] bump-render=height: FPS: 59 FrameTime: 16.949 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 59 FrameTime: 16.949 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 59 FrameTime: 16.949 ms
[pulsar] light=false:quads=5:texture=false: FPS: 59 FrameTime: 16.949 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 59 FrameTime: 16.949 ms
[desktop] effect=shadow:windows=4: FPS: 59 FrameTime: 16.949 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 6 FrameTime: 166.667 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 8 FrameTime: 125.000 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 7 FrameTime: 142.857 ms
[ideas] speed=duration: FPS: 40 FrameTime: 25.000 ms
[jellyfish] <default>: FPS: 59 FrameTime: 16.949 ms
[terrain] <default>: FPS: 19 FrameTime: 52.632 ms
[shadow] <default>: FPS: 59 FrameTime: 16.949 ms
[refract] <default>: FPS: 23 FrameTime: 43.478 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 59 FrameTime: 16.949 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 59 FrameTime: 16.949 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 59 FrameTime: 16.949 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 59 FrameTime: 16.949 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 59 FrameTime: 16.949 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 59 FrameTime: 16.949 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 59 FrameTime: 16.949 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 59 FrameTime: 16.949 ms
=======================================================
                                  glmark2 Score: 50 
=======================================================
```

geerlingguy, 2022-04-24:
Something else to test (when I get a chance)—Coreforge mentioned that the following `cmdline.txt` was necessary to prevent early crashes:

```
console=serial0,115200 console=tty1 root=PARTUUID=5dac9010-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait radeon.uvd=0  maxcpus=4 pci=noaer,nomsi radeon.msi=0 radeon.pcie_gen2=0 pcie_aspm=off radeon.aspm=0 radeon.runpm=0 radeon.dpm=0
```

Hugobros3, 2022-04-27:
> So that may be looking at the Pi's VPU and not the Radeon GPU.
neither:

```
deviceType     = PHYSICAL_DEVICE_TYPE_CPU
deviceName     = llvmpipe (LLVM 11.0.1, 128 bits)
```this is the software driver

Hugobros3, 2022-04-27:
additionally I believe (don't quote me on that) `radv` only supports the `amdgpu` DRM driver, so if you get something like a hd 7750, you'll need to use the `amdgpu` module instead of `radeon`

walterav1984, 2022-06-02:
> additionally I believe (don't quote me on that) `radv` only supports the `amdgpu` DRM driver, so if you get something like a hd 7750, you'll need to use the `amdgpu` module instead of `radeon`

Correct vulkan is only for the `amdgpu` driver supported devices!

Nice catch that you actually got a HD7470 GCN1 which can run both `radeon` or `amdgpu` opensource kernel drivers. Offcourse `radeon` kernel driver has been patched a lot already (5450 thread) before you got it running on the `pi` but did you try `amdgpu` already?

Just boot with kernel argument  `radeon.si_support=0 amdgpu.si_support=1` and make sure you have firmware in `/lib/firmware/amdgpu/` and that your pi kernel has been compiled with the `amdgpu` module . More options like `amdgpu.dc=1` for alternate method of initializing display connector output/modes  or `amdgpu.dpm=0` for disabling power management.

Also stumbled upon this [patch](https://gist.github.com/jnettlet/1f461487bee9c3e2a2d994f25441717d) for fixing graphic corruption with `amdgpu` on aarch64?

geerlingguy, 2025-09-29:
I'm testing this again using yanghaku's DRM patch (see #764), and so far, with kernel `rpi-6.15.y` and the radeon driver, I see:

```
$ dmesg | grep radeon
[    4.454170] [drm] radeon kernel modesetting enabled.
[    4.454298] radeon 0001:01:00.0: enabling device (0000 -> 0002)
[    4.454349] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[    4.788874] [drm:radeon_atombios_init [radeon]] *ERROR* Unable to find PCI I/O BAR; using MMIO for ATOM IIO
[    4.793993] radeon 0001:01:00.0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
[    4.794961] radeon 0001:01:00.0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
[    4.795142] [drm] radeon: 1024M of VRAM memory ready
[    4.795153] [drm] radeon: 1024M of GTT memory ready.
[    4.807381] [drm] radeon: dpm initialized
[    4.810757] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
[    4.849268] radeon 0001:01:00.0: WB enabled
[    4.849293] radeon 0001:01:00.0: fence driver on ring 0 uses gpu addr 0x0000000040000c00
[    4.849302] radeon 0001:01:00.0: fence driver on ring 3 uses gpu addr 0x0000000040000c0c
[    4.855941] radeon 0001:01:00.0: fence driver on ring 5 uses gpu addr 0x0000000000072118
[    4.856113] radeon 0001:01:00.0: radeon: MSI limited to 32-bit
[    4.856231] radeon 0001:01:00.0: arch assigned 64-bit MSI address 0xfffffff000 but device only supports 32 bits
[    4.856301] [drm] radeon: irq initialized.
[    5.257572] [drm] Initialized radeon 2.50.0 for 0001:01:00.0 on minor 1
[    5.562542] radeon 0001:01:00.0: [drm] fb0: radeondrmfb frame buffer device
```

Looking up `arch assigned 64-bit MSI address 0xfffffff000 but device only supports 32 bits` I found [this forum topic](https://forums.raspberrypi.com/viewtopic.php?t=374193), which suggests forcing 32 bit compatibility mode in `/boot/firmware/config.txt`:

```
dtparam=pciex1
dtoverlay=pcie-32bit-dma
dtoverlay=pciex1-compat-pi5,no-mip
```

geerlingguy, 2025-09-29:
With the compatibility mode enabled I see:

```
$ dmesg | grep radeon
[    4.735840] [drm] radeon kernel modesetting enabled.
[    4.735952] radeon 0001:01:00.0: enabling device (0000 -> 0002)
[    4.736001] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[    5.007867] [drm:radeon_atombios_init [radeon]] *ERROR* Unable to find PCI I/O BAR; using MMIO for ATOM IIO
[    5.013741] radeon 0001:01:00.0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
[    5.013756] radeon 0001:01:00.0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
[    5.014737] [drm] radeon: 1024M of VRAM memory ready
[    5.014754] [drm] radeon: 1024M of GTT memory ready.
[    5.026475] [drm] radeon: dpm initialized
[    5.044662] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
[    5.107436] radeon 0001:01:00.0: WB enabled
[    5.107443] radeon 0001:01:00.0: fence driver on ring 0 uses gpu addr 0x0000000040000c00
[    5.107446] radeon 0001:01:00.0: fence driver on ring 3 uses gpu addr 0x0000000040000c0c
[    5.114068] radeon 0001:01:00.0: fence driver on ring 5 uses gpu addr 0x0000000000072118
[    5.114232] radeon 0001:01:00.0: radeon: MSI limited to 32-bit
[    5.114327] radeon 0001:01:00.0: radeon: using MSI.
[    5.114374] [drm] radeon: irq initialized.
[    5.504118] [drm] Initialized radeon 2.50.0 for 0001:01:00.0 on minor 1
[    5.835407] radeon 0001:01:00.0: [drm] fb0: radeondrmfb frame buffer device

$ dmesg | grep drm
[    4.173781] systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
[    4.225777] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    4.225912] systemd[1]: Finished modprobe@drm.service - Load Kernel Module drm.
[    4.633449] v3d 1002000000.v3d: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[    4.643521] [drm] Initialized v3d 1.0.0 for 1002000000.v3d on minor 0
[    4.735840] [drm] radeon kernel modesetting enabled.
[    4.735975] [drm] initializing kernel modesetting (CAICOS 0x1002:0x6778 0x1028:0x2120 0x00).
[    4.736001] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[    5.007867] [drm:radeon_atombios_init [radeon]] *ERROR* Unable to find PCI I/O BAR; using MMIO for ATOM IIO
[    5.008032] [drm] GPU not posted. posting now...
[    5.013760] [drm] Detected VRAM RAM=1024M, BAR=256M
[    5.013762] [drm] RAM width 64bits DDR
[    5.014737] [drm] radeon: 1024M of VRAM memory ready
[    5.014754] [drm] radeon: 1024M of GTT memory ready.
[    5.014774] [drm] Loading CAICOS Microcode
[    5.021175] [drm] Internal thermal controller without fan control
[    5.026475] [drm] radeon: dpm initialized
[    5.043396] [drm] GART: num cpu pages 262144, num gpu pages 262144
[    5.044662] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
[    5.068797] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    5.075871] vc4-drm axi:gpu: bound 107c701400.hdmi (ops vc4_hdmi_ops [vc4])
[    5.083168] vc4-drm axi:gpu: bound 107c706400.hdmi (ops vc4_hdmi_ops [vc4])
[    5.083413] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    5.083556] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    5.083635] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.083755] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.085375] [drm] Initialized vc4 0.0.0 for axi:gpu on minor 2
[    5.088044] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    5.091619] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    5.095286] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    5.107290] [drm] PCIE GART of 1024M enabled (table at 0x0000000000162000).
[    5.114374] [drm] radeon: irq initialized.
[    5.131153] [drm] ring test on 0 succeeded in 2 usecs
[    5.131177] [drm] ring test on 3 succeeded in 8 usecs
[    5.311331] [drm] ring test on 5 succeeded in 2 usecs
[    5.311352] [drm] UVD initialized successfully.
[    5.311621] [drm] ib test on ring 0 succeeded in 0 usecs
[    5.311752] [drm] ib test on ring 3 succeeded in 0 usecs
[    5.488203] [drm] ib test on ring 5 succeeded
[    5.488783] [drm] Radeon Display Connectors
[    5.488792] [drm] Connector 0:
[    5.488794] [drm]   DP-1
[    5.488796] [drm]   HPD2
[    5.488797] [drm]   DDC: 0x6460 0x6460 0x6464 0x6464 0x6468 0x6468 0x646c 0x646c
[    5.488800] [drm]   Encoders:
[    5.488802] [drm]     DFP1: INTERNAL_UNIPHY1
[    5.488804] [drm] Connector 1:
[    5.488806] [drm]   DVI-I-1
[    5.488807] [drm]   HPD4
[    5.488809] [drm]   DDC: 0x6450 0x6450 0x6454 0x6454 0x6458 0x6458 0x645c 0x645c
[    5.488811] [drm]   Encoders:
[    5.488812] [drm]     DFP2: INTERNAL_UNIPHY
[    5.488814] [drm]     CRT1: INTERNAL_KLDSCP_DAC1
[    5.504118] [drm] Initialized radeon 2.50.0 for 0001:01:00.0 on minor 1
[    5.677603] [drm] fb mappable at 0x1800363000
[    5.677616] [drm] vram apper at 0x1800000000
[    5.677620] [drm] size 8294400
[    5.677622] [drm] fb depth is 24
[    5.677623] [drm]    pitch is 7680
[    5.835407] radeon 0001:01:00.0: [drm] fb0: radeondrmfb frame buffer device
[    6.121309] [drm] amdgpu kernel modesetting enabled.
```

So it's definitely seeing the display, but maybe there's an incompatibility with the Radeon driver. Maybe switching to X instead will help? I used `sudo raspi-config` and switched the advanced wayland manager over to `X11` instead of `labwc`.

geerlingguy, 2025-09-30:
Testing again today, after I noticed a newer Mesa version fixed up the 9070 XT...

I can get screen output! Though logging in, it just loops back to the login screen, so maybe the GUI/desktop environment doesn't want to run on this old card/driver?

[image]

geerlingguy, 2025-09-30:
Rebooted into X11 instead of labwc (switched via `raspi-config`), and it works!

The desktop at 4K is a little touchy (there are some artifacts on the far right of the screen that kind of flicker in and out, more heavily when there's a lot being rendered).

https://github.com/user-attachments/assets/5a38a7a6-4c7e-4f03-a8b1-f19ba61b793b

Some benchmarks:

  - glmark2-x11: 539 (yes, that's slower than the iGPU lol)
  - vkmark: TODO

geerlingguy, 2025-09-30:
lol https://gravitymark.tellusim.com/report/?id=e26f83ed60845e329f760c6dba4ebcf1459314e9

[image]