geerlingguy/raspberry-pi-pcie-devices #435: Add Waveshare CM4 Nano Base Board (CM4-Nano-B)

geerlingguy, 2022-04-25:
Waveshare recently introduced the [CM4 Nano Base Board (B)](https://www.waveshare.com/CM4-NANO-B.htm), which sits within the CM4 footprint and offers full GPIO, Ethernet, USB 2.0 Type A, mini HDMI, CSI/DSI, 3.5mm Audio Output, and USB-C input for power:

[image]

geerlingguy, 2022-04-25:
h/t to https://twitter.com/peter_mount/status/1518684491400847361

geerlingguy, 2022-05-05:
On the site: https://pipci.jeffgeerling.com/boards_cm/waveshare-cm4-nano.html

closed 2022-05-05 by geerlingguy

marcusstenbeck, 2022-12-14:
Any plans on testing this board? Ports and form factor is perfect for me but I'm seeing varying reviews on Amazon.

clomads, 2023-01-26:
I tried two variants of OpenWRT (Rooter Goldenorb & SmoothWAN) and both of them stalled right after ethernet check. I'm using a lite/no-wifi. Official 64-bit release from Rpi works just fine. Only had it for a day so more testing to do.