geerlingguy/raspberry-pi-pcie-devices #443: Add generic VT6307L 2+1 IEEE 1394 Firewire PCIe card

geerlingguy, 2022-05-19:
GitHub user @WACOmalt was able to get a [Generic VT6307L / ASM1083 2+1 IEEE 1394 Firewire PCIe card](https://www.ebay.com/itm/175158502561) working on the Compute Module 4:

[image]

He compiled the kernel with `firewire-ohci` enabled, and added `dtoverlay=pcie-32bit-dma` to `/boot/config.txt`, and was able to get DVGrab to capture footage live and from tapes.

He did report trouble initializing the camera shortly after boot, and said his only concern was reliability—there were a few dropped frames or failed initializations.

t5b6de, 2022-05-19:
Hi Jeff,

I did transfer a whole 90 minute cassette without any trouble or dropped frames. But i have to say that i am transferrring the Data not onto a sd card or usb drive. I use a asmedia pci switch (which you already tested: https://pipci.jeffgeerling.com/cards_adapter/ashata-pcie-1-to-4-usb3-switch.html)

There are 2 other cards plugged in, my samsung 950 pro nvme drive, and a 4 port usb3.0 vl805 USB Card.

My Camera i'm using is a Sony DCR-HC14E from about 2005 (i think)

i do not have any troubles with initilization or anything else. it just works as intendet. But i have more than just the ohci compiled within the kernel. I added some Basic PCI Bus functionality, since i tested a asm1083 PCI to PCIe adapter with my multistandard DVB-S2 (as mentioned in #429 ).

I have to search for the options of the kernel, since i have reinstalled my laptop where i crosscompiled the kernel.

kind regards
Thomas

WACOMalt, 2022-05-19:
I should add an update. I think the dropped frames were during tape cuts.  Also I found I got MUCH better reliability and more responsiveness from `dvgrab` by using the `-buffers` option. the default is `-buffers 100` and I found this causes very laggy response from the tool. `-buffers 20` gives smooth feedback on recording stats and much more reliability and speed in starting and stopping captures.  Unfortunately I'm now dealing with a flaky firewire port (tested on a normal PC with the same issues) so I'm working on fixing that before I continue testing and working on my project.

WACOMalt, 2022-05-30:
Figured I'd go ahead and test another Firewire card. The new one I tested is another tiny one available [on AliExpress](https://www.aliexpress.com/item/2255799941718037.html). This one is powered by the VT6315N which to my eyes is a native PCIe chip rather than having a separate PCI to PCIe chip on it. 

The board is tiny too so will fit in plenty of projects. The same setup as the previous card worked here though I found it to be more reliable in my experience. I'm thinking my reliability issues are down to a faulty IO board. The PCIe port is extremely loose. Currently working on my own carrier board design which will have a more sturdy side-mounted PCIe port on it so I'll update if that solves my reliability issues. Also I'm currently writing files direct to the onboard storage, so there could be bottlenecking going on there causing dropped frames after a while. Once my final setup is done I can clarify my findings. But all in all it's clear that these work!

geerlingguy, 2022-07-26:
@WACOMalt - Great find! I will add that card as a separate issue in case you want to add any more details. That way we can track it separately on the site so people can see there are other options as well: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/465

And this card (the VT6307L version) is now on the site here: https://pipci.jeffgeerling.com/cards_other/via-firewire-iim-vt6307l.html

closed 2022-07-26 by geerlingguy