geerlingguy/raspberry-pi-pcie-devices #452: > And I tested the latest OpenWRT image (`openwrt-bcm2711-rpi-4-ext4-rtl8168-factory.img.gz`, flashed with Etcher), and it booted up and let me add the WAN port / eth1 without issue:

David-McGuire, 2022-06-28:
> And I tested the latest OpenWRT image (`openwrt-bcm2711-rpi-4-ext4-rtl8168-factory.img.gz`, flashed with Etcher), and it booted up and let me add the WAN port / eth1 without issue:
> 
> [image]
> 
> Now if the software config for that OLED display gets sorted, I think this board will have everything fixed up out of the box except the PoE issue. I saw a comment about pulling resistors earlier, but it looks like that comment was removed. @yoyojacky are the hardware engineers still taking a look at that? Thanks!

Hi bro, 
Thanks for your testing, that's cool, and our products has been removed the resistor on board which cause the PoE issue. the product on stock is good to go.

_Originally posted by @yoyojacky in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/337#issuecomment-1155924019_

David-McGuire, 2022-06-28:
Help needed installing the openwrt on the Raspberry Pi CM4 Router Carrier Board by 52Pi

So ive tried to use the (openwrt-22.03.0-rc4-bcm27xx-bcm2711-rpi-4-ext4-factory.img) to load this and It will not load.

Steps I have taken to install openwrt

1. downloaded the (openwrt-22.03.0-rc4-bcm27xx-bcm2711-rpi-4-ext4-factory.img) from (https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/)
2. decompiled this file
3. Flashed to SD card using BalenEtcher
4. Inserted the SD card

 to [Raspberry Pi CM4 Router Carrier Board by 52Pi]
5. Enabled by jumper caps, to Device mode and Host mode both and same results.
6. Powered up 

Steps taken to install Raspberry Pi OS

1. I have also tried to install the Raspberry Pi OS using the steps n  (https://wiki.52pi.com/index.php/EP-0146#Operating_System) with the same results.

[image]

Any help here to get (openwrt) installed I would greatly appreciate.

yoyojacky, 2022-06-29:
SD:　card not detected... please check the TF card and try again?