geerlingguy/raspberry-pi-pcie-devices #471: Add Sangoma Technologies Corp. A10x T1/E1 AFT card

hharte, 2022-08-13:
The Sangoma A10x AFT series are 1-, 2-, 4-, 8-, and 16-port T1/E1 controllers.

```
# lspci -vvvvnnx
...
02:04.0 Network controller [0280]: Sangoma Technologies Corp. A104d QUAD T1/E1 AFT card [1923:0100]
	Subsystem: Device [a114:4213]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 255 (1250ns min, 3750ns max)
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 600000000 (32-bit, non-prefetchable) [size=128K]
00: 23 19 00 01 06 00 00 02 00 00 80 02 00 ff 00 00
10: 00 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 14 a1 13 42
30: 00 00 00 00 00 00 00 00 00 00 00 00 ff 01 05 0f
```
### Installation

Kernel driver: [Wanpipe](https://wiki.freepbx.org/display/DAS/Telephony+Card+Driver+Download)

Needs some changes for ARM.

To compile, run "./Setup dahdi --silent"

### Status

Not currently working, due to attempt to use IRQ 255.

hharte, 2022-08-13:
I beleive the reason why the IRQ is set to 255 is that the Sangoma Wanpipe driver uses the "old" method of PCI device probing [Reference 12.1.6](http://www.makelinux.net/ldd3/chp-12-sect-1.shtml), where it walks the list of PCI devices until it finds cards that it can support.

Manually calling pci_assign_irq() in the sdladrv.c code gets past the IRQ 255 issue, and IRQ 63 is assigned to the card.  It is not possible to call pci_assign_irq() directly, as this symbol is not exported by the kernel.  A hack can be used to call this function using the address noted in /proc/kallsyms:

```
# cat /proc/kallsyms |grep pci_assign_irq
ffffffc0086ab130 T pci_assign_irq
```

And then in sdladrv.c:
```
void (*hh_pci_assign_irq)(struct pci_dev *dev) = 0xffffffc0086ab130;
```
This gets the driver to proceed further, but it crashes later on in the intialization:

```
[ 4235.943895] WANPIPE(tm) Hardware Support Module  7.0.34.0 (c) 1994-2016 Sangoma Technologies Inc
[ 4235.946263] usbcore: registered new interface driver sdlausb
[ 4235.947587] wanrouter: no symbol version for sdla_hw_probe_free
[ 4235.949585] WANPIPE(tm) Interface Support Module 7.0.34.0 (c) 1994-2016 Sangoma Technologies Inc
[ 4235.966568] WANPIPE(tm) Multi-Protocol WAN Driver Module 7.0.34.0 (c) 1994-2016 Sangoma Technologies Inc
[ 4235.966617] wanpipe: Probing for WANPIPE hardware.
[ 4235.966748] pci 0000:00:00.0: enabling device (0000 -> 0002)
[ 4235.966789] pci 0000:01:00.0: enabling device (0000 -> 0002)
[ 4235.966826] pci 0000:02:04.0: enabling device (0000 -> 0002)
[ 4235.976566] wanpipe: AFT-A108-SH PCIe T1/E1 card found (HDLC (DS) rev.43), cpu(s) 1, line(s) 8, bus #2, slot #4, irq #255
[ 4235.976622] wanpipe: Allocating maximum 8 devices: wanpipe1 - wanpipe8.
[ 4235.977672] WANPIPE: TDM Codecs Initialized
[ 4235.998499] WANPIPE(tm) Socket API Module 7.0.34.0 (c) 1994-2016 Sangoma Technologies Inc
[ 4235.998543] NET: Registered PF_WANPIPE protocol family
[ 4236.036147] WANPIPE(tm) WANEC Layer 7.0.34.0 (c) 1995-2006 Sangoma Technologies Inc.
[ 4236.036182] wanec_create_dev: Registering Wanpipe ECDEV Device!
[ 4236.050811] wanpipe1: Starting WAN Setup

[ 4236.050850] Processing WAN device wanpipe1...
[ 4236.050862] wanpipe1: Locating: A101/1D/2/2D/4/4D/8/8D/16/16D card, CPU A, PciBus=2, PciSlot=4
[ 4236.050878] wanpipe1: Found: A101/1D/2/2D/4/4D/8/8D/16/16D card, CPU A, PciBus=2, PciSlot=4, Port=0
[ 4236.051005] wanpipe1: AFT PCI memory at 0x600000000
[ 4236.051024] wanpipe1: IRQ 63 allocated to the AFT PCI card
[ 4236.051037] setup: hharte: requesting IRQ 63
[ 4236.051088] wanpipe1: Starting AFT 2/4/8 Hardware Init.
[ 4236.051118] wanpipe1: AFT A108 HW RJ45 Port Map: DEFAULT [1,5] [2,6] [3,7] [4,8]
[ 4236.051129] wanpipe1: Enabling front end link monitor
[ 4236.051140] wanpipe1: Global Chip Configuration: used=1 used_type=1
[ 4236.051158] Unable to handle kernel paging request at virtual address ffffffc00a4a3bb4
[ 4236.051170] Mem abort info:
[ 4236.051177]   ESR = 0x96000021
[ 4236.051185]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 4236.051195]   SET = 0, FnV = 0
[ 4236.051203]   EA = 0, S1PTW = 0
[ 4236.051210]   FSC = 0x21: alignment fault
[ 4236.051218] Data abort info:
[ 4236.051225]   ISV = 0, ISS = 0x00000021
[ 4236.051233]   CM = 0, WnR = 0
[ 4236.051241] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001152000
[ 4236.051252] [ffffffc00a4a3bb4] pgd=100000007ffff003, p4d=100000007ffff003, pud=100000007ffff003, pmd=100000004750e003, pte=0068000077cb4703
[ 4236.051286] Internal error: Oops: 96000021 [#1] PREEMPT SMP
[ 4236.051295] Modules linked in: wanec(O) af_wanpipe(O) wanpipe(O) wanrouter(O) sdladrv(O) wct4xxp(O) oct612x(O) dahdi(O) crc_ccitt cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec binfmt_misc brcmfmac brcmutil cfg80211 v3d rfkill raspberrypi_hwmon gpu_sched i2c_brcmstb bcm2835_v4l2(C) bcm2835_isp(C) videobuf2_vmalloc bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc4 vc_sm_cma(C) cec videodev snd_bcm2835(C) mc drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[ 4236.051475] CPU: 3 PID: 13815 Comm: wanconfig Tainted: G         C O      5.15.32-v8+ #1538
[ 4236.051487] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[ 4236.051495] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 4236.051504] pc : a104_fifo_adjust+0x538/0x7c0 [wanpipe]
[ 4236.051572] lr : a104_global_chip_config+0x88/0x844 [wanpipe]
[ 4236.051615] sp : ffffffc00a4a3b60
[ 4236.051620] x29: ffffffc00a4a3b60 x28: ffffffc001321088 x27: ffffff8077f61000
[ 4236.051633] x26: ffffff8077f61c00 x25: ffffff8077f64000 x24: ffffffc0011eb000
[ 4236.051646] x23: ffffff8077f70000 x22: ffffff8077f70000 x21: ffffff8077f64000
[ 4236.051657] x20: ffffffc00a4a3bb4 x19: ffffff8077f60000 x18: 0000000000000010
[ 4236.051669] x17: 5b205d372c335b20 x16: 5d362c325b205d35 x15: 0764076507730775
[ 4236.051681] x14: 0720073a076e076f x13: 0764076507730775 x12: ffffffc0093a6670
[ 4236.051693] x11: 0000000000000003 x10: ffffffc00938e630 x9 : ffffffc0011c0d6c
[ 4236.051705] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[... cut ...]
[ 4236.051928]  wanrouter_ioctl+0x430/0x9b0 [wanrouter]
[ 4236.051942]  proc_reg_unlocked_ioctl+0xb0/0x100
[ 4236.051954]  __arm64_sys_ioctl+0xb0/0xf0
[ 4236.051963]  invoke_syscall+0x4c/0x110
[ 4236.051973]  el0_svc_common.constprop.3+0xfc/0x120
[ 4236.051982]  do_el0_svc+0x2c/0x90
[ 4236.051989]  el0_svc+0x24/0x60
[ 4236.051999]  el0t_64_sync_handler+0x90/0xb8
[ 4236.052007]  el0t_64_sync+0x1a0/0x1a4
[ 4236.052017] Code: c8017c40 35ffffa1 17fff6ee f9800051 (c85f7c40) 
[ 4236.052026] ---[ end trace aa5b96795f143dcf ]---
```

geerlingguy, 2022-08-17:
Card is on the site now: https://pipci.jeffgeerling.com/cards_other/sangoma-a10x.html

(@hharte - Feel free to close this issue if you feel like the info posted is complete!)

hharte, 2022-08-21:
This card is working now for 32-bit kernels, just submitted #475 to reflect that.  The patched driver for Raspberry Pi is available at: https://github.com/hharte/wanpipe-rpi-cm4

closed 2022-08-21 by hharte