geerlingguy/raspberry-pi-pcie-devices #474: cross-compile can't mount fusefs on ubuntu 22.04 LTS

MaciekMachni, 2022-08-20:
sshfs mount fails with the following error:
``` bash
# sshfs root@192.168.1.249:/ /mnt/pi-ext4
fuse: mount failed: Permission denied
```
Adding `privileged: true` to the docker-compose.yml seems to resolve the problem, but I'm not sure if that's the way to do it.

geerlingguy, 2022-08-22:
Interesting; typically all that's required are these options: https://github.com/geerlingguy/raspberry-pi-pcie-devices/blob/master/extras/cross-compile/docker-compose.yml#L13-L17

I'm not sure why it would require full `privileged` access just on Ubuntu 22.04 (I test on macOS / Docker Desktop mostly).

paulwratt, 2022-09-04:
It might be a permissions thing on '/mnt/pi-ext4', also remembering that you have to 'sudo mount' anything from the command line ..

marniks7, 2023-01-02:
Got the same issue and `privileged` helped. As well Ubuntu 22.04 (Mint).

Issue: https://github.com/vgist/dockerfiles/issues/16