geerlingguy/raspberry-pi-pcie-devices #477: Add MikroTik CCR2004-1G-2XS-PCIe 2x25G SFP28 SmartNIC

neg2led, 2022-08-30:
The MikroTik [CCR2004-1G-2XS-PCIe](https://mikrotik.com/product/ccr2004_1g_2xs_pcie) is a SmartNIC/Router-NIC/poor man's [DPU](https://en.wikipedia.org/wiki/Data_processing_unit) equipped with:

- 2x 25G SFP28 network ports
- 1x 10M/100M/1G management port
- An Annapurna Labs (Amazon) AL52400 SmartNIC SoC (quad-core Cortex-A57* @ 1.5GHz)
- 4GB of RAM + 128MiB parallel NAND (why, MikroTik? a 4-8GB eMMC costs the same, and would let you run containers on it...)

\* may actually be Cortex-A72

The card itself runs MikroTik RouterOS v7 (based on Linux 5.6.3 with a number of modifications). The three network ports are attached to the AL52400 SoC, which has a number of hardware packet processing accelerators, and is actually the same SoC used in the ~third generation of Amazon's [AWS Nitro System](https://aws.amazon.com/ec2/nitro/) network and storage virtualization cards.

Because the PCIe endpoint is actually the SoC, it has to emulate a PCIe device of some kind; MikroTik's choice here was a bit weird. Around a year ago they upstreamed some patches adding support for [multiple TX/RX queues](https://lore.kernel.org/all/20210527144423.3395719-1-gatis@mikrotik.com/t/#u) and [10G/25G link speeds](https://lore.kernel.org/all/20210513114326.699663-1-gatis@mikrotik.com/t/#u) to the kernel's `atl1c` driver - a driver for decade-old Atheros PCIe 1Gbps NICs. 

Okay, sure, why not.
The card itself presents as four AR8151s (I've snipped out the lspci details for the other ports they're largely identical):

```
01:00.0 Ethernet controller: Qualcomm Atheros AR8151 v2.0 Gigabit Ethernet
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 65
        Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x8, ASPM not supported
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s (downgraded), Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp+ ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [d0] Vital Product Data
                Not readable
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP- SDES+ TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [148 v1] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [158 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Capabilities: [178 v5] Extended Capability ID 0x905
        Kernel driver in use: atl1c
        Kernel modules: atl1c

01:00.1 Ethernet controller: Qualcomm Atheros AR8151 v2.0 Gigabit Ethernet
[...]
01:00.2 Ethernet controller: Qualcomm Atheros AR8151 v2.0 Gigabit Ethernet
[...]
01:00.3 Ethernet controller: Qualcomm Atheros AR8151 v2.0 Gigabit Ethernet
```

It's interesting that it presents as four NICs despite only having two ports; on the routerOS side, you get ports "eth1-pcie" through "eth4-pcie", "sfp28-1" and "sfp28-2", and "eth0" which is the 1Gbps port. Any of the four PCIe vNIC ports can be directly mapped to either of the SFP28 ports in "passthrough mode", leaving two ports free for host <-> router communication, or all six ports can be used as regular routed/bridged ports in RouterOS, same as you would with a regular MikroTik CCR device.

I've tested this card with a CM4 in my CM4IO board, and it does work:

[image]

(forgot to take a pic with a cable plugged in, but you get the idea.)

Unfortunately, the `atl1c` driver isn't enabled in the default Pi kernel config, so you'll need to build a kernel with `CONFIG_ATL1C` set to `m` (recommended) or `y`, and this card makes use of 32-bit DMA transfers so `dtoverlay=pcie-32bit-dma` needs to be added to `/boot/config.txt` otherwise you'll get an allocation error on attempting to bring a link up.

I put one of the ports into passthrough mode, attached it to a 10Gbps device running an iperf3 server, and was able to hit just over 3Gbps TCP with four threads:

[image]

Not quite the full 3.4Gbps the CM4 might be able to achieve, but the other end was a quad Cortex-A72 as well & the 32-bit DMA overlay does slow things down somewhat. For the record, I saw roughly the same performance in tests directly to (as i've taken to calling it) the mikroNIC's internal SoC using MikroTik's bwtest utility.

PR incoming ;)

geerlingguy, 2022-08-30:
Thanks for sharing! I had a few people asking about this card specifically, so I'm glad someone was able to acquire one and get it tested. A slightly odd device, but it would definitely be better suited to a faster computer :)

geerlingguy, 2022-08-30:
It's up on the site! https://pipci.jeffgeerling.com/cards_network/mikrotik-ccr2004-1g-2xs-pcie.html

closed 2022-08-30 by geerlingguy