geerlingguy/raspberry-pi-pcie-devices #486: Test SATA card (Aliexpress, 12/16/20 ports, ASM1064 chipset)

tyllmoritz, 2022-09-12:
Source: Aliexpress (search for Model number)

12 Port: SA3112J
16 Port: SA3116J
20 Port: SA3120J

I have the 12 Port variant.

The following data is from an armbian system on an Pine64 Quartz64a
(Armbian_22.11.0-trunk_Quartz64a_jammy_current_5.18.19)

```
robin@quartz64a:~$ dmesg | grep " ata"
[    3.489639] ata1: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082100 irq 100
[    3.490329] ata2: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082180 irq 100
[    3.491011] ata3: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082200 irq 100
[    3.491692] ata4: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082280 irq 100
[    3.492371] ata5: DUMMY
[    3.492707] ata6: DUMMY
[    3.492935] ata7: DUMMY
[    3.493191] ata8: DUMMY
[    3.493419] ata9: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082500 irq 100
[    3.494128] ata10: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082580 irq 100
[    3.494846] ata11: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082600 irq 100
[    3.495564] ata12: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082680 irq 100
[    3.496282] ata13: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082700 irq 100
[    3.497099] ata14: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082780 irq 100
[    3.497788] ata15: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082800 irq 100
[    3.498475] ata16: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082880 irq 100
[    3.499161] ata17: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082900 irq 100
[    3.499846] ata18: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082980 irq 100
[    3.500542] ata19: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082a00 irq 100
[    3.501229] ata20: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082a80 irq 100
[    3.501913] ata21: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082b00 irq 100
[    3.502598] ata22: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082b80 irq 100
[    3.503282] ata23: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082c00 irq 100
[    3.503966] ata24: SATA max UDMA/133 abar m8192@0x302082000 port 0x302082c80 irq 100
[    6.336595] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[    6.639344] ata1.00: ATAPI: HL-DT-STDVD-RAM GSA-H30N, 1.01, max UDMA/100
[    6.642493] ata1.00: configured for UDMA/100
[    7.028013] ata2: SATA link down (SStatus 0 SControl 300)
[    7.344297] ata3: SATA link down (SStatus 0 SControl 300)
[    7.660253] ata4: SATA link down (SStatus 0 SControl 300)
[    7.976247] ata9: SATA link down (SStatus 0 SControl 300)
[    8.292292] ata10: SATA link down (SStatus 0 SControl 300)
[    8.608065] ata11: SATA link down (SStatus 0 SControl 300)
[    8.924237] ata12: SATA link down (SStatus 0 SControl 300)
[    9.240212] ata13: SATA link down (SStatus 0 SControl 300)
[    9.556294] ata14: SATA link down (SStatus 0 SControl 300)
[    9.872286] ata15: SATA link down (SStatus 0 SControl 300)
[   10.188599] ata16: SATA link down (SStatus 0 SControl 300)
[   10.504252] ata17: SATA link down (SStatus 0 SControl 300)
[   10.820260] ata18: SATA link down (SStatus 0 SControl 300)
[   11.136259] ata19: SATA link down (SStatus 0 SControl 300)
[   11.452250] ata20: SATA link down (SStatus 0 SControl 300)
[   11.768260] ata21: SATA link down (SStatus 0 SControl 300)
[   12.084282] ata22: SATA link down (SStatus 0 SControl 300)
[   12.400296] ata23: SATA link down (SStatus 0 SControl 300)
[   12.716251] ata24: SATA link down (SStatus 0 SControl 300)
robin@quartz64a:~$ lspci
00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02)
```

QrchackOfficial, 2022-09-22:
JMB575 is not a "chipset". It's a port multiplier capable of splitting a single 6Gb/s port into a maximum of 5. They use the same ASM1064 "chipset"/SATA controller, which has 4x 6Gb/s (but with protocol overhead it's 4.8Gbps aka 600MB/s useful transfer speeds, per port) ports over PCIe 3.0 x1 (theoretically max 985MB/s total). So you have a little over 2x SATA ports worth of transfer speed for *all* drives connected.

The port multipliers are the 3 chips without a heatsink, and this means on a 12-port card each bank of 3 SATA ports share a maximum of 600MB/s to the controller, and then all 12 ports have 985MB/s total. On the 20 port one, it's 5 drives with 600MB/s shared, and again 985MB/s max for all 20 drives.

ThomasKaiser, 2022-10-02:
> PCIe 3.0 x1 (theoretically max 985MB/s total)

Unfortunately when attaching the ASM1064 to both CM4 or Quartz64 (RK3566) this will result only in Gen2 x1 (5GT/s with 8b10b coding) and as such those 12 SATA ports all share the bandwidth of less than 80% of a single SATA 6Gb/s port.

Based on my experiences with el cheapo port multipliers I would immediately buy some adhesive heatsinks to put on those JMB575 but most probably the artificial Gen2 x1 bottleneck already slows down the individual port multipliers so that data corruption caused by overheating shouldn't happen.

QrchackOfficial, 2022-10-02:
> Unfortunately when attaching the ASM1064 to both CM4 or Quartz64 (RK3566) this will result only in Gen2 x1
This is because both boards are only 2.0 x1 capable.

> most probably the artificial Gen2 x1 bottleneck already slows down the individual port multipliers so that data corruption caused by overheating shouldn't happen
Well, it's running at half the PCIe 3.0 x1 speed the card is capable of, aka 500MB/s shared across all the connected ports.

tyllmoritz, 2022-10-02:
Thanks a lot.

I'm aware oft the performance limitations and would bit recommend using this device for storage.

But for my use case - ripping up to 10 Audio-CDs simultaneously - 500MB/s should be enough.
If neccessary I'll limit the speed to 10x[playback speed].

PREngineer, 2023-08-24:
Are these hot plug capable?  I've not found any information regarding that for these.

QrchackOfficial, 2023-09-01:
The ASM1064 controller does: 
https://www.sybausa.com/index.php?route=product/product&product_id=1086

and so does the port multiplier:
https://www.jmicron.com/file/download/893/JMB575.pdf

so the answer should be yes