geerlingguy/raspberry-pi-pcie-devices #487: the docker project for cross-compiling the linux kernel does not run on asahi linux

monomycelium, 2022-09-21:
i have an m1 mac running arch linux on the asahi kernel. i followed [your steps to set up docker](https://www.jeffgeerling.com/blog/2022/installing-asahi-linux-alpha-on-my-m1-mac-mini), but i keep running into errors when trying to cross-compile using your docker thing [here](https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile).

<details>
<summary>docker logs</summary>

```
╭─mycelium@asahi:~/Documents/raspberry-pi-pcie-devices/extras/cross-compile ‹master› 
╰─🐧 $ docker-compose up -d
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3.10/site-packages/urllib3/connectionpool.py", line 398, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.10/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.10/http/client.py", line 975, in send
    self.connect()
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/lib/python3.10/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3.10/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/lib/python3.10/site-packages/urllib3/packages/six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "/usr/lib/python3.10/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3.10/site-packages/urllib3/connectionpool.py", line 398, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.10/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.10/http/client.py", line 975, in send
    self.connect()
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/api/client.py", line 214, in _retrieve_server_version
    return self.version(api_version=False)["ApiVersion"]
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/api/daemon.py", line 181, in version
    return self._result(self._get(url), json=True)
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/api/client.py", line 237, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/usr/lib/python3.10/site-packages/requests/sessions.py", line 600, in get
    return self.request("GET", url, **kwargs)
  File "/usr/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3.10/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3.10/site-packages/requests/adapters.py", line 547, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mycelium/.local/bin/docker-compose", line 8, in <module>
    sys.exit(main())
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/main.py", line 200, in perform_command
    project = project_from_options('.', options)
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/command.py", line 60, in project_from_options
    return get_project(
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/command.py", line 152, in get_project
    client = get_client(
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/docker_client.py", line 41, in get_client
    client = docker_client(
  File "/home/mycelium/.local/lib/python3.10/site-packages/compose/cli/docker_client.py", line 170, in docker_client
    client = APIClient(use_ssh_client=not use_paramiko_ssh, **kwargs)
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/api/client.py", line 197, in __init__
    self._version = self._retrieve_server_version()
  File "/home/mycelium/.local/lib/python3.10/site-packages/docker/api/client.py", line 221, in _retrieve_server_version
    raise DockerException(
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))
```

</details>

have you run a similar problem before? i really hope i can make use of my mac's superior performance to compile the raspberry pi kernel faster.

paulwratt, 2022-09-21:
there are some things I _dont_ the OS repo's for Box86, Box64, and Docker

I am not on Arch or Asahi, but it looks like you have a Docker Setup problem - _all_ those errors are fie/dir locations (including remote fetch)

monomycelium, 2022-09-21:
interesting… can i fix the file or directory location errors?

---

i already updated everything using `pacman -Syu`. docker installed correctly. i have already installed docker-compose using pip3 (and added it to my path).

paulwratt, 2022-09-21:
go to the "Official Docker" site and have a look at the scripts they use to install it, that might give you some ideas.

If you followed those original instructions and it still does not work, I would uninstall it and try the above mentioned install method instead (thats what I am using here).

monomycelium, 2022-09-21:
the script [here](https://github.com/docker/docker-install) to install docker failed:

```
ERROR: Unsupported distribution 'archarm'
```

so i went with arch linux's distribution of docker. for docker compose, i followed instructions [here](https://docs.docker.com/compose/install/other/#install-compose-standalone) and tweaked it for aarch64, and now, it's working! thanks for your support!

closed 2022-09-21 by monomycelium

paulwratt, 2022-09-23:
are you able to document how you got it to work here (ie. the specific "tweaks" you made)?

monomycelium, 2022-09-25:
to install docker-compose on arm64 linux, install the official binary from docker:

```
curl -SL https://github.com/docker/compose/releases/latest/download/docker-compose-linux-aarch64 -o /usr/local/bin/docker-compose
chmod +x /usr/local/bin/docker-compose
```