geerlingguy/raspberry-pi-pcie-devices #490: Add to CM4 board list: RAKwireless WisGate Connect

xoseperez, 2022-10-28:
[RAKwireless](https://www.rakwireless.com/en-us) is known for our LoRaWAN Gateways (and Helium hotspots). Now we have just released our first CM4-based gateway, the RAK7391 WisGate Connect. It has been designed to be a flexible edge gateway platform and support multiple wireless & wired connectivity options. I has been tested using LoRaWAN, 4G, Zigbee, WiFi HaLow, WiFi6, Modbus, Canbus,... and more in the future.

Some features:

- Accepts the complete range of CM4 modules
- 2x WisBlock IO connectors (https://store.rakwireless.com/pages/wisblock)
- 3x MiniPCIe slots (SPI, USB, I2C and PCIe interfaces)
- 1x M.2 B-Key interface (USB and PCIe interfaces)
- Flexible power supply modes such as DC terminal, Phoenix terminal and PoE (optional)
- Full size HDMI 2.0 connector
- 1 Gb Ethernet with PoE support (using an optional PoE module)
- 2.5 Gb Ethernet
- 1x USB2.0 port
- 2x USB3.0 ports
- MIPI DSI display FPC connector (22 pin 0.5 mm pitch cable)
- 2x MIPI CSI-2 camera FPC connectors (22 pin 0.5 mm pitch cable)
- Raspberry Pi HAT connector
- Onboard RTC, ADC, temperature sensor, fan driver and extra IO via two GPIO expanders.
- I2C connector for the OLED screen available in indoor enclosures

[image]

There are multiple modules and accessories compatible with the WisGate Connect. We have also 3 different enclosures: an aluminum desktop enclosure, a sheet metal one in 1U size to rack, and an outdoors enclosure.

[image]

[image]

[image]

Resources:

- Info about the product: https://docs.rakwireless.com/Product-Categories/WisGate/RAK7391/Overview/
- Store: https://store.rakwireless.com/products/wisgate-connect-base-kit-rak7391

geerlingguy, 2022-11-03:
[image]

I just received a WisGate Connect with an outdoor enclosure, and will begin testing it soon! It seems like a very well-built unit, and the WisBlock expansion capability makes it pretty plug-and-play for IoT endpoints. I'll probably include this in my next CM4 video.

geerlingguy, 2023-04-04:
Wow, been a while... but I'm finally going to release a video with this in it! Hopefully in a day or two.

geerlingguy, 2023-04-04:
This is now on the site: https://pipci.jeffgeerling.com/boards_cm/wisgate-connect.html

closed 2023-04-04 by geerlingguy