geerlingguy/raspberry-pi-pcie-devices #494: Add Intel I225 2.5 Gb ethernet controller

dmeziere, 2022-11-23:
I would like to test compatibility with an Intel I225-V 2,5 Gbits Ethernet card. 

I bought one, and it works in my PC using the igc kernel module under Ubuntu 22.04.01 LTS. This module is in the kernel, and also in the Raspberry Pi branch, but I am not able to install or compile it.

Would you have some tracks to follow, some links to give me, to help you add this controller to your list ?

geerlingguy, 2022-11-27:
Sure! The main thing would be to add the relevant drivers for the kernel and cross-compile it or recompile the kernel on the Raspberry Pi itself. See how I do it here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile

For a couple of similar cards, see: https://pipci.jeffgeerling.com/cards_network/intel-i340-t4-4-port-1g.html and https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/204 — specifically, you can see if there's a base driver you could install, for example:

  - https://www.intel.com/content/www/us/en/products/sku/184676/intel-ethernet-controller-i225v/downloads.html

But otherwise it seems like you'd need to enable the i225 driver in the kernel and recompile: https://github.com/torvalds/linux/tree/master/drivers/net/ethernet/intel/igc

dmeziere, 2022-11-27:
I've read all these links, and tried to jump in a first kernel compilation.

My environment is a CM4 on CM4IO with an AM1184e PCIe switch hosting an AM1166 SATA controller and the Intel I225-V (rev 03). The system is a 64bits openmediavault 6.0.46-5 (Shaitan) using kernel Linux 5.15.79-v8+. Not related, there is also an ATX power supply, an home made ATX power management circuit and an home made i2c fan controller.

I've installed the sources and ncurses for menuconfig :
`sudo apt install -y raspberrypi-kernel-headers libncurses-dev`

`cd /usr/src/linux-headers-5.15.76-v8+/`
`sudo make menuconfig`
In menuconfig, i found the proper module and checked it with an M.
`> Device Drivers > Network device support > Ethernet driver support > Intel(R) Ethernet Controller I225-LM/I225-V support`

But i'm not getting very far :
`sudo make`
`make[1]: *** No rule to make target 'arch/arm64/tools/gen-cpucaps.awk', needed by 'arch/arm64/include/generated/asm/cpucaps.h'.  Stop.`
`make: *** [arch/arm64/Makefile:173: archprepare] Error 2`

And i don't find any usable solution on the web

mi-hol, 2022-11-29:
@dmeziere did you see https://www.raspberrypi.com/documentation/computers/linux_kernel.html#kernel ?

dmeziere, 2022-11-29:
@mi-hol Thank you ! I really don't know where to start. I am a long time (~ten years) Linux-only user, but i never compiled a kernel ! Even in the 90s when i was trying old dead distros on a P133 and M68k.

dmeziere, 2022-11-29:
What i did :

- I followed these instructions to download, configure, and compile the kernel : https://www.raspberrypi.com/documentation/computers/linux_kernel.html#kernel
- Then i activated the igc module using `make menuconfig` and compiled again :) (it only compiled the module and kept the result from the first compilation)
- I installed like in the howto and rebooted
- The interface was active but not the link. I had to manually call `sudo dhclient enp5s0`

Here is the dmesg igc return :
`igc 0000:05:00.0: enabling device (0000 -> 0002)`
`igc 0000:05:00.0: PCIe PTM not supported by PCIe bus/controller`
`igc 0000:05:00.0 (unnamed net_device) (uninitialized): PHC added`
`igc 0000:05:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)`
`igc 0000:05:00.0 eth1: MAC: 88:c9:b3:bf:73:36`
`igc 0000:05:00.0 enp5s0: renamed from eth1`
`igc 0000:05:00.0 enp5s0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX`

I see SMB transfert rates about 1/3 faster than using the CM4IO on board Ethernet. For testing I copied a large (~50GB) file from the Raspberry Pi's HDD through SMB file sharing to my PC's NVME. My PC have the same Ethernet chipset on it's motherboard. But the switch is a gigabit one.

mi-hol, 2022-11-29:
good to hear, guess this is resolved, hence you should close this issue :)

closed 2022-11-29 by dmeziere

dmeziere, 2022-12-14:
I have closed the issue, but it doesn't appear on the devices list, and don't have any badge saying it is ongoing.

geerlingguy, 2022-12-16:
@dmeziere - Do you have a picture of the card or a link to the card you bought?

dmeziere, 2022-12-16:
This is the product link, but these products have a short TTL, you'd better take the picture from it :)

https://fr.aliexpress.com/item/1005003538822828.html?spm=a2g0o.order_list.order_list_main.15.21ef5e5bZyZptl&gatewayAdapt=glo2fra