geerlingguy/raspberry-pi-pcie-devices #495: Add Waveshare CM4-DUAL-ETH-MINI board

jhe2, 2022-12-01:
The [Waveshare CM4-DUAL-ETH-MINI](https://www.waveshare.com/wiki/CM4-DUAL-ETH-MINI) board is similar to the [DFRobot IoT Router Carrier Board Mini](https://pipci.jeffgeerling.com/boards_cm/dfrobot-cm4-iot-router.html), but with a 40-pin I/O header and a USB 2.0 type A port instead of USB-C.

zhuzhe1983, 2023-02-16:
It also comes with a CNC case

[image]

[https://www.aliexpress.us/item/3256804871861623.html](https://www.aliexpress.us/item/3256804871861623.html)

dcousens, 2023-11-18:
I think this board might be nearer to the [DFRobot IoT Router Carrier Board Mini](https://pipci.jeffgeerling.com/boards_cm/dfrobot-cm4-iot-router.html) than the other alternatives. [Waveshare documentation](https://www.waveshare.com/wiki/CM4-DUAL-ETH-MINI) seems to indicate that it is using `kmod-r8169` for the second ethernet port, similar to the DFRobot board.

@mayniklas maybe you could share your experience with this board, as I found your project https://github.com/MayNiklas/openwrt-pi-cm4? 🚀

MayNiklas, 2023-11-19:
> I think this board might be nearer to the [DFRobot IoT Router Carrier Board Mini](https://pipci.jeffgeerling.com/boards_cm/dfrobot-cm4-iot-router.html) than the other alternatives. [Waveshare [...]

The board uses a PCI-e NIC for the second NIC.
I personally really like it: it has great documentation and was quite cheap in Europe (30€). It's also easily available around the world.
A friend helped me designing a 3D printable case we will soon publish.

Running NixOS (declerative Linux distribution), everything worked out of the box.
Together with @pinpox, we are working on a NixOS router using this hardware: https://github.com/pinpox/nixos-raspberrypi-router 

My biggest complaint: the second NIC creates a high single core load. I noticed a big jump in CPU interrupts.
Compared to the integrated NIC, the CPU load this NIC causes is around 10-20 times higher.
-> around 850-900 mbit will be the limit without overclocking the Pi
-> might be able to figure out some tuning

Contact me in case you need any specific tests!

jhe2, 2023-11-19:
In case anyone is interested, I created a [3d-printable case](https://www.thingiverse.com/thing:6322858) for this board.

When it comes to networking performance and CPU load, I would recommend [this blog post](https://blog.zakkemble.net/routerpi-compute-module-4-router/). This is not about this exact board, but a similar board with the same PCIe Realtek NIC. In that post the author assigns the different interrupts to individual CPU cores to distribute the load more evenly. I did the suggested optimizations and can confirm the author's conclusion, that the CM4's internal NIC causes a much higher CPU load than the Realtek one, which is barely noticable. Overall I like the board and it has been working quite well for me.