geerlingguy/raspberry-pi-pcie-devices #500: Add EDATEC CM4 Sensing Computer

JohnZong66, 2023-01-09:
The [EDATEC CM4 Sensing Computer](https://www.edatec.cn/en/Product/Embedded_Computers/2021/0930/132.html) is an industrial computer designed for IoT & Data Acquisition  applications based on Raspberry Pi Compute Module 4.
It provides 4G/LTE module, 4 channels RS485,1 channel RS232, 1 channel CAN, 1 channel Gigabit Ethernet port, alarm buzzer ,battery-powered real-time clock, 2 channels USB2.0 and 1 channel Micro USB2.0 interface.  

[image]  

[image]  

[image]

JohnZong66, 2023-01-12:
Good day. I am John from EDATEC. Below is the link to my submit pull request. Could you help check it? Thank.
[Add EDATEC CM4 Sensing Computer](https://github.com/geerlingguy/raspberry-pi-pcie-devices/pull/501)

geerlingguy, 2023-04-04:
@JohnZong66 - Thanks! This is now on the site: https://pipci.jeffgeerling.com/boards_cm/edatec-cm4-sensing-computer.html

closed 2023-04-04 by geerlingguy