geerlingguy/raspberry-pi-pcie-devices #506: ASUS Xonar AE 7.1 - Tested & working

siraaris, 2023-02-06:
Tested the Xonar AE 7.1 PCIe sound card on a CM4 IO Board (4GB version).

Works with rpi-5.15.y branch (5.15.91-v8+ Kernel version).

Add "dtoverlay=pcie-32bit-dma" to /boot/config.txt

6by9, 2023-02-06:
Can you say which kernel module is required for that card? ```lspci -k``` will tell you that.
Knowing the PCI vendor and device ID also helps in identifying the known good (or bad) device. ```lspci -nn```

geerlingguy, 2023-02-07:
In addition—did you have to install any particular drivers, or recompile the kernel?

siraaris, 2023-02-07:
Nod. Yes, needed to enable additional kernel options to get USB bus and PCI to work and also sound modules. 

I've just been looking for a .config diff tool so I can outline what is needed. 

I'll provide all requested detail in the next couple of days.

siraaris, 2023-02-08:
Starting from Raspberry Pi OS: `2022-09-22-raspios-bullseye-arm64-lite.img`, Kernel 5.15.61-v8+

Compile custom kernel:

Git: https://github.com/raspberrypi/linux, Branch: rpi-5.15.y

I cross compiled:

`KERNEL=kernel8`
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig`

Update generated .config based on the following:

 LEDS_TRIGGER_AUDIO n -> m
 LOCALVERSION "-v8" -> "-v8-cexeaudio"
 PCIE_MICROCHIP_HOST n -> y
 PCI_ENDPOINT n -> y
 PCI_HOST_GENERIC n -> y
 PCI_QUIRKS y -> n
 SND_HDA_INTEL n -> m
 SND_SOC_CS35L32 n -> m
 SND_SOC_CS35L33 n -> m
 SND_SOC_CS35L34 n -> m
 SND_SOC_CS35L35 n -> m
 SND_SOC_CS35L36 n -> m
 SND_SOC_CS4234 n -> m
 SND_SOC_CS4270 n -> m
 SND_SOC_CS4271_SPI n -> m
 SND_SOC_CS42L42 n -> m
 SND_SOC_CS42L51_I2C n -> m
 SND_SOC_CS42L52 n -> m
 SND_SOC_CS42L56 n -> m
 SND_SOC_CS42L73 n -> m
 SND_SOC_CS43130 n -> m
 SND_SOC_CS4341 n -> m
 SND_SOC_CS4349 n -> m
 SND_SOC_CS53L30 n -> m
 USB_EHCI_HCD n -> y
 USB_HCD_BCMA n -> m
 USB_OHCI_HCD n -> y
 USB_UHCI_HCD n -> y
 USB_XHCI_DBGCAP n -> y
+PCIE_CADENCE_PLAT_EP n
+PCIE_DW_PLAT_EP n
+PCI_ECAM y
+PCI_ENDPOINT_CONFIGFS n
+PCI_EPF_NTB n
+PCI_EPF_TEST n
+PCI_HOST_COMMON y
+PCI_J721E_EP n
+SND_CTL_LED m
+SND_HDA m
+SND_HDA_CODEC_ANALOG m
+SND_HDA_CODEC_CA0110 n
+SND_HDA_CODEC_CA0132 n
+SND_HDA_CODEC_CIRRUS m
+SND_HDA_CODEC_CMEDIA m
+SND_HDA_CODEC_CONEXANT n
+SND_HDA_CODEC_CS8409 m
+SND_HDA_CODEC_HDMI n
+SND_HDA_CODEC_REALTEK m
+SND_HDA_CODEC_SI3054 n
+SND_HDA_CODEC_SIGMATEL n
+SND_HDA_CODEC_VIA n
+SND_HDA_CORE m
+SND_HDA_GENERIC m
+SND_HDA_GENERIC_LEDS y
+SND_HDA_HWDEP n
+SND_HDA_INPUT_BEEP n
+SND_HDA_INTEL_HDMI_SILENT_STREAM n
+SND_HDA_PATCH_LOADER n
+SND_HDA_POWER_SAVE_DEFAULT 0
+SND_HDA_RECONFIG n
+SND_INTEL_DSP_CONFIG m
+SND_SOC_CS42L51 m
+USB_EHCI_FSL n
+USB_EHCI_HCD_PLATFORM y
+USB_EHCI_PCI y
+USB_EHCI_ROOT_HUB_TT y
+USB_EHCI_TT_NEWSCHED y
+USB_OHCI_HCD_PCI y
+USB_OHCI_HCD_PLATFORM y
+USB_SISUSBVGA n

NOTE: Some of the above may not be required, but it results in a configuration that works.

Create custom kernel, deploy and set `dtoverlay=pcie-32bit-dma` to /boot/config.txt.

Final output on working system:

`root@cexe-BR:~# lspci -k`
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
	Subsystem: ASUSTeK Computer Inc. ASM1042A USB 3.0 Host Controller
	Kernel driver in use: xhci_hcd

`root@cexe-BR:~# aplay -l`
**** List of PLAYBACK Hardware Devices ****
card 0: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: CARD [XONAR SOUND CARD], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: CARD [XONAR SOUND CARD], device 1: USB Audio [USB Audio #1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: CARD [XONAR SOUND CARD], device 2: USB Audio [USB Audio #2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

And because this is presented via USB behind the PCI device:

`root@cexe-BR:~# lsusb`
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0b05:180f ASUSTek Computer, Inc. XONAR SOUND CARD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

siraaris, 2023-02-08:
Simple check:

root@cexe-BR:~# `speaker-test -D hw:CARD -r 192000 -c 8 -F S24_3LE`

speaker-test 1.2.4

Playback device is hw:CARD
Stream parameters are 192000Hz, S24_3LE, 8 channels
Using 16 octaves of pink noise
Rate set to 192000Hz (requested 192000Hz)
Buffer size range from 48 to 384000
Period size range from 24 to 192000
Using max buffer size 384000
Periods = 4
was set period_size = 96000
was set buffer_size = 384000
 0 - Front Left
 2 - Front Center
 1 - Front Right
 7 - Side Right
 5 - Rear Right
 4 - Rear Left
 6 - Side Left
 3 - LFE
Time per period = 20.227230

geerlingguy, 2023-02-09:
@siraaris - Thanks so much for all the details! We'll have to get this board added to the site :)

6by9, 2023-02-10:
I'm actually a little surprised that you needed any additional kernel config options.
```
01:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
Subsystem: ASUSTeK Computer Inc. ASM1042A USB 3.0 Host Controller
Kernel driver in use: xhci_hcd
```
xhci_hcd is the standard USB3 host interface module, also used with the Via VL805 on any Pi4.

```
USB ID 0b05:180f ASUSTek Computer, Inc. XONAR SOUND CARD
```
I would expect that to be a standards compliant USB audio interface, in which case CONFIG_SND_USB_AUDIO should cover it, and that is included in the standard Pi kernel builds.
https://linux-hardware.org/index.php?id=usb:0b05-180f seems to confirm that too.

siraaris, 2023-02-11:
Nod. I did try the default kernel but I'll try again just to see if I can work out what the minimal kernel mods are.

siraaris, 2023-12-28:
A bit of an update, hopefully clarifying some of the above.

Adding dtoverlay=pcie-32bit-dma to /boot/config.txt is sufficient to get the card working on a Debian kernel (linux-image-6.1.0-16-rt-arm64), as I suspect the same for Raspberry OS kernels.

Kernel panics occur every so often (1 in 2?) reboots, before root is mounted etc.

When there is a successful boot, dmesg shows a slew of the following:

[    3.846796] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[    3.846836] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[    3.846845] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00000080/00002000
[    3.846857] pcieport 0000:00:00.0:    [ 7] BadDLLP               

After boot, the card though seems to be present and able to be used.

Issue above seems similar to a kernel but report [here](https://github.com/raspberrypi/linux/issues/5659).