geerlingguy/raspberry-pi-pcie-devices #510: Add Intel Arc A750

geerlingguy, 2023-02-10:
I am testing an [Intel Arc A750 8GB](https://amzn.to/3IeEQKL) graphics card:

TODO: Image goes here.

It requires external PCIe power, so needs to be used with a suitable power supply. Intel seems to only officially support [Ubuntu 22.04](https://www.intel.com/content/www/us/en/download/747008/intel-arc-graphics-driver-ubuntu.html), and by default only AMD64/x86_64... Here's the full [Ubuntu installation guide for Arc drivers](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html).

Not sure if I can get it to work at all on Debian, and I also haven't confirmed if aarch64/ARM64 support is at all existing, planned, or not planned. Would like to find out, because I'll soon have another ARM64 system to test this in with a bit more firepower.

geerlingguy, 2023-02-10:
There's a good deal of coil whine on my card—when testing on the PC and it needs to render anything. At idle there isn't much, but when the fans stop spinning and the card is not displaying anything, or alternatively when the GPU is under heavy load, there's a lot of coil whine, enough to be pretty distracting.

geerlingguy, 2023-02-10:
On Twitter someone mentioned this commit: https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/gpu/drm/i915/Kconfig?h=next-20230210 — it looks like the i915 has some hardcoded x86 dependencies:

```
# SPDX-License-Identifier: GPL-2.0-only
config DRM_I915
	tristate "Intel 8xx/9xx/G3x/G4x/HD Graphics"
	depends on DRM
	depends on X86 && PCI
	depends on !PREEMPT_RT
```

geerlingguy, 2023-02-15:
lspci output:

```
01:00.0 PCI bridge: Intel Corporation Device 4fa0 (rev 01) (prog-if 00 [Normal decode])
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 610000000 (64-bit, prefetchable) [disabled] [size=8M]
	Bus: primary=01, secondary=02, subordinate=04, sec-latency=0
	I/O behind bridge: 0000f000-00000fff [disabled]
	Memory behind bridge: d8000000-d9bfffff [size=28M]
	Prefetchable memory behind bridge: 00000000c0000000-00000000cfffffff [size=256M]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Upstream Port, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 0.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 16GT/s, Width x16, ASPM L1, Exit Latency L1 <64us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP+ LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS+
			 AtomicOpsCap: Routing+ 32bit+ 64bit+ 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled,
			 AtomicOpsCtl: EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS+
		LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: Upstream Port
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [158 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [188 v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [1b8 v1] Lane Margining at the Receiver <?>
	Capabilities: [20c v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [214 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=14us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=29696ns
[... 244 lines cut ...]
[    1.871971] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 04
[    1.872036] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x617ffffff 64bit pref]
[    1.872056] pci 0000:00:00.0: BAR 8: assigned [mem 0x618000000-0x619bfffff]
[    1.872082] pci 0000:01:00.0: BAR 9: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.872099] pci 0000:01:00.0: BAR 0: assigned [mem 0x610000000-0x6107fffff 64bit pref]
[    1.872143] pci 0000:01:00.0: BAR 8: assigned [mem 0x618000000-0x619bfffff]
[    1.872162] pci 0000:02:01.0: BAR 9: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.872179] pci 0000:02:01.0: BAR 8: assigned [mem 0x618000000-0x6197fffff]
[    1.872194] pci 0000:02:04.0: BAR 8: assigned [mem 0x619800000-0x6198fffff]
[    1.872213] pci 0000:03:00.0: BAR 2: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.872258] pci 0000:03:00.0: BAR 0: assigned [mem 0x618000000-0x618ffffff 64bit]
[    1.872301] pci 0000:03:00.0: BAR 6: assigned [mem 0x619000000-0x6191fffff pref]
[    1.872318] pci 0000:02:01.0: PCI bridge to [bus 03]
[    1.872344] pci 0000:02:01.0:   bridge window [mem 0x618000000-0x6197fffff]
[    1.872366] pci 0000:02:01.0:   bridge window [mem 0x600000000-0x60fffffff 64bit pref]
[    1.872399] pci 0000:04:00.0: BAR 0: assigned [mem 0x619800000-0x619803fff 64bit]
[    1.872445] pci 0000:02:04.0: PCI bridge to [bus 04]
[    1.872470] pci 0000:02:04.0:   bridge window [mem 0x619800000-0x6198fffff]
[    1.872508] pci 0000:01:00.0: PCI bridge to [bus 02-04]
[    1.872531] pci 0000:01:00.0:   bridge window [mem 0x618000000-0x619bfffff]
[    1.872552] pci 0000:01:00.0:   bridge window [mem 0x600000000-0x60fffffff 64bit pref]
[    1.872580] pci 0000:00:00.0: PCI bridge to [bus 01-04]
[    1.872600] pci 0000:00:00.0:   bridge window [mem 0x618000000-0x619bfffff]
[    1.872618] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x617ffffff 64bit pref]
```

geerlingguy, 2023-02-15:
To test out Intel's open source kernel driver, I cloned the 6.2.y branch of Raspberry Pi's linux fork:

```
git clone -b rpi-6.2.y https://github.com/raspberrypi/linux.git && cd linux
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig
  --> TODO
make -j8 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
```

geerlingguy, 2023-02-15:
...except I can't select the Intel Graphics support using menuconfig, because of the dependencies:

```
Symbol: DRM_I915 [=n]                                                                                           │  
  │ Type  : tristate                                                                                                │  
  │ Defined at drivers/gpu/drm/i915/Kconfig:2                                                                       │  
  │   Prompt: Intel 8xx/9xx/G3x/G4x/HD Graphics                                                                     │  
  │   Depends on: HAS_IOMEM [=y] && DRM [=m] && X86 && PCI [=y] && !PREEMPT_RT [=n]                                 │  
  │   Location:                                                                                                     │  
  │     -> Device Drivers                                                                                           │  
  │ (1)   -> Graphics support                                                                                       │  
  │         -> Intel 8xx/9xx/G3x/G4x/HD Graphics (DRM_I915 [=n])                                                    │  
  │ Selects: INTEL_GTT [=n] && INTERVAL_TREE [=y] && SHMEM [=y] && TMPFS [=y] && DRM_DISPLAY_DP_HELPER [=n] && DRM_
```

closed 2023-02-15 by geerlingguy

Mis012, 2023-04-14:
Just FYI, the most likely reason x86 is listed as a dependency is that historically Intel only had iGPUs, and those can't exactly be plugged into an aarch64 system for obvious reasons. That being said, it seems quite likely that if those dependencies were removed (since they are for all we know completely bogus), chances are the driver would actually compile just fine. It seems quite unlikely that it uses assembly for anything, which is just about the only valid reason for it to not compile for aarch64

pimzand, 2024-03-31:
It looks like you could try again with Xe drivers in kernel 6.8:
https://www.phoronix.com/review/intel-xe-benchmark

DanaGoyette, 2024-05-01:
If you do decide to test this with 6.8, note that Ubuntu's Mesa doesn't have the Intel drivers enabled, and even if you check out the experimental branch of the mesa packaging from debian's "salsa" repo, it doesn't enable the Intel drivers for armhf or arm64/aarch64.  You'll want to edit `debian/rules`:

```
   # etnaviv, tegra, vc4 and v3d kernel support are only available on armhf and arm64
   ifneq (,$(filter $(DEB_HOST_ARCH), armhf arm64))
-       GALLIUM_DRIVERS += etnaviv panfrost svga tegra vc4 v3d
-       VULKAN_DRIVERS += broadcom freedreno
+       GALLIUM_DRIVERS += iris etnaviv panfrost svga tegra vc4 v3d
+       VULKAN_DRIVERS += intel intel_hasvk broadcom freedreno
   endif
```

On my LX2160A, the kernel driver for my A770 loads once I add xe.force_probe=56a0.  It complains about lack of resizable BAR, though.  With a manually rebuilt Mesa, Weston works, but gdm3 and sddm both glitch out and freeze, and autologin doesn't give a working desktop either.

```
[ 4908.671096] xe 0004:03:00.0: [drm] Timedout job: seqno=4294967224, guc_id=7, flags=0x0
[ 4908.679401] xe 0004:03:00.0: [drm] Engine reset: guc_id=7
```

If you can get the driver to work without glitching out, you'll also run into these *chirping crickets*:
https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/234

(I actually bought the A770 to use with arm64, but now it's just sitting unused.)

geerlingguy, 2024-05-23:
@karatekid430 - Sadly, arm64 driver support on Windows for anything besides USB devices is pretty much nonexistent so far. I'm not sure if any graphics card (let alone NICs and other simpler devices) will support PCIe (especially through Thunderbolt/external enclosures) any time soon. Would love to be proven wrong on that!

martinx72, 2024-11-12:
Some progress here with my Arc a750

```
[    4.093330] Registered IR keymap rc-cec
[    4.095023] rc rc0: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0
[    4.095088] input: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0/input8
[    4.160314] usbcore: registered new interface driver xpad
[    4.179470] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    4.180283] xe 0000:03:00.0: [drm] Found DG2/G10 (device ID 56a1) display version 13.00
[    4.187393] Bluetooth: Core ver 2.22
[    4.187417] NET: Registered PF_BLUETOOTH protocol family
[    4.187419] Bluetooth: HCI device and connection manager initialized
[    4.187424] Bluetooth: HCI socket layer initialized
[    4.187427] Bluetooth: L2CAP socket layer initialized
[    4.187430] Bluetooth: SCO socket layer initialized
[    4.190074] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.29.2
[    4.190221] ------------[ cut here ]------------
[    4.190223] kernel BUG at drivers/gpu/drm/drm_gem.c:155!
[    4.195556] Internal error: Oops - BUG: 00000000f2000800 [#1] PREEMPT SMP
[    4.202369] Modules linked in: sha1_ce bluetooth sha1_generic xe(+) snd_soc_rpi_simple_soundcard ecdh_generic raspberrypi_hwmon ecc binfmt_misc brcmfmac libaes i2c_algo_bit drm_buddy xpad brcmutil vc4(+) drm_suballoc_helper drm_gpuvm snd_soc_hdmi_codec ff_memless cec drm_exec rpivid_hevc(C) cfg80211 i2c_brcmstb drm_display_helper pisp_be drm_ttm_helper spi_bcm2835 ttm drm_dma_helper v4l2_mem2mem videobuf2_dma_contig videobuf2_memops designware_i2s hid_logitech_dj(+) v3d snd_soc_pcm5102a rfkill gpu_sched videobuf2_v4l2 videodev snd_soc_core snd_compress drm_shmem_helper gpio_keys videobuf2_common snd_pcm_dmaengine drm_kms_helper snd_pcm mc pwm_fan snd_timer snd raspberrypi_gpiomem rp1_adc nvmem_rmem uio_pdrv_genirq uio uinput drm i2c_dev fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[    4.274521] CPU: 1 UID: 0 PID: 327 Comm: (udev-worker) Tainted: G         C         6.11.7-v8-16k+ #1
[    4.283779] Tainted: [C]=CRAP
[    4.286749] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    4.292600] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.299587] pc : drm_gem_private_object_init+0xac/0xb8 [drm]
[    4.305309] lr : ___xe_bo_create_locked+0x10c/0x3c0 [xe]
[    4.310758] sp : ffffc00080b0b500
[    4.314077] x29: ffffc00080b0b510 x28: 0000000000000000 x27: ffff800007b61248
[    4.321240] x26: ffff800007b60000 x25: 0000000000012022 x24: 0000000000000001
[    4.328402] x23: 0000000000000000 x22: 0000000000000000 x21: 0000000000000000
[    4.335564] x20: 000000000001a000 x19: ffff80000a5e3000 x18: ffffffffffffffff
[    4.342727] x17: 622e30375f637567 x16: ffffd06fce74f7d8 x15: d50e1386eb679192
[    4.349889] x14: 984f45474329738b x13: d50e1386eb679192 x12: 984f45474329738b
[    4.357051] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06f8ab89654
[    4.364213] x8 : ffff80000a5e3400 x7 : 0000000000000000 x6 : 000000000000003f
[    4.371375] x5 : 0000000000000040 x4 : 0000000000000400 x3 : ffff80000a5e3340
[    4.378537] x2 : 000000000001a000 x1 : ffff80000a5e3000 x0 : ffff800007b60000
[    4.385699] Call trace:
[    4.388146]  drm_gem_private_object_init+0xac/0xb8 [drm]
[    4.393511]  __xe_bo_create_locked+0x194/0x318 [xe]
[    4.398490]  xe_bo_create_pin_map_at+0x74/0x1d0 [xe]
[    4.403552]  xe_managed_bo_create_pin_map+0x34/0x110 [xe]
[    4.409050]  xe_guc_log_init+0x44/0xb0 [xe]
[    4.413326]  xe_guc_init+0x9c/0x260 [xe]
[    4.417339]  xe_uc_init+0x24/0xc8 [xe]
[    4.421179]  xe_gt_init_hwconfig+0x6c/0xd8 [xe]
[    4.425802]  xe_device_probe+0x20c/0x578 [xe]
[    4.430251]  xe_pci_probe+0x68c/0x9c0 [xe]
[    4.434438]  local_pci_probe+0x48/0xb8
[    4.438198]  pci_device_probe+0xac/0x1b8
[    4.442129]  really_probe+0xc4/0x2a8
[    4.445712]  __driver_probe_device+0x80/0x140
[    4.450080]  driver_probe_device+0x44/0x170
[    4.454273]  __driver_attach+0x9c/0x1b0
[    4.458117]  bus_for_each_dev+0x80/0xe8
[    4.461963]  driver_attach+0x2c/0x40
[    4.465546]  bus_add_driver+0xec/0x218
[    4.469302]  driver_register+0x68/0x138
[    4.473145]  __pci_register_driver+0x54/0x68
[    4.477426]  xe_register_pci_driver+0x30/0x48 [xe]
```

xe drivers built, updated firmware. 
booted and it detected the card, need more time to trace into the root cause.

note:
i use the RPi  6.11.y , turn the intel xe on via make menuconfig,
and force it to probe the vid/pid (i manually add it in the .config)
and update the firmware manually (got it from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/snapshot/linux-firmware-20241110.tar.gz )

geerlingguy, 2024-11-12:
@martinx72 Excited to re-open this issue ;)

pimzand, 2024-11-13:
Fedora 41 for aarch64 has xe kernel support out of the box.

martinx72, 2024-11-13:
> Fedora 41 for aarch64 has xe kernel support out of the box.

@pimzand  Does this release support Pi 5?

And some more progress here, at least no crash in the driver

```
[    4.909408] Setting dangerous option force_probe - tainting kernel
[    4.910326] [drm] Initialized vc4 0.0.0 for axi:gpu on minor 1
[    4.912910] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    4.916179] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    4.916328] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    4.919959] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    4.921582] xe 0000:03:00.0: [drm] Found DG2/G10 (device ID 56a1) display version 13.00 stepping C0
[    4.965268] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    4.965517] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    4.976072] input: Microsoft Xbox Series S|X Controller as /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.0/input/input6
[    4.976364] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.29.2
[    4.977619] usbcore: registered new interface driver xpad
[    4.989267] xe 0000:03:00.0: [drm] *ERROR* GT0: GuC mmio request 0x4509: failure 0x306 hint 0x0
[    4.989276] xe 0000:03:00.0: [drm] *ERROR* GT0: Failed to enable GuC CT (-ENXIO)
[    5.021751] Bluetooth: Core ver 2.22
[    5.021783] NET: Registered PF_BLUETOOTH protocol family
[    5.021784] Bluetooth: HCI device and connection manager initialized
[    5.021790] Bluetooth: HCI socket layer initialized
[    5.021792] Bluetooth: L2CAP socket layer initialized
```

Page Size set as 4K, then it went to some error much easier for me to narrow down the crash issue, but, i think i probably have to stop here. As intel keep modifying their repo every single day here: https://gitlab.freedesktop.org/drm/xe/kernel

IMO, never can chase their steps by myself. :(

pimzand, 2024-11-13:
> @pimzand Does this release support Pi 5?

Not fully, apparently: https://discussion.fedoraproject.org/t/aarch64-support-for-raspberry-pi-5/134857

I tried an Arc 310 Eco in an Ampere Altra system while still Fedora 40.
Xe support was introduced with F40 kernel 6.11.
The driver loaded without errors, but the screen stayed black.

martinx72, 2024-11-14:
> Not fully, apparently: https://discussion.fedoraproject.org/t/aarch64-support-for-raspberry-pi-5/134857
> 
> I tried an Arc 310 Eco in an Ampere Altra system while still Fedora 40. Xe support was introduced with F40 kernel 6.11. The driver loaded without errors, but the screen stayed black.

Just check the link in the post, https://rpmfusion.org/Howto/RaspberryPi
It says 'using Raspberry Pi Foundation [downstream kernel](https://github.com/raspberrypi/linux) and components from Fedora userspace.'

Then it is 6.6,y, and it never can boot with xe drive at all.

DanaGoyette, 2024-11-14:
> @karatekid430 - Sadly, arm64 driver support on Windows for anything besides USB devices is pretty much nonexistent so far. I'm not sure if any graphics card (let alone NICs and other simpler [...]

Intel doesn't have any ARM64 Windows drivers for any of their NICs I've checked.  NVIDIA/Mellanox, on the other hand, *does* have in-the-box ARM64 Windows drivers for ConnectX-4 and newer.

I've used a Mellanox ConnectX-4 Lx 10-/25-gigabit ethernet NIC with my Snapdragon X machine on Windows ARM64, via USB4 / Thunderbolt.  It works, but oddly, one direction seems to give only 5 gigabits of throughput for some reason.

As far as I can tell, there are no discrete GPU drivers for Windows ARM64.  On ARM64 machines with UEFI, such as LX2160A, the closest you can get is loading AMD's GOP driver in UEFI to run the framebuffer with Microsoft Basic Graphics Adapter.

geerlingguy, 2024-11-14:
@DanaGoyette From what I've heard, Nvidia may have working drivers for Windows arm64, as they've worked on support for enterprise use in Azure... at least that's what I infer from tweets like [this](https://x.com/JoeSpeeds/status/1843061853586661546).

FibreFoX, 2024-11-24:
@martinx72 Can you add some steps, what you've done so far? Like what RaspiOS are you using, and are you trying X11 or Wayland? So far I understand you compiled the 6.11.y branch, selected Intel XE via menuconfig and added some firmware binaries to it. Did you, by any chance, create some small script to do all of that?

I have a spare 8GB Raspi5 with a spare Intel Arc 770 here for some experiments, maybe I can help, even tho I am very new to the kernel hacking game.

martinx72, 2024-11-25:
> @martinx72 Can you add some steps, what you've done so far? Like what RaspiOS are you using, and are you trying X11 or Wayland? So far I understand you compiled the 6.11.y branch, selected Intel XE [...]

Check the way to build kernel for Pi5 aarch64: https://www.raspberrypi.com/documentation/computers/linux_kernel.html
just few commands to type, i believe you dont need any script just for building the kernel

And use make menuconfig to enable the intel Xe driver, and pagesize as 4K. that's all. 

And you can also try 6.12.y, i tried both, all crashed there. and u can also try the intel xe driver source stream here: https://gitlab.freedesktop.org/drm/xe/kernel

pepijndevos, 2024-11-25:
There are a few more menuconfig entries in the AMD article that could be worth trying 
https://www.jeffgeerling.com/blog/2024/amd-radeon-pro-w7700-running-on-raspberry-pi

FibreFoX, 2024-11-25:
So far using a very naive approach does not bring me very far. I am using the current 6.12.y branch of the raspberry kernel and starting with a fresh installation of Raspberry Pi OS (64bit, desktop).

Some commands I am using as root user:
```sh
# making sure to have up2date system
apt-get update
apt upgrade

# installing some build dependencies (might contain unneeded stuff, copy-pasted a lot from several of my own projects)
apt-get install -y bc bison flex libssl-dev make git
apt-get install -y libncurses5-dev
apt-get install -y xz-utils kmod
apt-get install -y libelf-dev libssl-dev

# clone the raspberry kernel, at time of writing at commit 05678eba0b1fc51355b03b0d3f05f059068baf3e
cd /opt
git clone --depth=1 --branch rpi-6.12.y https://github.com/raspberrypi/linux
cd linux

# adjusting kernel compile configuration
make bcm2712_defconfig
make menuconfig
```

Then I select the following options (the ones from the AMD article are not there, because the kernel-fork introduces them):
```
# Kernel Features > Page Size > 4 KB
# Kernel Features > Virtual address space size > 48-bit (had to change this in a different project, so I hoped this would help too)
# Kernel Features > Kernel support for 32-bit EL0 > Fix up misaligned multi-word loads and stores in user space
# Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > Intel Xe Graphics (is disabled by default)
# Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > Intel Xe Graphics > Force probe xe for selected Intel hardware IDs > 56a0
```

The ID `56a0` matches my GPU from the list https://dgpu-docs.intel.com/devices/hardware-table.html

Then the boring compile stuff.

```sh
make -j$(nproc) Image.gz modules dtbs
make modules_install
# installing the generated files
cp arch/arm64/boot/dts/broadcom/*.dtb /boot/firmware/
cp arch/arm64/boot/dts/overlays/*.dtb* /boot/firmware/overlays/
cp arch/arm64/boot/dts/overlays/README /boot/firmware/overlays/
cp arch/arm64/boot/Image.gz /boot/firmware/kernel8-xe.img

# selecting the new compiled kernel
echo "kernel=kernel8-xe.img" | tee -a /boot/firmware/config.txt
```

Out of the box I get hit by outdated firmware files:
```
[    3.619110] vc4-drm axi:gpu: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
[    3.622657] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    3.622702] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    3.623992] Registered IR keymap rc-cec
[    3.624050] rc rc0: vc4-hdmi-0 as /devices/platform/soc@107c000000/107c701400.hdmi/rc/rc0
[    3.624104] input: vc4-hdmi-0 as /devices/platform/soc@107c000000/107c701400.hdmi/rc/rc0/input5
[    3.626312] input: vc4-hdmi-0 HDMI Jack as /devices/platform/soc@107c000000/107c701400.hdmi/sound/card0/input6
[    3.626939] xe 0000:03:00.0: [drm] Found DG2/G10 (device ID 56a0) display version 13.00 stepping C0
[    3.627343] vc4-drm axi:gpu: bound 107c701400.hdmi (ops vc4_hdmi_ops [vc4])
[    3.629688] Registered IR keymap rc-cec
[    3.629744] rc rc1: vc4-hdmi-1 as /devices/platform/soc@107c000000/107c706400.hdmi/rc/rc1
[    3.629785] input: vc4-hdmi-1 as /devices/platform/soc@107c000000/107c706400.hdmi/rc/rc1/input7
[    3.631235] input: vc4-hdmi-1 HDMI Jack as /devices/platform/soc@107c000000/107c706400.hdmi/sound/card1/input8
[    3.633675] vc4-drm axi:gpu: bound 107c706400.hdmi (ops vc4_hdmi_ops [vc4])
[    3.633871] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    3.633950] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    3.634013] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.634100] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[... 79 lines cut ...]
[    4.269663]  xe_gt_init_hwconfig+0x7c/0xd8 [xe]
[    4.274288]  xe_device_probe+0x204/0x570 [xe]
[    4.278739]  xe_pci_probe+0x69c/0xa20 [xe]
[    4.282928]  local_pci_probe+0x48/0xb8
[    4.286688]  pci_device_probe+0xc0/0x1c0
[    4.290620]  really_probe+0xc4/0x2a8
[    4.294205]  __driver_probe_device+0x80/0x140
[    4.298574]  driver_probe_device+0x44/0x170
[    4.302768]  __driver_attach+0x9c/0x1b0
[    4.306613]  bus_for_each_dev+0x80/0xe8
[    4.310458]  driver_attach+0x2c/0x40
[    4.314041]  bus_add_driver+0xec/0x218
[    4.317799]  driver_register+0x68/0x138
[    4.321642]  __pci_register_driver+0x54/0x68
[    4.325923]  xe_register_pci_driver+0x30/0x48 [xe]
[    4.330810]  xe_init+0x3c/0xb8 [xe]
[    4.334388]  do_one_initcall+0x60/0x2a0
[    4.338235]  do_init_module+0x68/0x250
[    4.341993]  load_module+0x1fe0/0x20c8
[    4.345749]  __do_sys_init_module+0x180/0x1f0
```

Not sure if this helps someone, just wanted to post this here so others can find this issue.

My goal was running my A770 on a Pi5 for video transcoding feature of my home Jellyfin server

pepijndevos, 2024-11-26:
> Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/xe/kernel/issues/new

ifdu, 2024-11-26:
An issue was just opened to track progress here: https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3594

geerlingguy, 2024-11-26:
Could... Intel actually beat Nvidia to supporting the Pi?! :D

HeyMeco, 2024-11-26:
> Could... Intel actually beat Nvidia to supporting the Pi?! :D

Back at Embedded World 24 Intel wouldn't want to acknowledge Arc x Arm to me but that didn't hold me back for trying the A310 with RK3588. Didn't get it to be recognized at all in linux but in uefi I was able to list it with the other PCIe devices. We never got graphics output though :/

FibreFoX, 2024-11-27:
@ifdu Thanks for opening the issue on that other project. Sadly even with your patch it is not yet working for me, neither with 6.11.y nor 6.12.y ... when playing with 4K page size I am always greeted with `firmware signature verification failed`.

When trying to use your patch **without touching pagesize nor VA size** on the `6.11.y` branch, I was greeted by `firmware production part check failure`: 
```
[    6.825035] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    6.825320] xe 0000:03:00.0: [drm] Found DG2/G10 (device ID 56a0) display version 13.00
[    6.832466] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.29.2
[    6.835837] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status = 0x40000056, time = 0ms, freq = 2400MHz (req 2400MHz), done = -1
[    6.835843] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status: Reset = 0, BootROM = 0x2B, UKernel = 0x00, MIA = 0x00, Auth = 0x01
[    6.835846] xe 0000:03:00.0: [drm] *ERROR* GT0: firmware production part check failure
[    6.835851] xe 0000:03:00.0: [drm] *ERROR* CRITICAL: Xe has declared device 0000:03:00.0 as wedged.
               IOCTLs and executions are blocked. Only a rebind may clear the failure
               Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/xe/kernel/issues/new
[    6.835865] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[    6.844694] Mem abort info:
[    6.844695]   ESR = 0x0000000096000005
[    6.844696]   EC = 0x25: DABT (current EL), IL = 32 bits
[    6.844697]   SET = 0, FnV = 0
[    6.844698]   EA = 0, S1PTW = 0
[    6.844699]   FSC = 0x05: level 1 translation fault
[    6.844700] Data abort info:
[    6.844700]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[    6.844702]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[    6.844703]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    6.845848] Registered IR keymap rc-cec
[    6.847502] user pgtable: 16k pages, 47-bit VAs, pgdp=00000001085a4000
[    6.851261] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    6.851265] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[    6.856595] Modules linked in: brcmfmac_wcc xe(+) spidev
[    6.904103]  hci_uart vc4(+)
[    6.912840]  btbcm bluetooth snd_soc_hdmi_codec drm_dma_helper snd_soc_core i2c_algo_bit ecdh_generic cec joydev drm_buddy drm_suballoc_helper aes_ce_blk aes_ce_cipher ghash_ce drm_gpuvm ecc gf128mul snd_compress sha2_ce sha256_arm64 libaes rpivid_hevc(C) sha1_ce brcmfmac snd_pcm_dmaengine drm_exec hid_logitech_dj(+) drm_display_helper drm_ttm_helper pisp_be snd_pcm v4l2_mem2mem ttm
[    6.919151]  brcmutil videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 sha1_generic cfg80211 snd_timer videodev rfkill snd v3d
[    6.927024]  raspberrypi_hwmon videobuf2_common gpu_sched i2c_brcmstb mc drm_shmem_helper spi_bcm2835 gpio_keys drm_kms_helper rp1_pio pwm_fan raspberrypi_gpiomem rp1_mailbox rp1 rp1_adc nvmem_rmem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[    7.066183] CPU: 2 UID: 0 PID: 331 Comm: (udev-worker) Tainted: G         C         6.11.10-v8-16k+ #1
[    7.066186] Tainted: [C]=CRAP
[    7.066187] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    7.066188] pstate: 804000c9 (Nzcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    7.066190] pc : xe_gt_tlb_invalidation_reset+0x68/0xf0 [xe]
[    7.097016] lr : xe_gt_tlb_invalidation_reset+0x4c/0xf0 [xe]
[    7.102785] sp : ffffc00080c6b5b0
[    7.102786] x29: ffffc00080c6b5b0 x28: 0000000000001248 x27: 431bde82d7b634db
[    7.102790] x26: 00000000001b0001 x25: ffff8001013682e8 x24: ffff800101368dc8
[    7.120433] x23: 00000000ffffffff x22: ffff800101369730 x21: ffff8001013682d8
[    7.120435] x20: ffff800101368080 x19: ffff800101368080 x18: ffffffffffffffff
[    7.127598] x17: 6568632074726170 x16: ffffd06fce4be988 x15: 2f3a737074746820
[    7.127600] x14: 74612074726f7065 x13: 2f3a737074746820 x12: 74612074726f7065
[    7.127603] x11: 77656e2f73657573 x10: 0000000000000000 x9 : ffffd06fce4be890
[    7.149083] x8 : 0000000000000000 x7 : ffffffffffffffff x6 : ffffc00080c6b550
[    7.156245] x5 : 0000000000000010 x4 : ffffc00080c6b550 x3 : 0000000000000000
[    7.170567] x2 : 00000000000fffff x1 : 0000000000000000 x0 : 00000000ffffffff
[    7.177729] Call trace:
[    7.177730]  xe_gt_tlb_invalidation_reset+0x68/0xf0 [xe]
[    7.185507]  xe_gt_declare_wedged+0x2c/0x48 [xe]
[    7.190231]  xe_device_declare_wedged+0xec/0x160 [xe]
[    7.195389]  __xe_guc_upload+0x48c/0x650 [xe]
[    7.199846]  xe_guc_min_load_for_hwconfig+0x4c/0xd8 [xe]
[    7.205271]  xe_uc_init_hwconfig+0x34/0x48 [xe]
[    7.209909]  xe_gt_init_hwconfig+0x7c/0xd8 [xe]
[    7.214556]  xe_device_probe+0x20c/0x578 [xe]
[    7.219016]  xe_pci_probe+0x68c/0x9c0 [xe]
[    7.219105]  local_pci_probe+0x48/0xb8
[    7.219109]  pci_device_probe+0xac/0x1b8
[    7.219111]  really_probe+0xc4/0x2a8
[    7.219114]  __driver_probe_device+0x80/0x140
[    7.219116]  driver_probe_device+0x44/0x170
[    7.219117]  __driver_attach+0x9c/0x1b0
[    7.219119]  bus_for_each_dev+0x80/0xe8
[    7.219122]  driver_attach+0x2c/0x40
[    7.219124]  bus_add_driver+0xec/0x218
[    7.219127]  driver_register+0x68/0x138
[... 14 lines cut ...]
[    7.219336] Code: aa0303e1 71000400 1a821000 b9025680 (f85b8422)
[    7.219338] ---[ end trace 0000000000000000 ]---
[    7.219339] note: (udev-worker)[331] exited with irqs disabled
[... cut ...]
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/dg2_guc_70.bin?id=9889ca654ef0a251e633c43ce51e86ad911479b9
mv ./dg2_guc_70.bin\?id\=9889ca654ef0a251e633c43ce51e86ad911479b9  /lib/firmware/i915/dg2_guc_70.bin
```

One time I was greeted with ` GT0: invalid key requested, header = 0x900007D3`, but still with `xe_gt_tlb_invalidation_reset` as the last call stacktrace entry, but not sure how or why I got that message, not reproducable sadly.

For me it feels "not yet ready", when I have more time, I will check against a different ARC card  (A380), maybe it's some weird combination.

Srandista, 2024-12-12:
Well, Arc B580 looks quite good according the initial reviews, hopefully Intel will get their drivers to work on RPi (or ARM in general), that'll be a big push for me to get one of their cards.

geerlingguy, 2024-12-12:
@Srandista - Yeah, so far I've been unsuccessful in getting anyone at Intel to give me the time of day, so I'll likely have to wait until January or later to find one at retail before I can test!

lucasdemarchi, 2024-12-13:
> when playing with 4K page size I am always greeted with `firmware signature verification failed`.

Note that this is after the step you are without that fix.... It doesn't even start verifying the signature if you are trying to load the wrong firmware (in this case a non-production firmware in a production device). And it hits that because it's reading garbage due to the 16KB page.

If you can set the page to 4K, it would be good to iron out other issues before trying to have a complete fix for this one.... we may need to align other buffers to PAGE_SIZE.

pepijndevos, 2024-12-13:
Ngl this issue might determine if my next GPU is an Intel or AMD but it'll be at least a few weeks before I have the hardware to test my A770 on arm to help debug this.

HeyMeco, 2024-12-14:
Just found these:

- A patch for the Xe driver https://github.com/chimera-linux/cports/blob/master/main/linux-stable/patches/xe-nonx86.patch

- And a report that intel Arc Alchemist at least was stable on non x86 with kernel 6.12 https://community.amperecomputing.com/t/intel-arc-on-ampere-altra-unstable-but-somewhat-working/745/11

geerlingguy, 2024-12-14:
@HeyMeco - Hmm... maybe time to test kernel 6.12 with that patch and see how it runs on the Pi again. Probably some other tweaks needed on the Pi, but I can also test on my Ampere system to see if it works on there.

Would love to test a B580, but so far can't find one anywhere.

HeyMeco, 2024-12-14:
> @geerlingguy but I can also test on my Ampere system to see if it works on there

When you try don't forget to make sure patch 9001 and 9002 from the xe patch folder are also applied (those are Ampere specific)

Srandista, 2024-12-14:
It's funny tho, that (al least according to q66 comments) it seems, that after absolutely broken PCIe implementation in CM4, current one in RPi5/CM5 is better then one implemented in Ampere.

HeyMeco, 2024-12-14:
> @Srandista better then one implemented in Ampere.

Ampere Altra maybe but Ampere One should have things fixed. I would still not be surprised if Broadcom cut some corners for anything more complex than M.2 Storage / AI Accelerators for the Raspberry Pi

martinx72, 2024-12-17:
I tried intel xe2 B580 just today.
latest 6.12.y from RPi linux kernel. (the last commit of intel xe driver there was 3 days ago)

crashed at same code as xe a770. 
carshed inside xe_gt_tlb_invalidation_reset(...)
so there must be same issue for two cards.

dmesg ( i added  drm.debug=0xep at the end of cmdline.txt to let it output more info )
```
[    7.254142] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    7.254377] xe 0000:03:00.0: [drm] Found BATTLEMAGE (device ID e20b) display version 14.01 stepping B0
[    7.254382] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] BATTLEMAGE  e20b:0000 dgfx:1 gfx:Xe2_LPG / Xe2_HPG (20.01) media:Xe2_LPM / Xe2_HPM (13.01) display:yes dma_m_s:46 tc:1 gscfi:0 cscfi:1
[    7.254634] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] Stepping = (G:A0, M:A1, B:**)
[    7.254730] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] SR-IOV support: no (mode: none)
[    7.257584] xe 0000:03:00.0: [drm] Using GuC firmware from xe/bmg_guc_70.bin version 70.36.0
[    7.259573] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 0] = 0x00469fd3
[    7.259669] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 1] = 0x00000000
[    7.259756] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 2] = 0x00000000
[    7.259842] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 3] = 0x00000003
[    7.259927] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 4] = 0x00000900
[    7.260012] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 5] = 0xe20b0000
[    7.260097] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 6] = 0x00000000
[    7.260181] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 7] = 0x00000000
[    7.260266] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 8] = 0x00000000
[    7.260350] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 9] = 0x00000000
[    7.260434] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[10] = 0x00000000
[    7.260518] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[11] = 0x00000000
[    7.260602] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[12] = 0x00000000
[    7.260686] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[13] = 0x00000000
[    7.260771] xe 0000:03:00.0: [drm:xe_wopcm_init [xe]] WOPCM: 4096K
[    7.260861] xe 0000:03:00.0: [drm:xe_wopcm_init [xe]] GuC WOPCM is already locked [6144K, 832K)
[    7.263248] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status = 0x40000056, time = 0ms, freq = 2150MHz (req 2133MHz), done = -1
[    7.263252] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status: Reset = 0, BootROM = 0x2B, UKernel = 0x00, MIA = 0x00, Auth = 0x01
[    7.263254] xe 0000:03:00.0: [drm] *ERROR* GT0: firmware production part check failure
[    7.263257] xe 0000:03:00.0: [drm] *ERROR* CRITICAL: Xe has declared device 0000:03:00.0 as wedged.
               IOCTLs and executions are blocked. Only a rebind may clear the failure
               Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/xe/kernel/issues/new
[    7.263266] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[    7.272090] Mem abort info:
[    7.274885]   ESR = 0x0000000096000005
[    7.278642]   EC = 0x25: DABT (current EL), IL = 32 bits
[    7.283970]   SET = 0, FnV = 0
[    7.287027]   EA = 0, S1PTW = 0
[    7.290172]   FSC = 0x05: level 1 translation fault
[    7.295063] Data abort info:
[    7.297946]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[    7.303447]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[    7.308513]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    7.313841] user pgtable: 4k pages, 39-bit VAs, pgdp=000000010242d000
[    7.320303] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    7.329038] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[    7.335327] Modules linked in: cfg80211 xe(+) spidev hci_uart btbcm bluetooth ecdh_generic ecc vc4(+) aes_ce_blk aes_ce_cipher ghash_ce gf128mul sha2_ce snd_soc_hdmi_codec i2c_algo_bit sha256_arm64 sha1_ce sha1_generic libaes cec drm_dma_helper raspberrypi_hwmon drm_buddy sg snd_soc_core drm_suballoc_helper spi_bcm2835 i2c_brcmstb drm_gpuvm drm_exec snd_compress drm_display_helper snd_pcm_dmaengine rfkill rpivid_hevc(C) drm_ttm_helper ttm pisp_be v4l2_mem2mem snd_pcm gpio_keys videobuf2_dma_contig v3d videobuf2_memops snd_timer pwm_fan videobuf2_v4l2 hid_logitech_dj(+) gpu_sched videodev drm_shmem_helper drm_kms_helper rp1_pio snd videobuf2_common mc rp1 rp1_mailbox raspberrypi_gpiomem rp1_adc nvmem_rmem uio_pdrv_genirq uio cuse drm uinput i2c_dev ledtrig_pattern fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6
[    7.409852] CPU: 3 UID: 0 PID: 344 Comm: (udev-worker) Tainted: G         C         6.12.5-v8-16k+ #2
[    7.419110] Tainted: [C]=CRAP
[    7.422081] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    7.427932] pstate: 804000c9 (Nzcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    7.434918] pc : xe_gt_tlb_invalidation_reset+0x68/0xf0 [xe]
[    7.440694] lr : xe_gt_tlb_invalidation_reset+0x4c/0xf0 [xe]
[    7.446454] sp : ffffffc0809135b0
[    7.449773] x29: ffffffc0809135b0 x28: 0000000000000001 x27: 431bde82d7b634db
[    7.456936] x26: 00000000001b0001 x25: ffffff8101d98328 x24: ffffff8101d98e08
[    7.464098] x23: 00000000ffffffff x22: ffffff8101d99770 x21: ffffff8101d98318
[... cut ...]
        Kernel driver in use: xe
        Kernel modules: xe

0000:04:00.0 Audio device: Intel Corporation Device e2f7
        Subsystem: Intel Corporation Device 1100
        !!! Unknown header type 7f
        Region 0: Memory at 1b81800000 (64-bit, non-prefetchable) [size=16K]

```
@HeyMeco  what i did long ago which avoid all crashed in xe drives is same as that none-x86 patch you mentioned. but still stopped  somewhere inside the driver after i patch some more code. but at least it never crashed, ha,,,

geerlingguy, 2024-12-17:
@martinx72 - I've opened a new issue for the B580, since it's a generation newer, and would be useful to debug separately (some driver quirks may be different between the two).

Also, that:

```
[    7.263266] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
```

Looks an awful lot like similar address bugs we had with amdgpu :/

FibreFoX, 2024-12-18:
I got some more time for testing again.

I've just checked with 4K alignment (which I had before too, just to clarify) against my A380 (Intel Arc A380 Challenger ITX 6GB OC) on a Raspi 5 with 4GB RAM (to shuffe some variables) which is connected via a direct connection with a Pineboard uPCIty Lite, but sadly with the same result. So nothing card specific, and not 4K alignment specific (because it was compiled with that option).

It feels like "just an alignment problem" in general, which is very sad, because it would unlock so many usecases for "low energy" ARC scenarios. 

The mentioned NULL pointer seems to come from this line: https://github.com/raspberrypi/linux/blob/391d18039d22bea61a426376da83059dcbf62670/drivers/gpu/drm/xe/xe_gt_tlb_invalidation.c#L153

pepijndevos, 2025-01-02:
[image]

The prototypes of my cm5 mini itx board arrived so if it's useful I could take the A770 out of my desktop to try it but it seems like we're stuck on the null pointer for now.

martinx72, 2025-01-10:
> The prototypes of my cm5 mini itx board arrived so if it's useful I could take the A770 out of my desktop to try it but it seems like we're stuck on the null pointer for now.
> 
> 
> 
> [Edit] lots of activity on https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3594 so I'll keep an eye on that I guess

That null pointer issue is sort of fixed here: https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3956

FibreFoX, 2025-01-10:
The auth issue still exists, we have to wait for https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3594 to get resolved. AFAIK it is some kind of "bits do not get transfered in the right format" (what I meant with `Is it "just" a math problem` before).

I would help debugging this more, but I am nowhere near hardware kernel level debugging with DMA data transfer. If I see some progress there, I will check again and report back here, we all are eager to have this miracle working some day ;) especially right now where Battlemage is out (looks like the same problem for me right now)

FibreFoX, 2025-01-14:
The fix for the NULL ptr has landed in the 6.13.y branch of the Raspberry Pi kernel :)
https://github.com/raspberrypi/linux/commit/9ab4981552930a9c45682d62424ba610edc3992d

The display (connected to the raspberry pi) now still works, yet the firmware auth issue still persists. The display connected to the card itself always stays black.

```
[    6.307507] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    6.307709] Bluetooth: hci0: BCM: features 0x2f
[    6.308842] Bluetooth: hci0: BCM4345C0
[    6.308845] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    6.309724] [drm] Initialized vc4 0.0.0 for axi:gpu on minor 1
[    6.311112] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    6.311781] xe 0000:03:00.0: [drm] Found dg2/g11 (device ID 56a5) display version 13.00 stepping C0
[    6.317108] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.29.2
[    6.324526] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status = 0x400000A0, time = 4ms, freq = 2400MHz (req 2450MHz), done = -1
[    6.324532] xe 0000:03:00.0: [drm] *ERROR* GT0: load failed: status: Reset = 0, BootROM = 0x50, UKernel = 0x00, MIA = 0x00, Auth = 0x01
[    6.324534] xe 0000:03:00.0: [drm] *ERROR* GT0: firmware signature verification failed
[    6.324538] xe 0000:03:00.0: [drm] *ERROR* CRITICAL: Xe has declared device 0000:03:00.0 as wedged.
               IOCTLs and executions are blocked. Only a rebind may clear the failure
               Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/xe/kernel/issues/new
[    6.384689] xe 0000:03:00.0: [drm] *ERROR* GT0: GuC mmio request 0x4100: no reply 0x4100
[    6.384715] xe 0000:03:00.0: probe with driver xe failed with error -110
[    6.387031] Console: switching to colour frame buffer device 240x67
[    6.404808] vc4-drm axi:gpu: [drm] fb0: vc4drmfb frame buffer device
```

martinx72, 2025-01-14:
> Some sneak peek:
> 
> [image]
> 
> 
> 
> What I have done:
> 
> * using current 6.13.y raspberry pi kernel branch
> 
> * having https://github.com/chimera-linux/cports/blob/master/main/linux-stable/patches/xe-nonx86.patch applied 
> 
> * having 4k page size alignment configured
> 
> * having XE driver enabled
> 
> * having `dtparam=pciex1_gen=3` set (works without too but needs nore restarts)
> 
> 
> 
> ... after some reboots (at least 3 with "gen 3" pcie) while the graphics card is still powered on between reboots, I finally can get a terminal output via HDMI 😮 haven't tested other features yet .... already posted my findings to https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3594

Glad to see you got something shows!

BTW, this i915 init patch is must since I tried 6.6 to 6.12.

Will try 6.13 and see what i can get tonight.

pimzand, 2025-01-14:
Is there anything special in the raspberry pi kernel that would be beneficial for getting this to work with an Ampere Altra?

martinx72, 2025-01-14:
I second that, it does shows fram buffer console mode when it runs correctly with my A750.

When it works. dmes gave us this, and it cannot be captured by UAP recorder as it trigger HDCP, 
```
pi@raspberrypi:~ $ dmesg | grep 'xe '
[    7.484794] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    7.493616] xe 0000:03:00.0: [drm] Found dg2/g10 (device ID 56a1) display version 13.00 stepping C0
[    7.498829] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.36.0
[    7.541718] xe 0000:03:00.0: [drm] Finished loading DMC firmware i915/dg2_dmc_ver2_08.bin (v2.8)
[    7.542341] xe 0000:03:00.0: [drm] Attempting to resize bar from 256MiB -> 8192MiB
[    7.542356] xe 0000:03:00.0: BAR 2 [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[    7.542425] xe 0000:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: can't assign; no space
[    7.542427] xe 0000:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: failed to assign
[    7.542485] xe 0000:03:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC). Consider enabling 'Resizable BAR' support in your BIOS
[    7.542491] xe 0000:03:00.0: BAR 2 [mem 0x1800000000-0x180fffffff 64bit pref]: assigned
[    7.542515] xe 0000:03:00.0: [drm] VISIBLE VRAM: 0x0000001800000000, 0x0000000010000000
[    7.543142] xe 0000:03:00.0: [drm] Small BAR device
[    7.543145] xe 0000:03:00.0: [drm] VRAM[0, 0]: Actual physical size 0x0000000200000000, usable size exclude stolen 0x00000001fc000000, CPU accessible size 0x0000000010000000
[    7.543149] xe 0000:03:00.0: [drm] VRAM[0, 0]: DPA range: [0x0000000000000000-200000000], io range: [0x0000001800000000-1810000000]
[    7.543151] xe 0000:03:00.0: [drm] VRAM: 0x0000000200000000 is larger than resource 0x0000000010000000
[    7.543154] xe 0000:03:00.0: [drm] Total VRAM: 0x0000001800000000, 0x0000000200000000
[    7.543155] xe 0000:03:00.0: [drm] Available VRAM: 0x0000001800000000, 0x00000001fc000000
[    7.708397] xe 0000:03:00.0: [drm] vcs1 fused off
[    7.708405] xe 0000:03:00.0: [drm] vcs3 fused off
[    7.708406] xe 0000:03:00.0: [drm] vcs4 fused off
[    7.708408] xe 0000:03:00.0: [drm] vcs5 fused off
[    7.708409] xe 0000:03:00.0: [drm] vcs6 fused off
[    7.708410] xe 0000:03:00.0: [drm] vcs7 fused off
[    7.708412] xe 0000:03:00.0: [drm] vecs2 fused off
[    7.708413] xe 0000:03:00.0: [drm] vecs3 fused off
[    7.758428] [drm] Initialized xe 1.1.0 for 0000:03:00.0 on minor 2
[    7.870274] xe 0000:03:00.0: [drm] Cannot find any crtc or sizes
[    7.926276] xe 0000:03:00.0: [drm] Cannot find any crtc or sizes
[   10.153409] xe 0000:03:00.0: [drm] not enough stolen space for compressed buffer (need 35389440 more bytes), disabling. Hint: you may be able to increase stolen memory size in the BIOS to avoid this.
[   10.255386] xe 0000:03:00.0: [drm] fb0: xedrmfb frame buffer device
```

And, when it shows fbcon, the HD-Audio works fine there also.
[image]

but!!!!
it only works like 20% to 25% for me, to see the fbcon , most of time i got this
[image]
```
[drm] *ERROR* GT0: GuC mmio request 0x4509: failure 0x306 hint 0x0
```
and sometimes i got different error code.
and sometimes, i got kernel panic crash info ( like 5%)

As each reboot, i got different situation, all i can guess is there is some initialization race condition in the drive.
and litghtdm won't work so far, still need some more fix for the xe driver, will test B570 in the next few days.

martinx72, 2025-01-15:
FYI,

added drm.debug=0xe at the end of cmdline.txt to produce more info, 
for that most happened error
```
[drm] *ERROR* GT0: GuC mmio request 0x4509: failure 0x306 hint 0x0
```

here is what i got
```
[    7.582802] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    7.596809] xe 0000:03:00.0: [drm] Found dg2/g10 (device ID 56a1) display version 13.00 stepping C0
[    7.596822] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] DG2 G10 56a1:0008 dgfx:1 gfx:Xe_HPG (12.55) media:Xe_HPM (12.55) display:yes dma_m_s:46 tc:1 gscfi:1 cscfi:0
[    7.596995] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] Stepping = (G:C0, M:C0, B:**)
[    7.597097] xe 0000:03:00.0: [drm:xe_pci_probe [xe]] SR-IOV support: no (mode: none)
[    7.610165] xe 0000:03:00.0: [drm] Using GuC firmware from i915/dg2_guc_70.bin version 70.36.0
[    7.613744] snd_hda_intel 0000:04:00.0: Force to snoop mode by module option
[    7.615438] input: vc4-hdmi-0 HDMI Jack as /devices/platform/soc@107c000000/107c701400.hdmi/sound/card1/input7
[    7.615658] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 0] = 0x0045d3d7
[    7.615791] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 1] = 0x00000000
[    7.615887] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 2] = 0x00000000
[    7.615995] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 3] = 0x00000003
[    7.616088] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 4] = 0x00000ae0
[    7.616177] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 5] = 0x56a10008
[    7.616266] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 6] = 0x00000000
[    7.616355] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 7] = 0x00000000
[    7.616446] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 8] = 0x00000000
[    7.616538] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[ 9] = 0x00000000
[    7.616628] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[10] = 0x00000000
[    7.616718] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[11] = 0x00000000
[    7.616806] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[12] = 0x00000000
[    7.616894] xe 0000:03:00.0: [drm:guc_print_params [xe]] GT0: GuC param[13] = 0x00000000
[    7.621506] xe 0000:03:00.0: [drm:xe_wopcm_init [xe]] WOPCM: 4096K
[    7.621689] xe 0000:03:00.0: [drm:xe_wopcm_init [xe]] GuC WOPCM is already locked [1440K, 544K)
[    7.624460] xe 0000:03:00.0: [drm:__xe_guc_upload [xe]] GT0: load still in progress, timeouts = 0, freq = 2400MHz (req 2400MHz), status = 0x00000072 [0x39/00]
[    7.633769] xe 0000:03:00.0: [drm:__xe_guc_upload [xe]] GT0: init took 9ms, freq = 2400MHz (req = 2400MHz), before = 2400MHz, status = 0x8002F034, timeouts = 0
[    7.634637] xe 0000:03:00.0: [drm] *ERROR* GT0: GuC mmio request 0x4509: failure 0x306 hint 0x0
[    7.634643] xe 0000:03:00.0: [drm] *ERROR* GT0: Failed to enable GuC CT (-ENXIO)
[    7.651138] Bluetooth: Core ver 2.22
```

trace the the src, that error produced via this code

[image]

and
```
#define GUC_ACTION_HOST2GUC_CONTROL_CTB			0x4509
```

@FibreFoX  Can you report this in your existed issue/thread?

FibreFoX, 2025-01-15:
@martinx72 The thing is, the error is different/random depending on each reboot, that's what I already reported to the Intel issue. The error line `GuG mmio request` kinda just is the side effect.

This tells me, that there is some kind of timing issue when the Pi is talking to the graphics card (or the other way around when DMA data transfer happens). The graphics card seems to not loose it's "unlock state" from each previous attempts, so each reboot kinda puts different code-fragments a bit further until the GuC firmware is fully transfered.
I also checked the behaviour with PCIe set to gen2 speed VS gen3 speed, and with gen3 it is more likely to work, which makes me even more think of a timing issue in the DMA communication area, but I'm no hardware/Intel guru 😺 

And this only "works" with 4k page alignment, so the xe driver still has several issues, which I hope Intel can find and fix (at least two Intel engineers are listening to this github issue, hehe)

FibreFoX, 2025-01-18:
The NULL ptr fix got backported to 6.12.y 😺 
https://github.com/raspberrypi/linux/commit/09b94ddc58c6640cbbc7775a61a5387b8be71488

I've found another hint. I just checked `dmesg` output again on a freshly recompiled 6.12 raspi kernel:
```
[    1.987630] brcm-pcie 1000110000.pcie: Forcing gen 3
[    2.096138] brcm-pcie 1000110000.pcie: clkreq-mode set to default
[    2.102254] brcm-pcie 1000110000.pcie: link up, 8.0 GT/s PCIe x1 (!SSC)
(...)
[    2.842084] pci 0000:03:00.0: [8086:56a5] type 00 class 0x030000 PCIe Endpoint
[    2.849355] pci 0000:03:00.0: BAR 0 [mem 0x00000000-0x00ffffff 64bit]
[    2.855831] pci 0000:03:00.0: BAR 2 [mem 0x00000000-0x0fffffff 64bit pref]
[    2.862753] pci 0000:03:00.0: ROM [mem 0x00000000-0x001fffff pref]
[    2.868983] pci 0000:03:00.0: ASPM: overriding L1 acceptable latency from 0x0 to 0x7
[    2.876828] pci 0000:03:00.0: PME# supported from D0 D3hot
[    2.882450] pci 0000:03:00.0: vgaarb: setting as boot VGA device
[    2.888476] pci 0000:03:00.0: vgaarb: bridge control possible
[    2.894240] pci 0000:03:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    2.902668] pci_bus 0000:03: busn_res: [bus 03-ff] end is updated to 03
```

I noticed ASPM and PME being mentioned, and as I suspect some timing issue (aside from the 4k page size alignment), because after some reboots the GuC gets loaded, I tried to add this at the end of my `/boot/firmware/cmdline.txt`:
```
pcie_port_pm=off pcie_aspm=off
```

This reduces the required reboot count to "mostly only one" when on PCIe gen 3, still requires several (two to four) reboots with gen 2 while NO monitor is connected. It is very strange, when my monitor is connected to the graphics card (via HDMI), I need more reboots than compared to without any display connected. Could not find a good way to reproduce this consistently.

Still no solution found yet 😿

FibreFoX, 2025-01-20:
Already shared this on the intel-issue, but for sake of documentation, I found a workaround, which kinda proofes (I hope) that this is a timing issue (aside of the 4K alignment):
```patch
diff --git a/drivers/gpu/drm/xe/xe_guc.c b/drivers/gpu/drm/xe/xe_guc.c
index 52df28032..6d2a8e92c 100644
--- a/drivers/gpu/drm/xe/xe_guc.c
+++ b/drivers/gpu/drm/xe/xe_guc.c
@@ -5,6 +5,8 @@

 #include "xe_guc.h"

+#include <linux/delay.h>
+
 #include <drm/drm_managed.h>

 #include <generated/xe_wa_oob.h>
@@ -719,6 +721,7 @@ static int __xe_guc_upload(struct xe_guc *guc)
        ret = guc_xfer_rsa(guc);
        if (ret)
                goto out;
+       msleep(1 * 1000);
        /*
         * Current uCode expects the code to be loaded at 8k; locations below
         * this are used for the stack.
@@ -944,6 +947,7 @@ int xe_guc_mmio_send_recv(struct xe_guc *guc, const u32 *request,
                  GUC_HXG_TYPE_REQUEST);

 retry:
+       msleep(1 * 500);
        /* Not in critical data-path, just do if else for GT type */
        if (xe_gt_is_media_type(gt)) {
                for (i = 0; i < len; ++i)
```

Took me quite some time to find this (and lots of kernel recompiles ... maybe I need one of these red shirt jeff shirts xD), but I am at the point that I can cold-boot, restart and "use" the system. Not yet tested any actual stuff, but maybe someone else comes up with a good solution.

FibreFoX, 2025-01-21:
I might have found "the next piece" 😄 

[image]

I will clean up my modifications and post them here (and report to intel too after that). Surely I just have workarounds, but at least one step closer 😸

FibreFoX, 2025-01-21:
I came up with these changes, that needed to be done for the XE and the i915 kernel driver to be loaded. This requires 4K page size alignment as commented above.

First problem (for me) was NULL pointer, which has been resolved with the current 6.12 and 6.13 raspberry linux kernel.

Luckily there is that nonx86 patch for the i915 driver, which I could reduce to a single line being excluded (not sure about the impact tho) from the file `drivers/gpu/drm/i915/display/intel_vga.c`:
```patch
diff --git a/drivers/gpu/drm/i915/display/intel_vga.c b/drivers/gpu/drm/i915/display/intel_vga.c
index 0b5916c15..e233c0ce2 100644
--- a/drivers/gpu/drm/i915/display/intel_vga.c
+++ b/drivers/gpu/drm/i915/display/intel_vga.c
@@ -94,7 +94,9 @@ void intel_vga_reset_io_mem(struct drm_i915_private *i915)
         * and error messages.
         */
        vga_get_uninterruptible(pdev, VGA_RSRC_LEGACY_IO);
+#if defined(CONFIG_VGA_CONSOLE)
        outb(inb(VGA_MIS_R), VGA_MIS_W);
+#endif
        vga_put(pdev, VGA_RSRC_LEGACY_IO);
 }
```

Then there is the timing-issue for loading the firmware BLOBs to the graphics card, the file `drivers/gpu/drm/xe/xe_guc.c` needs to have some "dumb timeouts" for the system having enough time for data transfer (I assume inside the driver there are several timing-assumptions, that can not be met on the pi, or are ARM64 specific):

```patch
diff --git a/drivers/gpu/drm/xe/xe_guc.c b/drivers/gpu/drm/xe/xe_guc.c
index 52df28032..6d2a8e92c 100644
--- a/drivers/gpu/drm/xe/xe_guc.c
+++ b/drivers/gpu/drm/xe/xe_guc.c
@@ -5,6 +5,8 @@

 #include "xe_guc.h"

+#include <linux/delay.h>
+
 #include <drm/drm_managed.h>

 #include <generated/xe_wa_oob.h>
@@ -719,6 +721,7 @@ static int __xe_guc_upload(struct xe_guc *guc)
        ret = guc_xfer_rsa(guc);
        if (ret)
                goto out;
+       msleep(1 * 1000);
        /*
         * Current uCode expects the code to be loaded at 8k; locations below
         * this are used for the stack.
@@ -944,6 +947,7 @@ int xe_guc_mmio_send_recv(struct xe_guc *guc, const u32 *request,
                  GUC_HXG_TYPE_REQUEST);

 retry:
+       msleep(1 * 500);
        /* Not in critical data-path, just do if else for GT type */
        if (xe_gt_is_media_type(gt)) {

```

The third file that needs to be changed is inside the i915 driver too, the file `drivers/gpu/drm/i915/display/intel_display.c` needs to be changed:
```patch
diff --git a/drivers/gpu/drm/i915/display/intel_display.c b/drivers/gpu/drm/i915/display/intel_display.c
index 15541e37a..d6d52b743 100644
--- a/drivers/gpu/drm/i915/display/intel_display.c
+++ b/drivers/gpu/drm/i915/display/intel_display.c
@@ -7652,6 +7652,7 @@ int intel_atomic_commit(struct drm_device *dev, struct drm_atomic_state *_state,
         * FIXME: Cut over to (async) commit helpers instead of hand-rolling
         * everything.
         */
+#if defined(CONFIG_VGA_CONSOLE)
        if (state->base.legacy_cursor_update) {
                struct intel_crtc_state *new_crtc_state;
                struct intel_crtc *crtc;
@@ -7660,6 +7661,7 @@ int intel_atomic_commit(struct drm_device *dev, struct drm_atomic_state *_state,
                for_each_new_intel_crtc_in_state(state, crtc, new_crtc_state, i)
                        complete_all(&new_crtc_state->uapi.commit->flip_done);
        }
+#endif

        ret = intel_atomic_prepare_commit(state);
        if (ret) {
```

I have no idea WHY these work, but I now have a full desktop system running on my raspi 5 (8GB RAM version).

😺 I hope this helps and that others can confirm this working

geerlingguy, 2025-01-21:
@FibreFoX - Going to try this out now! I've wrapped your patches into this branch / PR: https://github.com/geerlingguy/linux/pull/9

6by9, 2025-01-21:
`outb` and `inb` are specific to architectures that have independent I/O and memory buses. The old 0x378 address you may remember for LPT1 and 0x3f8/0x2f8 for COM1 and COM2 were I/O addresses, as are the original hooks for VGA displays, hence i915 using them.

ARM has no I/O bus, hence why `outb` and `inb` will cause problems. It's also why a 

As to the delays, you're probably right that something is just taking a little longer than Intel expected.

The legacy_cursor_update looks weird, and I can't even suggest a reason for needing to disable it.

FibreFoX, 2025-01-21:
@geerlingguy Just make sure to grab the `intel_display.c` patch too, otherwise the display stays black.

FibreFoX, 2025-01-21:
@6by9 Wihout disabling the kernel ran into another NULL ptr, because the makro  `for_each_new_intel_crtc_in_state` (https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/gpu/drm/i915/display/intel_display.h#L345) hits something it does not like. Probably some kind of lock, that wasn't there, at least that what I've traced it to.

FibreFoX, 2025-01-21:
Not sure if relevant, I've connected my monitor via HDMI currently to my A380. I am running a 6.12.y kernel with the patches above.

I will re-check if I am missing something, and will check 6.13.y too. But so far lightdm seems to start for me 😺 we will find the underlying issue.

HeyMeco, 2025-01-21:
@geerlingguy
> And then rebooted. I'm guessing the device ID is going to be different on every Pi? Kind of an annoying aspect of using the Xe driver!

that’s actually the PCIe device ID you can look them up here https://www.pcilookup.com/?ven=&dev=56a1&action=submit for example. Yours correlates to a A750 if the database is correct. These are not individual.

geerlingguy, 2025-01-21:
@pimzand - Even without that, I'm still getting the message :(

```
pi@pi5-pcie:~ $ dmesg | grep xe
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=2C:CF:67:C7:6E:8B vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=e70d1683-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles xe.force_probe='56a1' cfg80211.ieee80211_regdom=US
...
[    4.301926] xe 0000:03:00.0: Your graphics device 56a1 is not officially supported
               by xe driver in this kernel version. To force Xe probe,
               use xe.force_probe='56a1' and i915.force_probe='!56a1'
```

lfelipe, 2025-01-21:
Also, you can use '*' to be a catchall and not have to deal with individual PCI IDs. For BMG, they are all supported by Xe so no problem there, but for DG2 they are officially supported by i915 so no way of removing the force_probe without causing problems upstream, but as @pimzand mentioned for Raspberry Pis you would likely be using Xe anyways.

lfelipe, 2025-01-21:
@geerlingguy set DRM_XE_FORCE_PROBE to '*' on your kernel config

[75 more replies]

commit 2547a4f1 (2023-02-15): Fixes #510: Add Intel Arc A750 graphics card.