geerlingguy/raspberry-pi-pcie-devices #511: How about Intel M10 Optane SSD(PCIE)  with it ?

datour, 2023-02-18:
Intel Optane SSD with M10 (16GB)  is only $5.41 on https://www.aliexpress.com，Does it work with CM4 ？

skpio, 2023-02-21:
I tested an M10 and H10 on a Compute Blade RC2 w/ a CM4 8GB Lite (CM4008000). 

The M10 has 32GB of Optane memory, while the H10 has 16GB of Optane memory and a 256GB SSD (Intel 660p, I think?) onboard.

The M10 would not boot, but the H10 would.

However, *both* could be used as a secondary (non-boot) disk. The M10 as a 32GB disk, and the H10 as a 256GB disk. The 16GB of Optane memory on the H10 was not available. I still have some work to do in order to fully understand the difference between the two. From what I understand, since the H10 is bifurcated. The CM4 has 1x PCIe Gen 2 lane, so it can only support 1/2 of H10. 

I'll update here when I have more answers.

BarsMonster, 2024-04-12:
Tested on Raspberry Pi 5 - M10 16Gb does not boot (Failed to open device 'nvme'), but works as data drive, which is quite sad.
Prices continue to stop - M10 is very attractive for boot drive on Pi.