geerlingguy/raspberry-pi-pcie-devices #520: Nebra CM4 Flashstick

shawaj, 2023-04-12:
Thought this might be of interest. You can buy it, but it is also OSHW :-)

https://github.com/NebraLtd/FlashStick
https://www.nebra.com/products/nebra-raspberry-pi-cm4-flash-stick

shawaj, 2023-04-12:
@geerlingguy so it has two modes - usb boot or emmc boot (i.e. you can use it as a compute stick as well as a flasher).

But yes you still need to use the usbboot stuff. Just that Etcher and EtcherPro have that built in and automatically recognise the compute module

geerlingguy, 2023-04-12:
> Etcher and EtcherPro have that built in and automatically recognise the compute module

I didn't realize that! https://blog.balena.io/etcher-now-with-multi-write-and-compute-module-support/

I would love to have one of these on hand—would make it even easier than the PiTray mini that I currently use, as I could just slap this into the USB 3 port I have hanging off my Mac, when I need to mess with an eMMC CM4.

shawaj, 2023-04-12:
Sure thing @geerlingguy we have plenty of them in the warehouse, and CM3 ones too :-)

Can you ping an email with your address to support@nebra.com and a link to this github issue and then let me know the ticket number. Then I'll handle the rest.

Also let us know if there's any other products you might be interested in playing around with on nebra.com or pi-supply.com :-)

shawaj, 2023-04-12:
And re EtcherPro - yeah it's super handy. We use them in manufacturing for our Helium Miners to program thousands of SD cards and CM3+ ... Very handy!

geerlingguy, 2023-04-13:
@shawaj - Nice! I emailed that support address but haven't gotten a response.

shawaj, 2023-04-13:
No worries, I've found the ticket #417021 :-)

I sent a reply a bit earlier

shawaj, 2023-06-08:
Just checking you received the package ok @geerlingguy ?

geerlingguy, 2023-06-13:
@shawaj - I did! I have stashed it into a small box with a couple other CM4-based products I'm hoping to get to after I can get my new office buildout underway... right now too many things cluttering up my shelves to get to a compendium video!

geerlingguy, 2023-11-22:
Page: https://pipci.jeffgeerling.com/boards_cm/nebra-cm4-flashstick.html

closed 2023-11-22 by geerlingguy