geerlingguy/raspberry-pi-pcie-devices #525: Add Pi4GPU ('PiG') PCIe Raspberry Pi CM4 board

geerlingguy, 2023-04-13:
I just announced the prototype [Pi4GPU]() in a [YouTube video](https://www.youtube.com/watch?v=l9dItRUjQ0k):

[image]

[image]

The PCIe x4 (physical - only x1 connected signals) card is meant to slot into a special adapter board that allows plugging up to a x16 PCI Express card (mainly intended for graphics cards) directly into this card for testing purposes.

The card exposes 1 Gbps Ethernet, full-size HDMI, two USB 2.0 ports, and a 12v barrel plug on the back. It has a USB-C port for programming on the inside edge, as well as a 6-pin PCIe pinout power input so it can be powered internally or from the same power supply used to power the card under test.

geerlingguy, 2023-04-13:
It's on the site now! https://pipci.jeffgeerling.com/boards_cm/pi4gpu.html

closed 2023-04-13 by geerlingguy

micklen, 2025-03-04:
Where can I get PIG card?