geerlingguy/raspberry-pi-pcie-devices #526: Add Nvidia Quadro 2000

geerlingguy, 2023-04-13:
Thanks to Bishop DuBourg High School, I am now the proud owner of an Nvidia Quadro 2000:

[image]

[image]

I'd like to test it and see what happens with this older card from (I believe) 2010.

geerlingguy, 2024-10-30:
```
pi@pi5-pcie:~ $ sudo lspci -vvvv
...
0000:01:00.0 VGA compatible controller: NVIDIA Corporation GF106GL [Quadro 2000] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: Hewlett-Packard Company GF106GL [Quadro 2000]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 1b80000000 (32-bit, non-prefetchable) [disabled] [size=32M]
	Region 1: Memory at 1800000000 (64-bit, prefetchable) [disabled] [size=128M]
	Region 3: Memory at 1808000000 (64-bit, prefetchable) [disabled] [size=64M]
	Expansion ROM at 1b82000000 [virtual] [disabled] [size=512K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [78] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <256ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 128 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [b4] Vendor Specific Information: Len=14 <?>
	Capabilities: [100 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [128 v1] Power Budgeting <?>
	Capabilities: [600 v1] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Kernel modules: nouveau

0000:01:00.1 Audio device: NVIDIA Corporation GF106 High Definition Audio Controller (rev a1)
	Subsystem: Hewlett-Packard Company GF106 High Definition Audio Controller
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin B routed to IRQ 0
	Region 0: Memory at 1b82080000 (32-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [78] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <256ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 128 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
```

geerlingguy, 2024-10-30:
Booting the Pi with `nouveau` support compiled into the 6.6.y kernel:

```
[    4.572555] nouveau 0000:01:00.0: bios: version 70.06.41.00.06
[    4.582173] ------------[ cut here ]------------
[    4.582180] nouveau 0000:01:00.0: swiotlb addr 0x00000010fbffc000+16384 overflow (mask ffffffff, bus limit 1fffffffff).
[    4.582192] WARNING: CPU: 0 PID: 338 at kernel/dma/swiotlb.c:1511 swiotlb_map+0x2a4/0x2c0
[    4.582203] Modules linked in: hci_uart btbcm bluetooth aes_ce_blk aes_ce_cipher ghash_ce gf128mul brcmfmac_wcc sha2_ce sha256_arm64 sha1_ce ecdh_generic vc4(+) ecc libaes snd_soc_hdmi_codec cec nouveau(+) drm_dma_helper raspberrypi_hwmon hid_apple snd_soc_core brcmfmac snd_compress i2c_brcmstb brcmutil rpivid_hevc(C) snd_pcm_dmaengine spi_bcm2835 pisp_be drm_exec cfg80211 v4l2_mem2mem v3d pwm_fan i2c_algo_bit snd_pcm snd_timer rfkill drm_display_helper snd videobuf2_dma_contig gpu_sched gpio_keys drm_ttm_helper videobuf2_memops drm_shmem_helper ttm videobuf2_v4l2 videodev videobuf2_common drm_kms_helper mc joydev raspberrypi_gpiomem hid_multitouch rp1_adc nvmem_rmem binfmt_misc uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6
[    4.582293] CPU: 0 PID: 338 Comm: (udev-worker) Tainted: G         C         6.6.58-v8-16k+ #12
[    4.582297] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    4.582299] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.582303] pc : swiotlb_map+0x2a4/0x2c0
[    4.582307] lr : swiotlb_map+0x2a4/0x2c0
[    4.582311] sp : ffffc00080c7b600
[    4.582313] x29: ffffc00080c7b600 x28: 0000000000000001 x27: 0000000000000001
[    4.582319] x26: ffff800105ad2000 x25: 0000001105730000 x24: ffff80010147a0c0
[    4.582324] x23: ffff80010147a0c0 x22: 0000000000000000 x21: 0000000000000000
[    4.582329] x20: 0000000000004000 x19: 0000000105730000 x18: ffffffffffffffff
[    4.582334] x17: 7265766f20343833 x16: 36312b3030306366 x15: 6662663031303030
[    4.582339] x14: 3030307830207264 x13: 2e29666666666666 x12: 666666312074696d
[    4.582344] x11: 696c20737562202c x10: ffffd00085716b40 x9 : ffffd0008410fd24
[    4.582348] x8 : 00000000ffffefff x7 : ffffd00085716b40 x6 : 80000000fffff000
[    4.582353] x5 : ffff8001ffa92d88 x4 : 0000000000000000 x3 : ffffb0017a7a4000
[    4.582358] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff800107aaf600
[    4.582363] Call trace:
[    4.582365]  swiotlb_map+0x2a4/0x2c0
[    4.582369]  dma_map_page_attrs+0xb4/0x248
[    4.582374]  nvkm_fb_ctor+0xc4/0x100 [nouveau]
[    4.582589]  gf100_fb_new+0x68/0xa8 [nouveau]
[    4.582769]  nvkm_device_ctor+0x1a2c/0x3da0 [nouveau]
[    4.582947]  nvkm_device_pci_new+0xf8/0x298 [nouveau]
[    4.583122]  nouveau_drm_probe+0xd0/0x230 [nouveau]
[    4.583299]  local_pci_probe+0x48/0xb8
[    4.583305]  pci_device_probe+0xac/0x1c8
[    4.583308]  really_probe+0x150/0x2c0
[    4.583312]  __driver_probe_device+0x80/0x140
[    4.583315]  driver_probe_device+0x44/0x170
[    4.583318]  __driver_attach+0x9c/0x1b0
[    4.583321]  bus_for_each_dev+0x80/0xe8
[    4.583327]  driver_attach+0x2c/0x40
[    4.583330]  bus_add_driver+0xec/0x218
[    4.583333]  driver_register+0x68/0x138
[    4.583336]  __pci_register_driver+0x54/0x68
[    4.583339]  nouveau_drm_init+0x214/0x3ff8 [nouveau]
[    4.583527]  do_one_initcall+0x60/0x2c0
[    4.583532]  do_init_module+0x60/0x218
[    4.583538]  load_module+0x1de0/0x2090
[    4.583543]  __do_sys_init_module+0x19c/0x1e0
[    4.583546]  __arm64_sys_init_module+0x24/0x38
[    4.583548]  invoke_syscall+0x50/0x128
[    4.583555]  el0_svc_common.constprop.0+0xc8/0xf0
[    4.583560]  do_el0_svc+0x24/0x38
[    4.583564]  el0_svc+0x40/0xe8
[    4.583568]  el0t_64_sync_handler+0x100/0x130
[    4.583571]  el0t_64_sync+0x190/0x198
[    4.583574] ---[ end trace 0000000000000000 ]---
[    5.066761] nouveau 0000:01:00.0: fb: 1024 MiB GDDR5
[    6.449134] nouveau 0000:01:00.0: DRM: VRAM: 1024 MiB
[    6.449140] nouveau 0000:01:00.0: DRM: GART: 1048576 MiB
[    6.449144] nouveau 0000:01:00.0: DRM: TMDS table version 2.0
[    6.449147] nouveau 0000:01:00.0: DRM: DCB version 4.0
[    6.449149] nouveau 0000:01:00.0: DRM: DCB outp 00: 02000300 00000000
[    6.449151] nouveau 0000:01:00.0: DRM: DCB outp 01: 01000302 00020030
[    6.449153] nouveau 0000:01:00.0: DRM: DCB outp 02: 048113b6 0f220010
[    6.449155] nouveau 0000:01:00.0: DRM: DCB outp 03: 04011372 00020010
[    6.449157] nouveau 0000:01:00.0: DRM: DCB outp 04: 028223a6 0f220010
[    6.449159] nouveau 0000:01:00.0: DRM: DCB outp 05: 02022362 00020010
[    6.449161] nouveau 0000:01:00.0: DRM: DCB conn 00: 00001030
[    6.449163] nouveau 0000:01:00.0: DRM: DCB conn 01: 00820146
[    6.449164] nouveau 0000:01:00.0: DRM: DCB conn 02: 00410246
[... 40 lines cut ...]
[    6.743403]  pci_device_probe+0xac/0x1c8
[    6.747335]  really_probe+0x150/0x2c0
[    6.751006]  __driver_probe_device+0x80/0x140
[    6.755373]  driver_probe_device+0x44/0x170
[    6.759565]  __driver_attach+0x9c/0x1b0
[    6.763409]  bus_for_each_dev+0x80/0xe8
[    6.767253]  driver_attach+0x2c/0x40
[    6.770835]  bus_add_driver+0xec/0x218
[    6.774591]  driver_register+0x68/0x138
[    6.778434]  __pci_register_driver+0x54/0x68
[    6.782715]  nouveau_drm_init+0x214/0x3ff8 [nouveau]
[    6.787818]  do_one_initcall+0x60/0x2c0
[    6.791663]  do_init_module+0x60/0x218
[    6.795421]  load_module+0x1de0/0x2090
[    6.799178]  __do_sys_init_module+0x19c/0x1e0
[    6.803545]  __arm64_sys_init_module+0x24/0x38
[    6.808000]  invoke_syscall+0x50/0x128
[    6.811758]  el0_svc_common.constprop.0+0xc8/0xf0
[    6.816475]  do_el0_svc+0x24/0x38
[    6.819796]  el0_svc+0x40/0xe8
[    6.822855]  el0t_64_sync_handler+0x100/0x130
[    6.827221]  el0t_64_sync+0x190/0x198
[    6.830891] Code: d1000443 ea03003f 8a230021 54000040 (d50b7e21) 
[    6.837004] ---[ end trace 0000000000000000 ]---
```

geerlingguy, 2024-10-30:
Looks like they didn't do proprietary drivers back then for arm64 either:

[image]

closed 2024-10-30 by geerlingguy

6by9, 2024-10-30:
I also tried my nvidia card (ASUS GT710-4H-SL-2GD5) with nouveau on a Pi5 the other week. I had the same allocation failure on initial probe. I changed one allocation call which cleared it, but I don't have access to that tree this week to check which one. (Swiotlb is definitely not what you want for any drm device) 
That got the driver probing fully, however I didn't succeed in getting anything on the screen.

geerlingguy, 2024-10-30:
@6by9 - If you let me know, I can work on re-testing — I have my setup on a desk that will remain for a bit so I can quickly swap out a bunch of cards. I will hopefully get a recent-gen workstation card or two in soon too... these older cards have minimal support already, much less arm64 driver support when they were designed!

I am hopeful we could get a 30-series or 40-series card going on a Pi, but it seems down to memory allocations, which may be fixable on the Pi side.

Could be a rabbit hole like Coreforge has gone down, but with Nvidia's open driver being available, maybe worth pursuing now!

6by9, 2024-10-30:
I still don't have aces to my tree as I'm on holiday, but I think what I copied was from [gk20a_bar_new](https://elixir.bootlin.com/linux/v6.11.5/source/drivers/gpu/drm/nouveau/nvkm/subdev/bar/gk20a.c#L35) where it ends up setting 
```
(*pbar)->iomap_uncached = true;
```
I think my card used [gf100_bar_new](https://elixir.bootlin.com/linux/v6.11.5/source/drivers/gpu/drm/nouveau/nvkm/subdev/bar/gf100.c#L192), and I set the same iomap_uncached flag there, and that cleared the mapping issue.

Having searched again to remind myself over this, I see there are various other caching related flags which could all be affecting this. It could take a moderate amount of work to get this working.