geerlingguy/raspberry-pi-pcie-devices #548: Test official Raspberry Pi NVMe SSD HAT

geerlingguy, 2023-10-10:
I don't have one, but I want one:

[image]

[image]

Some mentions of the HAT:

  - [Introducing: Raspberry Pi 5!](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/)
  - [Heating and cooling Raspberry Pi 5](https://www.raspberrypi.com/news/heating-and-cooling-raspberry-pi-5/)

> From early 2024, we will be offering a pair of mechanical adapter boards which convert between this connector and a subset of the M.2 standard, allowing users to attach NVMe SSDs and other M.2-format accessories. The first, which conforms to the standard HAT form factor, is intended for mounting larger devices. The second, which shares the L-shaped form factor of the new PoE+ HAT, supports mounting 2230- and 2242-format devices inside the Raspberry Pi 5 case.

geerlingguy, 2023-10-10:
See also: https://twitter.com/aallan/status/1707415398646202636

[image]

geerlingguy, 2023-10-11:
It is on the site now: https://pipci.jeffgeerling.com/hats/pi-nvme-hat.html

reinisb, 2023-12-16:
More information https://www.raspberrypi.com/news/double-standards/

reinisb, 2024-04-27:
For those interested here's the latest update from Jan 14, 2024. It looks like it has been taking longer than planned.

https://github.com/geerlingguy/raspberry-pi-pcie-devices/assets/6346595/5f467cba-0374-43d4-88c9-238fe90397e2

geerlingguy, 2024-04-27:
Oh wow, didn't know you can add videos inline in comments like that! But yes, I've heard they hit a few delays in getting it launched. Hopefully it'll be out soon!

The nice thing is there are a ton of great alternatives too.

reinisb, 2024-05-14:
It's now available! https://www.raspberrypi.com/news/m-2-hat-on-sale-now-for-12/

geerlingguy, 2024-05-15:
@reinisb - Indeed! I just pushed up a new image since they have official images now, and I have a blog post and video for it:

  - Video: https://www.youtube.com/watch?v=GYN3ub8Qb_I
  - Blog post: https://www.jeffgeerling.com/blog/2024/microsd-cards-sbc-days-are-numbered

geerlingguy, 2024-05-15:
Some quick impressions / notes, for reference:

  - The HAT fits snugly inside the official Pi 5 case, but not with the fan installed. You can fit the Active Cooler underneath, but it'll struggle a little pulling air through the tiny gaps.
  - The production version has a cutout for DSI/CSI cables (prototypes did not).
  - It includes an impedance-matched FFC for better signal integrity, and the cable has connections on opposite sides to help make it more foolproof in installation (though FFCs are still a bit finicky regardless).
  - It has a power and activity LED, and the activity LED flashes at a steady rate while the NVMe SSD is being accessed.
  - When you shut down the Raspberry Pi, power to the board is disabled.
  - It has a GPIO passthrough header, but AFAICT it is not necessary for the GPIO pins to be connected at all. I even ran it off a HatBRICK! Commander multi-PCIe device, and it was fine.
  - It includes standoffs and a GPIO riser, but to use GPIO pins for stackable HATs, you may need a different type of riser (e.g. with higher pins), depending on what you're doing.
  - It has threaded mounting points for 2230 or 2242 NVMe SSDs, and comes with a little thumbscrew with a notch into which the M.2 SSD is slotted. It's not hard to use, but it's still not as fun or easy as the little plastic screwless retention clips on ASUSTOR's Flashstor device!
  - I didn't have any issues with either a Pineboards Pinedrive or Cytron MakerDisk, both operated at greater-than-800 MB/sec at PCIe Gen 3 with no bus errors.

closed 2024-05-15 by geerlingguy