geerlingguy/raspberry-pi-pcie-devices #549: Test EVGA Nvidia RTX 3080 Ti on Raspberry Pi 5

geerlingguy, 2023-10-11:
I have tested the 3080 Ti on my Raspberry Pi 5 using Raspberry Pi's prototype PCIe adapter. It required an external power supply to feed that adapter and the card itself, and I had to install Nvidia's latest arm64 driver for the 3080 Ti, but after all that, I got the dreaded `RmInitAdapter failed!` message.

[image]

geerlingguy, 2023-11-20:
Testing with `nouveau` now...

geerlingguy, 2023-11-20:
Trying proprietary driver again. First disabling nouveau by creating a file `/etc/modprobe.d/blacklist-nouveau.conf` with the content:

```
blacklist nouveau
options nouveau modeset=0
```

Then after a reboot, it was not loading the nouveau module.

_May I interject debugging is SO much more fun when the whole system doesn't freeze every time requiring me to reset power?_

geerlingguy, 2023-11-20:
Downloading the Nvidia driver from https://www.nvidia.com/en-us/geforce/drivers/results/214101/

```
wget https://us.download.nvidia.com/XFree86/aarch64/545.29.02/NVIDIA-Linux-aarch64-545.29.02.run
chmod +x NVIDIA-Linux-aarch64-545.29.02.run
sudo ./NVIDIA-Linux-aarch64-545.29.02.run
```

After installation was complete, I rebooted, and dmesg shows:

```
[    3.756735] nvidia: loading out-of-tree module taints kernel.
[    3.756757] nvidia: module license 'NVIDIA' taints kernel.
[    3.756760] Disabling lock debugging due to kernel taint
[    3.840532] nvidia-nvlink: Nvlink Core is being initialized, major device number 509

[    3.845903] nvidia 0000:01:00.0: enabling device (0000 -> 0002)
[    3.845947] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    3.890053] NVRM: loading NVIDIA UNIX aarch64 Kernel Module  545.29.02  Thu Oct 26 21:50:44 UTC 2023
[    3.918561] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  545.29.02  Thu Oct 26 20:50:54 UTC 2023
[    3.925431] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    3.925439] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 2
[    6.148023] NVRM: Chipset not recognized (vendor ID 0x14e4, device ID 0x2712)
[    6.148030] The NVIDIA GPU driver for AArch64 has not been qualified on this platform
               and therefore it is not recommended or intended for use in any production
               environment.
[   10.688237] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x25:0x65:1470)
[   10.688306] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   14.949123] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x25:0x65:1470)
[   14.949180] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

That fun, and dreaded `RmInitAdapter failed!`.

geerlingguy, 2023-11-20:
See possibly related issue: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/514

I need to test with the [open-gpu-kernel-modules](https://github.com/NVIDIA/open-gpu-kernel-modules) bits.

Radiicall, 2024-03-11:
Have you tested with the new nouveau vulkan driver? It might have more luck initializing than the normal nouveau gl driver (I dont really know but its just an idea)

SpudGunMan, 2024-10-19:
Did this ever become reality or only amd driver?

geerlingguy, 2024-10-21:
@SpudGunMan - Only AMD for now. Going to hopefully do another round of testing with Nvidia cards soon.

yeongbba, 2024-10-24:
```
# nvidia-smi
No devices were found
```

```
2024-10-23T18:08:55.806865+00:00 worker01 kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x25:0x65:1468)
2024-10-23T18:08:55.806888+00:00 worker01 kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
2024-10-23T18:09:00.050861+00:00 worker01 kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x25:0x65:1468)
2024-10-23T18:09:00.050881+00:00 worker01 kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

I ran an NVIDIA GPU (4060 Ti) on Ubuntu 24.04.1 LTS, which I installed using Imager. I downloaded the 535 and 560 drivers, but unfortunately, the 'RmInitAdapter failed' error still occurred. When I executed `uname -r`, it printed `6.8.0-1013-raspi`. I think there might be a compatibility issue between the OS and the driver. If I install the [Ubuntu OS Desktop](https://ubuntu.com/download/desktop), will the OS run normally without any hardware issues?

geerlingguy, 2024-10-30:
Trying again today with the `550.127.05` version of Nvidia's aarch64 driver...

```
wget https://us.download.nvidia.com/XFree86/aarch64/550.127.05/NVIDIA-Linux-aarch64-550.127.05.run
chmod +x NVIDIA-Linux-aarch64-550.127.05.run
sudo ./NVIDIA-Linux-aarch64-550.127.05.run
```

TODO...

geerlingguy, 2024-10-30:
Attempting to uninstall and reinstall the 'open' version of the driver after reading [this issue](https://forums.developer.nvidia.com/t/solved-rminitadapter-failed-to-load-530-41-03-or-any-nvidia-modules-other-than-450-236-01-linux-via-esxi-7-0u3-passthrough-pci-gtx-1650/253239):

```
sudo nvidia-uninstall
sudo ./NVIDIA-Linux-aarch64-550.127.05.run -m=kernel-open

geerlingguy, 2024-10-30:
@Coreforge - Just a shot in the dark, but could this also be related to the BAR assignments like maybe the RX 6700 XT? At least with the open kernel driver, I get something more than the dratted "RmInitAdapter" lol

geerlingguy, 2024-10-30:
I've uninstalled the driver, and am rebuilding the kernel with `nouveau` enabled, to see how that fares.

geerlingguy, 2024-10-30:
If I set `dtoverlay=pcie-32bit-dma-pi5` in config.txt and reboot, I get:

```
[    3.980643] nouveau 0000:01:00.0: enabling device (0000 -> 0002)
[    3.980867] nouveau 0000:01:00.0: NVIDIA GA102 (b72000a1)
[    4.328955] nouveau 0000:01:00.0: bios: version 94.02.71.80.74
[    4.449813] nouveau 0000:01:00.0: bios: M0203E type 0a
[    4.449823] nouveau 0000:01:00.0: fb: 12288 MiB of unknown memory type
[    4.686936] nouveau 0000:01:00.0: sec2(acr): mbox 00000007 00000000
[    4.686946] nouveau 0000:01:00.0: sec2(acr):AHESASC: boot failed: -5
[    4.686951] nouveau 0000:01:00.0: acr: init failed, -5
[    4.687108] nouveau 0000:01:00.0: init failed with -5
[    4.687112] nouveau: DRM-master:00000000:00000080: init failed with -5
[    4.687121] nouveau 0000:01:00.0: DRM-master: Device allocation failed: -5
[    4.933155] nouveau: probe of 0000:01:00.0 failed with error -5
```

In my searching around, I found https://github.com/raspberrypi/linux/pull/5661, but trying `dtoverlay=dwc2-pi5` resulted in the same error as earlier.