geerlingguy/raspberry-pi-pcie-devices #553: Samsung Electronics NVMe SSD Controller PM9B1

VikingGer, 2023-10-25:
Hi

Not sure, if this is the right spot to post my experience with this or indirectly asking for help/suggestions what to try, but I'll do so anyway.

Setup:
Samsung Electronics Co Ltd NVMe SSD Controller PM9B1 [144d:a80b], 256GB NVME (1.8A power consump.), MZAL4256HBJD-00BL2 (OEM, from Lenovo as best guess)
CM4-IO-BASE-B
CM4 WLAN 8GB RAM no eMMC

Latest Firmware (may 23), latest Kernel tried (6.5 from Ubuntu 23.10)

Device doesn't boot from SD-card and reiterates at boot (waited about 10 minutes): nvme nvme0: I/O 1018 QID 8 timeout, completion polled
With the numbers varying.

Blacklisting nvme_init in the kernel cmdline leads to a working systems, however, clearly without nvme. The disk is recognized by the bus/shown via lspci but, as no driver/kernelmodule is loaded, not available as a device.

Beeing aware of the board beeing somewhat picky concerning nvme drives I'm aware that this device might simply not work; however I'm absolutely willing to try further cmdline kernelparameters (or other stuff) if there are any suggestions. The following didn't help:

nvme_core.default_ps_max_latency_us=0
pcie_aspm=off
nvme.noacpi=1
nvme.sgl_threshold=1

VikingGer, 2023-10-28:
I got the chance to swap those NVMEs towards some older ones. Booted fawlessly after DD and partition resizing:
Union Memory AM620 M.2 RPJTJ256MMEE1OWX, 1.0A (256GB)

I'll keep the issue open for your database, if you like.

Moving4407, 2024-11-25:
Same experience with this drive, does not work on a CM4

ned14, 2025-01-16:
With a very new eeprom this drive is nearly working well at PCIe 2.0 speeds:

```
rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue Jan 14 00:16:48 UTC 2025 (1736813808)
    LATEST: Tue Jan 14 00:16:48 UTC 2025 (1736813808)
   RELEASE: beta (/lib/firmware/raspberrypi/bootloader-2712/beta)
            Use raspi-config to change the release.
```

I let `fwupdmgr get-upgrades` apply the Samsung firmware update which apparently is critical:

```
Model Number:                       PM9B1 NVMe Samsung 512GB
Serial Number:                      S6MZNE1W359309
Firmware Version:                   46305039
```

It clearly has some problem with ASPM, the kernel will auto-disable that for you shortly after boot when it first hangs. `nvme_core.default_ps_max_latency_us=0` into `/boot/firmware/cmdline.txt` skips the hang.

The other thing I noticed is that this drive gets **hot** when doing anything. Like really surprisingly hot. It's a particularly small package, so the heat is concentrated. Maybe it's coincidence, but when running sustained writes it does get hot and I wonder if it thermally trips?

The drive will pass NVMe extended self checks etc and of course the NVMe firmware update worked fine. Sustained reads appear to work fine. The problem is sustained writes - run a five minute sustained write the drive drops out with i/o error even in PCIe 2.0. I've tried three different M.2 hats. I've tried five different PCIe cables. Identical results in all cases. I think the Pi and this drive just don't like each other.

TBH I bought this drive because of the Samsung brand. I assumed it had a Samsung controller, and was like all the other Samsung SSDs I have which is to say trouble free. However, the PM9B1 is actually a **Marvell** controller with Samsung flash. It's not an inhouse Samsung SSD controller. I hadn't realised this, so that's €35 probably down the drain as this SSD just isn't stable under sustained writes, and it doesn't work at all at PCIe 3.0.

I have other Samsung M.2 SSDs in Raspberry Pi and they're flawless under PCIe 3.0. Difference is those have the inhouse Samsung made controller. Those are trouble free. Shame.

https://wolfpaulus.com/rp5 reports that the PM9B1 works fine in his setup, so maybe earlier or later drive revisions work right.