geerlingguy/raspberry-pi-pcie-devices #555: PCIe link down Edgelogix controller

Danny-VdH, 2023-11-02:
Hi,
I'm trying to get the ethernet switch working in the edgelogix, they are connected to the PCI express slot on the CM4.
The CM4 has 2GB of ram.
This is the result I get with dmesg after changing the overlay file /boot/bcm2711-rpi-cm4.dtb

`
pi@raspberrypi:~ $ dmesg | grep -i pci
[    0.310151] PCI: CLS 0 bytes, default 64
[    1.200516] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.200555] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.200641] brcm-pcie fd500000.pcie:      MEM 0xffffffffffffffff..0x007ffffffe -> 0x0000000000
[    1.200720] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    1.200743] brcm-pcie fd500000.pcie: non-prefetchable memory resource required
[    1.520185] brcm-pcie fd500000.pcie: link down
pi@raspberrypi:~ $
`
I changed the overlay as followed,
`
        scb {
                compatible = "simple-bus";
                #address-cells = <0x02>;
                #size-cells = <0x02>;
                ranges = <0x0 0x7c000000  0x0 0xfc000000  0x0 0x03800000>,
                         <0x0 0x40000000  0x0 0xff800000  0x0 0x00800000>,
                         <0x6 0x00000000  0x6 0x00000000  0x2 0x00000000>,
                         <0x0 0x00000000  0x0 0x00000000  0x0 0xfc000000>;
                dma-ranges = <0x00 0x00 0x00 0x00 0x04 0x00>;
                phandle = <0xe0>;

                pcie@7d500000 {
                        compatible = "brcm,bcm2711-pcie";
                        reg = <0x00 0x7d500000 0x00 0x9310>;
                        device_type = "pci";
                        #address-cells = <0x03>;
                        #interrupt-cells = <0x01>;
                        #size-cells = <0x02>;
                        interrupts = <0x00 0x93 0x04 0x00 0x94 0x04>;
                        interrupt-names = "pcie\0msi";
                        interrupt-map-mask = <0x00 0x00 0x00 0x07>;
                        interrupt-map = <0x00 0x00 0x00 0x01 0x01 0x00 0x8f 0x04>,
                                                 <0x00 0x00 0x00 0x02 0x01 0x00 0x90 0x04>, 
                                                 <0x00 0x00 0x00 0x03 0x01 0x00 0x91 0x04>,
                                                 <0x00 0x00 0x00 0x04 0x01 0x00 0x92 0x04>;
                        msi-controller;
                        msi-parent = <0x2d>;
                        ranges = <0x02000000 0x0 0x00000000>, // PCI address
                                            <0x5 0x00000000>, // CPU address
                                            <0x0 0x80000000>; // Size
                        dma-ranges = <0x2000000 0x00 0x00 0x00 0x00 0x00 0xc0000000>;
                        brcm,enable-ssc;
                        brcm,enable-l1ss;
                        phandle = <0x2d>;

                        pci@0,0 {
                                device-type = "pci";
                                #address-cells = <0x03>;
                                #size-cells = <0x02>;
                                ranges;
                                reg = <0x00 0x00 0x00 0x00 0x00>;
                        };
                };
`
Can you give me a helping hand to get this up and running?
Thanks

6by9, 2023-11-03:
As with #554, this repo is with for recording PCI-E devices that work with the CM4, not a support forum for CM4 products.

In the first case you should ask Seeed for support on their product.
From https://www.seeedstudio.com/EdgeLogix-RPI-1000-CM4108032-p-5488.html I see they provide what is presumably an image for use on their product at https://sourceforge.net/projects/edgelogix-rpi-1000/files/. Have you tried that image?

If that fails, then the [Raspberry Pi forums](https:///forums.raspberrypi.com/) are the best place to ask.

If pcie reports `link down`, then the CM4 can't see anything connected. Look at hardware stuff first (eg is the CM4 properly seated in the sockets).

solomonould, 2023-11-29:
hey did you get this working have a very similar issue with the edgelogix from seed, the other NIC card on the pci bus works on a hard reset or power cycle but if i do sudo reboot i get brcm-pcie fd500000.pcie: link down
[    1.746772] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.746819] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.747405] pcieport 0000:00:00.0: PME: Signaling with IRQ 31
[    1.747874] pcieport 0000:00:00.0: AER: enabled with IRQ 31
any ideas lol? im currently on ssh and not with the device but it only has one NIC showing up which luckily is the one i use to ssh it over vpn haha