geerlingguy/raspberry-pi-pcie-devices #558: Add Waveshare PCIe-TO-SATA-4P to storage

yanfali, 2023-11-15:
This is a very interesting board [Amazon](https://www.amazon.com/dp/B0B6WQFYC7) because it has 4 power outputs and comes with all the SATA cables and brass standoffs to attach it to the CM4 carrier board.

The CM4 recognizes the chipset as ASM1061 but I've had issues with the driver crashing the kernel on boot depending on one port is plugged in. If you use ports 4 and 3, only one out of 3 boots will succeed. If you port 1, it boots most of the time. Seems like an enumeration bug, though the device is accompanied by a number of errors on boot pci errors on boot. 

Once it's up and running, it's rock solid and I'm able to sustain 350MiB/s reads from the device using hdparm, all day long.

[image]

yanfali, 2023-11-15:
Example errors

```
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00000080/00002000
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:    [ 7] BadDLLP
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00000080/00002000
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:    [ 7] BadDLLP
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00000080/00002000
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:    [ 7] BadDLLP
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: can't find device of ID0000
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00000080/00002000
Nov 14 20:25:58 cm4carrier kernel: pcieport 0000:00:00.0:    [ 7] BadDLLP
```

yanfali, 2023-11-15:
[image]

Example kernel panic

yanfali, 2023-11-15:
```
01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
        Subsystem: ZyDAS Technology Corp. Device 2116
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 38
        Region 0: Memory at 600080000 (32-bit, non-prefetchable) [size=8K]
        Region 5: Memory at 600082000 (32-bit, non-prefetchable) [size=8K]
        Expansion ROM at 600000000 [virtual] [disabled] [size=512K]
        Capabilities: <access denied>
        Kernel driver in use: ahci
        Kernel modules: ahci
```

yanfali, 2023-11-15:
Enumeration is very weird:

```
[    3.003824] ahci 0000:01:00.0: version 3.0
[    3.003875] ahci 0000:01:00.0: enabling device (0000 -> 0002)
[    3.008625] ahci 0000:01:00.0: SSS flag set, parallel bus scan disabled
[    3.008904] ahci 0000:01:00.0: AHCI 0001.0301 32 slots 24 ports 6 Gbps 0xffff0f impl SATA mode
[    3.008945] ahci 0000:01:00.0: flags: 64bit ncq sntf stag pm led only pio sxs deso sadm sds apst
[    3.096138] scsi host0: ahci
[    3.098708] scsi host1: ahci
[    3.107818] scsi host2: ahci
[    3.108821] scsi host3: ahci
[    3.109762] scsi host4: ahci
[    3.110715] scsi host5: ahci
[    3.111627] scsi host6: ahci
[    3.112613] scsi host7: ahci
[    3.119598] scsi host8: ahci
[    3.121619] scsi host9: ahci
[    3.126738] scsi host10: ahci
[    3.132593] scsi host11: ahci
[    3.135090] scsi host12: ahci
[    3.136087] scsi host13: ahci
[    3.136977] scsi host14: ahci
[    3.137863] scsi host15: ahci
[    3.138708] scsi host16: ahci
[    3.139642] scsi host17: ahci
[    3.140580] scsi host18: ahci
[    3.141435] scsi host19: ahci
[    3.148212] scsi host20: ahci
[    3.152384] scsi host21: ahci
[    3.153431] scsi host22: ahci
[    3.154296] scsi host23: ahci
[    3.154661] ata1: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082100 irq 38
[    3.154698] ata2: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082180 irq 38
[    3.154725] ata3: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082200 irq 38
[    3.154751] ata4: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082280 irq 38
[    3.154776] ata5: DUMMY
[    3.154793] ata6: DUMMY
[    3.154810] ata7: DUMMY
[    3.154826] ata8: DUMMY
[    3.154844] ata9: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082500 irq 38
[    3.154869] ata10: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082580 irq 38
[    3.154895] ata11: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082600 irq 38
[    3.154921] ata12: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082680 irq 38
[    3.154945] ata13: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082700 irq 38
[    3.154970] ata14: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082780 irq 38
[    3.154994] ata15: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082800 irq 38
[    3.155019] ata16: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082880 irq 38
[    3.155043] ata17: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082900 irq 38
[    3.155068] ata18: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082980 irq 38
[    3.155092] ata19: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082a00 irq 38
[    3.155117] ata20: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082a80 irq 38
[    3.155141] ata21: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082b00 irq 38
[    3.155165] ata22: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082b80 irq 38
[    3.155190] ata23: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082c00 irq 38
[    3.155214] ata24: SATA max UDMA/133 abar m8192@0x600082000 port 0x600082c80 irq 38
[    6.467989] ata1: found unknown device (class 0)
[    6.468054] ata1: SATA link down (SStatus 0 SControl 300)
[    6.782528] ata2: SATA link down (SStatus 0 SControl 300)
[    7.094521] ata3: SATA link down (SStatus 0 SControl 300)
[    7.406513] ata4: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    7.406775] ata4.00: ATA-10: SPCC Solid State Disk, S9FM02.0, max UDMA/133
[    7.406863] ata4.00: 234441648 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    7.407205] ata4.00: configured for UDMA/133
[    7.407927] scsi 3:0:0:0: Direct-Access     ATA      SPCC Solid State 02.0 PQ: 0 ANSI: 5
[    7.410033] sd 3:0:0:0: [sda] 234441648 512-byte logical blocks: (120 GB/112 GiB)
[    7.410157] sd 3:0:0:0: [sda] Write Protect is off
[    7.410197] sd 3:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    7.410371] sd 3:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    7.410602] sd 3:0:0:0: [sda] Preferred minimum I/O size 512 bytes
[    7.413296] sd 3:0:0:0: [sda] Attached SCSI disk
[    7.722531] ata9: SATA link down (SStatus 0 SControl 300)
[    8.034594] ata10: SATA link down (SStatus 0 SControl 300)
[    8.346512] ata11: SATA link down (SStatus 0 SControl 300)
[    8.658510] ata12: SATA link down (SStatus 0 SControl 300)
[    8.970513] ata13: SATA link down (SStatus 0 SControl 300)
[    9.282511] ata14: SATA link down (SStatus 0 SControl 300)
[    9.594506] ata15: SATA link down (SStatus 0 SControl 300)
[    9.906505] ata16: SATA link down (SStatus 0 SControl 300)
[   10.218504] ata17: SATA link down (SStatus 0 SControl 300)
[   10.530509] ata18: SATA link down (SStatus 0 SControl 300)
[   10.842513] ata19: SATA link down (SStatus 0 SControl 300)
[   11.154522] ata20: SATA link down (SStatus 0 SControl 300)
[   11.466548] ata21: SATA link down (SStatus 0 SControl 300)
[   11.778510] ata22: SATA link down (SStatus 0 SControl 300)
[   12.090508] ata23: SATA link down (SStatus 0 SControl 300)
[   12.402510] ata24: SATA link down (SStatus 0 SControl 300)
```

geerlingguy, 2023-11-15:
You're not the first person to report weird PCIe errors with one of Waveshare's 90°-mounted boards—I have a suspicion their through-PCB connection is a bit disastrous for PCIe signaling. It's tough to make a good solder connection between PCBs at right angles like that :/

yanfali, 2023-11-15:
> You're not the first person to report weird PCIe errors with one of Waveshare's 90°-mounted boards—I have a suspicion their through-PCB connection is a bit disastrous for PCIe signaling. It's tough to make a good solder connection between PCBs at right angles like that :/

It's a shame as the forn factor is great. This is perfect for drive testing and copying media.

PluueeR, 2024-04-03:
Had the same issues. According to this post it should work with an older kernel: https://bugzilla.kernel.org/show_bug.cgi?id=217276