geerlingguy/raspberry-pi-pcie-devices #559: Test Pineberry Pi's HatDrive! Top and Bottom NVMe HATs

geerlingguy, 2023-11-15:
Pineberry Pi is a joint venture between Michał Gapiński and Mirosław Folejewski — both of whom have a history of making some pretty cool Pi products.

They just announced their "HatDrive!" Line of HATs (one for the top, one for the bottom) for the Raspberry Pi 5.

[image]

I have one of each and will be testing it, along with their impedance-controlled FFCs (they make two lengths):

  - [HatDrive! Top](https://pineberrypi.com/products/hat-top-2230-2240-for-rpi5) for 2230/2242 M.2 E-key
  - [HatDrive! Bottom](https://pineberrypi.com/products/hatdrive-bottom-2230-2242-2280-for-rpi5) for 2230/2242/2280 M.2 E-key

geerlingguy, 2023-11-15:
Redirect added, and pages are up here:

  - https://pipci.jeffgeerling.com/hats/pineberry-pi-hatdrive-top.html
  - https://pipci.jeffgeerling.com/hats/pineberry-pi-hatdrive-bottom.html

Bouni, 2023-11-16:
Do you know if any RPi5 case that can be used with the **HatDrive! Bottom**? 
Pineberry has a picture on their website but it shows the **HatDrive! Top** variant I think.

ubopod, 2023-11-16:
I can't seem to find exact board outline dimensions for the bottom board. I was wondering if it would fit my enclosure...Is that published somewhere? I order a set btw and can verify that once they arrive.

geerlingguy, 2023-11-16:
A few quite notes testing the Top Hat:

  - I noticed my Pi 5 bootloader was from September. I ran `sudo rpi-eeprom-update -d -a` to upgrade to the latest bootloader and rebooted. Apparently older bootloaders could have issues booting off NVMe...
  - I didn't get the full hardware kit, just the PCB, so I had to improvise some spacers. It seems like you don't need GPIO at all if you just want to use NVMe. I think GPIO connections on the Top Hat are just for signaling (some sort of power measurement?)
  - You still have to enable PCIe like I document here: [NVMe SSD boot with the Raspberry Pi 5](https://www.jeffgeerling.com/blog/2023/nvme-ssd-boot-raspberry-pi-5)

geerlingguy, 2023-11-17:
Another note: I could not get the Pi to boot off NVMe if I used `rpi-clone`. I was able to do that on an earlier revision of the Pi 5 bootloader with the Pi prototype board... but a LOT has changed since then. It would try to boot, then drop me into a BusyBox initramfs shell. Very strange.

It works best if you flash Pi OS directly to the NVMe SSD, using the Pi with a USB adapter, Pi Imager direct, or on another computer.

geerlingguy, 2023-11-17:
[image]

geerlingguy, 2023-11-17:
A few questions about the board that I have now that I've put in a few hours...

  - How is the 5V power input on the 'Bottom' version of the board to be used? Will some sort of adapter cable be included?
  - How do you access the power monitoring on the 'Top' version?
  - Will the full production kit include a GPIO extender or some other means to get GPIO pins to extend into or through the 'Top' version?
  - Are there any cases that are compatible with either version specifically? (The Pi 5 Case is a bit tight, and airflow is a concern on it).

jack-ma, 2023-11-17:
What is the pitch of the FPC cable connecting the PCIe signal? It looks like 0.5mm?

mikegapinski, 2023-11-17:
1. We'll stock it but I have not found a device that needs it yet. We are playing it safe in case someone needs it, that is why we have power monitoring in the first place.
2. If you want to access the current monitoring chip via I2C (HatDrive! TOP only for now!!!):

`dtparam=i2c_vc=on`

You should see the INA with i2cdetect:

```
mgapinski@coralpi:/dev $ i2cdetect 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    
```
3. We have ordered GPIO extenders for all of our TOP hats in the 1st batch.
4. I am working on adapting a acrylic design (2 elements on top/bottom, sides exposed, it will also work with other boards we are developing now) , it won't be ready in 2 weeks but I'll drop you something once we are done. We are not dealing with enough volume to do plastic injection but a lot can change in a month.

Cristianistrate, 2023-11-17:
The official active cooler can be used with the TOP Hat ?  
Doe's significatively impact the cooling of the fan ?

mikegapinski, 2023-11-17:
Yes, and it does not seem like it is having a huge impact. It did not throttle when building the kernel

ThomasKaiser, 2023-11-17:
@geerlingguy as for your [disk benchmark mentioned on X](https://twitter.com/geerlingguy/status/1725368223460151318).

The 777 MB/s do not represent the device's capabilities since a) `hdparm` uses a rather small block size when benchmarking (it's a tool from last century originally meant to deal with spinning rust) and b) [you already showed that ~900 MB/s are possible with Gen3 speeds on RPi 5](https://www.jeffgeerling.com/blog/2023/testing-pcie-on-raspberry-pi-5).

So either testing methodology or SSD in question is to blame for these low numbers (most probably both) :)

As for the 'testing methodology'... PiBenchmarks' `Storage.sh` relies on `hdparm` (limited block size, in the past it was 128K, with more modern versions it looks like 1M) or `dd` with just 4K blocksize. Quick test with a SATA SSD should show the problem:

    hdparm -Tt --direct /dev/sda
    
    /dev/sda:
     Timing O_DIRECT cached reads:   754 MB in  2.00 seconds = 376.91 MB/sec
     Timing O_DIRECT disk reads: 1318 MB in  3.00 seconds = 439.28 MB/sec
    
    dd if=/dev/zero of=test bs=4k count=130k conv=fsync
    133120+0 records in
    133120+0 records out
    545259520 bytes (545 MB, 520 MiB) copied, 1.63406 s, 334 MB/s

`Storage.sh` tells us about 334 MB/s sequential write speeds and 439.28 MB/s sequential reads. While when testing with e.g. 16M blocksize we're at 490/520 MB/s in reality:

    iozone -e -I -a -s 100M -r 1024k -r 16384k -i 0 -i 1

              kB  reclen    write  rewrite    read    reread
          102400    1024   107457   436871   441420   439296                                                                                  
          102400   16384   486231   493974   519500   521398

No idea how you measured your 'nearly 900 MB/sec' on the Pi 5... but if it was only with `Storage.sh` I would revisit this :)

geerlingguy, 2023-11-17:
A few test results from Pineberry, of various SSDs:

| Model | PiBenchmarks.com score |
| --- | --- |
| [Sabrent 500GB Rocket PCIe 4.0 2280](https://amzn.to/3G3bjlp) | [47002](https://twitter.com/pineberrypi/status/1725506228401406148?s=61&t=W_7kAy4-MglIxSSCZQOmNg) |
| [Crucial P3 Plus 1TB 2280](https://amzn.to/3sGv6Eo) | [40118](https://twitter.com/pineberrypi/status/1725503223954297206?s=61&t=W_7kAy4-MglIxSSCZQOmNg) |
| [PNY CS1030 1TB 2280](https://amzn.to/49RPnYm) | [42203](https://twitter.com/pineberrypi/status/1725501353575797024?s=61&t=W_7kAy4-MglIxSSCZQOmNg) |

https://twitter.com/pineberrypi/status/1725498655367463148?s=61&t=W_7kAy4-MglIxSSCZQOmNg
https://twitter.com/pineberrypi/status/1725493979750257030?s=61&t=W_7kAy4-MglIxSSCZQOmNg

geerlingguy, 2023-11-17:
> No idea how you measured your 'nearly 900 MB/sec' on the Pi 5... but if it was only with Storage.sh I would revisit this :)

@ThomasKaiser - I only use PiBenchmarks.com script because it's a fun way of gamifying the testing—for my actual reporting/testing I use my own script: https://github.com/geerlingguy/pi-cluster/blob/master/benchmarks/disk-benchmark.sh

It runs fio at 1024k block size 4 GB test, then iozone 1024k and 4k random read and write tests. It's quick but effective for most cases. I am looking at expanding the benchmarks a little now that so many devices are reaching into "GB/sec" range, and trying to go for more gigabytes could help squash any additional caching that's happening and throwing off the results.

geerlingguy, 2023-11-17:
> The official active cooler can be used with the TOP Hat ?

Yep, see the video I just posted! https://www.youtube.com/watch?v=EXWu4SUsaY8

closed 2023-11-17 by geerlingguy

ThomasKaiser, 2023-11-17:
> It runs fio at 1024k block size 4 GB test

Well, just 1024k is IMO nothing that represents today's sequential access patterns any more. Even network protocols like AFP or SMB dynamically tune their internal blocksizes [to more than this in a LAN environment for over a decade now](https://www.helios.de/web/EN/support/TI/157.html). See my SSD example above: with 1M it's 430/440 MB/s write/read while it needs 16M to get closer to the real sequential transfer rates: 490/520 MB/s

And a write test 4GB in size will result with el cheapo SSDs in an amount of data written to fast cache and once the cache capacity is exceeded a drastic drop in write performance can be observed which – with a 4GB test size – will result in a MB/s number neither representing the 'cache write performance' nor the 'sustained write performance once cache is full' but a crude mixture.

Firing up `fio` with just 1GB data size but 4 times in a row generates both numbers and also an idea of how large the cache area of a specific SSD approximately is. Needless to say that once the cache area is exceeded the later `iozone` tests will also suffer from this and generate almost meaningless numbers (at least 4k random I/O) :)

AEW2015, 2023-11-17:
I want to try my FPGA load tester and see what the upper limit is.  I can get about 25 Watts on the compute blade with the CM4.  It was important to cool the CM4 to supply stable power otherwise the CM4 got to 80C very fast.

lijoseph, 2023-11-17:
Does the pineberry hat resolve wifi issue with the prototype raspberry pi m.2 nvme hat?

mikegapinski, 2023-11-17:
Jeff used the old eeprom firmware with the OG prototype HAT. It had tons of issues

balupton, 2023-11-18:
Does the bottom nvme board fit within the raspi case, with the cooler on the top?

FriendlyNGeeks, 2023-11-18:
Now you didn,t hear this from me, a birdie told me if you buy a spark fun qicc shim and bypass the voltage regulator you can i2c 5volts from pi to any accessories you want(pisugar, m.2[bottom] board, etc..)

DJJo14, 2023-11-23:
Maybe a stupid question but is it mounted with the nvme to the rpi, or the nvme to the table? Can you make a picture?

mikegapinski, 2023-12-11:
@geerlingguy our docs with drive benchmarks are up on https://docs.pineberrypi.com/nvme-compatibility-list.

Some WD drives that did not work on the CM4 also fail on the Pi5, it is safe to assume most of them work.

MacGyverr, 2023-12-18:
Couldn't find a case, so I made a case (for the bottom hat).
https://www.thingiverse.com/thing:6382978
[image]
[image]

rjbrown99, 2023-12-23:
> @geerlingguy our docs with drive benchmarks are up on https://docs.pineberrypi.com/nvme-compatibility-list.
> 
> Some WD drives that did not work on the CM4 also fail on the Pi5, it is safe to assume most of them work.

Protip for anyone buying one of these hats - **read that link**! I didn't do that ahead of time, bought a Western Digital SN770M, and wasted a bunch of time and effort trying to make it work. Reordered a Sabrent which shows up tomorrow.

tim-hilt, 2023-12-23:
Where can I go to ask about hardware-issues related to the Pineberry Pi HatDrive! Top?

My [Transcend 430s](https://www.transcend-info.com/product/internal-ssd/mts430s) doesn't show up in `lsblk` or `fdisk -l` and I'm not sure, whether this is a hardware-incompatibility or just wrong configuration on my part (didn't have to deal with hardware-troubleshooting too much yet).

If this is a good place, I could post the `journalctl`-output related to pcie. I also don't want to spam inappropriately.

fantasticweb, 2023-12-28:
I got both hatdrive modules for my raspberry pi 5 but sometimes I have the problem, that the ssd is not recognized on boot.
I use a Western Digital SN530 wich is running perfectly if the hat recognizes it.
To solve the problem I have to bend the ribbon cable and then mostly the ssd is reconized.
I think the cables are too thin or defective. It occurs with both cables - the long one from hatdrive bottom and the shorter one from hatdrive top.
Have you guys here had similar experiences?

MacGyverr, 2024-01-01:
Mine has been on since I got it, I've rebooted it multiple times. and I run Recalbox, but I upgraded the firmware and changed the boot order with the official OS first. I have no SD and just boot Recalbox from the M2. I didn't have an M2 to USB adapter so I booted from the SD and copied over the install image and ran the Raspberry Pi imager from the SD and installed the Recalbox image to the M2 that way. Has worked fine since.

geerlingguy, 2024-01-03:
I never tested the full Bottom kit until today—a few notes:

  - The little M.2 standoff is a little annoying in that you have to take off all four screws fixing the board to the Pi, so you can flip up the board to unscrew the M.2 standoff since the retention is integral to the standoff. It would be simpler I think if it were the type you had a screw up top, so you could pop the M.2 SSD out without having to take off the entire Bottom board.
  - The included PCIe cable is great... except the length and rigidity makes it effectively impossible to get at the microSD card slot (without disconnecting the cable). It would be nice if it could be out of the way (I know Pimoroni has a special cable for this).
  - It's too bad the first hardware revision was too early for the HAT+ standard; the cables included with HatDrive! aren't hot swappable with cables from other manufacturers (though other manufacturers weren't shipping devices until much later—some haven't yet!)
  - I tried on a new Pi 5 to just flash Pi OS to NVMe and boot it up... but that didn't work. Still have to do all the steps to configure EEPROM and such. Supposedly HAT+ devices could be auto recognized, though I haven't been able to test that.
  - The [Pineberry Pi Documentation](https://docs.pineberrypi.com/tutorial-connecting-your-hatdrive) currently only covers the HatDrive! top, not the bottom. It would be nice if that section had two parts, one for Top and one for Bottom.

grching, 2024-01-13:
Just received my top and bottoms. It would be nice if the M.2 ssd could be screwed in from the top and not bottom, I hope that they will consider that in any future revisions. The supplied standoffs M2.5x6mm should be either M2.5x10mm or M2.5x6+6mm. This allows the bottom to then have extra stand offs on the bottom to keep the SSD off of the table/case or what ever you are using. Currently, for the bottom standoffs I am using M2.5x20 nylon standoffs since the SSD has a heat sink. 

for the bottom I have a Samsung 980 Pro + heatsink and for the top a micron 2400 2TB 2230.

The top came with a 28 pin gpio extender it would be nice if it included a 40 pin extender. Since I only have RPI 4's and RPI 5's I have no experience with the smaller gpios. 

The installation went pretty smooth thanks to Jeff Geerling. 
Pineberry Pi did a wonderful job with the design of both hats. 

Samsung 980 Pro Gen 3:
     Category                  Test                      Result     
HDParm                    Disk Read                 794.56 MB/sec            
DD                              Disk Write                371 MB/s                 
                          Score: 45184

darkmanlv, 2024-01-14:
tested 2 ssd with Hat Drive! Top on rpi5:

Union Memory not working, didn`t see nvme at all, size 2242

[image]

Kingston NV1, size 2280, is working, detectable, but size not correct  

[image]

will buy different 2242 size ssd to test...

shain, 2024-01-16:
The HatDrive! Top works perfectly fine until I reboot the pi. Once I reboot or shutdown, the Nvme drive is no longer recognised until I unplug the pi from the power supply and boot it up again. This has happened with all 3 HatDrives that I have. Does anybody else have this issue? I am using WD SN530 Nvme drives.

mikegapinski, 2024-01-17:
> The HatDrive! Top works perfectly fine until I reboot the pi. Once I reboot or shutdown, the Nvme drive is no longer recognised until I unplug the pi from the power supply and boot it up again. This [...]

WD drives are tricky - this particular issue with the SN530 is listed on our docs page in the NVME compatibility section

mathiasscheurer, 2024-01-17:
I have similar problems (ssd not recognized an boot) with varoius drives. 
Unfortunately, my emails are ignored and my request to send new cables too...
The drives work perfectly with the adapter from pimoroni.

mikegapinski, 2024-01-17:
> I have similar problems (ssd not recognized at boot) with varoius drives. 
> 
> Unfortunately, my emails are ignored and my request to send new cables too...
> 
> The drives work perfectly with the adapter from pimoroni.

It is not feasible to send spare ribbon cables knowing the costs of international shipping. We have sent our backlog now, so you'll get a reply soon - my inbox is flooded at the moment.

mathiasscheurer, 2024-01-17:
Hello Michal, I would have covered the costs, but of course it's fine this way - thank You very much!

shain, 2024-01-17:
> WD drives are tricky - this particular issue with the SN530 is listed on our docs page in the NVME compatibility section

Thanks @mikegapinski, I have been able to get it to boot from an SN530 on one of the HatDrives without any issues. Do you have an idea what the root cause of the boot issue is? If it is an issue that can be fixed in the [rpi-eeprom](https://github.com/raspberrypi/rpi-eeprom), I can give it a shot and release it for other HatDrive users who maybe facing this issue.

darkmanlv, 2024-01-19:
bought sn740 today, size 2230, Pineberry HAT Top doesn`t detect at all

[image]

mikegapinski, 2024-01-19:
All WD drives are very similar to one another, we have 3 different SKUs that use the same PCB. That is why I decided to update our docs a while ago to say that we do not recommend ANY Western Digital drives at the moment. I have two SN530 drives in our lab and one of them only works. Different revision it looks like, consumer drives do not have a fixed BOM so they change controllers/nand from batch to batch

darkmanlv, 2024-01-19:
> All WD drives are very similar to one another, we have 3 different SKUs that use the same PCB. That is why I decided to update our docs a while ago to say that we do not recommend ANY Western [...]

I think problem is in "pci gen4 x4", as i saw that sn520 wd drive working good with pineberry hat's, but it is "pci gen3 x2", already bought on ebay one of this 10, will test when will be received :)  

Also maybe all drives with key M doesn`t work, because all pci gen4 x4 are without key b, only with key m, but older ones with pci gen3 x2 have b+m key....

mathiasscheurer, 2024-01-19:
I will test some Samsung and Kingspec SSDs next week. I will report on the results of my tests.

mikegapinski, 2024-01-19:
Samsung devices are OK in general, but I've seen some complaints about Kingspec through our support channels. I don't know this brand at all, but we could test their lineup if they are popular

slagathor69, 2024-01-24:
@geerlingguy 
> How is the 5V power input on the 'Bottom' version of the board to be used? Will some sort of adapter cable be included?

@mikegapinski 
> 1. We'll stock it but I have not found a device that needs it yet. We are playing it safe in case someone needs it, that is why we have power monitoring in the first place.

I can't get Samsung 990 PRO w/ Heatsink PCIe 4.0 NVMe SSD 4TB to work with HatDrive! Bottom. This may be the first drive that needs the additional power, where can I order this?

pawelek-org, 2024-01-24:
Hi Jeff & All!

I finally finished my Pi 5 setup with Pineberry Pi HatDrive! Bottom and Lexar NM620 256GB.

It took me a while because at first I got stuck with WD SN530. I couldn't force this little guy to boot but it seems this issue it's already mentioned on PineberryPi website. So I ordered Lexar NM620, tried again and it worked.

Thanks to Jeff's [rpi-clone](https://github.com/geerlingguy/rpi-clone) script I was able to transfer everything from microSD card to Lexar NVMe and finally boot from it.

I am pretty happy with this setup so far!

Here are some outputs using [Storage.sh](https://raw.githubusercontent.com/TheRemote/PiBenchmarks/master/Storage.sh) for LEXAR NM620 256GB.

PS. Greetings from Poland 🇵🇱

```
Score: 39051

     Category                  Test                      Result     
HDParm                    Disk Read                 432.48 MB/sec            
HDParm                    Cached Disk Read          347.03 MB/sec            
DD                        Disk Write                277 MB/s                 
FIO                       4k random read            108936 IOPS (435744 KB/s)
FIO                       4k random write           76992 IOPS (307969 KB/s) 
IOZone                    4k read                   135655 KB/s              
IOZone                    4k write                  169752 KB/s              
IOZone                    4k random read            69211 KB/s               
IOZone                    4k random write           183361 KB/s 
```
[image]
[image]

fantasticweb, 2024-01-24:
To copy the microSD with Raspberry Pi OS You can use "SD Card Copier" from "Accessoires" in the start menu.

pawelek-org, 2024-01-24:
> To copy the microSD with Raspberry Pi OS You can use "SD Card Copier" under "Accessoires" in the start menu.

Yes, but only using the GUI. I needed a shell script in Raspberry Pi OS Lite and `rpi-clone` worked just fine.

KombatPL, 2024-01-26:
Might I suggest taking down WD SN530 from compatibility [list](https://docs.pineberrypi.com/nvme-compatibility-list) - or at least mark it with red color or something...
I just bought Pi5 + Pineberry NVME HAT + org. power supply + one of the first recommended drives (SN530) from the list only to have it not boot (of course the fault is mine cause I didn't read bottom of the page). Might save some frustration for future buyers.
Found some old SK Hynix that works fortunately.

mikegapinski, 2024-01-26:
> Data: piątek, 26 stycznia 2024 o 20:34
> Do: geerlingguy/raspberry-pi-pcie-devices ***@***.***>
> DW: Michał Gapiński ***@***.***>, Mention ***@***.***>

Our docs say:

WD SN530 has been problematic for some users when it comes to NVMe boot, especially in the 2242 variant. Our test drive was a 2280 SKU and it worked OK but we generally recommend staying away from Western Digital drives completely.

KombatPL, 2024-01-26:
I know I just got confused because this was one of the first positions on the compatibility list :) 
Like I said - fault is mine - I didn't read till the end before placing order (only read it after troubleshooting it for 2 days lol), but It could save someone else having to walk that path if it Was at least marked 'READ BELOW' :)

darkmanlv, 2024-01-26:
Bought SN520 256gb, also not working. Don`t even boot with it installed. 

[image]

So my not working list so far, can be added to compatibility list on wiki.

WD SN740, size 2230, not detected after boot from sd card
Union Memory, size 2242, not detected after boot from sd card
WD SN520, size 2230, not booting from sd card at all

PS: all worked with usb to nvme adapter from aliexpress

slagathor69, 2024-01-26:
> @geerlingguy
>
>
>
>
> @mikegapinski
>
>
>
>
> I can't get Samsung 990 PRO w/ Heatsink PCIe 4.0 NVMe SSD 4TB to work with HatDrive! Bottom. This may be the first drive that needs the additional power, where [...]

To be clear my Samsung 990 PRO w/ Heatsink PCIe 4.0 NVMe SSD 4TB works fine on my PC and even lights up when connected via HatDrive! Bottom but is never recognized by Pi even after changing dtparam settings. Formatting is ext4

grching, 2024-01-27:
> @geerlingguy
>
>
> @mikegapinski
>
>
> I can't get Samsung 990 PRO w/ Heatsink PCIe 4.0 NVMe SSD 4TB to work with HatDrive! Bottom. This may be the first drive that needs the additional power, where can I [...]

I've tried the same device  minus the heat sink. Same issue the drive works via m.2 USB interface and even with a USB SSD adapter.  So the Samsung 990 Pro works just fine on with the RPI 5. 

The issue with the Pineberry Pi bottom could be either power (amps) or it could be a timing issue. I am trying to find the TTR (Time To Ready) for this device but I haven't found it yet. So it would be nice to understand how Pineberry Pi handles long TTR. I was hoping also that the RPI 5 could adjust it's polling for longer TTR devices (if possible). 

I would also be interested in the power adapter too. Where does it plug into? the RPI GPIO pins or another power supply?

codebuk, 2024-01-28:
Samsung 990 PRO 1T works intermittently for me with bottom hat. I have been testing external 5V on the bottom hat. The hat works with other drives.  

 Video output is also intermittent. 

Unfortunately the 5V power connection on the PI hat also back powers the PI. 

Any suggestions welcome. 

Bad boot:

[14:43:12.200] RPi: BOOTLOADER release VERSION:e891ded6 DATE: 2024/01/22 TIME: 14:44:36
[14:43:12.207] BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1705934676 serial edc8bb12 boardrev c04170 stc 4530236
[14:43:12.216] AON_RESET: 00000003 PM_RSTS 00001000
[14:43:12.240] usb_pd_init status 3
[14:43:12.244] USB_PD CONFIG 0 41
[14:43:12.245] XHCI-STOP
[14:43:12.247] xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[14:43:12.253] USBSTS 11
[14:43:12.253] xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[14:43:12.258] xHC0 ports 3 slots 64 intrs 4
[14:43:12.268] XHCI-STOP
[14:43:12.270] xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[14:43:12.274] USBSTS 11
[14:43:12.275] xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[14:43:12.279] xHC1 ports 3 slots 64 intrs 4
[14:43:12.283] Boot mode: NVME (06) order f14
[14:43:12.465] USB2[2] 00020ae1 connected
[14:43:12.496] USB2[1] 00020ae1 connected
[14:43:12.526] USB-PD: src-cap PDO object1 0x0a11912c
[14:43:12.530] Current 3000 mA
[14:43:12.531] Voltage 5000 mV
[14:43:12.533] USB-PD: src-cap PDO object2 0x0012d12c
[14:43:12.536] Current 3000 mA
[14:43:12.537] Voltage 9000 mV
[14:43:12.540] USB-PD: src-cap PDO object3 0x0014b12c
[14:43:12.542] Current 3000 mA
[14:43:12.544] Voltage 15000 mV
[14:43:12.545] USB-PD: src-cap PDO object4 0x00164145
[14:43:12.549] Current 3250 mA
[14:43:12.552] Voltage 20000 mV
[14:43:12.553] USB2[2] 00200a03 connected enabled
[14:43:12.557] USB2 root HUB port 2 init
[14:43:12.558] USB2[1] 00200a03 connected enabled
[14:43:12.562] USB2 root HUB port 1 init
[14:43:12.564] DEV [01:00] 2.00 000000:02 class 0 VID 17ef PID 6019
[14:43:12.568] DEV [01:00] 1.16 000000:01 class 0 VID 413c PID 2107
[14:43:12.577] HID [01:00] 2.00 000000:02 register HID
[14:43:12.580] HID [01:00] 1.16 000000:01 register HID
[14:43:13.049] PCI1 init
[14:43:13.050] PCI1 reset
[14:43:13.105] PCIe scan 0000144d:0000a80c
[14:43:18.108] Timeout 00000000 00000000 00000000 00000000

Good example with a SAMSUNG MZVLB256HBHQ-000L7  :

[16:40:03.992] RPi: BOOTLOADER release VERSION:e891ded6 DATE: 2024/01/22 TIME: 14:44:36
[16:40:03.998] BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1705934676 serial edc8bb12 boardrev c04170 stc 4552407
[16:40:04.008] AON_RESET: 00000003 PM_RSTS 00001000
[16:40:04.032] usb_pd_init status 3
[16:40:04.033] USB_PD CONFIG 0 41
[16:40:04.038] XHCI-STOP
[16:40:04.039] xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[16:40:04.044] USBSTS 11
[16:40:04.044] xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[16:40:04.050] xHC0 ports 3 slots 64 intrs 4
[16:40:04.059] XHCI-STOP
[16:40:04.059] xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[16:40:04.065] USBSTS 11
[16:40:04.066] xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
[16:40:04.071] xHC1 ports 3 slots 64 intrs 4
[... 93 lines cut ...]
[17:05:45.724] MBR: 0x00000000,       0 type: 0x00
[17:05:45.728] Trying partition: 0
[17:05:45.729] type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261116 (4)
[17:05:45.734] rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
[17:05:45.741] FAT32 clusters 261116
[17:05:45.743] [nvme] autoboot.txt not found
[17:05:45.745] Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
[17:05:45.752] Trying partition: 0
[17:05:45.753] type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261116 (4)
[17:05:45.759] rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
[17:05:45.764] FAT32 clusters 261116
[17:05:45.767] Read config.txt bytes     1213 hnd 0x126
[17:05:45.771] [nvme] pieeprom.upd not found
[17:05:45.914] usb_max_current_enable default 0 max-current 3000
[17:05:45.922] Read bcm2712-rpi-5-b.dtb bytes    75197 hnd 0x1048f
[17:05:45.926] dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
[17:05:45.933] dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712
[17:05:45.937] Selecting USB low current limit

[17:05:51.614] NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
[17:05:51.618] NOTICE:  BL31: Built : 14:26:57, Jun 22 2023

[17:06:03.185] Debian GNU/Linux 12 zbmvme990pro ttyAMA10

[17:06:03.189] zbmvme990pro login:

mikegapinski, 2024-01-28:
The power connector on the bottom board is not meant to be used with an external power supply, it is meant to be wired to the 5V GPIO pins. We are removing it from a future revision - it is not necessary and we put it there before we had our hands on the first pi5

The Samsung 990 is also tricky on other HATs as far as I know

Wysyłane z aplikacji Outlook dla systemu iOS<https://aka.ms/o0ukef>

Wysłane: Sunday, January 28, 2024 2:25:06 AM

codebuk, 2024-01-28:
Suggest you document your comments on the 5V connector somewhere on your site.....

kitopopo, 2024-02-02:
Hi, dear friends.

I use a HatDrive! Bottom from Pineberry pi in a raspberry pi 5 8gb with a nvme lexar nm620 256Gb. The nvme boots correctly with dtparam=pciex1_gen=2 in config.txt. I can change the dtparm to dtparam=pciex1_gen=3 the first timeand the nvme boots correctly with gen 3 and i make a speed test with hdparm and i get 830 MB/s, but sometimes if i power off the raspberry pi 5 and after power on the nvme does not boot (HatDrive leds are on and rasp fan is with high speed), If I power off and power on again the raspberry the HatDrive leds are always off (unrecoverable). I can solve the problem but i need plug the nvme in a computer and change the dtparam from gen3 to gen 2 and all works correctly. Sometimes change again from gen2 to gen3 and sometimes boot and sometimes the boot crash.

I have the feeling that it could be a failure of the eeprom since the failure occurs just when the raspberry pi is turned on, at the same time I have opened an issue in rpi-eeprom to see if they can solve it. While doing tests to solve the problem, my previous Lexar nm620 hard drive broke and I had to buy another one. I have also tried with another HatDrive Bottom board but the same problem. Please does anyone have a solution? I've had this problem for weeks. Thanks in advance. a cordial greeting

mikegapinski, 2024-02-02:
> Hi, dear friends.
>
> I use a HatDrive! Bottom from Pineberry pi in a raspberry pi 5 8gb with a nvme lexar nm620 256Gb. The nvme boots correctly with dtparam=pciex1_gen=2 in config.txt. I can change [...]

Lexar drives sometimes have issues under gen3(firmware?, it is OK in our testing), looks bootloader related - the controller times out the initialization in gen3. I can recommend removing other boot order entries. The pi should reboot and try to reinit the drive

kitopopo, 2024-02-02:
> ¡El HatDrive! Top funciona perfectamente bien hasta que reinicio el pi. Una vez que reinicio o apago, la unidad Nvme ya no se reconoce hasta que desconecto el pi de la fuente de alimentación y lo inicio nuevamente. Esto ha sucedido con los 3 HatDrives que tengo. ¿Alguien más tiene este problema? Estoy usando unidades WD SN530 Nvme.

Hi dear friend, similar problem, please se my penultimate repply. Thanks

mikegapinski, 2024-02-02:
> Hi dear friend, similar problem, please se my penultimate repply. Thanks

SN530 is listed as incompatible with NVMe boot on our docs page

kitopopo, 2024-02-02:
> removing other boot order entries

Thanks for your answare, can you explain how i can removing other boot order entries? I don't understand what I can eliminate to solve the problem. In any case, the Lexar nm620 appears on the list of full compatibility, which is why I bought this model. Thank you

[54 more replies]