geerlingguy/raspberry-pi-pcie-devices #564: Test GPU (XFX AMD Radeon RX 460 4GB GDDR5)

geerlingguy, 2023-11-23:
The RX 460 is a Polaris era GPU, and @Coreforge has already done a good amount of work getting one running (maybe the 2GB variant) over in #6.

[image]

But I thought I'd create a separate issue for this card since the RX 550 issue is already a bit long, and we are both testing on a Raspberry Pi 5 now, where this card may have more opportunity to shine.

Using [Coreforge's 6.1.x kernel fork](https://github.com/Coreforge/linux/tree/rpi-6.1.y-gpu-pi5), if you recompile the kernel, you'll end up with a working HDMI output, with working console output:

```
pi@pi5:~ $ neofetch
       _,met$$$$$gg.          pi@pi5 
    ,g$$$$$$$$$$$$$$$P.       ------ 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 12 (bookworm) aarch64 
 ,$$P'              `$$$.     Host: Raspberry Pi 5 Model B Rev 1.0 
',$$P       ,ggs.     `$$b:   Kernel: 6.1.62-v8_16k+ 
`d$$'     ,$P"'   .    $$$    Uptime: 8 mins 
 $$P      d$'     ,    $$P    Packages: 1604 (dpkg) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.15 
 $$;      Y$b._   _,d$P'      Resolution: 1920x1080 
 Y$$.    `.`"Y$$$$P"'         Terminal: /dev/pts/0 
 `$$b      "-.__              CPU: (4) @ 2.400GHz 
  `Y$$                        GPU: AMD ATI Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X 
   `Y$$.                      Memory: 207MiB / 8053MiB 
     `$$b.
       `Y$$b.                                         
          `"Y$b._                                     
              `"""
```

I have not been able to get wayfire/lightdm working (it sits there on a blinking cursor screen, and the `wireplumber` process seems to get stuck on something under the `lightdm` user. Coreforge was running with X11 and seemed to be able to run `glmark2`, Minecraft, Portal 1 and 2, and some other games, but currently is running with a PCIe x1 Gen 1 connection.

geerlingguy, 2023-11-23:
To use `radeontop`:

```
sudo apt install -y libdrm-dev libncurses-dev libxcb-dri2-0-dev
git clone https://github.com/clbr/radeontop.git
cd radeontop
make
./radeontop
```

Since I'm having trouble getting into lightdm / wayfire, it's slightly less useful to me right now though :D

geerlingguy, 2023-11-23:
If I use `raspi-config` to boot to CLI instead of desktop, I try running:

```
$ wayfire-pi
II 23-11-23 12:51:00.366 - [src/main.cpp:280] Starting wayfire version 0.7.5
II 23-11-23 12:51:00.366 - [libseat] [libseat/backend/seatd.c:64] Could not connect to socket /run/seatd.sock: No such file or directory
II 23-11-23 12:51:00.366 - [libseat] [libseat/libseat.c:76] Backend 'seatd' failed to open seat, skipping
Bus error
```

And:

```
$ startx
... get logged errors ...

$ cat /home/pi/.local/share/xorg/Xorg.0.log 
...
[  1476.387] (II) Applying OutputClass "AMDgpu" options to /dev/dri/card2
[  1476.387] (==) modeset(G0): RGB weight 888
[  1476.387] (==) modeset(G0): Default visual is TrueColor
[  1476.387] (II) Loading sub module "glamoregl"
[  1476.387] (II) LoadModule: "glamoregl"
[  1476.387] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[  1476.387] (II) Module glamoregl: vendor="X.Org Foundation"
[  1476.387] 	compiled for 1.21.1.7, module version = 1.0.1
[  1476.387] 	ABI class: X.Org ANSI C Emulation, version 0.4
[  1476.394] (EE) 
[  1476.395] (EE) Backtrace:
[  1476.397] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x188) [0x5555b82fc668]
[  1476.397] (EE) unw_get_proc_info failed: no unwind info found [-10]
[  1476.397] (EE) 
[  1476.398] (EE) Bus error at address 0x7ffec3a78080
[  1476.398] (EE) 
Fatal server error:
[  1476.398] (EE) Caught signal 7 (Bus error). Server aborting
[  1476.398] (EE) 
[  1476.398] (EE)
```

geerlingguy, 2023-11-23:
I grabbed Coreforge's memcpy library:

```
wget https://gist.githubusercontent.com/Coreforge/91da3d410ec7eb0ef5bc8dee24b91359/raw/b4848d1da9fff0cfcf7b601713efac1909e408e8/memcpy_unaligned.c

gcc -shared -fPIC -o memcpy.so memcpy_unaligned.c
sudo mv memcpy.so /usr/local/lib/memcpy.so
sudo nano /etc/ld.so.preload

# Put the following line inside ld.so.preload:
/usr/local/lib/memcpy.so
```

That got much further with wayfire...

```
II 23-11-23 12:57:46.203 - [backend/drm/drm.c:1553] Found connector 'DVI-D-1'
II 23-11-23 12:57:46.203 - [backend/drm/drm.c:1614] connector HDMI-A-3: Requesting modeset
II 23-11-23 12:57:46.203 - [src/core/output-layout.cpp:1098] new output: HDMI-A-3
II 23-11-23 12:57:46.203 - [src/core/output-layout.cpp:537] loaded mode auto
II 23-11-23 12:57:46.231 - [backend/drm/drm.c:734] connector HDMI-A-3: Modesetting with 1920x1080 @ 60.000 Hz
(type equals variant: [type: string, value: toplevel] | (type equals variant: [type: string, value: x-or] & focusable equals variant: [type: bool, value: 1]))
type equals variant: [type: string, value: overlay]
false
false
false
app_id equals variant: [type: string, value: Kodi]
(type equals variant: [type: string, value: toplevel] & floating equals variant: [type: bool, value: 1])
II 23-11-23 12:57:46.288 - [backend/drm/drm.c:1502] Scanning DRM connectors on /dev/dri/card1
II 23-11-23 12:57:46.290 - [backend/drm/drm.c:1553] Found connector 'HDMI-A-1'
II 23-11-23 12:57:46.294 - [backend/drm/drm.c:1553] Found connector 'HDMI-A-2'
EE 23-11-23 12:57:46.294 - [types/wlr_cursor.c:875] Cannot map device "smart-kvm Multifunction USB Device" to output (not found in this cursor)
EE 23-11-23 12:57:46.294 - [types/wlr_cursor.c:875] Cannot map device "pwr_button" to output (not found in this cursor)
EE 23-11-23 12:57:46.294 - [types/wlr_cursor.c:875] Cannot map device "vc4-hdmi-0" to output (not found in this cursor)
EE 23-11-23 12:57:46.294 - [types/wlr_cursor.c:875] Cannot map device "vc4-hdmi-1" to output (not found in this cursor)
EE 23-11-23 12:57:46.296 - [render/allocator/gbm.c:147] gbm_bo_create failed
EE 23-11-23 12:57:46.296 - [render/swapchain.c:109] Failed to allocate buffer
```

`startx` also got further... but I'm not sure what's up, it just ends up not rendering a display through the RX 460 at the point I run it (the system is not locked up however).

geerlingguy, 2023-11-23:
On the site now: https://pipci.jeffgeerling.com/cards_gpu/xfx-radeon-rx460-4gb.html

Coreforge, 2023-11-23:
Was there anything in dmesg when running wayfire or x11? 
I saw you had some issues compiling in #6. `compat_alignment.c` might only get compiled if `Kernel Features -> Kernel support for 32bit EL0 -> Fix up misaligned multi-word loads and stores in user space` is enabled (I should probably move the code into a separate file, as that option is disabled by default).
There might be something in newer mesa versions that doesn't get entirely fixed by the memcpy library that's now causing issues with startx as well, as I could get that running before without additional alignment in the kernel. Wayfire was triggering the alignment trap a few times though, so that currently won't work without it. If it's still getting stuck somewhere (with the alignment trap), dmesg will likely get spammed full of essentially the same error over and over again. I'd need at least the `Faulting instruction:` and ideally the `Load/Store: op0....` line if it's there as well to add the relevant instruction(s). I'm currently just adding them as I encounter issues, as there are quite a lot of load/store instructions on arm64.
My card has 4gb of vram as well, so that's not an issue.

geerlingguy, 2023-11-27:
@Coreforge - indeed, after enabling that flag, I can compile (with a number of warnings), rebooting now...

Running `wayfire-pi`, while the environment initializes, I see:

```
[   40.300504] Alignment fixup
[   40.300510] Faulting instruction: 0xa9001444
[   40.300513] Load/Store
[   40.300515] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x1
[   40.300517] Storing 8 bytes (pair: 1) to 0x7fff5056016c
[   40.309090] Alignment fixup
[   40.309098] Faulting instruction: 0xa9000c22
[   40.309101] Load/Store
[   40.309102] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x3
[   40.309105] Storing 8 bytes (pair: 1) to 0x7fff50568e7c
[   41.159727] Alignment fixup
[   41.159732] Faulting instruction: 0xa9001444
[   41.159735] Load/Store
[   41.159737] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x1
[   41.159739] Storing 8 bytes (pair: 1) to 0x7fff5056056c
[   41.289474] Alignment fixup
[   41.289486] Faulting instruction: 0xa9001444
[   41.289490] Load/Store
[   41.289491] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x1
```

[image]

When I clicked on the Pi menu, I saw:

```
[   41.289494] Storing 8 bytes (pair: 1) to 0x7fff5056096c
[  132.284968] Alignment fixup
[  132.284976] Faulting instruction: 0xa9001444
[  132.284980] Load/Store
[  132.284983] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x1
```

When I opened up Chromium there were maybe 30 or so fixups. 

I ran `glmark2` and it was spitting out hundreds (?) of fixups per second—certainly a huge number was running through.

It did give the GPU some work, though! Not enough to kick in the internal fans it seems... I *think* they work :P (the fun of testing used hardware...).

The last time I installed Minecraft on a Pi I just used Pi-Apps — is there a preferred place where you grab it?

Coreforge, 2023-11-28:
Were there a bunch of fixup messages in dmesg without the`Storing %d bytes` message when glmark failed? (they might get mixed up a bit, but if it's getting stuck on an instruction, there should be a lot of other messages without that one)
The fans were occasionally spinning up on my card, but not very much, so depending on the fan curve, cooler, and power profile on your card it might just not get warm enough with these loads.
I'm just running minecraft from a technic install where I replaced the native libraries with arm versions and echoed out the actual launch command (since the launcher would otherwise overwrite the libraries again with x86 versions). I think some launchers directly support arm now, but I haven't tried any in a while.

qwertychouskie, 2023-11-28:
SuperTuxKart ox max settings would probably be a good benchmark for these cards, as it's arm64 native, OpenGL/GLES based, and in the Raspbian repos.

geerlingguy, 2023-11-28:
A few notes:

  - Fans spin up (slowly then fast briefly) at boot, so at least I know they work :)
  - Haven't tried games yet, but SuperTuxKart on high would be good to see, for sure (cc @qwertychouskie)
  - For some reason the mouse cursor disappears once I move the mouse at all. Makes using the GUI a bit fun :D (Does this happen to you too @Coreforge?)
  - I've pasted below the last few dozen messages from my last failed glmark2 run:

```
[  564.557038] Faulting instruction: 0xf8226865
[  564.557039] systemd-journald[2049]: /dev/kmsg buffer overrun, some messages lost.
[  564.557039] Load/Store
[  564.557041] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557043] Alignment fixup
[  564.557044] Faulting instruction: 0xf8226865
[  564.557046] Load/Store
[  564.557047] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557049] Alignment fixup
[  564.557051] Faulting instruction: 0xf8226865
[  564.557052] Load/Store
[  564.557053] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557055] Alignment fixup
[  564.557057] Faulting instruction: 0xf8226865
[  564.557058] Load/Store
[  564.557059] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557062] Alignment fixup
[  564.557063] Faulting instruction: 0xf8226865
[  564.557064] Load/Store
[  564.557065] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557068] Alignment fixup
[  564.557069] Faulting instruction: 0xf8226865
[  564.557071] Load/Store
[  564.557072] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557074] Alignment fixup
[  564.557075] Faulting instruction: 0xf8226865
[  564.557077] systemd-journald[2049]: /dev/kmsg buffer overrun, some messages lost.
[  564.557078] Load/Store
[  564.557146] Load/Store: op0 0xf op1 0x0 op2 0x0 op3 0x22 op4 0x2
[  564.557254] systemd-journald[2049]: /dev/kmsg buffer overrun, some messages lost.
[  564.560311] Alignment fixup
[  564.560317] Faulting instruction: 0xa9001444
[  564.560320] Load/Store
[  564.560321] Load/Store: op0 0xa op1 0x0 op2 0x2 op3 0x0 op4 0x1
[  564.560324] Storing 8 bytes (pair: 1) to 0x7fff3831296c
[  564.760582] systemd[1]: Started systemd-journald.service - Journal Service.
```

It was stuck on `[ideas]` both times I think. (The blue wireframey wavey one)

geerlingguy, 2023-11-29:
This thread may get a little more activity, I just posted a video: [You can use external GPUs on the Raspberry Pi 5](https://www.youtube.com/watch?v=BLg-1w2QayU).

theofficialgman, 2023-11-29:
> The last time I installed Minecraft on a Pi I just used Pi-Apps — is there a preferred place where you grab it?

@geerlingguy Pi-Apps to install Minecraft (Minecraft Bedrock, Minecraft Java with Prism Launcher, and Minecraft Pi) will work well. All of them are native ARM64. I would love to see Minecraft Java with Prism Launcher running on that setup hopefully with the Simply Optimized modpack or similar.

theofficialgman, 2023-11-29:
I need to try this myself at some point.

I guess one of those mining riser cards like this
[image]

https://www.amazon.com/BEYIMEI-VER010-X-Adapter-Bitcoin-Ethereum/dp/B09BVNSFN8?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&smid=A1BM86NEBPKXB0&th=1

plus the m.2 hat should do the trick
[image]

https://pineberrypi.com/products/hat-top-2230-2240-for-rpi5

geerlingguy, 2023-11-29:
Regarding the cursor, according to [this YouTube comment](https://www.youtube.com/watch?v=BLg-1w2QayU&lc=Ugy59inoCoPpveWSDud4AaABAg) I could add the environment variable `WLR_NO_HARDWARE_CURSORS=1` to use the software renderer, and that would hopefully keep it visible for now :)

Coreforge, 2023-11-29:
The faulting instruction seems to just be a 64bit store (which, since I haven't encountered them, I haven't added yet). I'll hopefully get it added in the next few days.

DanaGoyette, 2023-11-29:
If PCIe on the Pi is anything like on the LX2160A, the GPU might be falling off the bus when the PCIe link rate changes as a power-savings feature.  One way to work around that is to set `amdgpu.pcie_gen_cap` to `0x10001` for gen1, `0x20002` for gen2, or `0x40004` for gen3.  While you're at it, you might try `amdgpu.aspm=0`.  

There's also a double-negative `amdgpu.noretry=0` to enable retries of... something, I don't know exactly what.

I'm also curious if amdgpu's HDMI audio sounds correct on the Pi.  On the LX2160A, the audio comes out crackly and garbled.

DanaGoyette, 2023-11-29:
> SuperTuxKart ox max settings would probably be a good benchmark for these cards, as it's arm64 native, OpenGL/GLES based, and in the Raspbian repos.

Another good thing to try is the game Veloren.  The launcher for it is available in flatpak: `net.veloren.airshipper`
It's a game that has support for Vulkan, DX12, and Metal, with native ARM64 versions for Linux and Mac OS.

If you create a world with a fixed non-zero seed, and then Specate World, I think you should end up with the same map viewed from about the same place, possibly with different weather at a given moment.

stalkerg, 2023-11-29:
seems like AMD is also working on it, https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.7-rc3&id=ba0fb4b48c19a2d2380fc16ca4af236a0871d279 

Did you talk about it on amd-gfx mail list?

Coreforge, 2023-11-29:
I got the uPCity today (Thanks to the Pineberry people again!), but haven't had too much time to do benchmarks yet. 
`glmark2` is now also crashing for me on the ideas test with the same instruction, so I guess I just had some old library that got updated when I installed wayfire that did something differently. Minecraft has a similar issue, so until I get that added, I only have this partial run of glmark2. Since I didn't do another run with the old adapter on the same libraries, I don't think the numbers can be compared directly. I saw a GPU utilization of 95% through a good part of the run though, so it doesn't seem like it's getting CPU limited (and toning down the logging should also help with that).
```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      AMD
    GL_RENDERER:    AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.6, DRM 3.49, 6.1.61-v8_16k+)
    GL_VERSION:     4.6 (Compatibility Profile) Mesa 23.2.1-1~bpo12+rpt2
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   3840x2160 fullscreen
=======================================================
[build] use-vbo=false: FPS: 342 FrameTime: 2.929 ms
[build] use-vbo=true: FPS: 1799 FrameTime: 0.556 ms
[texture] texture-filter=nearest: FPS: 1824 FrameTime: 0.548 ms
[texture] texture-filter=linear: FPS: 1875 FrameTime: 0.533 ms
[texture] texture-filter=mipmap: FPS: 1866 FrameTime: 0.536 ms
[shading] shading=gouraud: FPS: 1750 FrameTime: 0.572 ms
[shading] shading=blinn-phong-inf: FPS: 1770 FrameTime: 0.565 ms
[shading] shading=phong: FPS: 1741 FrameTime: 0.574 ms
[shading] shading=cel: FPS: 1734 FrameTime: 0.577 ms
[bump] bump-render=high-poly: FPS: 1755 FrameTime: 0.570 ms
[bump] bump-render=normals: FPS: 1810 FrameTime: 0.553 ms
[bump] bump-render=height: FPS: 1821 FrameTime: 0.549 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 1411 FrameTime: 0.709 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 508 FrameTime: 1.972 ms
[pulsar] light=false:quads=5:texture=false: FPS: 1434 FrameTime: 0.698 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 523 FrameTime: 1.913 ms
[desktop] effect=shadow:windows=4: FPS: 904 FrameTime: 1.107 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 123 FrameTime: 8.158 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 136 FrameTime: 7.383 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 158 FrameTime: 6.364 ms

```
I'll try SuperTuxCart as well once I get the things I had running running again.

volyrique, 2023-11-29:
Hello, one benchmark that I would suggest is [GravityMark](https://gravitymark.tellusim.com/). It has support for modern features such as ray tracing and I think that it has a native AArch64 build for all major desktop operating systems - certainly for Linux (I have tried it myself on an Ampere eMAG machine).

Geekbench's compute benchmark is another option, covering a different GPU use case.

theofficialgman, 2023-11-29:
> Geekbench's compute benchmark is another option, covering a different GPU use case.

@volyrique Geekbench compute (at least the vulkan backend) isn't in the arm64 geekbench 5/6 builds. It is only in the x86_64 builds. I have been asking jfpoole to add it for a year now but they won't add it for some reason. You can run the geekbench 5/6 x86_64 compute benchmark through box64 though and it does work fully with (probably) minimal overhead. That is what I did to get the geekbench vulkan compute benchmark results here -> https://forums.raspberrypi.com/viewtopic.php?p=2144650#p2140061 on Pi4 and Pi5

geerlingguy, 2023-12-02:
@Coreforge Awesome, thanks! Could you also try running it windowed inside the `wayfire-pi` environment too? (If you get a chance).

theofficialgman, 2023-12-02:
Just for a frame of reference, on my RX 570 on x86_64 desktop I get ~3200 points at the same resolution. So I think you are getting the full performance in that benchmark since the RX 460 is supposed to be a bit less than 1/2 as powerful as the RX 570. Do you have a reference test of that RX 460 on a desktop x86_64 computer?

theofficialgman, 2023-12-02:
Also just a suggestion, I imagine you have trouble using the raspberry pi patched and built chromium with anything other than the pi4/pi5 videocore gpus. You will probably have better luck with vanilla chromium either from building chromium from source, using the chromium flatpak (which is vanilla chromium without any real patches), or using my chromium debs that you can find here -> https://github.com/theofficialgman/testing/releases/tag/gmans-releases (note I don't suggest using my chromium debs all the time since I have specifically patched out libVPX support to use ffmpeg for vp9 hardware decoding on nvidia tegra systems).

Coreforge, 2023-12-02:
I didn't do benchmarks on x86 with the card, but since it was showing 100% utilization in most tests, that should be the full performance in those. I haven't run any browsers so far, but I'll keep that in mind if I do run into any issues.

Coreforge, 2023-12-07:
Fixing SIMD instructions seems to be more complicated, as the SIMD registers don't get saved as part of the `pt_regs` struct. I'm currently using `fpsimd_save_state()` to save them to a temporary location and just assume that nothing in between the exception and that part in the handler changes those registers (which should be the case), but even though unity games work somewhat now, they segfault sometimes, and not always at the same point. 
The issues always occur in the `UnityGfxDeviceW` thread, which isn't a big surprise, but I'm not exactly sure what's causing them, I need to do more debugging on that. It does look like I'm not properly fixing some instructions though, as there are graphical glitches in the menu of Getting Over It, and it's even worse in-game. I have it narrowed down to a 128bit unsigned immediate `str`, but other than maybe not getting the correct data to write to memory (if `fpsimd_save_state()` doesn't work properly for this), I'm not really sure what's causing these problems.

Coreforge, 2023-12-08:
What I was doing before was definitely not working.
I changed to read the vector registers from `current->thread.uw.fpsimd_state.vregs`, and call `kernel_neon_begin()` before reading the data to ensure it's saved (and then `kernel_neon_end()` afterwards), and it's certainly looking better (at least the main menu of Getting Over It is), but Getting Over It still crashes in the same place. Looking around with GDB, I couldn't find much obvious either. It is related to SIMD instruction fixes, as I saw a null pointer dereference when I changed it to just write 0 for all SIMD fixups, but I'm not sure where exactly I'm doing something wrong. I pushed my current code again though.

I tried a few more games that I could run as well:
DOOM 2016 with OpenGL can at least get to the menu (I haven't tried further), but the performance isn't too great. (CPU bound, might also be affected by the logging going on). Both DOOM 2016 and DOOM Eternal don't launch with Vulkan, they complain about being unable to initialize vulkan (although I know both work with this card, and vulkan works fine on the pi too).
The Talos Principle runs, although performance isn't too great here either. It's also CPU bound, and it wasn't spamming dmesg (though it could still be triggering the alignment trap a lot). This game uses Vulkan.

I also did two runs of GravityMark, one on [vulkan](https://gravitymark.tellusim.com/report/?id=053eb14f5d50a36f1f6abb42447a9595e5988034) and one on [opengl](https://gravitymark.tellusim.com/report/?id=7be02e29e1d522e06a4ed1ca7ae2108a2853516c). The vulkan one ran and looked just fine, the opengl one was triggering the alignment trap a lot again with SIMD instructions, and the earth was flickering or not there at all for parts of the run. 

I saw that there were (likely) two instructions causing problems:
```
[ 6964.423429] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f68438038
[ 6964.457330] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f6842529c
[ 6964.491163] SIMD: storing 0x40300000402 40100000000 (64 bits) at 0x0000007f68401f9c
[ 6964.491173] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f68401fb8
[ 6964.526794] SIMD: storing 0x40300000402 40100000000 (64 bits) at 0x0000007f6840cc1c
[ 6964.526804] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f6840cc38
[ 6964.560762] SIMD: storing 0x40300000402 40100000000 (64 bits) at 0x0000007f6841981c
[ 6964.560773] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f68419838
[ 6964.594807] SIMD: storing 0x40a00000409 40800000407 (128 bits) at 0x0000007f6842869c
[ 6964.628703] SIMD: storing 0x40300000402 40100000000 (64 bits) at 0x0000007f684353f4
[ 6964.664459] SIMD: storing 0x40300000402 40100000000 (64 bits) at 0x0000007f68411f9c

```
I've been mostly focused on the 128bit stuff for now (as one of the instructions I saw causing a lot of issues was a 128bit str), so I suspect the 64bit one might not get handled properly.

I also gave monado a try (I wasn't expecting anything too special, as it mainly just needs vulkan, which works fine), and it works fine, though I haven't been able to get anything other than hello_xr to run.

Coreforge, 2024-01-12:
Apparently I forgot to read the second data register for SIMD instructions, so instructions `stp qx, qx, [xx]` only stored the first register, and not the second one. I also added some code to keep track of which instructions have already been handled at some point and which haven't, to make it easier to find potentially bad ones.

I added a bunch of instruction tests to my [gpu memory access tests](https://github.com/Coreforge/pi-gpu-tests) (all instructions I've seen get handled when launching Getting Over It or The Long Dark), which helped me find the `stp` issue. All instructions being tested are now being handled correctly. However, Unity games are still segfaulting, and I don't know why. I might try rendering them with llvmpipe instead to make sure it's actually related to the GPU, although I think it is.

Coreforge, 2024-01-15:
Most load and store instructions should work now (at least I haven't come across any that didn't seem to so far), except for loads with sign extension, for which the sign extension is currently being silently ignored (if it's not causing other decoding issues).

Apart from removing debug code, there's still a good bit of optimization needed though. DOOM Eternal was running at 15-40fps at 1920x1080, mostly around 20fps, and while that's just about playable, GPU utilization was as low as 40% sometimes (though it also went up to 90%) while the CPU was basically pinned at 100% on all cores. This card didn't run the game too well even in my PC, but it was better.

I still need to figure out why some unity games are crashing (but also not all, some work), but I'd also be interested if someone has other things they'd like to be tested (I still need to try geekbench), as I've been mostly just trying to run games so far.

geerlingguy, 2024-10-03:
@Coreforge - Just popping into this thread as I'm going to take another stab with the RX 460 after seeing [Pineboards running 4K gaming demos on theirs last month](https://pineboards.io/blogs/news/maker-faire-hannover-2024-is-a-wrap).

Did you or anyone else work on rebasing your branch on [`rpi-6.6.y`](https://github.com/raspberrypi/linux), which is the current stable branch for Pi OS? (cc @mikegapinski)

Coreforge, 2024-10-03:
I haven't tried it, but if not too much changed in the relevant parts of the amdgpu driver, it shouldn't be too complicated to do.

geerlingguy, 2024-10-03:
I'm rebasing `coreforge/rpi-6.1.y-gpu-pi5` on top of `rpi-6.6.y`; there are some conflicts, so I'll step through them and see if I can get it working.

Coreforge, 2024-10-03:
I can probably also give it a try tomorrow if you run into any snags.  
I also recently got an orange pi 5 and should get an M.2 adapter for it in a few weeks, so I can then see if there's any relevant differences there (I'm mostly just expecting some improvements in regards to CPU performance, and having double the RAM should help too).

Coreforge, 2024-10-04:
Looks like I moved everything to [6.7.y](https://github.com/Coreforge/linux/tree/rpi-6.7.y-gpu) at some point and just didn't upload it (last time I touched it was for some raspberry pimax shenanigans). I didn't do a proper rebase though, so I'll check if I made any relevant changes (mostly in the alignment trap) and then do a proper rebase to 6.6.y.

geerlingguy, 2024-10-04:
@Coreforge - Thanks! I'm trying to get a demo set up of SuperTuxKart at 4K for a livestream I'm working on tomorrow (the point of the livestream is mostly to talk about various PCIe things, not just GPU), but I figured it would be good to update people on the state of the patches, and how little is required at this point to get basic AMD graphics working.

Coreforge, 2024-10-04:
Got it properly rebased to [6.6.y](https://github.com/Coreforge/linux/tree/rpi-6.6.y-gpu) now. 
Make sure to enable `Kernel Features -> Kernel support for 32-bit EL0 -> Fix up misaligned multi-word loads and stores in user space`, as otherwise, the alignment trap doesn't get compiled, but since I haven't added the necessary preprocessor macros yet, the linker will then fail.

geerlingguy, 2024-10-04:
@Coreforge - Do I still need to [patch in the memcpy library](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/564#issuecomment-1824828828) before recompiling?

Coreforge, 2024-10-04:
I still have it preloaded on my pi. It might not be absolutely necessary (though I haven't done any testing i quite a while), but there was an issue with xorg without it if I remember correctly.  
It should also be a lot more performant than relying on the alignment trap.

geerlingguy, 2024-10-04:
@Coreforge Do you remember what kind of FPS you were getting with SuperTuxKart (if you had tested that)? It seems like I'm still stuck at 17fps windowed, which is the same as it was on the Pi's internal GPU. Running `glmark2-es2`, I'm seeing `radeontop` hitting 50-80% GPU utilization—supertuxkart wasn't doing that, it was 10-20%.

Coreforge, 2024-10-04:
Just tested it again, I got 30-60fps, full screen at 4k. I didn't add any special options though. Can you check the log output if it's using the right GPU? Also maybe check the CPU utilization if it is.

geerlingguy, 2024-10-04:
Weird - glmark2-es2 was also running on the internal GPU, not on the AMD GPU, even connected through the GPU.

Here's `dmesg`:

```
pi@pi5-pcie:~ $ dmesg | grep amdgpu
[    5.097704] [drm] amdgpu kernel modesetting enabled.
[    5.097858] amdgpu 0000:01:00.0: enabling device (0000 -> 0002)
[    5.339556] amdgpu 0000:01:00.0: amdgpu: Fetched VBIOS from ROM BAR
[    5.339567] amdgpu: ATOM BIOS: 113-BAFFIN_PRO_160513_D5_4GB_MIC_0_W82
[    5.339612] amdgpu 0000:01:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[    5.339616] amdgpu 0000:01:00.0: amdgpu: PCIE atomic ops is not supported
[    5.823739] amdgpu 0000:01:00.0: BAR 2: releasing [mem 0x1810000000-0x18101fffff 64bit pref]
[    5.823745] amdgpu 0000:01:00.0: BAR 0: releasing [mem 0x1800000000-0x180fffffff 64bit pref]
[    5.823776] amdgpu 0000:01:00.0: BAR 0: assigned [mem 0x1800000000-0x18ffffffff 64bit pref]
[    5.823784] amdgpu 0000:01:00.0: BAR 2: assigned [mem 0x1900000000-0x19001fffff 64bit pref]
[    5.823802] amdgpu 0000:01:00.0: amdgpu: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
[    5.823805] amdgpu 0000:01:00.0: amdgpu: GART: 256M 0x000000FF00000000 - 0x000000FF0FFFFFFF
[    5.823891] [drm] amdgpu: 4096M of VRAM memory ready
[    5.823894] [drm] amdgpu: 4026M of GTT memory ready.
[    5.850612] amdgpu: hwmgr_sw_init smu backed is polaris10_smu
[    6.364917] amdgpu 0000:01:00.0: amdgpu: SE 2, SH per SE 1, CU per SH 8, active_cu_number 14
[    6.368499] amdgpu 0000:01:00.0: amdgpu: Using BACO for runtime pm
[    6.368800] [drm] Initialized amdgpu 3.54.0 20150101 for 0000:01:00.0 on minor 2
[    6.391209] amdgpu 0000:01:00.0: [drm] fb0: amdgpudrmfb frame buffer device
```

I'm just using the default Pi 5 OS install though, with Wayland. Is there anything I need to do to force it to use the AMD GPU?

geerlingguy, 2024-10-04:
I've installed `mesa-utils`, and here's `glxinfo`:

pi@pi5-pcie:~ $ DISPLAY=:0 glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
...
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
...
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Broadcom (0x14e4)
    Device: V3D 7.1 (0xffffffff)
    Version: 23.2.1
    Accelerated: yes
    Video memory: 8052MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 3.1
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Broadcom
OpenGL renderer string: V3D 7.1
OpenGL core profile version string: 3.1 Mesa 23.2.1-1~bpo12+rpt3
```

Hmm... how do I get OpenGL to use amdgpu?

Coreforge, 2024-10-04:
I'm on xorg, could that have something to do with it (though I tried wayland a while ago too)? 
Check if there are multiple entries in `/dev/dri`.
`xrandr --listproviders` should also list multiple cards.  
If both cards are available, `DRI_PRIME=1` should select the second one. 

Oddly enough though, on my system, the integrated GPU isn't available. That probably has something to do with 
```
[    0.348449] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    0.356149] bcm2708_fb: probe of soc:fb failed with error -2
```

Coreforge, 2024-10-04:
Try `DRI_PRIME=1` and `DRI_PRIME=2` then. Having `card0`, `card1` and `card2` there is a good sign, as well as `renderD129`, as that looks like it's just both GPUs working and the default one for some reason being the internal one.

geerlingguy, 2024-10-04:
@Coreforge - I've tried `DRI_PRIME=1 supertuxkart` in the terminal, but is that enough to force it? (Also tried `DRI_PRIME=2` but same result - still using Broadcom). Do I need to set that up as an environment variable or system-wide?

Coreforge, 2024-10-04:
It should be enough as an environment variable. Another option is to disable the internal GPU, though I don't really know how to ideally do that (somehow I apparently kinda did it, but I'm not sure how. The closest I found was that I set GPU memory to `16` in config.txt). Blacklisting the `bcm2708_fb` or the `vc4` module might do it.  
Or you could try xorg again.

geerlingguy, 2024-10-04:
Well, one thing that _doesn't_ work is commenting out `dtparam=vc4-kms-v3d` in config.txt.

I added `blacklist vc4` to `/etc/modprobe.d/blacklist-vc4.conf`, and rebooted.

The display output now stops loading with a cursor just sitting there... and dmesg shows:

```
[    6.275757] [drm] Display Core v3.2.247 initialized on DCE 11.2
[    6.304411] [drm] UVD and UVD ENC initialized successfully.
[    6.405341] [drm] VCE initialized successfully.
[    6.405862] amdgpu 0000:01:00.0: amdgpu: SE 2, SH per SE 1, CU per SH 8, active_cu_number 14
[    6.410215] amdgpu 0000:01:00.0: amdgpu: Using BACO for runtime pm
[    6.410619] [drm] Initialized amdgpu 3.54.0 20150101 for 0000:01:00.0 on minor 1
[    6.416386] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
```

Coreforge, 2024-10-04:
that's really weird. The amdgpu shouldn't care about vc4. But that message is also really odd considering there is some display output, as there shouldn't be any if it can't find any displays.

Coreforge, 2024-10-04:
I doubt it's some missing packages, but just in case, the output of `apt list --installed  | grep amd`
```
firmware-amd-graphics/now 1:20230210-5+rpt2 all [installed,upgradable to: 1:20230625-2+rpt2]
libdrm-amdgpu1/stable,now 2.4.114-1+b1 arm64 [installed,automatic]
libdrm-amdgpu1/stable,now 2.4.114-1+b1 armhf [installed,automatic]
libteamdctl0/stable,now 1.31-1 arm64 [installed,automatic]
ricks-amdgpu-utils/stable,stable,now 3.8.0-1 all [installed]
xserver-xorg-video-amdgpu/stable,now 23.0.0-1 arm64 [installed,automatic]
```

geerlingguy, 2024-10-04:
@Coreforge 

```
firmware-amd-graphics/stable,stable,now 1:20230625-2+rpt3 all [installed]
libdrm-amdgpu1/stable,now 2.4.114-1+b1 arm64 [installed,automatic]
libteamdctl0/stable,now 1.31-1 arm64 [installed,automatic]
xserver-xorg-video-amdgpu/stable,now 23.0.0-1 arm64 [installed,automatic]
```

geerlingguy, 2024-10-04:
Woah.

```
pi@pi5-pcie:~ $ sudo systemctl restart lightdm
pi@pi5-pcie:~ $ DISPLAY=:0 glxinfo | grep "OpenGL renderer"
OpenGL renderer string: AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.6, DRM 3.54, 6.6.54-v8-16k+)
```

Completely random shot in the dark after reading this SE answer lol: https://raspberrypi.stackexchange.com/q/94650

geerlingguy, 2024-10-04:
Confirmed, now all the apps are using the AMD GPU instead of internal GPU!

[image]

No clue how restarting `lightdm` fixes it... maybe something in the `amdgpu` driver is taking too long to load, so Wayland chooses the internal GPU by default? (Still weird that I can't override per app, but I'm not a Wayland expert, just good at plugging in random stuff I find online lol.)

Coreforge, 2024-10-04:
That would make sense. `amdgpu` seems to load fairly late (it finishes at about 6s for me), while bcm2708_fb loads much earlier (and it looks like I can't blacklist that driver. Maybe setting the max number of FBs in config.txt to 0 could work, but I can't really try it as the driver always fails for me).

geerlingguy, 2024-10-04:
GLMark2 running on the 460:

```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      AMD
    GL_RENDERER:    AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.6, DRM 3.54, 6.6.54-v8-16k+)
    GL_VERSION:     OpenGL ES 3.2 Mesa 23.2.1-1~bpo12+rpt3
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 343 FrameTime: 2.919 ms
[build] use-vbo=true: FPS: 2272 FrameTime: 0.440 ms
[texture] texture-filter=nearest: FPS: 2243 FrameTime: 0.446 ms
[texture] texture-filter=linear: FPS: 2247 FrameTime: 0.445 ms
[texture] texture-filter=mipmap: FPS: 2234 FrameTime: 0.448 ms
[shading] shading=gouraud: FPS: 2251 FrameTime: 0.444 ms
[shading] shading=blinn-phong-inf: FPS: 2249 FrameTime: 0.445 ms
[shading] shading=phong: FPS: 2320 FrameTime: 0.431 ms
[shading] shading=cel: FPS: 2378 FrameTime: 0.421 ms
[bump] bump-render=high-poly: FPS: 2341 FrameTime: 0.427 ms
[bump] bump-render=normals: FPS: 2409 FrameTime: 0.415 ms
[bump] bump-render=height: FPS: 2388 FrameTime: 0.419 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 2422 FrameTime: 0.413 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 2400 FrameTime: 0.417 ms
[pulsar] light=false:quads=5:texture=false: FPS: 2375 FrameTime: 0.421 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 2296 FrameTime: 0.436 ms
[desktop] effect=shadow:windows=4: FPS: 2129 FrameTime: 0.470 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 120 FrameTime: 8.334 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 132 FrameTime: 7.592 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 174 FrameTime: 5.771 ms
[ideas] speed=duration: FPS: 1746 FrameTime: 0.573 ms
[jellyfish] <default>: FPS: 2243 FrameTime: 0.446 ms
[terrain] <default>: FPS: 856 FrameTime: 1.168 ms
[shadow] <default>: FPS: 2179 FrameTime: 0.459 ms
[refract] <default>: FPS: 1388 FrameTime: 0.720 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 2408 FrameTime: 0.415 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 2411 FrameTime: 0.415 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 2270 FrameTime: 0.441 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 2264 FrameTime: 0.442 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 2381 FrameTime: 0.420 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 2388 FrameTime: 0.419 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 2393 FrameTime: 0.418 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 2406 FrameTime: 0.416 ms
=======================================================
                                  glmark2 Score: 1970 
=======================================================
```

(Got around 1800 on the VideoCore on the Pi.)

Getting over 110 fps in SuperTuxKart, and solid 60 fps in dhewm / Doom 3 (how did you install Doom Eternal?).

Coreforge, 2024-10-04:
Doom Eternal was through box64 and steam. When I tried it earlier today again though, I couldn't get anything that needed wine (or proton) to run unfortunately (seems like either something on my system is messed up, as I probably didn't update something correctly, or a regression in box64).

geerlingguy, 2024-10-04:
Heh, always fun to run a combination of software + hardware + architecture configuration that's totally unsupported using bleeding-edge patches that only you and maybe 2-4 other people on Earth have tested!

Coreforge, 2024-10-04:
The commits I updated from where `ba017657` for box86 (probably less relevant) and `067cf75c` for box64. Unfortunately, those versions aren't able to run the steam client anymore, but it might either be an option to not use the steam client (if the steam API doesn't out to be a huge issue, lutris would be a decent option for a launcher, or just raw wine), or to run steam on the current version and run the games with the older version by setting the launch arguments appropriately in steam  
(since the UI doesn't work properly on arm, that can be done in the file `$steamroot/steam/userdata/<userID>/config/localconfig.vdf` by adding the line `"LaunchOptions"		"<launchoptions>"` to the block under `Software/Valve/Steam/apps/<appid>`). 

I probably won't be able to bisect this in time, so those are probably the best options (if it's not something just limited to my install).

geerlingguy, 2024-10-04:
@Coreforge - regarding box86/64, just to confirm, are you on a 4k page size for your kernel? I noticed when I installed Pi-Apps and tried installing Steam, it gave me a warning since I'm on the default 16k page size right now.

Coreforge, 2024-10-04:
I changed it to 4k pages

Coreforge, 2024-10-05:
Box64 at commit `088720db` works. The current one (`2a2f82a7`) just locks up, but this one is from yesterday, so not very old.  
I'd highly recommend having a few GB of swap though (steam takes up a bit over 2GB of memory, which gets quite tight with only 8GB in total) and overclocking the CPU as high as it still runs stable (I have mine at 3GHz), as Doom Eternal ends up CPU limited most of the time. I got about 17-25fps, though it spiked up to 40fps when the CPU was loaded less (the alignment issues are mostly at fault here, as htop shows a good bit of time is spent in the kernel).

Coreforge, 2024-10-08:
Weird. For me it seems to work.
```
pi@raspberrypi:~ $ DISPLAY=:0 vainfo 
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Mesa Gallium driver 23.2.1-1~bpo12+rpt3 for AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.6, DRM 3.54, 6.6.51-v8-16k+)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
```
It also works without setting `DISPLAY` over ssh, though it will complain about not being able to connect to the X server.

[61 more replies]