geerlingguy/raspberry-pi-pcie-devices #565: Add Pi 5 HAT: Pimoroni NVMe Base

geerlingguy, 2023-12-01:
Pimoroni announced their new [NVMe Base for Raspberry Pi 5](https://shop.pimoroni.com/products/nvme-base?variant=41201623695443), which seems to fit within the Pi 5's footprint (underneath) supporting 2230, 2242, 2260, and 2280-sized NVMe SSDs.

[image]

It's listed at £13.50 but is not yet available for sale.

DJM0, 2023-12-14:
Is now available for [pre-orders](https://shop.pimoroni.com/products/nvme-base) (shipping early Jan 2024).

Scream85, 2023-12-22:
I hope you can put your hands on one of these.

geerlingguy, 2023-12-23:
@Scream85 - I can, and I am soon going to ;)

roblatour, 2023-12-31:
just ordered one, figures crossed!

dancingbear1965, 2024-01-01:
Mine arrived a couple of days ago. So far so good. Nice design and a well thought-out flex cable to connect two boards together.

mathiasscheurer, 2024-01-14:
Mine arrived last week. No problems so far. Good choice. I also own the two pineberry hatdrives (top and bottom) and hat cable problems with both (ssd is not recognized on boot). Unfortunately, the pineberry support did not get back to me after my request to send me new cables…

geerlingguy, 2024-01-14:
This board is on the site now: https://pipci.jeffgeerling.com/hats/pimoroni-nvme-base.html

geerlingguy, 2024-01-15:
Closing, but will add any more information to this issue as I see fit!

closed 2024-01-15 by geerlingguy

kmm0, 2024-01-15:
> Mine arrived this week. No problems so far with Western Digital SN530. Good choice. I also own the two pineberry hatdrives (top and bottom) and had cable problems with both (ssd is not recognized on boot). Unfortunately, the pineberry support did not get back to me after my request to send me new cables…

How are the NVME temps with the drive sitting underneath the Pi?

Exaga, 2024-01-23:
> How are the NVME temps with the drive sitting underneath the Pi?

That would depend on the make/model of the M.2 NVMe SSD used. On https://shop.pimoroni.com/products/nvme-base?variant=41219587178579 Pimoroni states "Samsung 970 EVO Plus. Runs at a higher temp. Needs to be quantified." So, I'm assuming the same will be applicable for other drives, all with differing thermodynamic properties.

How much heat is generated by the RPi5 during operation may also be a contributing factor for consideration.

geerlingguy, 2024-01-29:
I'm testing with a Kioxia XG6, and it picked right up, no issue whatsoever. I like how they go overboard with the included screws and standoffs, so you can mount it in a variety of scenarios. Also including four little rubber feet is nice—I can set it on my desk with no case and not worry about it sliding around—or shorting anything out.

At default PCIe Gen 2 settings:

```
     Category                  Test                      Result      
HDParm                    Disk Read                 413.42 MB/sec            
HDParm                    Cached Disk Read          410.13 MB/sec            
DD                        Disk Write                246 MB/s                 
FIO                       4k random read            107225 IOPS (428900 KB/s)
FIO                       4k random write           49829 IOPS (199318 KB/s) 
IOZone                    4k read                   123129 KB/s              
IOZone                    4k write                  123445 KB/s              
IOZone                    4k random read            49038 KB/s               
IOZone                    4k random write           136288 KB/s              

                          Score: 29847                                       
```

And PCIe Gen 3:

```
     Category                  Test                      Result      
HDParm                    Disk Read                 770.86 MB/sec            
HDParm                    Cached Disk Read          760.47 MB/sec            
DD                        Disk Write                350 MB/s                 
FIO                       4k random read            163840 IOPS (655360 KB/s)
FIO                       4k random write           51979 IOPS (207918 KB/s) 
IOZone                    4k read                   162446 KB/s              
IOZone                    4k write                  144622 KB/s              
IOZone                    4k random read            48896 KB/s               
IOZone                    4k random write           180953 KB/s              

                          Score: 38457                                       
```

One thing I would like to see is at least a 'power' LED on the board near the front edge so I can see if it's powered on. Additionally, as a bonus it would be nice to have an SSD activity LED to see when it's being accessed. I can live without, but I do like having those for debugging purposes :)

geerlingguy, 2024-01-29:
Note: I'm using the NVMe BASE in tandem with the [Waveshare PoE HAT (F) for Raspberry Pi 5](https://amzn.to/3Ohhhn9).

sgansky, 2024-02-02:
I've just installed a Pimoroni bottom on a pi5 w/ 4 GB memory. I'm using a Sabrent Rocket Nano NVMe PCIe M.2 Gen3 - 1 TB SSD.  It works OK.  However on a pi5 with 8 Gb memory and a Pineberry top hat and the same SSD I get very different speed checks.  The 8 GB yields around 800 and 4 GB yields around 400.  Can I assume that the speed difference is due to the difference in memory between the 2 pi's ?

geerlingguy, 2024-02-04:
@sgansky If you mean 800 MB/sec and 400 MB/sec (let me know if that's not what you mean), then that sounds a lot like PCIe Gen 3 was enabled on one Pi and Gen 2 was on the other (you can check by running `sudo lspci -vvv` and then searching for the `LinkSta` line for the NVMe drive).

Check your `/boot/firmware/config.txt` file to see if pcie is set to gen 3.

sgansky, 2024-02-04:
Running hdparm -t —direct /dev/nvme0n1 on 8MB pi: Timing 0_Direct disk reads: 2218 MB in 3.00 seconds = 738.77 MB/sec
on 4MB pi yields: 1264 MB in 3.00 seconds = 421.17 MB/sec

I’’ve attached the output of  the config.txt files as well as the lspci - which does not contain the LinkSta line.

Many thanks for the help

Steve

￼￼￼￼

Stephen Gansky
***@***.***
C) 215-990-5300

> @sgansky <https://github.com/sgansky> If you mean 800 MB/sec and 400 MB/sec (let me know if that's not what you mean), then that sounds a lot like PCIe Gen 3 was enabled on one Pi and Gen 2 was on [...]

# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
dtparam=pciex1
dtparam=pciex1_gen=3

# For more options and information see
# http://rptl.io/configtxt
[... 121 lines cut ...]

0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 45
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

0001:01:00.0 Ethernet controller: Device 1de4:0001
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 45
	Region 0: Memory at 1f00410000 (32-bit, non-prefetchable) [size=16K]
	Region 1: Memory at 1f00000000 (32-bit, non-prefetchable) [virtual] [size=4M]
	Region 2: Memory at 1f00400000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: rp1

sgansky, 2024-02-22:
JeffNever saw a reply to my response to your questions.ThanksSteveSent from sgansky iPad(C) 215-990-5300

git-homme, 2024-07-11:
using the Pimoroni Hat with a SKHynix BC901 M.2 NVMe drive and it worked flawlessly. At 1st for compatibility I did not enable the PCIe Gen3 after a couple of benchmark noticed I was "slower" than others so I did remembered I had not put the config.txt line : dtparam=pciex1_gen=3.

Got the great speeds : https://pibenchmarks.com/benchmark/90837/

jjacobsson, 2024-08-14:
I have 4 of these running flawlessly. Two with WD Green SN350 1Tb drives that everyone including the docs claim is not supposed to work, but they work fine and boot without issue after updating the eeprom on the raspberry pi. The other two are using KINGSTON SNV2S1000G 1Tb drive, also working without any issues.