geerlingguy/raspberry-pi-pcie-devices #566: Add Pi 5 HAT: 52Pi

geerlingguy, 2023-12-06:
According to [this tweet](https://twitter.com/tltangliang/status/1732244831160942708), 52Pi has built a PCIe x1 open-ended slot HAT for Pi 5, which allows switching between internal power (leeched off the GPIO pins it seems), or external power (using a 4-pin molex floppy power connector like on the CM4 IO Board).

[image]

No details on release or pricing just yet.

tltangliang, 2023-12-22:
[image]

[image]

[image]

tltangliang, 2023-12-22:
[image]

tltangliang, 2023-12-22:
I used it to test a standard PCIe 2.5G network card, the chip is RTL8125bg.
[image]

The network card was recognized correctly, the IP was obtained successfully。
[image]

The speed handshake was normal when checked with ethtool, and the network speed was normal when tested with iperf3.
[image]
[image]

leobsky, 2024-01-07:
I'm looking forward to get one, please let us know when is ready for sale.

tltangliang, 2024-01-10:
> I'm looking forward to get one, please let us know when is ready for sale.

@leobsky 52pi says it'll ship around the beginning of February.

geerlingguy, 2024-01-14:
The board has been added to the site here: https://pipci.jeffgeerling.com/hats/52pi-pcie-x1-slot-hat.html

geerlingguy, 2024-01-15:
Closing, but will add any more information to this issue as I see fit!

closed 2024-01-15 by geerlingguy