geerlingguy/raspberry-pi-pcie-devices #569: Add 52Pi N05 M.2 2242 PCIe to NVMe Top

yoyojacky, 2023-12-11:
N05 is an NVME M.2 SSD(NVMe 2230, 2242 GEN 3) PIP (PCIe Peripheral Board) for the Raspberry Pi 5 that uses the new PCIE interface of the Raspberry Pi 5 to utilise the NVME M2 SSDs for fast data transfers and ultra-fast booting.

N05 is compatible with the standard HAT size of Raspberry Pi and supports two sizes of NVMe SSD: 2230 and 2242. The included SSD mounting screws make it convenient for users to switch between the two sizes at any time.

N05 is compact in size and can be used with 52Pi's two cases, ZP-0167-black and C-0047.
[image]
[image]
[image]

here is the wiki of that PCIe board:  [ https://wiki.52pi.com/index.php?title=EP-0209 ]

yoyojacky, 2023-12-11:
The features are here: 
Designed for RPI 5: Dedicated to the latest single-board computer, the Raspberry Pi 5.
NVMe Capability: Supports installing NVMe (M-key) drives in M.2 format sizes 2230 and 2242, extra custom CNC SSD mount screw, no soldering required.
PCIe Support: PCIe x1 interface in both Gen2 & Gen3 standards.
Hollow design: Hollow design provides excellent ventilation airflow for cooling.
Gold-plated PCB: Gold-plated PCBs offer superior electrical performance.
PCIe ultra-short signal lines:The short trace routing of PCIe is more reliable and faster, fully meeting the signal requirements of PCIe 3.0.
LED Indicators: Features LED light for M.2 disk activity (“ACT”).
Compatible Form Factor:Board format is compatible with the original Raspberry Pi HAT dimensions of 65 x 56.50 mm.
M.2 Standard-Compliant Power System: Integrated voltage regulator delivering up to 3A for the 3.3V power rail, compliant with M.2 (NGFF) standard.
All Necessary Accessories Included: Comes with a FPC PCIe ribbon, M2 and M2.5 screws, spacers, and extra custom CNC SSD mount screw.
Camera/Display fpc cutout: The openings in the PCB allow for the routing of two Display/Camera FPC cables.

yoyojacky, 2023-12-11:
Here are the features of that PCIe extension board: 

- Designed for RPI 5: Dedicated to the latest single-board computer, the Raspberry Pi 5.
- NVMe Capability: Supports installing NVMe (M-key) drives in M.2 format sizes 2230 and 2242, extra custom CNC SSD mount screw, no soldering required.
- PCIe Support: PCIe x1 interface in both Gen2 & Gen3 standards.
- Hollow design: Hollow design provides excellent ventilation airflow for cooling.
- Gold-plated PCB: Gold-plated PCBs offer superior electrical performance.
- PCIe ultra-short signal lines:The short trace routing of PCIe is more reliable and faster, fully meeting the signal requirements of PCIe 3.0.
- LED Indicators: Features LED light for M.2 disk activity (“ACT”).
- Compatible Form Factor:Board format is compatible with the original Raspberry Pi HAT dimensions of 65 x 56.50 mm.
- M.2 Standard-Compliant Power System: Integrated voltage regulator delivering up to 3A for the 3.3V power rail, compliant with M.2 (NGFF) standard.
- All Necessary Accessories Included: Comes with a FPC PCIe ribbon, M2 and M2.5 screws, spacers, and extra custom CNC SSD mount screw.
- Camera/Display fpc cutout: The openings in the PCB allow for the routing of two Display/Camera FPC cables.

geerlingguy, 2024-01-15:
Thank you! The board is now on the site: https://pipci.jeffgeerling.com/hats/52pi-nvme-hat-no5.html

And I have one to test... will hopefully do so soon!

closed 2024-01-15 by geerlingguy