geerlingguy/raspberry-pi-pcie-devices #570: Add 52Pi N04 M.2 2280 PCIe to NVMe Top

yoyojacky, 2023-12-11:
Hi Geerlingguy, 
I have new product of raspberry pi 5 pcie device, it is 52Pi N series product, it called :` N04 M.2 2280 PCIe to NVMe Top.  `
(N04 for short)

## Descriptions 
N04 is an NVME M.2 SSD(NVMe 2230, 2242, 2260, 2280 GEN 3) PIP (PCIe Peripheral Board) for the Raspberry Pi 5 that uses the new PCIe interface of the Raspberry Pi 5 to utilise the NVME M2 SSDs for fast data transfers and ultra-fast booting.

N04 is compatible with the standard HAT size of Raspberry Pi and supports quad sizes of NVMe SSD: 2230, 2242, 2260 and 2280. The included SSD mounting screws make it convenient for users to switch between the two sizes at any time. 

## Features be like: 

- Designed for RPI 5: Dedicated to the latest single-board computer, the Raspberry Pi 5.
- NVMe Capability: Supports installing NVMe (M-key) drives in M.2 format sizes 2230, 2242, 2260 and 2280, extra custom CNC SSD mount screw, no soldering required.
- PCIe Support: PCIe x1 interface in both Gen2 & Gen3 standards.
- Ventilation hole design: Ventilation hole design provides excellent ventilation airflow for cooling.
- Camera/Display fpc cutout: The openings in the PCB allow for the routing of two Display/Camera FPC cables.
- Gold-plated PCB: Gold-plated PCBs offer superior electrical performance.
- PCIe ultra-short signal lines: The short trace routing of PCIe is more reliable and faster, fully meeting the signal requirements of PCIe 3.0.
- LED Indicators: Features LED light for M.2 disk activity (“ACT”).
- M.2 Standard-Compliant Power System: Integrated voltage regulator delivering up to 3A for the 3.3V power rail, compliant with M.2 (NGFF) standard.
- All Necessary Accessories Included: Comes with a FPC PCIe ribbon, M2 and M2.5 screws, spacers, and extra custom CNC SSD mount screw.
## Gallery 
[image]
[image]
[image]

yoyojacky, 2023-12-12:
I have tested on my RPi5，it assemble like this : 
[image]
and I am using Kingston NV2 PCIe 4.0 NVME M.2 SSD on it, tested it by using pibenchmarks. the result is : 
[image]
My friend has tested it on Fanxiang 2280 SSD which is better than mine. 
[image]
his score is : 
[image]

yoyojacky, 2023-12-12:
Samsung NVMe SSD PM981  with 52Pi N04 M.2 2280 PCIe to NVMe Top board, testing result is: 
[image]
it is the best result i have ever had. LoL.

geerlingguy, 2024-01-15:
Very cool, and thanks for posting the board, and those results!

I have added the board to the site: https://pipci.jeffgeerling.com/hats/52pi-nvme-hat-no4.html

I'll close this issue but post any other interesting results here when I can!

closed 2024-01-15 by geerlingguy

Earnest-Williams, 2024-05-15:
> Very cool, and thanks for posting the board, and those results!
> 
> I have added the board to the site: https://pipci.jeffgeerling.com/hats/52pi-nvme-hat-no4.html
> 
> I'll close this issue but post any other interesting results here when I can!

Got yourself a typo.

https://pipci.jeffgeerling.com/hats/52pi-nvme-hat-n04.html

clintcronin, 2024-05-19:
Using this hat with a combination of a 500gb WD Blue SN570 NVMe ssd results in the Raspberry Pi 5 8gb model hanging for 10- 20+ seconds intermittently and essentially making it pretty much unusable. I'll try another SSD later on today.  I'm using the official RPi 5 power supply. 

SSD I used https://www.amazon.com/Western-Digital-SN570-Internal-Solid/dp/B09HKG6SDF?th=1
Kernel:  Linux BATOCERAPI 6.6.20 #1 SMP Thu May  2 07:39:40 Europe 2024 aarch64 GNU/Linux

dmesg output
[  525.318738] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10
[  525.318748] nvme nvme0: Does your device have a faulty power saving mode enabled?
[  525.318752] nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off" and report a bug
[  525.355580] nvme0n1: I/O Cmd(0x2) @ LBA 476504, 8 blocks, I/O Error (sct 0x3 / sc 0x71)
[  525.355588] I/O error, dev nvme0n1, sector 476504 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[  525.372105] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[  525.375417] nvme nvme0: 4/0/0 default/read/poll queues

lspci -vvv entry for the ssd

0000:01:00.0 Non-Volatile memory controller: Sandisk Corp WD Blue SN570 NVMe SSD 1TB (prog-if 02 [NVM Express])
        Subsystem: Sandisk Corp WD Blue SN570 NVMe SSD
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 40
        Region 0: Memory at 1b00000000 (64-bit, non-prefetchable) [size=16K]
        Region 4: Memory at 1b00004000 (64-bit, non-prefetchable) [size=256]
        Capabilities: [80] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [90] MSI: Enable- Count=1/32 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [b0] MSI-X: Enable+ Count=17 Masked-
                Vector table: BAR=0 offset=00002000
                PBA: BAR=4 offset=00000000
        Capabilities: [c0] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <1us, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr+ FatalErr- UnsupReq+ AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <8us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s, Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
                         EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq+ ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr+ BadTLP- BadDLLP- Rollover+ Timeout+ AdvNonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 14, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 40000001 0000000f 00001018 c1020000
        Capabilities: [150 v1] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [1b8 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Capabilities: [300 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: LaneErr at lane: 0
        Capabilities: [900 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1- ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
                          PortCommonModeRestoreTime=32us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: nvme

yoyojacky, 2024-06-06:
Power Saving Mode: The device may have enabled some kind of power-saving mode, which could lead to instability or compatibility issues with the operating system.
Hardware Failure: "controller is down" may indicate a hardware issue, which could be due to a malfunction or improper configuration.
Driver or Firmware Issues: It's possible that there is a compatibility issue or a bug with the NVMe driver or SSD firmware in relation to the current operating system version.
The first step in resolving this issue is to follow the kernel message's suggestion to try disabling the power-saving mode and see if the problem persists. If the problem continues, further hardware diagnostics may be necessary to check the compatibility of the SSD and motherboard, or to update the SSD firmware and operating system kernel. Additionally, if this is a known hardware or software issue, looking for relevant bug reports or community discussions may provide more solutions.

yoyojacky, 2024-06-14:
Today I have test it with hailo-8 AI co-processor, it works fine !
[image]

WhoIsAvailable, 2024-07-12:
Last week, I bought a 52Pi N04 top HAT which I want to install onto my Pi 5 and active cooler. However, the instructions that came with the HAT are incomplete and out of date. I tried to get support from the 52Pi website through their support email address and chat line with no response.

My main problem is the FPC / FFC cable(s) that is intended to connect the HAT to the PCIe slot on the Pi 5. The instructions tell me that the contacts on the cable should face one way, which is easy enough to achieve, even though the contacts are now silver where the instructions say they are gold. The problem is that there is no guidance on which end of the cable should be inserted into which slot – especially given that there are now distinguishing marks on the two ends of the cable (a digit 1 on one end and digits 6 and 8 on the other).

I did find a video on youtube showing the installation of a 52Pi N04 HAT, but that clearly shows that my cables are different from the ones in the video.

Perhaps it does not matter which way round the cable is inserted, but I have no inclination to experiment and possibly damage the Pi given the ease with which electronic components can be damaged.

Can anyone here please either give me guidance on which way round the cable should be inserted, or confirm that it really does not matter ?

Thanks in advance for any help . . .

geerlingguy, 2024-07-12:
Hid the above comment - it was also posted separately, and is here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/656?converting=1 — this site isn't support for 52Pi's products, I would ask them about it first :)

WhoIsAvailable, 2024-07-13:
I first asked my supplier - The Pi Hut.

They suggested contacting 52Pi directly which I did - I am still waiting for an acknowledgment after 4 days, let alone a helpful reply . . .

yoyojacky, 2024-07-20:
> I first asked my supplier - The Pi Hut. They suggested contacting 52Pi directly which I did - I am still waiting for an acknowledgment after 4 days, let alone a helpful reply . . .
> […](#)

who did you contact for?

WhoIsAvailable, 2024-07-20:
I have no idea how to insert the cable between the Pi 5 and the N04. It appears to be double ended and the instructions say nothing about which end goes in which device, and there is nothing to suggest that the cable is or is not keyed to fit a certain way.

In the last 10 days, I have submitted three separate requests for information to 52Pi and have received absolutely nothing. If 52Pi refuse to offer support, then how on earth am I supposed to know how to move forward ?

geerlingguy, 2024-07-21:
@WhoIsAvailable - Please see the discussion over in https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/656

jjacobsson, 2024-08-13:
Aum, sorry to bother, but why does the board use 10 of the GPIO pins? Anyone know? Or is it just done that way for stability or something and the pins are actually unused?

yoyojacky, 2024-08-21:
> I have no idea how to insert the cable between the Pi 5 and the N04. It appears to be double ended and the instructions say nothing about which end goes in which device, and there is nothing to [...]

I mean who did you contact in the last 10 days?  Could you please send your email to tech@52pi.net