geerlingguy/raspberry-pi-pcie-devices #572: Test Samsung PM991a 256GB SSD

thmang82, 2023-12-12:
I have just build up a CM4 based mini-server with a Samsung PM991a 2242 M2 265 GB SSD.
SAMSUNG MZALQ256HBJD-00BL2

**Important**: The quite similar named Samsung PM9B1 does not work (Kernel Error when booted from SD-Card, and mass-storage-gadget does not detect it). Maybe it's because it is PCIe 4.0, instead PCIe 3.0 for the PM991a.

System is based on a WaveShare CM4-IO-BASE-B board.

Speed seems to be quite good, see below.

I got the drive for about 16€ from ebay. They are not available "new" any more, likely stems from some unsold laptops. Was only used for less than 4 hours according to smartctl.

IOZone Speed Test with 4k/64k Block size and 1MB:
```
	Auto Mode
	Record Size 4 kB
	Record Size 64 kB
	File size set to 1024 kB
	Command line used: iozone -a -i 0 -i 1 -i 2 -r 4 -r 64 -s 1024
	Output is in kBytes/sec
	Time Resolution = 0.000001 seconds.
	Processor cache size set to 1024 kBytes.
	Processor cache line size set to 32 bytes.
	File stride size set to 17 * record size.
                                                              random    random                                
              kB  reclen    write  rewrite    read    reread    read     write
            1024       4   264944   634055   975063  1061060   667977   519543                                                                
            1024      64   346765  1211964  2275110  2497356  2236023  1193442
```

SmartCTL result:
```
sudo smartctl -a /dev/nvme0n1p2
smartctl 7.3 2022-02-28 r5338 [aarch64-linux-6.1.0-rpi7-rpi-v8] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SAMSUNG MZALQ256HBJD-00BL2
Serial Number:                      S65FNX0Txxxxxx
Firmware Version:                   5L2QFXM7
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 256,060,514,304 [256 GB]
Unallocated NVM Capacity:           0
Controller ID:                      5
NVMe Version:                       1.4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          256,060,514,304 [256 GB]
Namespace 1 Utilization:            6,782,369,792 [6.78 GB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            002538 d821b10cc0
Local Time is:                      Tue Dec 12 19:12:18 2023 GMT
Firmware Updates (0x16):            3 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x0057):     Comp Wr_Unc DS_Mngmt Sav/Sel_Feat Timestmp
Log Page Attributes (0x0f):         S/H_per_NS Cmd_Eff_Lg Ext_Get_Lg Telmtry_Lg
Maximum Data Transfer Size:         512 Pages
Warning  Comp. Temp. Threshold:     81 Celsius
Critical Comp. Temp. Threshold:     87 Celsius
Namespace 1 Features (0x10):        NP_Fields

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     5.35W       -        -    0  0  0  0        0       0
 1 +     5.35W       -        -    1  1  1  1        0       0
 2 +     5.35W       -        -    2  2  2  2        0     500
 3 -   0.0500W       -        -    3  3  3  3      210    1200
 4 -   0.0050W       -        -    4  4  4  4     1000    9000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0
```

thomas-lentz, 2024-05-24:
Hi, 
this is really interesting! My search for a problem with my Raspberry CM4 with Samsung PM991 (MZALQ256HAJD) braught me here. I'm using the (almost) exact setup, but I can't get them to run. The moment I'm inserting the SSD into the WaveShare CM4-IO-BASE-B, the boards stops booting until it runs into some sort of timeout and then continues to boot. Using Windows with an USB to M2 Adapter the P991 are running fine (I bought 5 of them). The problem is related to these entries and I can't figure out, what the problem is.

```
[    1.288572] nvme nvme0: pci function 0000:01:00.0
[    1.288621] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[   62.431806] nvme nvme0: I/O 16 QID 0 timeout, completion polled
[  123.871767] nvme nvme0: I/O 17 QID 0 timeout, completion polled
[  123.871804] nvme nvme0: Shutdown timeout set to 8 seconds
[  185.311746] nvme nvme0: I/O 18 QID 0 timeout, completion polled
```

I attached the dmesg messages in case you have an idea what the problem might be
[dmesg2.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/15429732/dmesg2.txt)

I appreciate your help.

Kind regards
 Tom

ned14, 2025-01-16:
Note that the official Raspberry Pi NVMe SSD is underneath a Samsung PM991a (the 'a' is important). Source: https://www.techpowerup.com/ssd-specs/raspberry-pi-ssd-512-gb.d2227

geerlingguy, 2025-01-16:
@ned14 - Indeed! That's what we found while testing the Pi SSD here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/679

ned14, 2025-01-16:
Looks like at the time of me writing this right now, used eBay/Aliexpress PM991a SSDs are very similarly priced to the official Raspberry Pi SSD new if you exclude delivery. The difference appears to be basically the delivery charge for new vs used and potentially dodgy vs guaranteed working.

BTW Jeff love your blog have done for ages. Been occasionally following you since you were more a religious blog (I am also a Catholic). Then you seemed to suddenly go into tech, and I was "oh I do that too!". Keep doing what you're doing!