geerlingguy/raspberry-pi-pcie-devices #573: Test Pineberry Pi's HatDrive! Bottom (RPI5)

xy8000, 2023-12-27:
I am using the HatDrive! **Bottom** with a TS256GMTE110S (and official Power-Adapter). Unfortunately, I have observed that the performance does not meet my expectations. While the drive is successfully listed in lsblk, several functionalities seem to be compromised.

Initially, I encountered an error when attempting to adjust rpi-eeprom-config after connecting the drive. The command `rpi-eeprom-config --edit` failed with the following error: ERROR: `rpi-eeprom-update -b timeout`.

Upon further investigation, I discovered that the latency of the drive is excessively high. Even a basic operation like touch can take up to 30 seconds.
```
time touch /mnt/nvme0n1p2/home/XXX/writeTest

real	0m25,986s
user	0m0,001s
sys	0m0,000s
```
As anticipated, the results of `hdparm` indicate low performance:
```
hdparm -Tt /dev/nvme0n1p1

/dev/nvme0n1p2:
 Timing cached reads:     2 MB in 33.79 seconds =  60.61 kB/sec
 Timing buffered disk reads:   2 MB in 33.88 seconds =  60.45 kB/sec
```
Reviewing the system logs with `journalctl revealed a significant number of timeout and abortion messages associated with the NVMe drive:

```
nvme nvme0: I/O 79 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 80 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 81 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 82 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 83 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 84 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 85 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: I/O 86 (I/O Cmd) QID 4 timeout, aborting
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
nvme nvme0: Abort status: 0x0
```
Additionally, I have checked the SMART data, and the reported information appears to be normal.

It's noteworthy that my drive is not included in the [nvme-compatibility-list](https://docs.pineberrypi.com/nvme-compatibility-list). Therefore it is not officially supported. But it seems like nobody encountered this issue jet. Therefore i want to report that behavior. It seems worth to check your drive properly before booting from it.

geerlingguy, 2023-12-27:
Strange—I wonder if the cable might not be fully inserted (try reseating it), or if you downgrade the PCIe connection to Gen 1, maybe that could help?

xy8000, 2023-12-27:
> Strange—I wonder if the cable might not be fully inserted (try reseating it), or if you downgrade the PCIe connection to Gen 1, maybe that could help?

I re-plugged everything. Even checked the connection points on the cable. Those are looking good to me.
I even checked the PCIe Gen Version. Results don't change much between them.

Results: PCIe Gen 2:
```
 Timing cached reads:     2 MB in 33.80 seconds =  60.60 kB/sec
 Timing buffered disk reads:   2 MB in 33.82 seconds =  60.55 kB/sec
```
Results: PCIe Gen 3:
```
 Timing cached reads:     2 MB in 31.74 seconds =  64.53 kB/sec
 Timing buffered disk reads:   2 MB in 32.84 seconds =  62.35 kB/sec
```
After conducting tests with [nvmi.cli](https://github.com/linux-nvme/nvme-cli), I discovered that waking up the drive before use is necessary in my case. 
For instance, I used the command `watch nvme transcend healthvalue /dev/nvme0` to call the drive every two seconds. Once i received the first response `Transcend NVME heath value: 100%`, I reran the benchmark using `hdparm -Tt /dev/nvme0n1p1`. The results were astonishing (and reproducible):

Results: PCIe Gen 2 with manual nvme-cli call:
```
 Timing cached reads:   2106 MB in  2.00 seconds = 1053.42 MB/sec
 Timing buffered disk reads: 512 MB in  1.16 seconds = 443.09 MB/sec
```
I'm uncertain about the root cause of this issue and need to investigate further. However, I can confidently say it's unlikely to be related to cables.

geerlingguy, 2023-12-27:
@xy8000 ah, nice catch. Honestly I've never run into that in my testing—all my NVMe SSDs either worked (at full speed) or didn't (at all!).

tltangliang, 2024-01-07:
Maybe the hardware went into hibernation, need to check how the wake# signal is handled by the pcie on the hat. 
If the pull-up resistor is too small, it may affect the pcie's wake# to work properly.
 This may not work properly on some pcie disks.

xy8000, 2024-01-07:
Finally, I tested with an officially supported drive (SDBPNPZ-256G-10SB). Unfortunately, it didn't work either. I believe I have encountered a hardware issue. I will close this ticket and return the HAT.

@tltangliang: It could be. Unfortunately, I do not have the expertise to address that.

closed 2024-01-07 by xy8000