geerlingguy/raspberry-pi-pcie-devices #575: Add Pi 5 HAT: Geekworm X1000/X1001/X1002

geerlingguy, 2024-01-04:
The [GeekWorm X1001](https://wiki.geekworm.com/X1001) is the model that interests me most, but Geekworm has introduced three models, X1000, X1001, and X1002, supporting Top/2230/2242, Top/2230/2242/2260/2280, and Bottom/2230/2242/2260/2280, respectively.

[image]

It works with the Pi Active Cooler, and seems to give wide clearance to the GPIO header so other HATs could be used. There's also an external 5V input, in case you need it, and activity LEDs for power/SSD activity.

leobsky, 2024-01-07:
I'm using the  models X1000 and X1001 with OpenWrt combined with a m.2 to pcie adapter to connect a dual 2.5Gb network card 
https://github.com/openwrt/openwrt/pull/13987#issuecomment-1872227588
I changed the FCC cable with a longer one,150mm to make easy the test.
[image]

kbftech, 2024-01-11:
Bought one for 35$ on amazon canada.
[image]
I'm going to pair it with a 2280 "cheap" NVME ssd:
[image]

Should receive everything by saturday and will report back on how it went,

kmm0, 2024-01-14:
Bought a X1001 from Amazon for $15.99.

Working fine at gen 3 with a cheap Inland TN320 256GB NVME ($20 open box at Microcenter).

Noteworthy, the the Inland TN320 256GB NVME has a drive activity LED.
[image]

kbftech, 2024-01-14:
Works fine. SSD runs very hot.

[image]

geerlingguy, 2024-01-15:
@kbftech I haven't seen any fan splitter for the tiny connector on the Pi board (could be interesting, but might also cause issues). I think the best option would either be a heatsink on the M.2 drive (can find some on Amazon), or a little fan off the 5v GPIO pins.

msgilligan, 2024-01-15:
I got one (also for $15.99 on Amazon) and my SSD is not visible. Could this be because I don't have the RPi 27W (or equivalent) power supply? Documentation for the X1001 is slim and Raspberry Pi's documentation (and the warning message displayed at boot) talk about not having power for USB devices, but does this also apply to the PCIe slot?

Only one of the photos above shows a power cable attached and it sure looks like the RPi 27W cable...

kmm0, 2024-01-15:
> I got one (also for $15.99 on Amazon) and my SSD is not visible. Could this be because I don't have the RPi 27W (or equivalent) power supply? Documentation for the X1001 is slim and Raspberry Pi's documentation (and the warning message displayed at boot) talk about not having power for USB devices, but does this also apply to the PCIe slot?
> 
> Only one of the photos above shows a power cable attached and it sure looks like the RPi 27W cable...
> 
> I'm also wondering what the threshold is for using the 5V connector on the X1001 and what you're supposed to attach the other end to - 5V/GND on the GPIO?

I am using the the 27W PSU, but I just tested with a Pi4 PSU and a powerbank and the Pi 5 booted to NVME with both.

Confirmed power with:
dtc /proc/device-tree/chosen/power/ -f 

max_current = <0x0384> (that is .9A).

The 27W PSU shows max_current = <0x1388> (that is 5A)

Did you add dtparam=pciex1_gen=3 to config.txt ?

kmm0, 2024-01-15:
> Works as expected, boots fast (need to remove SD card). SSD runs very hot. (gen 3 speed)

Did you set the boot order?  I can boot to NVME with an SD card installed.

How hot is very hot?

Using smartctl -a /dev/nvme0 I'm seeing temps in the low 30 degrees C.

kbftech, 2024-01-15:
Fyi it works on my end with a pi4 psu (4amps). 3amps might not cut it; I did not test it but will check if I have one when I get back home.

Also good to note that not all drives are created equal; some probably require More power.

kbftech, 2024-01-15:
>  ... I think the best option would either be a heatsink... 

A small heatsink doesn't seem to cut it during heavy use. Tested with a 512gb image the heatsink was very hot to the touch. I would definitely recommend active cooling at least with this specific drive, especially if you plan on keeping the pi in any kind of case.

msgilligan, 2024-01-15:
> I am using the the 27W PSU, but I just tested with a Pi4 PSU and a powerbank and the Pi 5 booted to NVME with both.

Thanks, that's good to know.

> Did you add dtparam=pciex1_gen=3 to config.txt ?

I added `dtparam=pciex1`

I've verified that the NVMe SSD itself works correctly in another device. I receive a 27W (not Raspberry Pi, but this one) tomorrow, so I can try it. But at this point I'm suspecting either my X1001 or the cable is bad.

paulwratt, 2024-01-16:
in one of the original RPi5 YT reviews (Explaining Computers maybe? or ETA Prime?), you can see a closeup of the text on the  "official RPi5 power supply": ​its a PD (as in phone charging standard) power supply - PD 3.0? The RPi4 power supply will work but there is only so much current available. The PD power supply on the other hand can up the input voltage (from 5v to 9v & 12v) for the RPi5 so that its power circuitry can down-convert it back to 5 volts (5v) with enough AMP's to properly power _every_ port and bus on the SBC, if they were _all_ being used at the same time.

Having watched Jeff's RPi5 NVME related videos, I would be using a seperate power delivery for the NVME, even if its controlled from the same switch.

As for "heat": You are going to have to wait for someone else to compile a list of "low power" (ie low heat generating) NVME devices suitable for SBC & laptops, unless you want to do your own "research" (ie. experiments), or find a PC review platform that included thermals and/or operating temps for NVME devices.

kbftech, 2024-01-16:
Rpi5 does not have a dc to dc converter and does not use any other voltage than 5v.

The PD psu offered for the pi has an arguably out of spec option that can provide 5amps at 5v. This is not in violation of the spec per say, but sure is a very odd thing to do.

Plz do not mislead people in buying "standard" pd psu to run the pi5.

geerlingguy, 2024-01-16:
I have tested the Pi 5 with both the Radxa 30W adapter and Raspberry Pi's older 5v 3A adapter, and both seemed to boot at least the little Kingspec NVMe SSD I had without issue.

The Pi 5 really doesn't burn through a huge amount of power, it's mostly if you have a ton of peripherals attached, or an NVMe or PCIe device that sucks down a lot of power (it's in spec for M.2 devices to use up to 15W, yikes!). Raspberry Pi's own FFC guidance recommends an external supply for PCIe devices if you can use one, but states up to around 5W is possible straight through the FFC.

And it is not arguably out of spec, just uncommon. There are a lot of things uncommon about the Pi, like it's bootloader :)

leobsky, 2024-01-17:
I'd tried the official rpi4/5 PS and both work with the x1001, also there is a model x1004 that support dual sdd http://suptronics.com/Raspberrypi/Storage/x1004-v1.1.html so I think power supply shouldn't be an issue.
[image]

msgilligan, 2024-01-17:
Updates:

1. The [BOXPUT](https://www.amazon.com/dp/B0CNRBLQM1)) power supply arrived and seems to work properly.  (The warning message went away, but I have not actually tested it with anything but the X1001 and NVMe SSD.)

2. The X1001 and NVMe SSD are working correctly now. It looks like I neglected to update the eeprom, because after I did that it showed up correctly.

kbftech, 2024-01-18:
I ordered a USB Hub and another PSU (official 27w). I'm going to try and add a secondary fan aswell. I decided against having the NVME drive under the usb hub as I originally planned due to heat. To sandwish the USB Hub between the pi and the X1001 NVME, I need a longer ribbon cable so waiting on that to test everything out. Once I have the whole thing, I want to test:
 - NVME + Hub powered solely by the PI
 -  --> Without periferals
 -  --> With USB/Keyboard directly on the PI
 -  --> With USB/Keyboard on the hub
 -  --> Mix of different periferals (USB wireless controllers, wired controllers, external drives)
 -  --> Charging through every port
 - Same thing with multiple USB-C chargers

I'd be very interrested in the dual slot NVME but couldn't find it in Canada. Also, I have yet to find a POE Hat that will provide sufficent power to the pi 5 and it's periferals. The best I found so far is a waveshare that claims max 4.5a on 5v.
 
This is probably not the best place to discuss this though. I feel like I am polluting this thread. Feel free to move this comment where it should be.

kbftech, 2024-01-19:
Thigh fit, but it does fit & work. 
[image]
[image]
[image]

Side note: During "normal" operations, the drive runs cool to the touch as opposed to when writing the image.

I'll test and will post results.

eric-becker, 2024-01-26:
Seems to run fine with the waveshare PoE HAT (F), but I'm only powering the X1001 and a small USB zigbee dongle for Home Assistant. The Samsung 990 NVMe drive is overkill, but it's temporary. It's significantly slower on HAOS vs Raspberry Pi OS (300 MB/s vs 800MB/s), so I'll swap back and run HA in a container.  
[image]

UberGeekDude, 2024-10-13:
I got the X1001 for my RPi 5.  And I am using a 1-5 PCIe adapter (B14 Quad FPC PCIe HAT) that also connects a 5 port SATA hat (Geekworm X1009 PCIe to 5-Port SATA). 

I came across a problem when I plugged in a USB 3.0 (or 3.1) HDD, where the USB Drive did not appear as a device in Linux (default Linux distro for RPi 5).

I have a WD 1TB SSD, plus 3 HDDs.  When unplugging one of the HDDs, then plugging in the USB drive, the USB appears and works fine.  So, my final configuration is: M.2 WD 1TB SSD, two HDDs and the USB HDD.  I don't think it is a power issue.  I have a high wattage PS for the SATA card: 12 Volts at 120 watts.  The SATA card converts 12 V to 5V to run the entire system.  I had both USB-C power and the SATA power but learned I do not need both.  The system is under powered with just the USB-C power plugged in.  I am guessing the problem is a RPi 5 limitation.  Anyone know the limit for the number of drives?  Before I added the M.2 SSD, I had three HDDs and the USB HDD all working fine (4 drives).

I use the RPi 5 to replace an older Intel PC that ran as my Linux system and shared drives for backup and whatnot.

thestumbler, 2025-03-07:
Just want to note something here. I have installed one of these X1001 boards on an RPi5 stack-up. My first time using an NVME module on an RPi. In addition to the usual reasons to avoid using SD cards, in my application mounted on a 5-in LCD display, it is nigh impossible to access / change the SD card without dismantling the whole stack.

Anyway, @geerlingguy mentions in his blog post on these modules:
```
# Add the following line if using a non-HAT+ adapter:
PCIE_PROBE=1
```
But it wasn't entirely clear to me whether this hat conformed to the Hat+ specification or not. I took a change that it did not, and included the above line. It all seems to work for me. The RPi5 is running fine, without an SD card installed. I didn't time it, but the boot-up time is remarkably faster than before. Thanks, Jeff, for documenting all this for us engineers who don't work with the bleeding edge daily.

[image]

thestumbler, 2025-03-07:
It's hard to see the NVME board in there, the others are a power management board and an rs485 interface hat as well