geerlingguy/raspberry-pi-pcie-devices #576: VL805 and µPD720201 not working on Pi 5

tltangliang, 2024-01-05:
I'm testing vl805 and µPD720201 on pi5 and the usb adapter card doesn't work.

# VL805
The pci card can be enumerated, but the usb device cannot be found and takes a very long time to boot up.
[image]
[image]

#µPD720201 
µPD720201 directly fails to boot the system, causing kernel errors
[image]

julienrobin28, 2024-06-19:
Hi @tltangliang 

Had the exact same issue on the exact same hardware (I believe, PCIe x1 Slot for RPI 5 from 52PI, but the PCB is black, and exact same looking USB3 card with VL805 on it).

I ended up with the VL805 card working, by adding this line into `/boot/firmware/config.txt`:

		dtoverlay=pcie-32bit-dma-pi5

I tried a lot of different PCIe options (both by dtoverlay, dtparam, and cmdline.txt) but this one is the only / single one that made it work (others options aren't needed). Works fine, around 400 MiB/s without warnings from `dmesg`.

From `/boot/firmware/overlays/README` the only options I didn't try is `pciex1-compat-pi5`, on which `l1ss`, `no-l0s` and `no-mip` sub-parameters are available (but I still don't know what they do). I also have to get information about `CLKREQ#` things.

It looks like having PCIe devices working with the Pi 5 isn't as plug and play as expected (at least as I expected...), but everything seems to turn around these options.

Hope this may help!

tltangliang, 2024-06-19:
That's awesome! Upon seeing your message, I immediately tested it, and adding this configuration made it work. 
But the system cannot boot from USB through VL805.
However, the µPD720201 still cannot be used. 

Thank you very much!!!

mikeneko007, 2025-04-26:
Report on operation with raspberryPI Kernel 6.12.22
RaspberryPI 5B on µPD720201 card

It does not work when connected directly to µPD720201.
Adding an ASM1184 bridge in the middle makes it work.

RaspberryPI5 - µPD720201
[    1.056972] brcm-pcie 1000110000.pcie: host bridge /axi/pcie@1000110000 ranges:
[    1.056976] brcm-pcie 1000110000.pcie:   No bus range found for /axi/pcie@1000110000, using [bus 00-ff]
[    1.056983] brcm-pcie 1000110000.pcie:      MEM 0x1b00000000..0x1bfffffffb -> 0x0000000000
[    1.056987] brcm-pcie 1000110000.pcie:      MEM 0x1800000000..0x1affffffff -> 0x0400000000
[    1.056992] brcm-pcie 1000110000.pcie:   IB MEM 0x0000000000..0x0fffffffff -> 0x1000000000
[    1.056996] brcm-pcie 1000110000.pcie:   IB MEM 0x1000131000..0x1000131fff -> 0xfffffff000
[    1.060891] brcm-pcie 1000110000.pcie: PCI host bridge to bus 0001:00
[    1.060895] pci_bus 0001:00: root bus resource [bus 00-ff]
[    1.060897] pci_bus 0001:00: root bus resource [mem 0x1b00000000-0x1bfffffffb] (bus address [0x00000000-0xfffffffb])
[    1.060899] pci_bus 0001:00: root bus resource [mem 0x1800000000-0x1affffffff pref] (bus address [0x400000000-0x6ffffffff])
[    1.060909] pci 0001:00:00.0: [14e4:2712] type 01 class 0x060400 PCIe Root Port
[    1.060914] pci 0001:00:00.0: PCI bridge to [bus 00]
[    1.060917] pci 0001:00:00.0:   bridge window [mem 0x1b00000000-0x1b000fffff]
[    1.060919] pci 0001:00:00.0:   bridge window [mem 0x1b00000000-0x1b000fffff 64bit pref]
[    1.060936] pci 0001:00:00.0: PME# supported from D0 D3hot
[    1.061522] pci 0001:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.168746] brcm-pcie 1000110000.pcie: clkreq-mode set to safe
[    1.168748] brcm-pcie 1000110000.pcie: link up, 5.0 GT/s PCIe x1 (!SSC)
[    1.168782] pci 0001:01:00.0: [1912:0014] type 00 class 0x0c0330 PCIe Endpoint
[    1.168807] pci 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b00001fff 64bit]
[    1.168949] pci 0001:01:00.0: PME# supported from D0 D3hot D3cold
[    1.170755] pci_bus 0001:01: busn_res: [bus 01-ff] end is updated to 01
[    1.170764] pci 0001:00:00.0: bridge window [mem 0x1b00000000-0x1b000fffff]: assigned
[    1.170766] pci 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b00001fff 64bit]: assigned
[    1.170780] pci 0001:00:00.0: PCI bridge to [bus 01]
[    1.170781] pci 0001:00:00.0:   bridge window [mem 0x1b00000000-0x1b000fffff]
[    1.170784] pci_bus 0001:00: resource 4 [mem 0x1b00000000-0x1bfffffffb]
[    1.170785] pci_bus 0001:00: resource 5 [mem 0x1800000000-0x1affffffff pref]
[    1.170787] pci_bus 0001:01: resource 1 [mem 0x1b00000000-0x1b000fffff]
[    1.170790] pci 0001:00:00.0: Max Payload Size set to  128/ 512 (was  128), Max Read Rq  512
[    1.170800] pci 0001:01:00.0: Max Payload Size set to  128/ 128 (was  128), Max Read Rq  512
[    1.170857] pcieport 0001:00:00.0: enabling device (0000 -> 0002)
[    1.170921] pcieport 0001:00:00.0: PME: Signaling with IRQ 36
[    1.170993] pcieport 0001:00:00.0: AER: enabled with IRQ 36
[    1.171030] pci 0001:01:00.0: enabling device (0000 -> 0002)
[    1.171048] Unable to handle kernel paging request at virtual address ffffc00082d98103
[    1.171050] Mem abort info:
[    1.171050]   ESR = 0x0000000096000021
[    1.171051]   EC = 0x25: DABT (current EL), IL = 32 bits
[    1.171053]   SET = 0, FnV = 0
[    1.171054]   EA = 0, S1PTW = 0
[    1.171054]   FSC = 0x21: alignment fault
[    1.171055] Data abort info:
[    1.171056]   ISV = 0, ISS = 0x00000021, ISS2 = 0x00000000
[    1.171057]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[    1.171058]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    1.171059] swapper pgtable: 16k pages, 47-bit VAs, pgdp=00000000014d4000
[    1.171060] [ffffc00082d98103] pgd=1000000001968003, p4d=1000000001968003, pud=1000000001968003, pmd=10000000032ac003, pte=0068001b00000713
[    1.171066] Internal error: Oops: 0000000096000021 [#1] PREEMPT_RT SMP
[    1.171068] Modules linked in:
[    1.171071] CPU: 1 UID: 0 PID: 102 Comm: kworker/u18:2 Not tainted 6.12.22+ #14
[    1.171074] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    1.171075] Workqueue: events_unbound deferred_probe_work_func
[    1.171080] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    1.171083] pc : quirk_usb_early_handoff+0x65c/0x9f0
[    1.171089] lr : quirk_usb_early_handoff+0x59c/0x9f0
[    1.171092] sp : ffffc00082acb8f0
[    1.171092] x29: ffffc00082acb900 x28: 0000000000000000 x27: 0000000000000000
[    1.171095] x26: 0000000045b1fe6a x25: ffffc00080a0f438 x24: ffffc000816d0030
[    1.171097] x23: ffffc00082d980ff x22: ffffc00082d98103 x21: ffffc00082d98000
[    1.171099] x20: 000000000007a0f0 x19: ffff800003242000 x18: ffffc000816dac26
[    1.171102] x17: ffff80000230aa80 x16: ffffc0008178c000 x15: 0000000000000801
[    1.171104] x14: 0140000000000000 x13: 00003ffff00020b6 x12: 0000401a7d268000
[... 238 lines cut ...]
[    1.300120] usb usb4: Manufacturer: Linux 6.12.22+ xhci-hcd
[    1.300121] usb usb4: SerialNumber: xhci-hcd.0
[    1.300202] hub 4-0:1.0: USB hub found
[    1.300207] hub 4-0:1.0: 1 port detected
[    1.300470] xhci-hcd xhci-hcd.1: xHCI Host Controller
[    1.300476] xhci-hcd xhci-hcd.1: new USB bus registered, assigned bus number 5
[    1.300905] xhci-hcd xhci-hcd.1: hcc params 0x0240fe6d hci version 0x110 quirks 0x0000808000000810
[    1.300941] xhci-hcd xhci-hcd.1: irq 145, io mem 0x1f00300000
[    1.301006] xhci-hcd xhci-hcd.1: xHCI Host Controller
[... cut ...]
[    1.301136] hub 5-0:1.0: USB hub found
[    1.301143] hub 5-0:1.0: 2 ports detected
[    1.301251] usb usb6: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.12
[    1.301253] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.301254] usb usb6: Product: xHCI Host Controller
[    1.301255] usb usb6: Manufacturer: Linux 6.12.22+ xhci-hcd
[    1.301256] usb usb6: SerialNumber: xhci-hcd.1
[    1.301339] hub 6-0:1.0: USB hub found
[    1.301344] hub 6-0:1.0: 1 port detected