geerlingguy/raspberry-pi-pcie-devices #585: Add 52Pi 2.5G PCIe Network Adapter

tltangliang, 2024-01-16:
RTL8125BG 10/100/1000M/2.5G PCIe Ethernet for Raspberry Pi 5 

[image]

geerlingguy, 2024-01-16:
Oh wow, love this design! I would love to test one too. It's on the site now: https://pipci.jeffgeerling.com/hats/52pi-2.5g-pcie-network-adapter-hat.html

One other version that would be very interesting is basically this dual-2.5G card, but in HAT form: https://pipci.jeffgeerling.com/cards_network/syba-dual-25gbe-pcie-nic.html (uses 2x realtek chips + an asmedia bridge).

tltangliang, 2024-01-17:
I can contact 52pi to send you a piece, please send me the shipping address to my email: tltangliang@gmail.com

As far as I know they are mass producing single port 1000, dual port 1000M version, dual port 2.5G is in design, regular dual port 2.5G using ASMedia 1182 chip, can't fully utilize the performance of two 2.5 network ports, they may use the latest pcie 3.0 switch, but the cost is expensive.